evcc-io/evcc #15063: Easee-Wallbox (Single): Switch phases to 3p fails

eviw, 2024-07-26:
### Describe the bug

I have just a single easee home wallbox.
Whenever loadpoint lp-1 exeeds 3p threshold, it starts scel3p switch, but easee does not physically switch from 1p->3p. It remains 1p. Later a "charger logic error: phases mismatch (got 1, expected 3)" is logged.
[image]
[evcc-20240726-094945-trace.log](https://github.com/user-attachments/files/16388971/evcc-20240726-094945-trace.log)

### Steps to reproduce

1. Connect car to easee
2. EVCC mode=pv
3. Available power > 3p threshold
4. switch to 3p fails
...

### Configuration details

```yaml
Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  easee: trace

# unique installation id
plant: *****

interval: 10s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: tesla-powerwall
  usage: grid
  host: 192.168.0.151
  password: *****
  name: gridTesla
- type: template
  template: tesla-powerwall
  usage: pv
  host: 192.168.0.151
  password: *****
  name: pvTesla
- type: template
  template: tesla-powerwall
  usage: battery
  host: 192.168.0.151
  password: *****
  name: batteryTesla

vehicles:
- type: template
  template: mg2mqtt
  title: MG4
  user: *****
  vin: *****
  capacity: 77
  timeout: 1h
  name: mg4

chargers:
- type: template
  template: easee
  user: *****
  password: *****
  charger: EC3SUNEP
  timeout: 10s
  name: wallboxEasee

loadpoints:
- title: Garage
  charger: wallboxEasee
  vehicle: mg4
  mode: pv
  soc:
    poll:
      mode: connected
      interval: 30m
    estimate: true

site:
  title: Forstern
  meters:
    grid: gridTesla
    pv:
    - pvTesla
    battery:
    - batteryTesla
  residualPower: 100

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.295
  feedin:
    type: fixed
    price: 0.082
  planner:
    type: grünstromindex
    zip: 85659

mqtt:
  broker: 192.168.0.91:1887
  topic: evcc
  clientid: evcc
  user: *****
  password: *****

Version: `0.128.4`
```

### Log details

```text
see attached log
[evcc-20240726-094945-trace.log](https://github.com/user-attachments/files/16388971/evcc-20240726-094945-trace.log)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.128.4

GrimmiMeloni, 2024-07-26:
Bonus hint on your config: You should rather go with a 20s timeout for the Easee. The entire client <-> cloud API <-> charger roundtrip regularly takes more than 10s.

GrimmiMeloni, 2024-07-26:
I checked the provided log, evcc is sending the commands as expected, so nothing in the logic wrong. However, looking at the explicit values, I am wondering if your system has been properly setup by your electrician. Your circuit settings indicate 1000A per phase as maximum current. Please correct me, but to me that looks like an utterly unrealistic value.

My working theory is, that as we use the max Current to switch phases, but the value is unrealistic, the command gets ignored by the charger.

Can you please double check what the max amps show for the circuit (either on the easee web portal or via the app)?

```
[easee ] TRACE 2024/07/26 09:48:22 GET https://api.easee.com/api/sites/715783/circuits/676200/settings
[easee ] TRACE 2024/07/26 09:48:23 {"allowOfflineMaxCircuitCurrent":false,"equalizerId":null,"masterChargerId":"EC3SUNEP","useDynamicMaster":false,"dynamicCircuitCurrentP1":1000.0,"dynamicCircuitCurrentP2":1000.0,"dynamicCircuitCurrentP3":1000.0,"enableIdleCurrent":false,"maxCircuitCurrentP1":1000.0,"maxCircuitCurrentP2":1000.0,"maxCircuitCurrentP3":1000.0,"offlineMaxCircuitCurrentP1":25,"offlineMaxCircuitCurrentP2":25,"offlineMaxCircuitCurrentP3":25}
[easee ] TRACE 2024/07/26 09:48:23 POST https://api.easee.com/api/sites/715783/circuits/676200/settings
[easee ] TRACE 2024/07/26 09:48:23 {"dynamicCircuitCurrentP1":1000,"dynamicCircuitCurrentP2":1000,"dynamicCircuitCurrentP3":1000}
[lp-1  ] DEBUG 2024/07/26 09:48:23 switched phases: 3p
```

GrimmiMeloni, 2024-07-26:
@premultiply just quick sanity check - are there installations that big (i.e. 1000A) or can we already consider this a misconfiguration on site/circuit level?

premultiply, 2024-07-26:
As far as I know germany the limit for grid connections to the lowest grid layer ("Niederspannungsnetz") is ~250 kVA using 3 ~400A fuses at the transformer.

eviw, 2024-07-26:
@GrimmiMeloni Can you please double check what the max amps show for the circuit (either on the easee web portal or via the app)?:
Yes, it really shows 1000A:
[image]

eviw, 2024-07-26:
What would be the proper setting value?

eviw, 2024-07-26:
By the way, switching from 1p->3p works, if I switch EVCC mode to OFF and back to PV. I have attached this log as well.
But I see a log entry for mqtt syntax error:
**[mqtt  ] ERROR 2024/07/26 13:22:18 set evcc/loadpoints/1/phases/set: strconv.Atoi: parsing "NaN": invalid syntax**
[image]

GrimmiMeloni, 2024-07-27:
> What would be the proper setting value?

I am no electrician, so I don‘t know for sure. I _think_ this depends on the fuses on the circuit your wallbox is attached to. Whoever installed the charger for you should know.

closed 2024-08-08 by github-actions[bot]