evcc-io/evcc #15064: Charger Weidmüller AC-Smart

tobi119, 2024-07-26:
### Describe the bug

Verbindung des Fahrzeugs wird nicht erkannt.
Von der Wallbox gelesene Werte (S. Log) werden fehlerhaft interpretiert

[weidmüller] TRACE 2024/07/25 15:42:47 modbus: send 26 fd 00 00 00 06 ff 03 01 2d 00 01
[weidmüller] TRACE 2024/07/25 15:42:47 modbus: recv 26 fd 00 00 00 05 ff 03 02 00 43
[lp-1 ] ERROR 2024/07/25 15:42:47 charger status: invalid status:

Die Ladeleistung hat einen unrealistischen Wert

[image]

Vermutung: die Reihenfolge der gelesenen Bytes /Words wird fehlerhaft interpretiert.
In meinem Smart Home (ioBroker) muss ich die Werte für Fahrzeugstatus als "String(Little Endian)" und die Leistung als "Unsigned32 (Big Endian Word Wrap)" definieren

Log:
[image]

Mein Versuch der Aufschlüsselung:
[image]

### Steps to reproduce

1. Neustart des Programms
2. Log zeigt [lp-1 ] ERROR 2024/07/26 08:49:20 charger status: invalid status: 
3. gelesene daten s.o. sind für mich richtig; der Fahrzeugstatus "A" bzw "B" bzw "C" ist mit den Werten 41 / 42 /43 hex im Modbus Recv richtig enthalten

Gene erstelle ich weitere Tests.
Da ich nicht in GO programmieren kann, kann ich leider keine eigenen Änderungen vornehmen.

### Configuration details

```yaml
chargers:
  - name: my_charger
    type: template
    template: weidm  ller
    host: 192.168.178.25 # IP-Adresse oder Hostname 

Weidmüller AC-Smart CH-W-S-A11-P-V
```

### Log details

```text
[main ] INFO 2024/07/26 09:20:00 evcc 0.128.4
[main ] INFO 2024/07/26 09:20:00 using config file: /etc/evcc.yaml
[db ] INFO 2024/07/26 09:20:00 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2024/07/26 09:20:00 listening at :7070
[weidmüller] TRACE 2024/07/26 09:20:00 modbus: send 00 01 00 00 00 06 ff 03 02 be 00 01
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: recv 00 01 00 00 00 05 ff 03 02 00 08
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: send 00 02 00 00 00 06 ff 03 01 c9 00 02
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: recv 00 02 00 00 00 07 ff 03 04 00 00 00 00
[lp-1 ] DEBUG 2024/07/26 09:20:01 charge total import: 0.000kWh
[site ] INFO 2024/07/26 09:20:01 site config:
[site ] INFO 2024/07/26 09:20:01 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/07/26 09:20:01 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2024/07/26 09:20:01 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2024/07/26 09:20:01 pv 2: power ✓ energy ✓ currents ✗
[site ] INFO 2024/07/26 09:20:01 pv 3: power ✓ energy ✓ currents ✗
[site ] INFO 2024/07/26 09:20:01 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[lp-1 ] INFO 2024/07/26 09:20:01 loadpoint 1:
[lp-1 ] INFO 2024/07/26 09:20:01 mode: pv
[lp-1 ] INFO 2024/07/26 09:20:01 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2024/07/26 09:20:01 meters: charge ✓
[lp-1 ] INFO 2024/07/26 09:20:01 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2024/07/26 09:20:01 set buffer soc: 90
[site ] DEBUG 2024/07/26 09:20:01 set buffer start soc: 90
[site ] DEBUG 2024/07/26 09:20:01 set priority soc: 75
[site ] DEBUG 2024/07/26 09:20:01 set residual power: 100
[lp-1 ] DEBUG 2024/07/26 09:20:01 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/26 09:20:01 phase timer inactive
[lp-1 ] DEBUG 2024/07/26 09:20:01 pv timer inactive
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: send 00 03 00 00 00 06 ff 03 02 be 00 01
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: recv 00 03 00 00 00 05 ff 03 02 00 08
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: send 00 04 00 00 00 09 ff 10 02 be 00 01 02 00 3c
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: recv 00 04 00 00 00 06 ff 10 02 be 00 01
[lp-1 ] DEBUG 2024/07/26 09:20:01 max charge current: 6A
[site ] DEBUG 2024/07/26 09:20:01 ----
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: send 00 05 00 00 00 06 ff 03 01 a2 00 02
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: recv 00 05 00 00 00 07 ff 03 04 00 00 00 00
[lp-1 ] DEBUG 2024/07/26 09:20:01 charge power: 0W
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: send 00 06 00 00 00 06 ff 03 01 96 00 06
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: recv 00 06 00 00 00 0f ff 03 0c 00 00 00 00 00 00 00 00 00 00 00 00
[lp-1 ] DEBUG 2024/07/26 09:20:01 charge currents: [0 0 0]A
[site ] DEBUG 2024/07/26 09:20:01 pv power: 7512W
[site ] DEBUG 2024/07/26 09:20:01 battery soc: 68%
[site ] DEBUG 2024/07/26 09:20:01 battery power: -211W
[site ] DEBUG 2024/07/26 09:20:01 grid meter: -6392W
[site ] DEBUG 2024/07/26 09:20:01 grid currents: [-12.6 -7.4 -7.47]A
[site ] DEBUG 2024/07/26 09:20:01 battery has priority at soc 68% (< 75%)
[site ] DEBUG 2024/07/26 09:20:01 site power: -6292W
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: send 00 07 00 00 00 06 ff 03 01 c9 00 02
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: recv 00 07 00 00 00 07 ff 03 04 00 00 00 00
[lp-1 ] DEBUG 2024/07/26 09:20:01 charge total import: 0.000kWh
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: send 00 08 00 00 00 06 ff 03 01 2d 00 01
[weidmüller] TRACE 2024/07/26 09:20:01 modbus: recv 00 08 00 00 00 05 ff 03 02 00 42
[lp-1 ] ERROR 2024/07/26 09:20:01 charger status: invalid status:
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.128.4

premultiply, 2024-07-26:
Bitte mal die Ausgabe vom `evcc charger --log trace --diagnose` hier einstellen.

tobi119, 2024-07-26:
root@EVCC:~# evcc charger --log trace --diagnose 
[main  ] INFO 2024/07/26 11:33:00 evcc 0.128.4
[main  ] INFO 2024/07/26 11:33:00 using config file: /etc/evcc.yaml
[db    ] INFO 2024/07/26 11:33:00 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/07/26 11:33:00 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/07/26 11:33:00 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/07/26 11:33:00 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/07/26 11:33:00 SELECT * FROM `settings` 15 <nil>
[db    ] TRACE 2024/07/26 11:33:01 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/07/26 11:33:01 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/07/26 11:33:01 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/07/26 11:33:01 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/07/26 11:33:01 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/07/26 11:33:01 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/07/26 11:33:01 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/07/26 11:33:01 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/07/26 11:33:01 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT        fk_configs_details      %") -1 <nil>
[db    ] TRACE 2024/07/26 11:33:01 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/07/26 11:33:01 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT        fk_devices_details      %") -1 <nil>
[db    ] TRACE 2024/07/26 11:33:01 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "% device_id        %") -1 <nil>
[db    ] TRACE 2024/07/26 11:33:01 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: send 00 01 00 00 00 06 ff 03 02 be 00 01
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: recv 00 01 00 00 00 05 ff 03 02 00 08
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: send 00 02 00 00 00 06 ff 03 01 a2 00 02
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: recv 00 02 00 00 00 07 ff 03 04 00 00 00 00
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: send 00 03 00 00 00 06 ff 03 01 c9 00 02
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: recv 00 03 00 00 00 07 ff 03 04 00 00 00 00
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: send 00 04 00 00 00 06 ff 03 01 96 00 06
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: recv 00 04 00 00 00 0f ff 03 0c 00 00 00 00 00 00 00 00 00 00 00 00
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: send 00 05 00 00 00 06 ff 03 01 90 00 06
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: recv 00 05 00 00 00 0f ff 03 0c 89 05 00 03 00 00 00 00 00 00 00 00
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: send 00 06 00 00 00 06 ff 03 01 2d 00 01
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: recv 00 06 00 00 00 05 ff 03 02 00 42
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: send 00 07 00 00 00 06 ff 03 02 be 00 01
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: recv 00 07 00 00 00 05 ff 03 02 00 08
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: send 00 08 00 00 00 06 ff 03 03 e8 00 0b
[weidmüller] TRACE 2024/07/26 11:33:01 modbus: recv 00 08 00 00 00 19 ff 03 16 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Power:          0W
Energy:         0.0kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 2.3e+06V 0V 0V
Charge status:  invalid status: 
Enabled:        true
Identifier:     <none>

PR #15071 (2024-07-27): Weidmüller: Fix uint32 and status encoding
Fix #15064
closed 2024-07-27 by andig