evcc-io/evcc #15077: Add GoodWe DT Inverters Modbus Protocol

Lennart-O, 2024-07-26:
The GoodWe DT (also called SDT) series inverters feature a different Modbus protocol than other GoodWe inverters. In order to get proper data from those inverters, the protocol needs to be implemented. I'm looking forward to someone willing to do this.

I already found a rep from someone which documents the protocol:
https://github.com/MiG-41/Modbus-GoodWe-DT

If someone points me on where and how to implement this properly, I'm willing to assist and do as much as possible myself.

Thanks.

andig, 2024-07-27:
Hi @Lennart-O we already have Goodwe Modbus ET/EH/BH/BT, see https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/goodwe-hybrid.yaml. Is DT different? Then you'll need to implement a similar template. 

For testing- before bundling as template- do something like this:

```yaml
meters:
- name: goodwe-dt-pv
  type: custom
  power:
    source: modbus
    uri: <ip:502>
    register: # manual non-sunspec register configuration
      address: 36025 # MTTotalActivepower Pmeter
      type: holding
      decode: int32
    scale: -1
...
```

Lennart-O, 2024-07-28:
Yes the DT series is different. The values I get with the Goodwe Modbus ET/EH/BH/BT config are rubbish.

So the suggestion you made can just be added to my general EVCC config file for testing? I saw that you apparently also added a template for the SDT/DT series already. Or should I wait for that one to be released?

andig, 2024-07-28:
Just try the PR?

Lennart-O, 2024-07-28:
What is the PR?

As the DT series only has RS485, I'm planning to use a RS485 - Modbus TCP/IP converter. That should work right? Or does that change somehing?

closed 2024-07-28 by andig