evcc-io/evcc #15085: Multiple battery inputs on Goodwe Hybrid Inverters

UserChristian, 2024-07-27:
**Is your feature request related to a problem? Please describe.**
I can connect the second battery chanel on my hybrid inverter.
Unfortunately, I destroyed one channel in an accident, so I am now using the second input. 
This now works perfectly in the inverter, but EVCC does not recognize it.

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
It looks as if only the first battery input is being queried via EVCC.

**Describe the solution you'd like**
It would be nice if the corresponding battery input could already be selected in the YAML configuration file during configuration.
I would suggest 
1 = Only battery input 1
2 = Only battery input 2
3 = Both battery inputs  (That would be nice, but not absolutely necessary for me)
In the future, several batteries will probably be used more often, as the capacities and loads with one battery quickly reach their limits.

**Describe alternatives you've considered**
Theoretically, I can also change the registers directly, but my correction will probably be overwritten with the next update.

**Additional context**
I found this document on the protocol on the Internet and the registers for the individual batteries are described from page 155 onwards.
https://www.photovoltaikforum.com/core/attachment/434409-bess-modbus-protocol-map-v1-1-20231030-pdf/

andig, 2024-07-27:
Battery 2 needs be added to the `goodwe-hybrid` template. Happy to take a PR.

VolkerK62, 2024-07-29:
In dem Dokument gibt es nur "Battery1 power (35182)"
Darüberhinaus gibt es nur noch "Battery Total Power (10414)"

@UserChristian  
probier doch mal bitte aus, ob es mit Register 10414 funktioniert.

UserChristian, 2024-07-29:
Hallo @VolkerK62,
bin leider ein blutiger Anfänger, aber gerne teste ich das aus.
Kannst du mir sagen, welche Datei ich an welcher Stelle anpassen soll?
Wie gesagt, bei mir läuft das in einem Docker Container auf meiner Synology.
Dort gibt es auch ein Terminalfenster, wo ich einen Befehl absetzen kann, nur leider weiß ich nicht mit welchen Befehl und auch nicht den Pfad bzw. Dateinamen. 
Gerne stelle ich auch meinen PC für den Remotezugriff bereit.

VolkerK62, 2024-07-30:
du müsstest in der yaml dein battery meter durch ein custom meter ersetzen.
Hier ein Beispiel wie es (ungefähr) aussehen soll https://github.com/evcc-io/evcc/issues/15077#issuecomment-2254140222
Details kannst du dir aus dem Template holen https://github.com/evcc-io/evcc/blob/a49188712a6292879a1e248ffe9c2880e5cf7df9/templates/definition/meter/goodwe-hybrid.yaml#L73
dabei dann für `power` das andere Register nutzen.

andig, 2024-08-10:
Looks like nobody wants to grab this, closing.

UserChristian, 2024-08-11:
Hallo @andig,
Entschuldigung für meine etwas verspätete Antwort, allerdings musste ich mich erst mit Modbus Poll vertraut machen.
Also die Abfrage der Adresse 10414 liefert kein Ergebnis, daraufhin habe ich versucht die korrekten Adressen für die Batterie 2 zu finden.
Folgendes habe ich herausgefunden:
35264 = Leistung Batterie 2
39005 = SOC Batterie 2
Falls noch andere Angaben benötigt werden, einfach kurz Bescheid geben.
Danke

andig, 2024-08-11:
Das hilft. Dann brauchts allerdings einen Wahlschalter, denn die socs darf man bei unterschiedlicher Größe nicht addieren.

UserChristian, 2024-08-11:
Mit den Kapazitätsangaben der beiden Batterien könnte man den SOC wieder berechnen.

UserChristian, 2024-08-11:
Mir würde ein einfacher Wahlschalter völlig ausreichen.

andig, 2024-08-11:
Ich sehe gerade, dass wir auch Energy haben:

```yaml
  power:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register: # manual non-sunspec register configuration
      address: 35182 # Battery1 Power
      type: holding
      decode: int32
  soc:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register: # manual non-sunspec register configuration
      address: 37007 # SOC
      type: holding
      decode: uint16
  energy:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register: # manual non-sunspec register configuration
      address: 35209 # Energy-Battery Discharge
      type: holding
      decode: uint32
    scale: 0.1
```

Findest Du den auch noch raus?

UserChristian, 2024-08-11:
Hab das Register kontrolliert und dieses scheint für beide Batterien zu gelten.

Hier der Auszug aus dem Protokoll
[image]

Daten aus meiner App:
[image]

Und noch die Modbus-Abfrage:
[image]

andig, 2024-08-11:
Das ist schlecht. Denn dann würden bei 2 Konfigurierten Batterien die Energie 2x gezählt :(. Ich hab keine gute Idee, wie das zu lösen wäre. Es sei denn, wir liessen es bei Batterie 2 dann weg.

UserChristian, 2024-08-11:
Vermutlich habe ich das Script nicht ganz verstanden, ich würde behaupten die Energie menge wir immer pauschal abgefragt, denn diese ändert sich nicht über die Konfiguration 1/2 bzw. 1+2 nicht.
Bei der Leistung ist es eine Addition bzw. Wechsel der beiden Werte.  Der SOC ist sicherlich in der 1+2 Variante ein Problem, denn dieser lässt sich nur mit Kapazitätsangaben berechnen.

andig, 2024-08-11:
...und genau deshalb müssten die beiden Batterien einzeln konfiguriert werden. Dabei darf die Energie dann aber nur 1x berücksichtigt werden. 

>  Ich hab keine gute Idee, wie das zu lösen wäre. Es sei denn, wir liessen es bei Batterie 2 die Energie dann weg.

UserChristian, 2024-08-11:
Vermutlich habe ich das Script noch nicht ganz verstanden, aber die Energiemenge ist doch unabhängig von der Batteriekonfiguratiin.
Leistung in der 1+2 Variante ist nur eine Addition.
Der SOC ist nur in der 1/2 Variante direkt nutzbar, denn für die 1+2 Variante würde man die Kapazität der einzelnen Batterien noch benötigen.

UserChristian, 2024-08-11:
.. so müsste es funktionieren.

andig, 2024-08-11:
Mach gerne einen PR. Das Template zu ändern ist nicht so kompliziert.

andig, 2024-08-12:
Einfach die entstprechende Template-Datei in Github bearbeiten- das mündet dann am Ende im PR.

closed 2024-08-12 by andig

UserChristian, 2024-08-12:
Vielen Dank @andig

UserChristian, 2024-08-17:
Hallo @andig, darf ich dir noch eine Anfängerfrage stellen?
Der Pull Request ist nun erstellt und abgeschlossen und nun würde ich diesen auch gerne nutzen. Nun habe ich erwartet, dass die Änderung über eine neue Version des Docker Containers kommt, aber irgendwie steht das kein Update an.

[image]

Also mir ist nun leider nicht klar, wie das Update in meinen Docker Container kommt?
Sag schon mal Danke für die Unterstützung.