evcc-io/evcc #15087: 1p/3p Erkennung Amperfied connect.solar

maatinh, 2024-07-27:
### Describe the bug

If the Amperfied connect.solar is restarted (Firmware-Update, Power-loss,...) it is using 3p for charging. When evcc is configured with automatic 1p/3p and was previously charging at 1p it looks like it is "assuming" the WB is also still using 1p and starts charging when there is PV energy. However, the WB starts charging with 3p and uses much more power so evcc stops charging. 
After delay/guardduration the same starts again. evcc is not changing the charger back to 1p.

### Steps to reproduce

1. Charge with 1p
2. Disconnect Car
3. Restart WB or externally change it to use 3p
4. Connect Car
5. evcc tries to charge at 1p and stops and restarts, ...
...

### Configuration details

```yaml
...
loadpoints:
  - title: Parkplatz
    circuit: leitung_garage
    charger: connect.solar
    vehicle: Zoe
    mode: pv
    soc:
      poll:
          mode: charging
          interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    phases: 0
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: -1200 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: -100 # maximum import power (W)
    guardDuration: 2m # switch charger contactor not more often than this (default 5m)
...
```

### Log details

```text
[lp-1  ] INFO 2024/07/27 12:04:50 car connected
[lp-1  ] INFO 2024/07/27 12:06:50 start charging ->
[lp-1  ] INFO 2024/07/27 12:07:50 stop charging <-
[lp-1  ] INFO 2024/07/27 12:09:50 start charging ->
[lp-1  ] INFO 2024/07/27 12:10:50 stop charging <-
[lp-1  ] INFO 2024/07/27 12:12:50 start charging ->
[lp-1  ] INFO 2024/07/27 12:14:50 stop charging <-
[lp-1  ] INFO 2024/07/27 12:15:50 start charging ->
[lp-1  ] WARN 2024/07/27 12:16:50 charger logic error: phases mismatch (got 3 measured, expected 1)
[lp-1  ] INFO 2024/07/27 12:17:50 stop charging <-
[lp-1  ] INFO 2024/07/27 12:24:50 start charging ->
[lp-1  ] INFO 2024/07/27 12:25:50 stop charging <-
[lp-1  ] INFO 2024/07/27 12:32:50 start charging ->
[lp-1  ] WARN 2024/07/27 12:33:50 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2024/07/27 12:33:50 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2024/07/27 12:33:50 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] INFO 2024/07/27 12:34:50 stop charging <-
[lp-1  ] WARN 2024/07/27 12:34:54 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2024/07/27 12:34:54 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2024/07/27 12:34:54 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2024/07/27 12:35:50 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2024/07/27 12:35:50 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2024/07/27 12:35:50 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] INFO 2024/07/27 12:36:50 start charging ->
[lp-1  ] WARN 2024/07/27 12:36:50 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2024/07/27 12:36:50 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2024/07/27 12:36:50 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] INFO 2024/07/27 12:37:50 stop charging <-
[lp-1  ] WARN 2024/07/27 12:37:59 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2024/07/27 12:37:59 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2024/07/27 12:37:59 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2024/07/27 12:38:50 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2024/07/27 12:38:50 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2024/07/27 12:38:50 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] INFO 2024/07/27 13:04:50 start charging ->
```

### What type of operating system are you running?

Docker container

### Version

0.128.4

andig, 2024-07-27:
Das Log ist etwas merkwürdig, denn eigentlich sollte das passieren:

```go
				if chargerPhases, err = pg.GetPhases(); err == nil {
					if chargerPhases > 0 && chargerPhases != phases {
						lp.log.WARN.Printf("charger logic error: phases mismatch (got %d, expected %d)", chargerPhases, phases)
						lp.setPhases(chargerPhases)
					}
				} else {
					if errors.Is(err, api.ErrNotAvailable) {
						return nil
					}
					return fmt.Errorf("charger get phases: %w", err)
				}
```

evcc merkt sich intern 3p und sollte dann wieder auf 1p schalten. Es braucht bitte mal ein vollständiges Log inkl. WB auf trace um rauszufinden warum das nicht der Fall ist.

maatinh, 2024-07-27:
Ich muss mir das nochmal genauer anschauen. Da gerade nicht mehr ausreichend Sonne scheint, wollte ich es mit min+pv reproduzieren, das scheint aber nicht zu klappen.
Ich vermute auch, dass die Ursache woanders liegt: Ich stelle in evcc den Charger als 1p charger ein, weil ich das Problem habe, dass minCurrent für 3p höher sein müsste, damit er überhaupt lädt. Setze ich minCurrent auf 13A damit er mit 3P lädt, würde ich bei wenig Sonne, wenn er mit 1P laden soll einiges verschenken, weil er nicht bis auf 6A runter gehen kann.
Somit stand der Charger in evcc wahrscheinlich auf 1p, der Charger ist neu gestartet und lief mit 3p. Evcc erkennt das dann auch, stellt aber nicht auf 1p zurück weil evcc ja nicht für automatischen wechsel konfiguriert ist.
Macht das Sinn?

Ich habe in der evcc.yml lp-1: trace gestellt, sehe aber keinen trace output. Mache ich etwas falsch?

andig, 2024-07-27:
> Evcc erkennt das dann auch, stellt aber nicht auf 1p zurück weil evcc ja nicht für automatischen wechsel konfiguriert ist.
Macht das Sinn?

Ja, das kann gut sein. Aber warum überhaupt erst auf 1p stellen? evcc macht das doch wunderbar von alleine. Ich mache mal zu bis Du das Problem klarer fassen kannst.

closed 2024-07-27 by andig

maatinh, 2024-07-29:
Ich muss auf 1p einstellen, weil ich wie gesagt phasenspzifische minCurrent-Settings bräuchte. Mit 7A minCurrent würde evcc auf 3p@7A umschalten - das ist aber zu wenig für den Zoe. 1p@7A funktioniert allerdings.

Eine Frage noch: Wenn evcc erkennt, dass die WB mit 3phasen läuft, das angeschlossene Auto aber nur mit 1p konfiguriert ist, würde evcc dann den Charger auf 1p einstellen?