evcc-io/evcc #15089: Can't connect to the Renault API

carygravel, 2024-07-27:
### Describe the bug

I've included my Zoe using the `renault` template in the `evcc.yml`, using the VIN, username and password which work with the MyRenault app, but evcc simply says `The vehicle was not reachable`

### Steps to reproduce

1. Configure as below
2. Start
3. See the logs as below
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ccd2cb451e024d70612b78afe35c738f8fd3d548679673b1a17cecfba8dc8201

interval: 30s # control cycle interval

sponsortoken: xxx

meters:
- type: template
  template: senec-home 
  usage: grid  
  host: 192.168.0.xx
  schema: https  
  name: grid1
- type: template
  template: senec-home 
  usage: pv  
  host: 192.168.0.xx
  schema: https  
  name: pv2
- type: template
  template: senec-home 
  usage: battery  
  host: 192.168.0.xx
  schema: https  
  name: battery3

chargers:
- type: template
  template: abl 
  id: 1  
  host: 192.168.0.xx 
  port: 8899  
  modbus: rs485tcpip  
  timeout: 0  
  name: wallbox4

vehicles:
- type: template
  template: renault
  title: Zoe
  vin: xxxxx
  capacity: 52
  name: ev1
  user: xxxxx
  password: xxxxx

loadpoints:
- title: Garage
  charger: wallbox4
  vehicle: ev1
  mode: now # or pv|minpy|off

site:
  title: My home
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```
```

### Log details

```text
[main  ] INFO 2024/07/27 17:00:14 evcc 0.128.4
[main  ] INFO 2024/07/27 17:00:14 using config file: /etc/evcc.yaml
[db    ] INFO 2024/07/27 17:00:14 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] ERROR 2024/07/27 17:00:16 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'renault': Get "https://accounts.eu1.gigya.com/accounts.login?apiKey=3_7PLksOyBRkHv126x5WhHb-5pqC1qFR8pQjxSeLB6nhAnPERTUlwnYoznHSxwX668&loginID=xxxx&password=xxxx": dial tcp 0.0.0.0:443: connect: connection refused
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'renault': Get "https://accounts.eu1.gigya.com/accounts.login?apiKey=3_7PLksOyBRkHv126x5WhHb-5pqC1qFR8pQjxSeLB6nhAnPERTUlwnYoznHSxwX668&loginID=xxxx&password=xxxx": dial tcp 0.0.0.0:443: connect: connection refused
Capacity: 52.0kWh
Features: [Offline Retryable]
```

### What type of operating system are you running?

Linux

### Version

0.128.4

StefanSchoof, 2024-07-27:
> dial tcp 0.0.0.0:443

The IP looks wrong. Does DNS works without any problem for you? Do have a DNS blocker like pihole, that can block this IP.

carygravel, 2024-07-27:
Thanks. That was it.

closed 2024-07-27 by carygravel