evcc-io/evcc #15090: Feature request: UI interval shorter than regulation interval (+ average/trend)

dglaude, 2024-07-27:
Borrowing this great idea _Originally posted by @MarkusMit in https://github.com/evcc-io/evcc/discussions/14955_:

"Would it be possible to have the UI update more frequently than the configured control cycle intervall (`interval`)?"

I really love the bar UI, more than some graph like "Power Flow+" or other graph I made in Home Assistant, but I would like it to refresh more quickly than the regulation interval. Right now this is just for me eye pleasure and to stare at the evolution. What could be interesting is a countdown for the next regulation decision, so that it is clear on when the decision was taken.

Now sometime, especially when cooking, the electricity usage change multiple time per minute. Depending on when you take the measure, you could believe 1.5 kW is available, or 4 kW is consumed. For my setup, this could mean charging at 6A (1 phase 230V) or stopping the charge (if in PV mode), and the only criteria is when the measurement is taken, so it is a bit random.

So I am wondering if evcc should not average the value taken at shorter interval and use that information at the regulation interval. So this is where it fit with the UI refresh I am talking about. I don't know the best math for that moving average, it could even estimate the trend and guess where things are going.

I understand that the regulation interval should not be too short, either to avoid oscillating or changing the car/charger setting too frequently, but maybe with averaging it is less of an issue. You also discuss about latency but maybe evcc could measure/learn that latency. I don't know how knowledge the latency can be part of the regulation formula and improve the decision... but by measuring more frequently, evcc would be able to know that latency.

It feel like a lot of math, and I could hardly suggest the right mean or formula, but that is something that can be experimented (and I would love the UI part).

andig, 2024-07-27:
> Now sometime, especially when cooking, the electricity usage change multiple time per minute. Depending on when you take the measure, you could believe 1.5 kW is available, or 4 kW is consumed.

That sounds like an argument for load averaging over the grid meter, not for multiple intervals? Did you try the `movingaverage` meter type?

dglaude, 2024-07-27:
`movingaverage` seems to work, but dinner time is finished, so I will have to check later. ;-)

I was more considering the use of the average value for decision and not for display. There is still the issue that there is only one interval, the same for UI and for regulation. If you do an average, you may want more sample than the frequency at witch you change the setting.

In current code, it is like having:
`measurement_interval = regulation_interval = interval`

I am suggesting to separate those two concepts to permit things.

PS: There are other mean than the movingaverage, things where the recent past has more weight.

andig, 2024-07-27:
The moving average meter has exponential valuation. Seems this has been answered?

closed 2024-07-30 by andig

dglaude, 2024-08-01:
The `average meter` is considered as the real meter, so the things displayed in the UI do not make sense anymore:
* Sometime my house disappear and only the car seems to consume (obviously not possible)
* If a big consumer is stopped, it stay visible

So while it smooth things for the estimator, it also give false visual information in the UI. I think this is why I wanted to have the interval to be different...
For me it is important to have real, accurate, recently refreshed value visible, but the estimator could use average or prediction based on recent evolution to take it's decisions.

PS: I know this is closed. ;-)

premultiply, 2024-08-02:
The estimator uses energy only.
There isn't anything smoother.

andig, 2024-08-02:
There are no plans to change this right now, especially since this affects visuals only.