evcc-io/evcc #15095: Hoymiles HMS-800W-2T Template hinzufügen

ZaphodB72, 2024-07-28:
Der HMS-800W-2T hat ein eigenes WLAN-Modul, über das er auch zB im Homeassistant eingebunden werden kann. Per Modbus ist er nicht erreichbar.
Ein Template um die Web-API abzugreifen und den Hoymiles zu EVCC hinzufügen zu können, wäre schön.

Ich entschuldige mich schonmal dafür, wenn ich hier nicht die richtige Form für so eine Anfrage eingehalten habe, ist mein erstes Feature-Request auf Git...

andig, 2024-07-28:
Gibts Informationen, _wie_ das API aussieht? Sonst würde es sich anbieten, das einfach aus HA über ein custom meter zu integrieren. Es macht eigentlich wenig Sinn, jedes Gerät in evcc nachzubauen, das in HA schon funktioniert.

ZaphodB72, 2024-07-28:
Ich finde keine Infos über die API, in Homeassistant funktioniert es aber einwandfrei.
Kannst du das mit dem custom meter genauer erklären?

SolarPower2024, 2024-07-28:
Am einfachsten ist es wenn du in Home Assistant eine Automatisierung erstellst die per mqtt die Werte vom Hoymiles schickt und in evcc erstellst du ein pv Meter mit der source "mqtt"

ZaphodB72, 2024-07-28:
MQTT läuft und der Hoymiles schickt seine Werte da bereits rein, ich habe auch eine Entität für die aktuelle Leistung: sensor.hoymiles_gateway_solarh_2929650_real_power
Aber wie erstelle ich jetzt daraus das zusätzliche PV-Meter?

SolarPower2024, 2024-07-28:
Dann ist der Weg nicht mehr weit.

https://docs.evcc.io/docs/devices/meters#manuell

Ein manuelles Meter anlegen und die source als mqtt festlegen. Topic eintragen und schon läuft es.