evcc-io/evcc #15101: BMW iX3 und Alfen Eve warten auf Verbindung

Danman800, 2024-07-28:
### Describe the bug

Ich habe EVCC konfiguriert wie in der Anleitung und den Tipps zur Alfen EVe beschrieben. Leider lädt mein BMW iX3 jedoch nicht. Die Verbindung scheint zu stehen, Akku % wird korrekt angezeigt. Nur startet das Laden nicht.

[image]

### Steps to reproduce

1. Start EVCC
2. BMW iX3 an Alfen Eve anschließen
3. warten.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: f85d05c7e7bd5e7937202603275b0dbf497f5fc63fde66cbbe998e3a6f5cd44c

interval: 30s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoiRGFubWFuODAwIiwiZXhwIjoxODE2Mzc2NDAwLCJpYXQiOjE3MjE3Njg0MDAsInNyYyI6ImdoIn0.VSokUje6wAy-eIRTGmgLXGCELbyRVlgEP55QZT0HLHA

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.178.85  
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 30s  
  name: grid1
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.178.85  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 30s  
  name: pv2
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.178.85  
  port: 502  
  usage: battery  
  modbus: tcpip  
  timeout: 30s  
  name: battery3

chargers:
- type: template
  template: alfen 
  id: 1  
  host: 192.168.178.68  
[... 976 lines cut ...]
[site  ] DEBUG 2024/07/28 14:12:59 ----
[lp-1  ] DEBUG 2024/07/28 14:13:00 charge power: 0W
[lp-1  ] DEBUG 2024/07/28 14:13:00 charge currents: [0 0 0]A
[site  ] DEBUG 2024/07/28 14:13:00 pv power: 5100W
[site  ] DEBUG 2024/07/28 14:13:00 battery soc: 100%
[site  ] DEBUG 2024/07/28 14:13:00 battery power: 0W
[site  ] DEBUG 2024/07/28 14:13:00 grid meter: -5938W
[site  ] DEBUG 2024/07/28 14:13:00 grid powers: [-2104 -1623 -2209]W
[site  ] DEBUG 2024/07/28 14:13:00 grid currents: [-6.6 -5.3 -7.2]A
[site  ] DEBUG 2024/07/28 14:13:00 site power: -5938W
[lp-1  ] DEBUG 2024/07/28 14:13:00 charge total import: 1882.603kWh
[lp-1  ] DEBUG 2024/07/28 14:13:00 charger status: B
[lp-1  ] DEBUG 2024/07/28 14:13:01 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/28 14:13:01 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/28 14:13:01 pv charge current: 8.61A = 0A + 8.61A (-5938W @ 3p)
[lp-1  ] DEBUG 2024/07/28 14:13:01 max charge current: 8.61A
```

### What type of operating system are you running?

Linux

### Version

0.128.4

Danman800, 2024-07-28:
Korrektur, ist WIndows, nicht Linux!

mfuchs1984, 2024-07-28:
Im Log sieht es so aus als würde das Fahrzeug die Ladung nicht starten. Oben rechts im Screenshot sieht man auch eine Fehlermeldung (rotes Dreieck). Ist ggf. ein Ladelimit im Auto eingestellt?

Danman800, 2024-07-28:
> Im Log sieht es so aus als würde das Fahrzeug die Ladung nicht starten. Oben rechts im Screenshot sieht man auch eine Fehlermeldung (rotes Dreieck). Ist ggf. ein Ladelimit im Auto eingestellt?

Nein. Ich bekomme ab und zu diese Meldung:

Alfen WB: wake-up vehicle: unexpected status: 403 (Forbidden)

mfuchs1984, 2024-07-28:
Übrigens steht oben dein Benutzername und Passwort für den BMW drin, würde ich an deiner Stelle ändern.

Danman800, 2024-07-28:
> Übrigens steht oben dein Benutzername und Passwort für den BMW drin, würde ich an deiner Stelle ändern.

Danke, korrigiert.

mfuchs1984, 2024-07-28:
Das Aufwecken des Fahrzeugs über die API wird meines Wissens nach nur verwendet wenn der Ladevorgang nicht von alleine startet. Konntest du ohne evcc erfolgreich mit der Wallbox laden? Was passiert wenn du absteckst, wieder absteckst und versuchst, im Schnellmodus zu laden?

Danman800, 2024-07-28:
> Das Aufwecken des Fahrzeugs über die API wird meines Wissens nach nur verwendet wenn der Ladevorgang nicht von alleine startet. Konntest du ohne evcc erfolgreich mit der Wallbox laden? Was passiert wenn du absteckst, wieder absteckst und versuchst, im Schnellmodus zu laden?

Ja

mfuchs1984, 2024-07-28:
Ok, mit der alten kenne ich mich nicht aus, aber ich denke die nächste Frage hier wird die nach einem trace log der Wallbox und Fahrzeugkommunikation sein.

Danman800, 2024-07-28:
> Ok, mit der Alfen kenne ich mich nicht aus, aber ich denke die nächste Frage hier wird die nach einem trace log der Wallbox und Fahrzeugkommunikation sein.
> 
> Vorausgesetzt das hier wurde alles erfolgreich umgesetzt
> 
> "Die "Active load balancing" Lizenz wird benötigt um die Wallbox via Modbus extern zu steuern. In den Einstellungen muss "Active Load Balancing" aktiviert und "Energy Management System" als Data Source ausgewählt werden. Es wird empfohlen "ValidityTime" (Menu "TCP/IP EMS") auf 300s einzustellen. Falls die "Double"-Box verwendet wird müssen beide Ladepunkte getrennt voneinander hinzugefügt werden. Der erste Port (oder einzelne Port) ist unter ID 1 zugänglich, der zweite unter ID 2."

Ja, das ist alles so eingestellt.

mfuchs1984, 2024-07-28:
Letzte Idee meinerseits: RFID Identifikation aktiv? Ansonsten helfen wohl nur die trace logs und ein Alfen Kenner.

Danman800, 2024-07-28:
Also es klappt jetzt so in etwa, ist noch hekelig, aber zumindest tut sich was. Der Grund war (vermutlich) folgender:

Wenn man keinen Smart Meter hat, muss man im ACE alles, was irgendwas mit Solar Charging zu tun hat abschalten trotz Einstellung EMS. Ansonsten scheint es so zu sein, dass das Lastmanagement der Alfen EVCC übersteuert. Im einzelnen habe ich folgendes abgeschalten bzw. ausgewählt:

Charging profiles => Override charging profiles (direct start)
Solar Charging => Charging Mode OFF (ich vermute wenn man da ein Profil auswählt übersteuert das EVCC)

Noch ne andere Frage: Gibt es eine Empfehlung für die Einstellung Connectivity => Heartbeat? Da steht aktuell 900s.