evcc-io/evcc #15111: Keine Ladevorgang in der App & es wird nur mit 4,1 kWh geladen

DragoWob, 2024-07-28:
### Describe the bug

Leider sehe ich in der App weder einen Ladevorgang noch mein Fahrzeug, dass es lädt. Es wird leider auch nur mit 4,1 kWh geladen obwohl mehr Strom an der PV ist

Modus ist PV
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Meters:
- Type: Template 
  Template: huawei-dongle-powersensor
  Id: 1
  Host: xxxxx
  Port: 502
  Usage: Grid
  Modbus: tcpip
  Timeout: 60s
  Name: grid1
-Type: Template
  Template: Hinweis-dongle-powersensor
  Id: 1
  Host: xxxx
  Port: 502
  Usage: pv
  Modbus: tcpip
  Timeout: 60s
  Name: pv2
-Type: Template
  Template: Hinweis-dongle-powersensor
  Id: 1
  Host: xxxx
  Port: 502
  Usage: Battery
  Modbus: tcpip
  Timeout: 60s
  Name: battery3

Chargers:
-Type: Template
  Template: ellipro 
  Ski: xxxxx
  Name: wallbox5

Vehicles:
- Type: Template
  Template: id
  Titel: passathybrid
  User: xxxx
  Password: xxxx
  Vin: xxxx
  Capacity: 22
  Timeout: 60s
  Name: ev4

Loadpoints:
- Title: garage
  Enable:
    Delay: 1m
    Threshold: -2000
  Disable:
    Delay: 5m
    Threshold: 2000
  Charger: wallbox5
  Mode: pv 

Site:
  Title: SweetHome
  Meters:
    Grid: grid1
    Pv: - pv2
    Battery: - Battery3
```

### Log details

```text
Keine Warnungen oder errors
```

### What type of operating system are you running?

Linux

### Version

V0.128.4

andig, 2024-07-28:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-07-28 by andig

DragoWob, 2024-07-28:
Here a detailed log
[evcc-20240728-181904-debug.log](https://github.com/user-attachments/files/16404482/evcc-20240728-181904-debug.log)

StefanSchoof, 2024-07-28:
4,1 kw ist die Notlandung der Elli Box, wenn die Box gepairt ist, aber die Verbindung unterbrochen wurde. Das kann passieren, wenn die Firmware der Box abstürzt oder die Netzwerk Verbindung unterbrochen wird. Um die Anzahl der Abstürze der Box zu verringern ist es wichtig evcc nur zu starten oder zu aktualisieren, wenn kein Auto mit der Wallbox verbunden ist.

DragoWob, 2024-07-28:
Kann es an der gelben Leuchte an der Wallbox beim Haus liegen?
Habe schon die Box ausgemacht. Evcc gestoppt. Die Box neu gestartet. Evcc neu gestartet, aber leider kein Erfolg. 
auch keine Ahnung, wie ich das gelbe Signal an der wallbox weg bekomme