evcc-io/evcc #15112: Keine Ladevorgänge in der App & kein Laden über 4,1 kWh

DragoWob, 2024-07-28:
### Describe the bug

Ich kann leider keine Ladevorgänge in der App sehen, obwohl der Wagen lädt.
Leider wird der Wagen auch nicht erkannt.
Seit Evcc lädt er nur noch mit 4,1 kWh anstelle mit Max 11 egal wie viel Strom an der PV anliegt.

evcc läuft im PV Modus (aber auch bei den anderen ist es nicht anders)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Meters:
- Type: Template 
  Template: huawei-dongle-powersensor
  Id: 1
  Host: xxxxx
  Port: 502
  Usage: Grid
  Modbus: tcpip
  Timeout: 60s
  Name: grid1
-Type: Template
  Template: Hinweis-dongle-powersensor
  Id: 1
  Host: xxxx
  Port: 502
  Usage: pv
  Modbus: tcpip
  Timeout: 60s
  Name: pv2
-Type: Template
  Template: Hinweis-dongle-powersensor
  Id: 1
  Host: xxxx
  Port: 502
  Usage: Battery
  Modbus: tcpip
  Timeout: 60s
  Name: battery3

Chargers:
-Type: Template
  Template: ellipro 
  Ski: xxxxx
  Name: wallbox5

Vehicles:
- Type: Template
  Template: id
  Titel: passathybrid
  User: xxxx
  Password: xxxx
  Vin: xxxx
  Capacity: 22
  Timeout: 60s
  Name: ev4

Loadpoints:
- Title: garage
  Enable:
    Delay: 1m
    Threshold: -2000
  Disable:
    Delay: 5m
    Threshold: 2000
  Charger: wallbox5
  Mode: pv 

Site:
  Title: SweetHome
  Meters:
    Grid: grid1
    Pv: - pv2
    Battery: - Battery3
```

### Log details

```text
Logs kommen im Kommentar
```

### What type of operating system are you running?

Linux

### Version

V0.128.4

DragoWob, 2024-07-28:
[evcc-20240728-181904-debug.log](https://github.com/user-attachments/files/16404556/evcc-20240728-181904-debug.log)

DragoWob, 2024-07-28:
[evcc-20240728-184721-trace.log](https://github.com/user-attachments/files/16404557/evcc-20240728-184721-trace.log)

horstvogel, 2024-07-28:
Moin, da steht doch im Log bei Charge Power immer 0 Watt?

horstvogel, 2024-07-28:
[image]

https://docs.evcc.io/docs/devices/chargers#elli

DragoWob, 2024-07-28:
Er lädt ja dennoch.
Das heißt ich brauch ein weiteres Gerät?

StefanSchoof, 2024-07-28:
Bitte keine zwei Stellen mit dem gleichen Thema. Würde vorschlagen, dass wir unter #15115 weiter machen.

closed 2024-07-28 by andig