evcc-io/evcc #15117: Zwei BMWs werden nur zufällig richtig erkannt

Frintrop, 2024-07-28:
### Describe the bug

Wir haben zwei Wallboxen Commander 2 und zwei BMWs als Vehicles mit VIN und unterschiedlichen BMW Accounts konfiguriert. Wenn beide gleichzeitig angeschlossen sind, ist die Erkennung eher willkürlich, entweder werden sie vertauscht erkannt oder eins als Gastfahrzeug. Wird der BMW ix alleine angeschlossen, scheint die Erkennung zuverlässiger zu sein.

### Steps to reproduce

2 BMWs gleichzeitig anschließen.

### Configuration details

```yaml
vehicles:
  - name: bmwix
    type: template
    template: bmw
    title: BMW iX M60
    user: 
    password: 
    vin: 
    capacity: 115 
    mode: pv
    priority: 10
    #cache: 12m
  - name: bmw3
    type: template
    template: bmw
    title: BMW 3er Touring
    user: 
    password:
    vin:
    capacity: 12 
    mode: pv
    priority: 9
```

### Log details

```text
Wie heisst der Logger für den Trace Level ?
```

### What type of operating system are you running?

Linux

### Version

0.129

andig, 2024-07-28:
>  Wenn beide gleichzeitig angeschlossen sind, ist die Erkennung eher willkürlich, entweder werden sie vertauscht erkannt oder eins als Gastfahrzeug

Wie soll das auch gehen? Works as designed.

closed 2024-07-28 by andig

Frintrop, 2024-07-28:
Ich dachte über die VIN oder die Batteriekapazität ? @andig