evcc-io/evcc #15122: UI Experimental - vehicles: Eintrag bei "RFID identifiers" unter "advanced settings" führt zu Crash beim Starten von evcc

An-Wi, 2024-07-29:
### Describe the bug

Ich habe 3 Fahrzeuge angelegt:
**Ariya** und **Gast-NFC** über die yaml. 
**Gast**-Fahrzeug über UI.
[image]

Sobald ich über die Oberfläche bei **Gast** eine RFID-Nummer, die ich zuvor beim Gast-NFC in der yaml hinterlegt hatte, dort eintrage und anschließend den erforderlichen Neustart ausführe, kommt folgende Fehlermeldung:
`cannot create vehicle 'db:1': cannot create vehicle type 'template': template: eebus.tpl:7:3: executing "eebus.tpl" at <include "vehicle-identify" .>: error calling include: template: vehicle-identify.tpl:19:10: executing "vehicle-identify" at <.identifiers>: range can't iterate over '[N-04038101c00303-card2]' `

[image]

Die evcc.db habe ich dann aus einer Sicherung wiederherstellen müssen damit evcc wieder fehlerfrei startet.

### Steps to reproduce

1. Konfiguration öffnen
2. Fahrzeug bearbeiten - Advanced settings
3. RFID eintragen
[image]
4. Prüfen & speichern
5. Notwendigen Neustart durchführen

### Configuration details

```yaml
...

vehicles:
  - name: ev1
    type: template
    #template: nissan 
    template: nissan-ariya
    title: Ariya  
    user: ***  
    password: ***
    vin: JN1***88
    capacity: 87  # kWh
    cache: 5m   #default: 15m
    identifiers:
    - N-0402fb013d4903-card1

    # onIdentify: # set defaults when vehicle is identified
    #   mode: minpv # enable PV-charging when vehicle is identified
    #   minSoc: 20 # immediately charge to 0% regardless of mode unless "off" (disabled)
    #   targetSoc: 90 # limit charge to 90%

  - name: ev2
    title: Gast-NFC
    type: custom
    icon: van
    identifiers:
    - N-04038101c00303-card2

...

loadpoints:
  - title: Einfahrt # display name for UI
    charger: wallbox1 # charger
    # meter: charge # charge meter
    mode: pv   # minpv  # set default charge mode, use "off" to disable by default if charger is publicly available
    vehicle: ev1 # set default vehicle (disables vehicle detection)
    # resetOnDisconnect: true # set defaults when vehicle disconnects
    phases: 0   #3   #0 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    # minCurrent: 6 # minimum charge current (default 6A)
    # maxCurrent: 16 # maximum charge current (default 16A)

...
```

[... 48 lines cut ...]
[script] DEBUG 2024/07/28 23:19:41 /fupcService/pvCurrentPower.bat: 0
[script] DEBUG 2024/07/28 23:19:47 /fupcService/pvCurrentPower.bat: 0
[script] DEBUG 2024/07/28 23:19:52 /fupcService/pvCurrentPower.bat: 0
[script] DEBUG 2024/07/28 23:19:57 /fupcService/pvCurrentPower.bat: 0
[script] DEBUG 2024/07/28 23:20:02 /fupcService/pvCurrentPower.bat: 0
[script] DEBUG 2024/07/28 23:20:07 /fupcService/pvCurrentPower.bat: 0
[script] DEBUG 2024/07/28 23:20:13 /fupcService/pvCurrentPower.bat: 0
[script] DEBUG 2024/07/28 23:20:18 /fupcService/pvCurrentPower.bat: 0
[script] DEBUG 2024/07/28 23:20:23 /fupcService/pvCurrentPower.bat: 0
[script] DEBUG 2024/07/28 23:20:29 /fupcService/pvCurrentPower.bat: 0
[script] DEBUG 2024/07/28 23:20:33 /fupcService/pvCurrentPower.bat: 0
[script] DEBUG 2024/07/28 23:20:39 /fupcService/pvCurrentPower.bat: 0
[script] DEBUG 2024/07/28 23:20:43 /fupcService/pvCurrentPower.bat: 0
[script] DEBUG 2024/07/28 23:20:49 /fupcService/pvCurrentPower.bat: 0
[script] DEBUG 2024/07/28 23:20:54 /fupcService/pvCurrentPower.bat: 0
[main  ] INFO 2024/07/28 23:20:58 evcc was stopped by user. OS should restart the service. Or restart manually.
```

### What type of operating system are you running?

Windows

### Version

0.129.0

andig, 2024-07-29:
> führt zu Crash beim Starten von evcc

Ist im Log nicht zu erkennen?

An-Wi, 2024-07-29:
"Crash" war vielleicht etwas unglücklich formuliert.
-> evcc ist nicht "arbeitsfähig", da keine Fahrzeuge mehr verfügbar sind (siehe 2. Screenshot) - auch nicht die beiden aus der yaml.
[image]

Herbert23, 2024-07-30:
Das Fahrzeug wird mMn vom UI richtig angelegt, allerdings kann das Backend das array als string nicht mehr zu einem Array umwandeln.
Payload der Rest-Anfrage:
[image]
Es wird dann wie folgt aus der DB in NewFromConfig geladen
[image]
Daraus könnte man mit Hilfe von string Bearbeitung dann ein Array machen, wodurch der Fehler gelöst wäre.
Zum Beispiel so in NewFromConfig:
```
	if stringarray, found := cc.Other["identifiers"].(string); found {
		stringarray = strings.ReplaceAll(stringarray, "[", "")
		stringarray = strings.ReplaceAll(stringarray, "]", "")
		array := strings.Split(stringarray, " ")
		cc.Other["identifiers"] = array
	}
```

naltatis, 2024-07-30:
@andig das ist ein Serialisierungsthema in der API. Betrifft Properties from Typ `stringlist`. Aktuell nutzt den nur `identifiers`. Wie wollen wir damit umgehen? Momentan sendet und erwartet die UI ein String-Array. Ich seh hier zwei Optionen:

1. Lesen und Schreiben in `config-details` so anpassen, dass der Slice sauber zu String (De-)Serialisiert wird.
2. UI liest und schreibt nur einen String (kein Array). Es gibt ein Trennzeichen, das in UI und Go identisch ist (bspw. `\n`) und man müsste 'ne Sonderlocke bei der Initialisierung (ähnlich Vorschlag von @Herbert23) bauen.

Mir war das Thema bei der UI Entwicklung schon aufgefallen, hatte da aber vergessen das entsprechende Todo in den PR zu packen.

andig, 2024-07-30:
Wie sieht denn der Request aus? Endpoint erwartet yaml?

naltatis, 2024-07-30:
> Endpoint erwartet yaml?

Nein, bei den Devices wird JSON gesprochen.

```
curl 'http://localhost:7071/api/config/devices/vehicle/79' \
-X 'PUT' \
-H 'Content-Type: application/json' \
--data-binary '{"template":"offline","icon":"car","maxCurrent":"6","minCurrent":"5","phases":"2","priority":"2","title":"My Car","capacity":22,"identifiers":["token1","token2"]}'
```

andig, 2024-07-30:
> Lesen und Schreiben in config-details so anpassen, dass der Slice sauber zu String (De-)Serialisiert wird.

Ich verstehe das Problem noch nicht. Was landet denn im Moment in der DB? Ist die Serialisierung das Problem oder die Deserialisierung?

naltatis, 2024-07-30:
Aktuell landet der die Go String Repräsentation vom Slice in der DB. Das ist auf jeden Fall falsch :D

[image]

andig, 2024-07-30:
> Das ist auf jeden Fall falsch :D

Egtl genau wie erwartet. Jetzt müssen wir nur noch raus finden, warum das nicht deserialisiert.

naltatis, 2024-07-30:
> Egtl genau wie erwartet.

Nein, das, was du da in der Datenbank siehst, sind keine sauber serialisierten Daten. Das ist die Slice-String-Repräsentation von Go. Die ist mehrdeutig. Die Werte `["foo", "bar", "baz"]` und `["foo bar", "baz"]` würden beide gleich als `[foo bar baz]` persistiert.

andig, 2024-07-30:
Dann sind das doch 2 Probleme. Das wäre ja auch YAML das man wieder deserialisieren können sollte. Ich schaus mir an.

andig, 2024-07-31:
Die ConfigDetails sind einfache key/val. Das schlägt bei einem Slice fehl:

```go
// detailsFromMap converts map to device details
func detailsFromMap(config map[string]any) []ConfigDetail {
	res := make([]ConfigDetail, 0, len(config))
	for k, v := range config {
		res = append(res, ConfigDetail{Key: k, Value: fmt.Sprintf("%v", v)})
	}
	return res
}
```

Das Problem hier ist, dass wir JSON als Go codieren um es hinterher wieder als YAML einzulesen :O

andig, 2024-07-31:
Template:

    identifiers: []interface {}{"foo bar"} ([]interface {})

DB:

	identifiers: "[foo bar ]" (string)

Bleibt die Frage, wie wir in der DB ein Slice erkennen und ablegen könnten?

closed 2024-08-03 by naltatis