evcc-io/evcc #15126: Charging Mode "Now" uses 3P even Car is configured with 1P

maatinh, 2024-07-29:
### Describe the bug

If evcc is connected to a 3p charger and configured for automatic phase change it switches to 3P when using "now" charging mode even the vehicle is configured with 1P only.
evcc does not switch back to 1P then.

https://github.com/evcc-io/evcc/issues/15087

### Steps to reproduce

1. Configure evcc for automatic phase change
2. Configure vehicle with 1P
3. Switch from PV to now mode
...

### Configuration details

```yaml
see https://github.com/evcc-io/evcc/issues/15087
```

### Log details

```text
See https://github.com/evcc-io/evcc/issues/15087
```

### What type of operating system are you running?

Docker container

### Version

0.128.4

StefanSchoof, 2024-07-29:
https://docs.evcc.io/en/docs/reference/configuration/vehicles#phases

> The maximum number of phases this vehicle can use (possibly including the charging cable).

If your car can 3p you can not set this value to 1.

maatinh, 2024-07-29:
Da sich minCurrent nicht phasenabhängig einstellen läßt, war das der hier irgendwann mal genannte Workaround: Das Auto zweimal anlegen - einmal mit 1P und einmal mit 3P und jeweils verschiedenen minCurrent-Settings.

andig, 2024-07-29:
`minCurrent` hat nichts mit `now` zu tun. Das funktioniert genau wie erwartet- `fast` lädt so schnell wie möglich.