evcc-io/evcc #15128: OCPP: version 0.129 always charge at full power even using minPV and PV

amtssp, 2024-07-29:
### Describe the bug

With 0.128.4 evcc worked fine when started using minPV - it would charge with around 4.1 kW - but after upgrading to 0.129 it always charge at max output (around 11 kW) no matter what I use (PV, minPV or fast charging).
I use an OCPP charger (Orbis viaris combi+) 

### Steps to reproduce

1. start evcc
2. start charging using minPV
3. evcc start charging at max power 11 kW
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 2b73f895a7b0cf7769a7059e78bd84229ba5a52ebc502d7ce4002bc62cab9da4

interval: 15s # control cycle interval

sponsortoken: XXXXXXXX

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.0.110  
  port:   
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.0.110  
  port:   
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.0.110  
  port:   
  name: battery3
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.0.110  
  port:   
  capacity: 7.5  
  token: 32eb664b-3b9b-42ef-bf7e-cb9b1415eeaf  
  name: battery4

site:
  title: Huset_no2
  meters:
    grid: grid1
    pv:
    - pv2
[... 330 lines cut ...]
[lp-1 ] DEBUG 2024/07/29 14:07:04 !! session: chargeRater.chargedEnergy=5.4 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/07/29 14:07:04 charge total import: 8494.051kWh
[lp-1 ] DEBUG 2024/07/29 14:07:04 charger status: C
[lp-1 ] DEBUG 2024/07/29 14:07:04 charger disable
[lp-1 ] DEBUG 2024/07/29 14:07:04 wake-up timer: stop
[site ] DEBUG 2024/07/29 14:07:06 ----
[lp-1 ] DEBUG 2024/07/29 14:07:06 charge power: 0W
[lp-2 ] DEBUG 2024/07/29 14:07:06 charge power: 0W
[site ] DEBUG 2024/07/29 14:07:06 pv power: 3652W
[site ] DEBUG 2024/07/29 14:07:07 battery soc: 77%
[site ] DEBUG 2024/07/29 14:07:07 battery power: 1854W
[site ] DEBUG 2024/07/29 14:07:07 grid meter: 5733W
[site ] DEBUG 2024/07/29 14:07:07 site power: 7687W
[lp-2 ] DEBUG 2024/07/29 14:07:07 !! session: chargeRater.chargedEnergy=3.5 - chargedAtStartup=0.0
[lp-2 ] DEBUG 2024/07/29 14:07:07 charge total import: 2398.249kWh
[lp-2 ] DEBUG 2024/07/29 14:07:07 charger status: A
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.129.0

premultiply, 2024-07-29:
There were no changes regarding OCPP between 0.128.4 and 0.129.0.

closed 2024-07-29 by premultiply

amtssp, 2024-07-29:
Thank you for your comment.
That made me think what else could be changed between the sessions - and I remembered that I changed "Residual power from 0 to 100.
And bingo when I change this value back to 0 it is again able to charge at 4.1 kW using minPV. (using evcc version 0.129)

However, I don't understand why changing the residual power can result in such dramatic effect? I have a Sonnen Hybrid battery system so it is recommended in the docs to use a value of 100 - but in my case it clearly is not correct?