evcc-io/evcc #15141: MyEnergi Integration Zappi Wallbox 2.1.etc.

Stevin13, 2024-07-29:
Zappi 2.1 provides a good 22kw Wallbox with many PV Charging Options on Board. However, I can't Inegrate DC Sensors because of missing Space and it has no connection to Cars. So EVCC could be powerful combination.

Myenergi also is a hub to connect heating, batteries, etc. So in combination with evcc and car connection it could the most powerful solution on the market.

**Describe the solution you'd like**
Integrationof the Zappi v.2.1 Wallbox and if you like of the other Ecosystem Parts, too. But the wallbox most important, because since the wallbox would be integrated, it could manage the other possible connections in wireless slave mode alone. 

**Describe alternatives you've considered**
I also tried it with the Solution "Clever-PV" but their OCPP is beta and doesn't work. Also everything "smart" like pv Charging would need subscription...

The included Cablesensors do not fit in my Zählerkasten / Meterbox.

**Additional context**
MyEnergi has many connection features and comes with an similar looking dashboard and system, so it would be an impressive combination. Please consider to integrate it.🙏🏻

https://www.myenergi.com/de/zappi/

andig, 2024-07-30:
> I also tried it with the Solution "Clever-PV" but their OCPP is beta and doesn't work. 

What's missing if the Box supports OCPP?

StefanSchoof, 2024-07-30:
Cross ref: https://github.com/evcc-io/evcc/issues/461

Stevin13, 2024-07-30:
It rarely stops and starts but doesn't seem to control the power level according to pv levels.
So it seems that the ocpp on both sides, but mainly the "clever-pv" ocpp has issues.

Zappi has api and ocpp options so it would have many more options together with evcc if you can add. (Thank you for your product in general!)

andig, 2024-07-30:
Repeating the question: did you try ocpp with evcc?

closed 2024-08-02 by andig