evcc-io/evcc #15146: unterschiedliches Verhalten von "evcc -c evcc.yaml" und "systemctl start evcc"

walburgf, 2024-07-30:
### Describe the bug

Im evcc.ayml ist als Loadpoint 3 eine My-PV Thor 9s Heizstab implementiert.
Wenn evcc als Systemdienst gestartet wird, wird der Heizstab nicht aktiviert, obwohl PV Überschuss vorhanden ist.
Beim Start von evcc mit "evcc -c evcc.yaml" wird der Heizstab jedoch aktiviert.
Im Logfile sieht man den Unterschied: beim Start mit "evcc -c" sind die folgenden Einträge enthalten, die beim Systemstart fehlen:
[lp-3 ] DEBUG 2024/07/30 09:32:26 max charge current: 1A
[lp-3 ] DEBUG 2024/07/30 09:32:26 charger enable

Dies bedeutet, dass der Charger beim Starten als Systemdienst nie auf "enable" gesetzt wird, bzw. der "max charge current" nicht gesetzt wird.

Hier die jeweiligen Logfiles des Loadpoints:
[log mit evcc -c evcc.yaml.txt](https://github.com/user-attachments/files/16424209/log.mit.evcc.-c.evcc.yaml.txt)
[log mit evcc-service.txt](https://github.com/user-attachments/files/16424214/log.mit.evcc-service.txt)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc-pve.fritz.box:7070
network:
  schema: http
  host: evcc-pve.fritz.box # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  lp-3: trace
  script: trace
  cache: error

# unique installation id
plant: yyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyy
telemetry: true
sponsortoken: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
interval: 20s # control cycle interval, default is 30s

meters:
- name: grid1
  type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.1.246

- name: BYD_battery
  type: template
  template: sma-hybrid
  usage: battery
  modbus: tcpip
  id: 3
  host: 192.168.1.247
  port: 502

- name: SMA_Hybrid_WR
  type: template
  template: sma-hybrid
  usage: pv
  # Modbus TCP
  modbus: tcpip
  id: 3
  host: 192.168.1.247 # Hostname
  port: 502 # Port 

- name: AC_Thor_9s_Meter
  type: custom
  power:
    source: modbus
    id: 1
    uri: 192.168.1.231:5021  # IP-Adresse oder Hostname 
    timeout: 2s
    register:
      address: 1000
      type: holding # Holding Register lesen
      decode: uint16
  soc:
    source: http
    uri: http://192.168.1.248/data.jsn
    method: GET # default HTTP method
    headers:
      - content-type: application/json
    insecure: false # set to true to trust self-signed certificates
#    jq: "[.temp1, .temp2, .temp3] | map(. | select (. != null)) | min" # parse response json, get the lowest temperature
    jq: "[.temp1, .temp2, .temp3] | map(. | select (. != null)) | add / length" # parse response json, get the median temperature
    scale: 0.1 # floating point factor applied to result, e.g. for kW to W conversion
    timeout: 10s # timeout in golang duration format, see https://golang.org/pkg/time/#ParseDuration

vehicles:
- name: vw.id.4
  title: VW ID.4
  type: template
  template: vw
  user: walburg.frank@gmail.com
  password: Hammer99!
  capacity: 77
  timeout: 10s

- name: Warmwasser
  title: Warmwasser
  type: custom
  icon: waterheater
  phases: 3
  soc:
    source: http
    uri: http://192.168.1.248/data.jsn
    method: GET # default HTTP method
    headers:
      - content-type: application/json
    insecure: false # set to true to trust self-signed certificates
#    jq: "[.temp1, .temp2, .temp3] | map(. | select (. != null)) | min" # parse response json, get the lowest temperature
    jq: "[.temp1, .temp2, .temp3] | map(. | select (. != null)) | add / length" # parse response json, get the median temperature
    scale: 0.1 # floating point factor applied to result, e.g. for kW to W conversion
    timeout: 10s # timeout in golang duration format, see https://golang.org/pkg/time/#ParseDuration

site:

- title: Salierstr
  meters:
    grid: grid1
    pv:
    - SMA_Hybrid_WR
    battery:
    - BYD_battery
#    aux:
#    - AC_Thor_9s_Meter
  residualPower: 100

chargers:

- name: Charger_rechts
  type: template
  template: easee
  user: walburg.frank@gmail.com # Emailadresse
  password: Hammer99! # wie Login für Easee App oder Web Portal (https://easee.cloud)
  charger: EHJZ7EM3
  timeout: 20s # optional

- name: Charger_links
  type: template
  template: easee
  user: walburg.frank@gmail.com # Emailadresse
  password: Hammer99! # wie Login für Easee App oder Web Portal (https://easee.cloud)
  charger: EHA4RSE2
  timeout: 20s # optional

- name: AC_Elwa-2_waterheater
  type: template
  template: ac-elwa-2
  modbus: tcpip
  id: 1
  host: 192.168.1.231
  port: 5021

- name: AC_Thor_9s_waterheater
  type: custom
  icon: waterheater
  power:
    source: modbus
    id: 1
    uri: 192.168.1.231:5021  # IP-Adresse oder Hostname 
    timeout: 2s
    register:
      address: 1000
      type: holding # Holding Register lesen
      decode: uint16
  wakeup:
    source: script
    cmd: /home/evcc/AC_Thor_9s/wakeup.sh
    timeout: 5s
  enabled: # charger enabled state (true/false or 1/0)
    source: script
    cmd: /home/evcc/AC_Thor_9s/get_enabled.sh #0 or 1
    timeout: 5s
  enable: # set charger enabled state (true/false or 0/1)
    source: script
    cmd: /home/evcc/AC_Thor_9s/set_enable.sh ${enable}
    timeout: 5s
  maxcurrent:
    source: script
    cmd: /home/evcc/AC_Thor_9s/set_maxcurrent.sh ${maxcurrent}
    timeout: 5s
  status: # charger status A..F
    source: script
    cmd: /home/evcc/AC_Thor_9s/get_status.sh #B or C always connected
    timeout: 5s
  features: [integrateddevice, heating]

loadpoints:
- title: Garage_rechts # display name for UI
  charger: Charger_rechts # Wallbox rechts Referenz
  vehicle: vw.id.4 # Referenz auf Standardfahrzeug
  mode: pv # charge mode (off, now, minpv, pv)
  priority: 6

- title: Garage_links # display name for UI
  charger: Charger_links  # Wallbox links Referenz 
  vehicle:  # Referenz auf Standardfahrzeug
  mode: pv # charge mode (off, now, minpv, pv)
  priority: 4

- title: AC Thor 9s
  charger: AC_Thor_9s_waterheater
  vehicle: Warmwasser
  mode: pv
  meter: AC_Thor_9s_Meter
  priority: 2
  soc:
    poll:
      mode: charging
      interval: 5m
    estimate: true
  enable:
    threshold: -100
    delay: 10s
  disable:
    threshold: -50
    delay: 60s

modbusproxy:
- port: 5021
  uri: 192.168.1.248:502  # IP des AC Thor 9s
- port: 5022
  uri: 192.168.1.248:502
```

### Log details

```text
siehe oben
```

### What type of operating system are you running?

Linux

### Version

0.129.0

OlliLu, 2024-07-30:
Editier mal alle Benutzerkonten, Benutzeremails und Passworte raus.

andig, 2024-07-30:
Damit wir nicht suchen müssen- wo unterscheidet sich das Verhalten?

walburgf, 2024-07-30:
wie oben geschrieben wird im Fall evcc -c evcc.yaml anscheinend ein "set maxcurrent" getriggert und ein enable=true gesetzt....., beim service jedoch nicht.

In der custom configuration des chargers "AC_Thor_9s_waterheater" werden die Scripte

  enable: # set charger enabled state (true/false or 0/1)
    source: script
    cmd: /home/evcc/AC_Thor_9s/set_enable.sh ${enable}
    timeout: 5s

und 

  maxcurrent:
    source: script
    cmd: /home/evcc/AC_Thor_9s/set_maxcurrent.sh ${maxcurrent}
    timeout: 5s

im ersten Fall "evcc -c" aufgerufen, im Fall "systemctl start evcc" jedoch nicht....

andig, 2024-07-30:
> Editier mal alle Benutzerkonten, Benutzeremails und Passworte raus.

Alle ändern- die sind verbrannt!

andig, 2024-07-30:
> werden die Scripte

Bitte relevante Teile aus den Logfiles mit ausreichend Kontext.

walburgf, 2024-07-30:
Die Logfiles hängen im 1. Eintrag an

StefanSchoof, 2024-07-30:
Bist du dir sicher das evcc.yaml in dem Ordner wo du `evcc -c evcc.yaml` ausführst identisch mit der in `/etc/evcc yaml` (die der Service nutzt) ist?

walburgf, 2024-07-30:
ja, ich hab das extra mehrfach kontrolliert, bevor ich das issue aufgemacht hatte....

walburgf, 2024-07-30:
noch ein Punkt:   wenn ich in der Oberfläche minpv auswähle, startet der Ladevorgang (es wird ein enable=true und maxcurrent=xA getriggert). Wenn ich dann wieder zurückschalte auf pv, wird weiter geladen.....

Wenn ich dann den Service stoppe und wieder starte (mit mode=pv) bleibt der Ladevorgang aus.... -> enable und maxcurrent werden nicht getriggert

andig, 2024-07-30:
> Bitte relevante Teile aus den Logfiles mit ausreichend Kontext.

walburgf, 2024-07-30:
**### logfile (bei evcc -c evcc.yaml:**

[lp-3 ] INFO 2024/07/30 09:31:45 loadpoint 3:
[lp-3 ] INFO 2024/07/30 09:31:45 mode: pv
[lp-3 ] INFO 2024/07/30 09:31:45 charger: power ✓ energy ✗ currents ✗ phases ✗ wakeup ✓
[lp-3 ] INFO 2024/07/30 09:31:45 meters: charge ✓
[lp-3 ] INFO 2024/07/30 09:31:45 charge: power ✓ energy ✗ currents ✗
[lp-3 ] DEBUG 2024/07/30 09:31:45 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-3 ] DEBUG 2024/07/30 09:31:45 phase timer inactive
[lp-3 ] DEBUG 2024/07/30 09:31:45 pv timer inactive
[lp-3 ] INFO 2024/07/30 09:31:45 vehicle updated: unknown -> Warmwasser
[lp-3 ] DEBUG 2024/07/30 09:31:45 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
[lp-3 ] DEBUG 2024/07/30 09:31:45 charge power: 0W
[lp-3 ] DEBUG 2024/07/30 09:32:05 charge power: 0W
[lp-3 ] DEBUG 2024/07/30 09:32:25 charge power: 0W
[lp-3 ] DEBUG 2024/07/30 09:32:25 charger status: B
[lp-3 ] INFO 2024/07/30 09:32:25 car connected
[lp-3 ] DEBUG 2024/07/30 09:32:25 pv timer elapse
[lp-3 ] DEBUG 2024/07/30 09:32:25 pv timer inactive
[lp-3 ] DEBUG 2024/07/30 09:32:26 vehicle soc: 48%
[lp-3 ] DEBUG 2024/07/30 09:32:26 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
[lp-3 ] DEBUG 2024/07/30 09:32:26 pv charge current: 7.55A = 0A + 7.55A (-5212W @ 3p)
[lp-3 ] DEBUG 2024/07/30 09:32:26 site power -5212W <= -100W enable threshold
[lp-3 ] DEBUG 2024/07/30 09:32:26 pv enable in 0s
[lp-3 ] DEBUG 2024/07/30 09:32:26 pv enable timer elapsed
**[lp-3 ] DEBUG 2024/07/30 09:32:26 max charge current: 1A    <<-- hier wird der Charge Vorgang gestartet
[lp-3 ] DEBUG 2024/07/30 09:32:26 charger enable**
[lp-3 ] DEBUG 2024/07/30 09:32:26 wake-up timer: start
[lp-3 ] DEBUG 2024/07/30 09:32:45 charge power: 104W
[lp-3 ] DEBUG 2024/07/30 09:33:05 charge power: 104W
[lp-3 ] DEBUG 2024/07/30 09:33:25 charge power: 105W
[lp-3 ] DEBUG 2024/07/30 09:33:26 charger status: C
[lp-3 ] INFO 2024/07/30 09:33:26 start charging ->
[lp-3 ] DEBUG 2024/07/30 09:33:26 wake-up timer: stop
[lp-3 ] DEBUG 2024/07/30 09:33:26 vehicle soc: 48%
[lp-3 ] DEBUG 2024/07/30 09:33:26 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
[lp-3 ] DEBUG 2024/07/30 09:33:26 pv charge current: 8.42A = 1A + 7.42A (-5119W @ 3p)
[lp-3 ] DEBUG 2024/07/30 09:33:26 pv timer reset
[lp-3 ] DEBUG 2024/07/30 09:33:26 pv timer inactive
[lp-3 ] DEBUG 2024/07/30 09:33:26 max charge current: 8A
[lp-3 ] DEBUG 2024/07/30 09:33:45 charge power: 5335W
[lp-3 ] DEBUG 2024/07/30 09:34:05 charge power: 5339W
[lp-3 ] DEBUG 2024/07/30 09:34:25 charge power: 5346W
[lp-3 ] DEBUG 2024/07/30 09:34:25 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-3 ] DEBUG 2024/07/30 09:34:25 charger status: C
[lp-3 ] DEBUG 2024/07/30 09:34:26 vehicle soc: 48%
[lp-3 ] DEBUG 2024/07/30 09:34:26 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
[lp-3 ] DEBUG 2024/07/30 09:34:26 pv charge current: 7.34A = 8A + -0.659A (455W @ 3p)
[lp-3 ] DEBUG 2024/07/30 09:34:26 max charge current: 7A

### **logfile bei systemctl evcc start**

[lp-3 ] INFO 2024/07/30 09:25:13 loadpoint 3:
[lp-3 ] INFO 2024/07/30 09:25:13 mode: pv
[lp-3 ] INFO 2024/07/30 09:25:13 charger: power ✗ energy ✗ currents ✗ phases ✗ wakeup ✓
[lp-3 ] INFO 2024/07/30 09:25:13 meters: charge ✗
[lp-3 ] DEBUG 2024/07/30 09:25:13 set smart cost limit: 0
[lp-3 ] DEBUG 2024/07/30 09:25:13 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-3 ] DEBUG 2024/07/30 09:25:13 phase timer inactive
[lp-3 ] DEBUG 2024/07/30 09:25:13 pv timer inactive
[lp-3 ] INFO 2024/07/30 09:25:13 vehicle updated: unknown -> Warmwasser
[lp-3 ] DEBUG 2024/07/30 09:25:13 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
[lp-3 ] DEBUG 2024/07/30 09:25:13 charge power: 0W
[lp-3 ] DEBUG 2024/07/30 09:25:33 charge power: 0W
[lp-3 ] DEBUG 2024/07/30 09:25:53 charge power: 0W
[lp-3 ] DEBUG 2024/07/30 09:25:53 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
[lp-3 ] DEBUG 2024/07/30 09:25:53 charger status: B
[lp-3 ] INFO 2024/07/30 09:25:53 car connected
[lp-3 ] DEBUG 2024/07/30 09:25:53 pv timer elapse
[lp-3 ] DEBUG 2024/07/30 09:25:53 pv timer inactive
[... 4 lines cut ...]
[lp-3 ] DEBUG 2024/07/30 09:25:54 site power -4537W <= -100W enable threshold
**[lp-3 ] DEBUG 2024/07/30 09:25:54 pv enable in 0s
[lp-3 ] DEBUG 2024/07/30 09:25:54 pv enable timer elapsed     <<-- jetzt müsste der Charge Vorgang gestartet werden, wird er aber nicht**
[lp-3 ] DEBUG 2024/07/30 09:26:13 charge power: 0W
[lp-3 ] DEBUG 2024/07/30 09:26:33 charge power: 0W
[lp-3 ] DEBUG 2024/07/30 09:26:53 charge power: 0W
[lp-3 ] DEBUG 2024/07/30 09:26:53 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
[lp-3 ] DEBUG 2024/07/30 09:26:53 charger status: B
[lp-3 ] DEBUG 2024/07/30 09:26:54 vehicle soc: 48%
[lp-3 ] DEBUG 2024/07/30 09:26:54 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
[lp-3 ] DEBUG 2024/07/30 09:26:54 pv charge current: 6.26A = 0A + 6.26A (-4316W @ 3p)
[lp-3 ] DEBUG 2024/07/30 09:26:54 site power -4316W <= -100W enable threshold
[lp-3 ] DEBUG 2024/07/30 09:26:54 pv enable in 0s
[lp-3 ] DEBUG 2024/07/30 09:26:54 pv enable timer elapsed
[lp-3 ] DEBUG 2024/07/30 09:27:13 charge power: 0W
[lp-3 ] DEBUG 2024/07/30 09:27:33 charge power: 0W
[lp-3 ] DEBUG 2024/07/30 09:27:53 charge power: 0W
[... cut ...] hysical 0p charger)
[lp-3 ] DEBUG 2024/07/30 09:27:53 charger status: B
[lp-3 ] DEBUG 2024/07/30 09:27:53 vehicle soc: 48%
[lp-3 ] DEBUG 2024/07/30 09:27:53 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
[lp-3 ] DEBUG 2024/07/30 09:27:53 pv charge current: 6.8A = 0A + 6.8A (-4690W @ 3p)
[lp-3 ] DEBUG 2024/07/30 09:27:53 site power -4690W <= -100W enable threshold
[lp-3 ] DEBUG 2024/07/30 09:27:53 pv enable in 0s
[lp-3 ] DEBUG 2024/07/30 09:27:53 pv enable timer elapsed

andig, 2024-07-31:
> pv enable timer elapsed <<-- jetzt müsste der Charge Vorgang gestartet werden, wird er aber nicht

Im Log fehlt das 

> [lp-3 ] DEBUG 2024/07/30 09:32:26 charger enable

Warum das so ist lässt sich nicht sagen da das Log irgendwo in der Mitte anfängt.

Generell hat das Elwa Template ein Problem mit dem Heartbeat. Das muss erstmal gelöst werden, siehe https://github.com/evcc-io/evcc/discussions/12761#discussioncomment-10190302. Ich kann nicht sagen ob das auf Thor auch zutrifft.

walburgf, 2024-07-31:
was meinst Du mit: ... da das Log irgendwo in der Mitte anfängt....?

Die Logs sind vom Start von evcc bis zu dem Zeitpunkt, an dem man sieht, dass im System Fall das Laden starten müsste...
Oder benötigst Du das vollständige Log incl. aller anderen Loadpoints, Site, Scripts, etc... ?

Mit dem Thema Heartbeat hat das Problem aber nichts zu tun.
Ich habe die Zeit im AC Thor (die Steuerung entspricht der Steuerung des Elwa-2, nur kann bis 9kW anstelle von 3kW geheizt werden) auf 600s hochgesetzt und einen Cronjob angelegt, der alle 5 min prüft, ob durch evcc eine maxpower Vorgabe gab. Wenn es so ist, schicke ich diese Vorgabe alle 5 min an den AC Thor....

Das Problem ist, dass evcc den AC Thor nicht initial vom Status B auf den Status C umschaltet.
D.h. es fehlt erste Setzen von "enable" und "maxpower"

andig, 2024-07-31:
> Die Logs sind vom Start

Wirklich? Mir fehlt schon die Ausgabe der evcc Version/Config im Log.

walburgf, 2024-07-31:
ja, wirklich vom Anfang....aber wie gesagt, gefiltert auf den [LP-3]... die evcc-Version steht in {main]
Brauchst Du den kompletten Log?

StefanSchoof, 2024-07-31:
They use different databases

> [db ] INFO 2024/07/31 10:11:37 using sqlite database: /root/.evcc/evcc.db

> [db ] INFO 2024/07/31 10:05:40 using sqlite database: /var/lib/evcc/evcc.db

Are you getting the same behavior, when you set the DB path in your Config?

```yaml
database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db
```

walburgf, 2024-07-31:
This was a good hint.... If I use /var/lib/evcc/evcc.db, the behaviour is the same... It's not working....

How ca I find out the differences in the sqlite-db?

andig, 2024-07-31:
Try

    evcc settings get

with the right DB. I guess you have different `minCurrent` settings (via UI?) and setting leads to evcc not executing an update because it thinks value is already applied. `minCurrent=0` somewhere?

walburgf, 2024-07-31:
I found the reason for the issue.

I counfigured 0.125A as the minimum current, which was stored in the sqlite-DB in /var/lib/evcc.
In the sqlite-DB in /home/root/.evcc 1A was configured as minimum current.
[image]

evcc seem to start a charging session by assigning at first the minimum current, before assigning the calculated maxcurrent.
This minimum current is derived from the minimum current by just using the integer value.
This means, that 0.125A is set to 0A, which leads the charging session to never start.
Setting the minimum to 1A works then of course.....

My proposal would be, that the maxpower value should be a float type, to be able to use currents lower than 1A, as this is possible to be used from the GUI.

walburgf, 2024-07-31:
Probably a new issue should be opened:

Charging session will not be started, when minimum current is set to 0,125A

What do you think?

andig, 2024-07-31:
@naltatis `ac-elwa-2` can only do full amps. 0,125 is not possible- should we prevent this?

walburgf, 2024-07-31:
ac-elwa-2 and AC-Thor and AC-Thor-9s are steering via modbus register with address 1000. Here a Power value is expected which must be integer from 1W to 3kW or 9kW(9s). This means, that 1W would be 1W / 230V = 0.004A
This means, that it's possible to use 0,125A as this is 0,125A * 230V = 28,75W -> 29W.

So probably it would be better if it would be possible to define W(att) in the configuration instead of A(mpere) for heater....
What do you think?

Or it would even be great if you can decide in the configuration, if power = W or current = A should be used.....

andig, 2024-07-31:
> So probably it would be better if it would be possible to define W(att) in the configuration instead of A(mpere) for heater....
What do you think?

That's not available right now...

naltatis, 2024-07-31:
> can only do full amps. 0,125 is not possible- should we prevent this?

We would have to add this info to the device templates. Currently, this does not exist. The options are always the same. This is similar to the issue where people try to use < 6A minCurrent to reduce the minimum charging power of their vehicle/charger. I am not sure if this is worth the extra effort.

closed 2024-07-31 by andig