evcc-io/evcc #15165: Elli Wallbox charge power: not available

Kludi2019, 2024-07-31:
### Describe the bug

Hallo, mit dem neuen nightly Update wird keine Ladeleistung mehr erkannt.

### Steps to reproduce

1.update auf letzte nightly Version 129.0
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 6d0cd584651b8821c9f9c4c885404a80de75220ae4fed45689c3c651a401e207

interval: 10s # control cycle interval

meters:
#- type: template
  #template: sonnenbatterie 
  #usage: grid  
  #host: 192.168.178.33  
  #port: 8080  
  #name: grid1

- type: template
  template: tasmota
  usage: grid
  host: 192.168.178.61
  user: xxxx
  password: xxxxxxxxx
  name: grid1
- type: template
  template: sonnenbatterie
  usage: pv
  host: 192.168.178.33
  port: 8080
  name: pv2
- type: template
  template: sonnenbatterie
  usage: battery
  host: 192.168.178.33
  port: 8080
  name: battery3

- type: template
  name: Mobilladegerat_Pwr
  template: shelly-1pm
  user: xxxx
  password: xxxxx
  usage: charge
  host: 192.168.178.26

chargers:
- type: template
  template: ellipro
  ski: EE38 62DB 0242 7083 xxxx xxxx 
  ip: 192.168.178.74
  name: wallbox

- type: template
  template: shelly
  host: 192.168.178.26
  standbypower: 5
  user: xxxx
  password: xxxxx
  name: Mobilladegeraet

vehicles:
- type: template
  template: fiat
  title: Paulchen
  user: xxxxxx
  password: 'xxxxx'
  vin: xxxxxxx
  capacity: 42
  name: ev4

- type: template
  template: id
  title: Id5
  user: xxxxxx
  password: 'xxxxxx'
  vin: WVGxxxxxxx
  capacity: 77
  timeout: 10s
  name: ev1

loadpoints:
- title: Elli
  charger: wallbox
  mode: off
  
- title: mobil
  charger: Mobilladegeraet
  meter: Mobilladegerat_Pwr
  mode: off
  
mqtt:
  broker: 192.168.178.32:1887
  topic: evcc
  clientid: foo
  user: xxx
  password: 'xxxxxx'

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.368

  feedin:
    type: fixed
    price: 0.082

site:
  - title: Im Schxxxxx
    meters:
      grid: grid1
      pv: pv2
      battery: battery3
    
eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          MHcCAQEEIL0mhWia5hk/EDBVPMXzsK+G2ScD58mnFybVMre1i3WCoAoGCCqGSM49
          AwEHoUQDQgAEb0t9jj7FWNin99dLvJfrqtbryKXcqbHML0erDTZ1uIAIMhrpvtI8
          eeLDq9JIMbQqCvMs1Hpzp9do6qxEJrH8GQ==
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          MIIBhzCCAS2gAwIBAgIBATAKBggqhkjOPQQDAjAzMQswCQYDVQQGEwJERTENMAsG
          A1UEChMERVZDQzEVMBMGA1UEAwwMRVZDQ19IRU1TXzAxMB4XDTIyMTAzMTA5MzQx
          MFoXDTMyMTAyODA5MzQxMFowMzELMAkGA1UEBhMCREUxDTALBgNVBAoTBEVWQ0Mx
          FTATBgNVBAMMDEVWQ0NfSEVNU18wMTBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IA
          BG9LfY4+xVjYp/fXS7yX66rW68il3KmxzC9Hqw02dbiACDIa6b7SPHniw6vSSDG0
          KgrzLNR6c6fXaOqsRCax/BmjMjAwMA8GA1UdEwEB/wQFMAMBAf8wHQYDVR0OBBYE
          FCNlU4HENAR0BV4oWM6B8o8QrM9UMAoGCCqGSM49BAMCA0gAMEUCIQDE9HHTldX/
          N2CSc9SxXT0qsUKYehin6OiebycLNUEfkwIgQIhWCZH1CaNmCLK0ClxEqzrfx2Xo
          ougJGx5M07FNtqE=
          -----END CERTIFICATE-----
```

### Log details

```text
[mqtt ] ERROR 2024/07/31 11:52:22 set evcc/loadpoints/1/mode/set: invalid value: NaN
[lp-1 ] ERROR 2024/07/31 11:53:03 charge power: not available
[lp-1 ] ERROR 2024/07/31 11:53:05 charger disable: no limits available
[lp-1 ] ERROR 2024/07/31 11:53:06 charge power: not available
[lp-1 ] ERROR 2024/07/31 11:53:06 charger disable: no limits available
[lp-1 ] ERROR 2024/07/31 11:53:13 charge power: not available
[lp-1 ] ERROR 2024/07/31 11:53:23 charge power: not available
[lp-1 ] ERROR 2024/07/31 11:53:23 charger disable: no limits available
[lp-1 ] ERROR 2024/07/31 11:53:33 charge power: not available
[lp-1 ] ERROR 2024/07/31 11:53:43 charge power: not available
[lp-1 ] ERROR 2024/07/31 11:53:43 charger disable: no limits available
[lp-1 ] ERROR 2024/07/31 11:53:53 charge power: not available
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.129.0 (1515a67f)

DerAndereAndi, 2024-07-31:
Das Log ist leider nicht aussagekräftig genug. Bitte das Auto abstecken, 30s warten und wieder anstecken und schauen ob es dann funktioniert.

Kludi2019, 2024-07-31:
Sorry, ich muss nochmals ergänzen. Es war das nightly von Vorgestern. Ich habe das ganz aktuelle von vor 1 Std eben nochmals geladen. Damit kommt die Leistung wieder aber es kommen jetzt noch errors:

[mqtt ] ERROR 2024/07/31 12:42:06 set evcc/loadpoints/1/mode/set: invalid value: NaN
[lp-1 ] ERROR 2024/07/31 12:44:26 charger status: timeout
[lp-1 ] ERROR 2024/07/31 12:44:46 charger status: timeout
[lp-1 ] ERROR 2024/07/31 12:45:06 charger status: timeout
[lp-1 ] ERROR 2024/07/31 12:45:26 charger status: timeout
[lp-1 ] ERROR 2024/07/31 12:45:46 charger status: timeout
[lp-1 ] ERROR 2024/07/31 12:46:06 charger status: timeout
[lp-1 ] ERROR 2024/07/31 12:46:26 charger status: timeout

und folgende debug Meldung:

[site ] DEBUG 2024/07/31 12:53:06 ----
[lp-1 ] DEBUG 2024/07/31 12:53:06 charge power: 3595W
[lp-1 ] DEBUG 2024/07/31 12:53:06 charge currents: [15.6 0 0]A
[lp-2 ] DEBUG 2024/07/31 12:53:06 charge power: 0W
[site ] DEBUG 2024/07/31 12:53:06 pv power: 5342W
[site ] DEBUG 2024/07/31 12:53:06 battery soc: 100%
[site ] DEBUG 2024/07/31 12:53:06 battery power: 1W
[site ] DEBUG 2024/07/31 12:53:06 grid meter: -1084W
[site ] DEBUG 2024/07/31 12:53:06 site power: -983W
[lp-1 ] DEBUG 2024/07/31 12:53:06 detected active phases: 1p
[lp-1 ] DEBUG 2024/07/31 12:53:06 !! session: chargeRater.chargedEnergy=0.3 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/07/31 12:53:06 charger status: C
[lp-1 ] DEBUG 2024/07/31 12:53:06 soc estimated: 93.70% (vehicle: 93.00%)
[lp-1 ] DEBUG 2024/07/31 12:53:06 vehicle soc: 94%
[lp-1 ] DEBUG 2024/07/31 12:53:06 vehicle range: 290km
[lp-1 ] DEBUG 2024/07/31 12:53:06 !! active phases: 1p = min(1p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/07/31 12:53:06 pv charge current: 20.3A = 16A + 4.27A (-983W @ 1p)
[eebus ] DEBUG 2024/07/31 12:53:07 Send: d:_i:47859_Elli-Wallbox-2021A0QGB5 notify 224 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/31 12:53:09 Send: d:_i:47859_Elli-Wallbox-2021A0QGB5 notify 225 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/31 12:53:11 Send: d:_i:47859_Elli-Wallbox-2021A0QGB5 notify 226 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/31 12:53:11 Recv: d:_i:47859_Elli-Wallbox-2021A0QGB5:Measurement to Measurement notify 359 MeasurementListData
[eebus ] DEBUG 2024/07/31 12:53:11 ski 136c5261edefb7e69d4b8472f4a564b948529f20 event cem-evcem-DataUpdateCurrentPerPhase
[eebus ] DEBUG 2024/07/31 12:53:13 Send: d:_i:47859_Elli-Wallbox-2021A0QGB5 notify 227 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/07/31 12:53:13 Recv: d:_i:47859_Elli-Wallbox-2021A0QGB5:Measurement to Measurement notify 360 MeasurementListData
[eebus ] DEBUG 2024/07/31 12:53:13 ski 136c5261edefb7e69d4b8472f4a564b948529f20 event cem-evcem-DataUpdateCurrentPerPhase
[eebus ] DEBUG 2024/07/31 12:53:15 Send: d:_i:47859_Elli-Wallbox-2021A0QGB5 notify 228 DeviceDiagnosisHeartbeatData
[site ] DEBUG 2024/07/31 12:53:16 ----

DerAndereAndi, 2024-07-31:
Bitte das Auto abstecken und wieder anstecken. Niemals bei angestecktem Auto ein Update von evcc machen. Denn jedes Mal wenn eine neue Verbindung zur Wallbox hergestellt wird, meldet diese keine korrekten Informationen. Da ist nichts zu machen.

Und falls das nicht hilft, bitte ein komplettes Log file und keinen Ausschnitt. Hier fehlen wieder wichtige Informationen wie es zu diesen Meldungen kommt. Nur dass ein Error erscheint, hilft niemals.

Kludi2019, 2024-07-31:
[evcc-20240731-132701-debug.log](https://github.com/user-attachments/files/16440549/evcc-20240731-132701-debug.log)

Momentan kommt keine neue Fehlermeldung. Das Fahrzeug habe ich beim Update nie angesteckt und hierbei musste ich sogar die Elli neu booten. Kann es sein, dass sich das Ganze nun in Wohlgefallen aufgelöst hat?

Danke für Eure Unterstützung

DerAndereAndi, 2024-07-31:
Ja, das kann gut sein.

closed 2024-07-31 by andig

andig, 2024-08-18:
https://github.com/evcc-io/evcc/releases/tag/0.130.0 contains a refactor of the EEbus charger that might or might not help here.