evcc-io/evcc #15171: Laden starten trotz Einspeisesperre

ZaphodB72, 2024-07-31:
Ich habe eine Einspeisesperre für meine Anlage. Das heißt, wenn kein Auto geladen wird und die Batterie voll ist und Energie ins Netz gehen würde, regelt der Wechselrichter die Module ab.
Wenn ich jetzt mein Auto ans Ladekabel hänge, startet in der Einstellung „PV“ das laden nicht, da EVCC denkt, es sei kein Überschuss vorhanden. Ebenso würde in der Einstellung „min+PV“ nur die Minimalladung abgenommen.
Eine schlaue Lösung für das Problem fällt mir nicht ein… vielleicht müsste es einen Schalter geben, der es erlaubt, für eine gewisse Zeit die Ladung mit voller Geschwindigkeit zu starten. Die Zeit muss so gewählt werden können, dass der Wechselrichter die PV wieder auf das maximal mögliche hochrecken kann… oder es muss eine Zeit gewählt werden können, für die Netzbezug erlaubt wird…
Für bessere Lösungen für dieses Problem bin ich offen…

danke schonmal,
ZAP

goebelmeier, 2024-07-31:
Weshalb ist denn die Einspeisung gesperrt? Es würde ja schon reichen gering einzuspeisen um sichtbar zu machen dass Überschuss da ist. Dann liesse sich das mit `residualPower` und `bufferStartSoc` lösen

ZaphodB72, 2024-08-01:
Das ist ne Vorgabe des Netzbetreibers und soll hier gar nicht Gegenstand der Diskussion sein, da arbeite ich schon an anderer Front dran…
Nur gering einzuspeisen würde doch auch nichts bringen, da EVCC das laden doch erst ab 1,4kW Überschuss starten würde. Also müsste ja schon mindestens das an Einspeisung kommen plus der Verbrauch vom Haus…?

PeterPablo, 2024-08-01:
Würde hier nicth das "welcomecharge"-Feature helfen, dass in jedem Fall einen Ladevorgang vorübergehend startet. Das führt im beschriebenen Szenario zunächst zu Netzbezug und anschließend regelt der Wechselrichter die PV-Erzeugung hoch. In der Folge kann evcc auch im PV-Modus entscheiden wieviel Leistung dem Ladepunkt bereitgestellt werden kann.
Ich habe dieses Feature selbst noch nicth genutzt, deswegen kann ich nicht mit Sicherheit sagen, ob es  von allen Modi (PV, PV+min, now) angesprochen wird.
ref. https://github.com/evcc-io/evcc/pull/14873