evcc-io/evcc #15174: Charging starts with minimum current (6A) although mode "now" ("Schnell"-Modus) is selected

Sumpfdotter, 2024-07-31:
### Describe the bug

I am using evcc with one wallbox (Elli Charger Pro), two vehicles in “mode: now” (fast mode) and without PV: My current use case is controlling the target SOC.

evcc should therefore start charging processes immediately with maximum charging current. This worked in older releases, and the procedure when it was still working was that the vehicle started with 6A (min current, default behavior of Elli Charger Pro when being coupled with an energy manager). A few seconds later, when the vehicle was recognized by evcc, evcc increased to 16A (max current). 

Since a few weeks, starting with a certain 1.128 nightly build, this increase to 16A does not happen anymore. The charge remains at 6A. Workaround: If I change the mode once manually in the frontend (e.g. to “Off” or “min+PV”), and then back to “now” ("Schnell"), then evcc pulls the current up to 16A (max current) as desired.

But sometimes it still works. I have done various tests and I think I can reproduce it so that, charging works after a fresh coupling of wallbox with evcc (16A). However, in everyday operation (already on the second charge process), charging remains at 6A as described. A reboot of the wallbox and/or evcc usually does not help.

If of interest, I can try to identify the version that introduced this behavoir more exactly. Let me know. 

### Steps to reproduce

Precondition:
1. The car (in my case ID.Buzz) is configured via evcc.yaml in "mode: now"
2. Wallbox (in my case Elli Charger Pro in RFID mode) and evcc are connected via LAN (no WiFi)
3. No car is connected to the wallbox
4. Wallbox is off (for happy path) and evcc is down
5. evcc in my case is configured without PV and only a virtual grid meter of type "constant 300W"
6. Does'nt matter if the wallbox internal meter is used or an external ABB B23 (same result)

Steps to reproduce (happy path):
1. Bootup wall box
2. Bootup evcc (in my case connected to elli charger pro via LAN, no WiFi)
3. Enter Elli Charger Pro UI, decouple and re-couple evcc
4. Connect car to wallbox
7. Start charging via RFID tag
==>  evcc is in "Schnell" mode, charging starts and raises to 16A (11kW) --> everything as expected

Steps to reproduce (bad path, can be repeated multiple times):
8. Stop active charging
9. Optinally plug off and an the car from and to the wallbox
10. Start charging via RFID tag
==>  evcc is in "Schnell" mode, BUT charging starts and raises to 6A (4,1kW) only

--> expected behavior: charging starts and raises to 16A (11kW)

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 30s # control cycle interval

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.39
  co2:
    type: grünstromindex
    zip: 11111

vehicles:
- name: up
  title: e!up
  icon: car
  type: template
[... 599 lines cut ...]
[site  ] DEBUG 2024/08/01 00:53:06 site power: 300W
[lp-1  ] DEBUG 2024/08/01 00:53:06 !! session: chargeRater.chargedEnergy=0.5 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/01 00:53:06 charger status: B
[lp-1  ] INFO 2024/08/01 00:53:06 stop charging <-
[lp-1  ] DEBUG 2024/08/01 00:53:06 wake-up timer: start
[lp-1  ] DEBUG 2024/08/01 00:53:06 !! session: chargedEnergy=0.5 > chargedEnergy=0.4
[lp-1  ] DEBUG 2024/08/01 00:53:06 !! session: start=0.000 stop=0.000 chargedEnergy=0.486
[lp-1  ] DEBUG 2024/08/01 00:53:06 vehicle status: B
[lp-1  ] DEBUG 2024/08/01 00:53:06 vehicle soc: 56%
[lp-1  ] DEBUG 2024/08/01 00:53:06 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/08/01 00:53:06 vehicle range: 188km
[eebus ] DEBUG 2024/08/01 00:53:07 Send: d:_i:47859_Elli-Wallbox-22XXXXXXXX notify 272 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/01 00:53:09 Send: d:_i:47859_Elli-Wallbox-22XXXXXXXX notify 273 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/01 00:53:11 Send: d:_i:47859_Elli-Wallbox-22XXXXXXXX notify 274 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/01 00:53:13 Send: d:_i:47859_Elli-Wallbox-22XXXXXXXX notify 275 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/01 00:53:15 Send: d:_i:47859_Elli-Wallbox-22XXXXXXXX notify 276 DeviceDiagnosisHeartbeatData
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.129.0

Sumpfdotter, 2024-07-31:
[evcc-20240801-005317-debug-black.log](https://github.com/user-attachments/files/16448117/evcc-20240801-005317-debug-black.log)
Full log file

Hakr23, 2024-08-01:
I have the same issue with the following setup:

Car: Audi Q4 etron
Wallbox: ABB Terra AC 22kw, connected via Modbus TCP/IP
evcc version: 0.129.0
OS: Ubuntu Server 24.04 LTS

Sumpfdotter, 2024-08-01:
Okay, sorry. Ich vergaß zu erwähnen: Ich hatte ein halbwegs aktuelles Nightly vorher und bin extra zurück auf Stable um das Verhalten dort zu verifizieren. Hab jetzt aber nochmal das aktuellste nightly genommen (0.129.0+1722479088) statt (0.129.0) und nochmal eine ganze Testreihe gemacht. Ich bekomme jetzt offenbar erfolgreiches Schnellladen bei jedem ersten Laden nach einem evcc Reboot. Also ich muss nicht die Kopplung aufheben.

Also Verhalten mit aktuellem Nightly:

1. Reboot evcc,
2. erste Ladung startet mit 6A und fährt dann nach wenigen Sekunden hoch auf 16A. -> wie gewünscht
3. Jede weitere Ladung startet mit 6A und verbleibt da. -> nicht wie gewünscht

Also muss ich für jede Ladung (außer die erste nach Reboot) manuell den Modus einmal wechseln z.B. PV+min und zurück auf Schnell. Nach dem Wechsel zurück auf SChnell fährt er hoch auf 16A.

Im log bekomme ich als Fehlermeldung zwischen zwei Ladevorängen (?) "Carport: charger out of sync: expected disabled, got enabled". Falls relevant kann ich noch mehr Tests machen und ggf. mehr Logfiles. Würde gerne genauer wissen, worauf ich achten könnte, denn die Tests sind immer ein wenig aufwendig für mich.

andig, 2024-08-01:
Bitte noch Logfile fürs nightly.

Sumpfdotter, 2024-08-01:
Okay. Kommt. Das kann ich erst morgen mitschneiden, wenn die Autos wieder SoC eingebüßt haben. Beide Autos sind über Default-Target SoC. In dem Zustand tritt der Fähler nicht auf. :-) Ich habe in einer älteren Version mal den Target-SoC der Autos passend eingestellt (80% und 85%) und nun bekomme ich das nicht mehr verändert. Ich finde das Setting nicht mehr um das mal temporär hochzustellen.

Sumpfdotter, 2024-08-03:
Anbei das Log mit dem aktuellen Nightly 0.129.0+1722651905

[evcc-20240803-072016-debug.log](https://github.com/user-attachments/files/16478192/evcc-20240803-072016-debug.log)

Zur Info: Nach dem Installieren und Restart zeigte die Wallbox im Webfrontend an, dass evcc noch nicht mit ihr verbunden sei. Um Zeit zu gewinnen habe ich manuell im Wallbox-Webfrontend entkoppelt und neu gekoppelt.

Aber: Mir ist noch was aufgefallen!

Es ist ja so, dass die Elliwallbox über eine HEMS-Lampe am Gerät verfügt. Diese leuchtet grün, wenn das HEMS verbunden ist und kein Auto angeschlossen ist. Wenn ich das Auto anschließe springt sie zunächst auf gelb, was bedeutet, dass der Ladestrom reduziert ist (default-verhalten der Box, sie wartet auf die Ladestromvorgabe vom HEMS).

Beim ersten Ladevorgang stöpsele ich das Auto an, die HEMS-Lampe springt auf gelb, aber schon bevor ich den Ladevorgang via RFID-Karte starte, springt die Lampe wieder auf grün. Ich vermute (das konnte ich leider nicht kontrollieren), dass evcc in der Zwischenzeit das Auto identifizieren konnte.

Ab dem zweiten Ladevorgang stöspele ich das Auto an, die HEMS-Lampe bleibt aber auf gelb. Ich habe extra mal länger gewartet und ins evcc-Frontend geschaut: evcc versuchte noch immer, das Auto zu identifizieren (Anzeige Gastfahrzeug). Erst als ich das Laden dann gestartet habe (RFID-Karte) konnte evcc kurze Zeit später das Auto identifizieren. Allerdings wurde dann der Ladestrom eben nicht mehr erhöht, sondern das passierte erst, nachdem ich von "Schnell" auf "min+PV" und "Schnell" zurückgewechselt habe.

Der eine Fehlereintrag im Log ist übrigens neu (neues nightly?). Den hatte ich bisher noch nie, AFAIK.

Sumpfdotter, 2024-08-03:
Habe den Test wiederholt, habe den Kommentar von vor 12 Minuten entsprechend präzisiert.

andig, 2024-08-03:
Textaufgabe. Wo zeigt das Logfile welches Problem?

Sumpfdotter, 2024-08-03:
Echt jetzt? Für mich sind das auch mehrere Stunden Test gewesen und ich hab mir anfangs die Logfiles angeschaut. Wenn Du was gefunden hast, sag es bitte.

andig, 2024-08-03:
> Echt jetzt? 

Ja! Bitte konkret: welches Problem? Und bei welchem Timestamp? Du kannst jetzt nicht erwarten, dass ich mir das selbst ausdenke oder dem Logfile raus suche. Es gibt noch mehr Issues als dieses hier, daher braucht es bitte ein bisschen Kontext.

DerAndereAndi, 2024-08-03:
Allgemeines: Wichtig ist bei einem Log das Verhalten mit Zeitpunkten zu beschreiben, so dass man das im Log nachvollziehen kann. Und dann auch wo ein nicht erwartetes Verhalten auftritt. Vermutungen helfen überhaupt nicht, sondern machen es noch schwieriger das Problem zu verstehen. Vor allem wenn man gar nicht weiss wie das interne Vorgehen ist.

Ein Intervall von 30s ist in der Konfigurationsdatei hinterlegt, d.h. Änderungen werden auch nur alle 30s gemeldet. Was in der Zwischenzeit passiert, ist meist unbekannt!

1. Ladevorgang:
  - 07:10:48: Auto wird angesteckt erkannt, und werden keine Ladeströme gemeldet -> es lädt nicht
  - 07:11:17: Es wird ein Ladestrom von 10777W gemeldet, das Auto lädt also
  - 07:11:48: Der ID. BUZZ wird als angesteckt erkannt und der Lademodus auf "now" gestellt
  - 07:12:17: Es wird ein Ladestrom von 10758W gemeldet
  - ....
  - 07:14:18: Das Auto wird als abgesteckt erkannt

2. Ladevorgang:
  - 07:14:47: Auto wird angesteckt erkannt, und werden keine Ladeströme gemeldet -> es lädt nicht
  - 07:16:17: Es wird ein Ladestrom von 4112W gemeldet, das Auto lädt also
  - 07:16:18: Der ID. BUZZ wird als angesteckt erkannt und der Lademodus auf "now" gestellt
  - 07:16:47: Es wird ein Ladestrom von 4125W gemeldet
  - 07:17:14: Der Lademodus wird auf MinPV umgestellt
  - 07:17:14: Es wird ein Ladestrom von 4125W gemeldet 
  - 07:17:37: Der Lademodus wird auf Now umgestellt
  - 07:17:37: Ein neues Ladelimit wird zur Wallbox gesendet (Wert unbekannt)
  - 07:17:47: Es wird ein Ladestrom von 10730W gemeldet
 
Das Problem scheint folgendes zusein:
- Im 2. Ladevorgang beginnt das Auto mit 4kW zu laden, das erfolgt nicht aufgrund von evcc Vorgaben!
- Als das Fahrzeug erkannt wird, und der Lademodus damit auf "now" gestellt wird, wird damit keine Änderung des Ladestrom auf 11kW durchgeführt.

andig, 2024-08-03:
> Als das Fahrzeug erkannt wird, und der Lademodus damit auf "now" gestellt wird, wird damit keine Änderung des Ladestrom auf 11kW durchgeführt.

Vmtl. ist evcc der Meinung, dass noch 16A eingestellt sind da es zwischendurch auch nichts anderes gesetzt hat. Aufgrund eines Bugs mit dem Lastmanagement haben wir das in https://github.com/evcc-io/evcc/pull/15218 geändert. Vielleicht hilft das ja auch hier? Ich trigger mal ein nightly...

andig, 2024-08-03:
Bitte morgen nochmal testen. Falls Problem noch auftritt bitte Logfile.

Sumpfdotter, 2024-08-03:
> > Echt jetzt?
> 
> Ja! Bitte konkret: welches Problem? Und bei welchem Timestamp? Du kannst jetzt nicht erwarten, dass ich mir das selbst ausdenke oder dem Logfile raus suche. Es gibt noch mehr Issues als dieses hier, daher braucht es bitte ein bisschen Kontext.

@andig, bitte nicht ärgern. Ich will doch auch nur helfen so gut es geht. Ich habe ein Issue gemeldet, was bei jeder Sequenz dieser Art unabhängig von den Timings auftritt. Von daher hatte ich erwartet, dass ihr nur auf die Reihenfolge der Events schauen müsst und nicht auf die Timings. Außerdem habe ich mir vor dem Erstellen des Issues Eure [Anleitung](https://github.com/evcc-io/evcc/blob/4f0739f2bbba0c5ed189517f11af102edc64f394/CONTRIBUTING.md) dazu angeschaut. Dort steht nicht drin, dass ihr immer das tatsächliches und erwartetes Verhalten bezogen auf die Log-Zeit genannt haben wollt. Wenn das wichtig für Euch ist, dann macht es vielleicht Sinn, das dort zu ergänzen. Ich mache das auch gerne, allerdings ist es super schwierig, wenn man draußen im Freien steht, und evcc irgendwo remote werkelt. Ich schlage vor, dass man vielleicht eine Möglichkeit schaffen könnte, die evcc-Zeit im Frontend einzublenden. Dann könnte man mit dem Handy auf die Uhr schauen und die Trigger zu leicht merk-, protokollier- und analysierbaren Zeiten setzen. Das wäre tatsächlich hilfreich (z.B. "Anstecken um xx:xx:00"). Nachdem ich das Issue gefiled hatte, hast Du mich gebeten genau den Test mit den Nightly zu wiederholen. Das habe ich getan und nicht mehr, Du hattest nicht nach mehr gefragt. Woher soll ich also wissen, dass ich Dir das Logfile genauer voranalysieren soll? Deshalb bitte nicht "Textaufgabe" schreiben. Ich glaube wir wollen beide das Gute und das ärgert dann mich, wenn ich keine klare Anweisung bekomme aber trotzdem einen Flame kassiere. :-) Natürlich habt ihr viele Issues und ich will dabei helfen, Euren Aufwand zu minimieren. Aber das Logfile sah jetzt für mich absolut passig aus zu dem, was ich beobachtet und beschrieben hatte. Also müsst ihr glaube ich schon selbst reinschauen, wo genau jetzt der Fehler passiert sein könnte. Also: Nein, ich erwarte nicht, dass Du da selbst im Detail reinschaust, hier ist alles irgendwo freiwillig. Nur wird es auch nicht weitergehen, wenn keiner mal genauer reinschaut. Denn ich habe nix gefunden.

Völlig richtig ist auch: Ich hatte in meiner Beschreibung, dass die Wallbox-Lampe sofort auf grün springt, dann eine falsche Vermutung getroffen, dass dieses Ereignis von evcc getriggert sein müsste. Einfach weil die Anleitung der Wallbox das so vermuten lässt. Ich werde da in Zukunft versuchen präziser sein. Es ist präzise gesprochen so, dass die Wallbox beim ersten Anstecken nach dem evcc-reboot bzw. erneutem Koppeln, aus irgendeinem Trigger heraus (intern oder extern) auf 16A stellt, und zwar noch bevor das Laden gestartet wird, und bei jedem Folge-Ladevorgang stellt sie auf 6A. In jedem Falle verbleibt sie bei diesem Wert, trotz Modus "Schnell".

Danke auch an Dich, @DerAndereAndi, dass Du das Logfile schon durchgegangen bist und die Uhrzeiten herausgeschrieben hast. Aus Zeitgründen konnte ich das nicht vor Dir tun. Ich unterstütze Euch gerne und mache das beim nächsten Mal. Wie gesagt: Ich nehme immer noch an, dass es ein kausales und kein temporales Problem ist, und dass die Timings daher unerheblich sind. Aber das könnt nur ihr wissen.

Den Test mit dem neuen Nightly mache ich schnellstmöglich (hoffentlich morgen). Viele Grüße!

Sumpfdotter, 2024-08-04:
> Bitte morgen nochmal testen. Falls Problem noch auftritt bitte Logfile.
Test wiederholt mit: evcc 0.129.0+1722738220

Hatte dieses Mal den e!up nehmen müssen, der ist 2-phasig, aber sonst kein Unterschied.

Problem tritt noch auf.

Logfile:
[evcc-20240804-140701-debug-black.log](https://github.com/user-attachments/files/16487468/evcc-20240804-140701-debug-black.log)

Logfile-Analyse und Korrespondenz zu ist/soll-Timestamps plane ich heute abend anzufügen.

DerAndereAndi, 2024-08-04:
Das Verhalten ist identisch.

andig, 2024-08-05:
> Problem tritt noch auf.

Welches Problem? Und v.a. wo? Bitte- aus tiefstem Herzen- Logfile mit konkretem Fehler: was war erwartet, was passiert stattdessen, bei welchem Timestamp?
Wieder 10min meiner Zeit weg in der ich mir andere Fehler hätte anschauen können. Es skaliert einfach nicht wenn ich mir die Probleme aus tausenden Zeilen Logfiles selbst raus suchen muss...

Sumpfdotter, 2024-08-05:
> Welches Problem? Und v.a. wo? Bitte- aus tiefstem Herzen- Logfile mit konkretem Fehler: was war erwartet, was passiert stattdessen, bei welchem Timestamp? Wieder 10min meiner Zeit weg in der ich mir [...]

Nach welches Problem wird es wohl es? Es ist immer noch das Problem, das im Issue beschrieben ist. Die Beschreibung hält sich zu 100% an Euer Template und enthält eine klare Beschreibung welches das Ist-Verhalten ist und welches das erwartete Verhalten ist, und wie man es reproduziert. Zitat:

> 10. Start charging via RFID tag
> ==> evcc is in "Schnell" mode, BUT charging starts and raises to 6A (4,1kW) only
> --> expected behavior: charging starts and raises to 16A (11kW)
 
Außerdem hängt wie gefordert das Logfile anbei, schon im ersten Post. Sorry, mein Leben muss auch skalieren, und aus meiner Sicht lässt Du mich nun zum dritten Mal einen Test mit immer neueren Nightlies wiederholen, und jetzt sagst Du mir, Du hast das Problem nicht verstanden, d.h. ich teste zum dritten Mal obwohl es gar keine relevante Änderung in der Software gab? Das skaliert so auch nicht.

Wer soll denn sonst das Logfile lesen, wenn nicht der Entwickler? Das muss ich bei den Nutzern meiner Software doch auch machen. Der Nutzer weiß doch garnicht die interna, die das System da so loggt. Ich kann und werde wie Dir natürlich gerne wie auch von mir versprochen die Timestamps zum Log liefern, da Du sagst, dass Dir das hilft. Die Hilfe bekommst Du! Nur muss ich dazu eben auf meiner Seite "skalieren", und dann kommt das auch.

Nichtsdestotrotz nahm ich natürlich an, Du hättest irgendeinen Fix committed, sonst hättest Du es mich ja nicht zum dritten Mal testen lassen, oder? Und da wollte ich Dir schnelle Rückmeldung geben, ob der Fix was gebracht hat. Wenn es einen gab: Nein, hat er nicht. Das war die Message. Wenn es keinen gab: Warum muss ich dann den Test wiederholen?

Es gibt jetzt schon drei Tests mit drei verschiedenen Versionen, also immer das gleiche Problem. Auch ein anderer User hat es bestätigt. Es gibt auch dank @DerAndereAndi eines Logfile, das mit den Zeiten korreliert wurde. Und ich kann Dir und werde Dir das auch wie versprochen noch für das dritte Logfile liefern. Nur habe ich auch ein Leben und muss das halt irgendwo zeitlich einbauen. Jetzt zum Beispiel.

Sumpfdotter, 2024-08-05:
> Test wiederholt mit: evcc 0.129.0+1722738220
>
> Hatte dieses Mal den e!up nehmen müssen, der ist 2-phasig, aber sonst kein Unterschied.
>
> Problem tritt noch auf.
>
> Logfile: [...]

Vorbedingung:
evcc konfiguriert auf Modus "Schnell", evcc läuft nicht

Korrespondenz Logfile-Zeiten zu Ereignissen:

1. 2024/08/04 13:58:04: evcc gestartet, wallbox mit evcc gekoppelt, Energie-Manager-Lampe an Wallbox ist auf grün
2. 2024/08/04 14:02:00: Auto an Wallbox angeschlossen (e!up, 2 Phasig), Energie-Manager-Lampe springt auf gelb
3. ca. 2024/08/04 14:02:03-14:02:08: Energie-Manager-Lampe an Wallbox springt auf grün
4. 2024/08/04 14:02:15: Laden per RFID-Tag gestartet (Energie-Manager-Lampe an Wallbox bleibt auf grün)
--> Auto lädt wie gewünscht im Schnellmodus mit 2x16A (grüne Energie-Manager-Lampe)

5. 2024/08/04 13:03:00: Ladevorgang gestoppt durch Abziehen des Kabels
6. 2024/08/04 14:04:00: Auto erneut an Wallbox angeschlossen, Energie-Manager-Lampe springt auf gelb
--> Feststellung: Dieses Mal springt die Energie-Manager-Lampe nicht auf grün
8. 2024/08/04 14:04:15: Laden per RFID-Tag gestartet (Energie-Manager-Lampe an Wallbox bleibt auf gelb)
--> TATSÄCHLICHES VERHALTEN: Auto lädt trotz Schnellmodus mit nur 2x6A (gelbe Energie-Manager-Lampe)
--> ERWARTETES VERHALTEN: Auto lädt im Schnellmodus mit 2x16A (grüne Energie-Manager-Lampe)

Sumpfdotter, 2024-08-05:
Ich habe zur besseren Verdaulichkeit mal die beiden Ladevorgänge aus dem Logfile separiert:

Ladevorgang 1 (direkt nach Reboot, Fehler tritt nicht auf):
[evcc-20240804-140701-debug-black-Laden1.log](https://github.com/user-attachments/files/16502828/evcc-20240804-140701-debug-black-Laden1.log)

Ladevorgang 2 (Folgevorgang, Fehler tritt auf):
[evcc-20240804-140701-debug-black-Laden2.log](https://github.com/user-attachments/files/16502843/evcc-20240804-140701-debug-black-Laden2.log)

Das ist das selbe Log, nur freigeschnitten. Wie schon gesagt: Ich erkenne leider keinen wirklichen Unterschied. Vielleicht passt die Vermutung von @andig, dass evcc noch von 16A ausgeht, und daher nichts neues anweist (war nicht meine Vermutung, ich soll ja nichts vermuten ;-)). Will nur sagen: bei älteren Versionen (0.128) war das Fehlverhalten wie gesagt noch nicht aufgetreten, also darf man annehmen, dass evcc was am Verhalten geändert hat.

DerAndereAndi, 2024-08-07:
Danke @Sumpfdotter.

Der Unterschied ist wie folgt

Beim 1. Ladevorgang:
```
[lp-1  ] WARN 2024/08/04 14:02:12 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2024/08/04 14:02:12 charger enable
[eebus ] DEBUG 2024/08/04 14:02:12 Send: d:_i:47859_Elli-Wallbox-22XXXIF write 50 LoadControlLimitListData
[lp-1  ] DEBUG 2024/08/04 14:02:12 max charge current: 16A
```

Beim 2. Ladevorgang:
```
[lp-1  ] WARN 2024/08/04 14:04:13 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2024/08/04 14:04:13 charger enable
```

Es wird beim 2. kein maximaler Ladestrom aktiv gesetzt.

Das Problem bei den Elli Boxen: Sie informieren nicht immer korrekt über den aktuellen Zustand. Normalerweise sollte eine (EEBUS) Wallbox nach der Freigabe den maximalen Ladestrom von sich aus erlauben. Und von außen muss man nur eingreifen wenn man eben diesen nicht haben will.

Was die Box genau meldet, kann ich leider nur mit Logdaten sagen, welche die EEBUS Daten im "trace" level  enthalten.

DerAndereAndi, 2024-08-07:
Ergänzung: Ein weiteres Log mit `trace` level für eebus würde sehr helfen. Brauchst es nicht stückeln. Hauptsache es enthält alles ab dem 1. Anstecken. Danke!

Sumpfdotter, 2024-08-07:
> Danke @Sumpfdotter.
>
> Der Unterschied ist wie folgt
>
> Beim 1. Ladevorgang:
>
> ```
> [lp-1  ] WARN 2024/08/04 14:02:12 charger out of sync: expected disabled, got enabled
> [lp-1  ] DEBUG 2024/08/04 14:02:12 [...]

Das Logfile ist kein Problem, danke Euch @DerAndereAndi und @andig.

[evcc-20240807-164239-trace-black.log](https://github.com/user-attachments/files/16530499/evcc-20240807-164239-trace-black.log)

Verflixt, das Logfile enthält jetzt ja nur noch eebus messages. Ich hoffe, ich bekomme die Zeiten zusammen:
16:37:00(?) Elli abgekoppelt (Über Elli Webfrontend)
16:37:30(?) Elli angekoppelt (Über Elli Webfrontend)
16:38:00 Ladestecker angesteckt
16:39:00 Ladestecker angesteckt
16:39:20 1. Laden gestartet mit RFID-Karte (lädt mit 16A --> erwartet sind 16A)
16:40:20 Ladestecker abgesteckt
16:41:00 Ladestecker angesteckt
16:41:20 2. Laden gestartet mit RFID-Karte (lädt nur mit 6A --> erwartet sind 16A)
16:41:50 auf Modus "min+PV" umgestellt
16:42:20 auf Modus "Schnell" umgestellt (lädt nun mit 16A --> erwartet sind 16A)
16:42:40 Ladestecker abgesteckt

Ich bin mir mit den Zeiten bei den ersten beiden Events nicht mehr sicher, die sind aber vermutlich nicht kriegsentscheidend. Bei den anderen Events bin ich hoffentlich in der Minute nicht verrutscht. Wenn die Zuordnung nicht klappt, oder das so nicht hilft, mache ich den Test gerne noch mal. Lass es mich wissen.

Sumpfdotter, 2024-08-07:
> Die anderen Ausgaben wären auch sehr wichtig. Kommst du da evtl. noch ran? Denn das eine ist was über EEBUS geht/kommt, das andere was evcc damit macht.

Ja, das hab ich befürchtet. Mir fehlt das Bedienwissen. Ich hatte zuerst im yaml auf "trace" gestellt, das hat nichts bewirkt. Wenn in der Webgui von "Trace" gehe und "eebus" auswähle, kommt nur noch eebus. Also würde ich "Trace - alle" machen, aber das könnte dann sehr viel sein? Oder gibt es eine Möglichkeit alle auf Debuglevel zu haben und nur eebus auf trace?

DerAndereAndi, 2024-08-07:
Änderungen an der Konfigurationsdatei erfordern einen Neustart von evcc und das hilft dann auch nur wenn man sich das Log vom Betriebssystem holt.

Daher ist Trace für alle wohl der einfachste Weg.

Sumpfdotter, 2024-08-07:
> Änderungen an der Konfigurationsdatei erfordern einen Neustart von evcc und das hilft dann auch nur wenn man sich das Log vom Betriebssystem holt.
> 
> Daher ist Trace für alle wohl der einfachste Weg.

Ah, das war sehr hilfreich. Weil ich im yaml auf trace umgestellt hatte (und reboot ist klar), konnte ich die Messages jetzt zusammenbauen. Hier sind alle DEBUG-Messages plus "eebus" auf "trace":

[evcc-20240807-164239-trace-evcc-black-classified.log](https://github.com/user-attachments/files/16531526/evcc-20240807-164239-trace-evcc-black-classified.log)

Sumpfdotter, 2024-08-07:
Hab die Zeiten oben korrigiert. Die waren um 1 Minute verrutscht - sorry.

Sumpfdotter, 2024-08-07:
Ich glaube ich hab was. Sehe ich das richtig, dass manchmal "16" **Ampere** gesendet werden und manchmal "76652" **Milliampere**? Ich vermute es müssten "16000" (mA) gesendet werden statt 16 "mA"?

Aug 07 16:38:11 kammerpi evcc[26471]: [lp-1  ] DEBUG 2024/08/07 16:38:11 max charge current: **16A**
Aug 07 16:38:11 kammerpi evcc[26471]: [eebus ] TRACE 2024/08/07 16:38:11 Recv: 055_SKI_ELLI_e73 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-22ELLIF"},{"entity":[1,1]},{"feature":10}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-evcc-01.saw.snueq"},{"entity":[1]},{"feature":7}]},{"msgCounter":135425},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":false},{"value":[{"number":16},{"scale":0}]}],[{"limitId":1},{"isLimitActive":false},{"value":[{"number":16},{"scale":0}]}],[{"limitId":2},{"isLimitActive":false},{"value":[{"number":**16**},{"scale":0}]}]]}]}]]}]}]}}]}

Aug 07 16:41:20 kammerpi evcc[26471]: [lp-1  ] DEBUG 2024/08/07 16:41:20 max charge current: **7.67A**
Aug 07 16:41:20 kammerpi evcc[26471]: [eebus ] TRACE 2024/08/07 16:41:20 Send: 055_SKI_ELLI_e73 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_n:EVCC_HEMS-evcc-01.saw.snueq"},{"entity":[1]},{"feature":7}]},{"addressDestination":[{"device":"d:_i:47859_Elli-Wallbox-22ELLIF"},{"entity":[1,1]},{"feature":10}]},{"msgCounter":160},{"cmdClassifier":"write"},{"ackRequest":true}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":76652},{"scale":-4}]}],[{"limitId":1},{"isLimitActive":true},{"value":[{"number":**76652**},{"scale":-4}]}],[{"limitId":2},{"isLimitActive":true},{"value":[{"number":**76652**},{"scale":-4}]}]]}]}]]}]}]}}]}

Sumpfdotter, 2024-08-07:
16:36:00 EVCC gestartet
16:37:00 Elli abgekoppelt (Über Elli Webfrontend)
16:37:30 Elli angekoppelt (Über Elli Webfrontend)
16:38:00 Ladestecker angesteckt --> Log File 16:38:10 **max charge current: 16A**
16:38:20 1. Laden gestartet mit RFID-Karte (lädt mit 16A --> erwartet sind 16A)
16:39:20 Ladestecker abgesteckt
16:40:00 Ladestecker angesteckt --> **_KEIN_** Log-File Eintrag für "**_max charge current: 16A_**"
16:40:20 2. Laden gestartet mit RFID-Karte (lädt nur mit 6A --> erwartet sind 16A)
16:41:20 auf Modus "min+PV" umgestellt --> Log File **max charge current: 7.67A** & 16:41:40 max charge current: 7.23A
16:41:50 auf Modus "Schnell" umgestellt (lädt nun mit 16A --> erwartet sind 16A) --> Log File **max charge current: 16A**
16:42:20 Ladestecker abgesteckt

DerAndereAndi, 2024-08-07:
Die Beschreibung ist korrekt. Und evcc hat auch keine Chance von den 6A etwas mitzubekommen, weil die Wallbox dies auch nicht mitteilt. Obwohl sie nach den aktuellen Limits nach dem Anstecken direkt gefragt wird.

@andig Bei EEBUS scheint es daher notwendig zu sein, dass evcc das Ladelimit nach jedem Ansteckvorgang auch durchführt.

Sumpfdotter, 2024-08-07:
> Die Beschreibung ist korrekt. Und evcc hat auch keine Chance von den 6A etwas mitzubekommen, weil die Wallbox dies auch nicht mitteilt. Obwohl sie nach den aktuellen Limits nach dem Anstecken direkt [...]

@DerAndereAndi: Beim Write antwortet die Box korrekt mit den Limits:
{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":false},{"value":[{"number":16},{"scale":0}]}],[{"limitId":1},{"isLimitActive":false},{"value":[{"number":16},{"scale":0}]}],[{"limitId":2},{"isLimitActive":false},{"value":[{"number":16},{"scale":0}]}]]}]}]]}]}]}}

Beim den Reads antwortet sie jeweils auch, aber die Limit-Datensätze sind leer:
{"loadControlLimitData":[[{"limitId":0}],[{"limitId":1}],[{"limitId":2}],[{"limitId":3}],[{"limitId":4}],[{"limitId":5}]]}]}]]}]}]}

Wir senden das Read immer kurz nach dem Anstecken (1-2 Sekunden später). Ich kenne EEBUS jetzt leider noch nicht. Was bedeutet denn leer? Bedeutet leer "keine Information" oder bedeutet leer Default = kein Limit.  Ich frage aus folgenden Grund: Wir müssen annehmen, dass die Elli-Box das Limit von selbst reinhaut. Wir wissen aber nicht, wann sie das tut. Wäre es möglich, dass die Wallbox das erst ein paar Sekunden nach unserer Read-Abfrage tun, und wir deshalb das ganze verpassen.

Es gibt noch einen weiteren Ansatz: Laut Doku haut die Wallbox das Limit rein, wenn sie die Verbindung zum Energiemanagement-System verloren hat. Was auch immer die Elli-Kollegen damit meinen. In dem Falle geht die gelbe Lampe an. Und die gelbe Lampe geht beim Anstecken ja tatsächlich auch an. Senden wir beim Anstecken irgendetwas etwas an die Box, was die Box als eine Art Disconnect/Reconnect auffassen könnte?

Ich werde jetzt mal noch eine Hausaufgabe machen: Wenn die gelbe Lampe wirklich zeitnah und korrekt das Vorhandensein des Limits anzeigt (wie im Handbuch beschrieben), dann kann ich anhand der Lampe zeitlich eingrenzen welches Ereignis das ausgelöst hat.

Sumpfdotter, 2024-08-07:
Die Messauflösung ist wegen des evcc-Logs nicht so gut wie ich gehofft hatte. Per Video hab ich auf eine 15tel Sekunde genau die gelbe Lampe gemessen (23:55 Uhr und 25,23 Sekunden gemessen in der Zeit des evcc-Hosts). Ungünstigerweise löst das evcc-Logfile nur auf Sekunden auf.

Die Elliwallbox hat entweder auf eigenen Trigger die gelbe Lampe reingehauen, oder aufgrund eines der EEBUS-Kommunikationsschritte in der fraglichen Sekunde:

Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Recv: d:_i:47859_Elli-Wallbox-22ELLIF:NodeManagement to NodeManagement notify 5505 NodeManagementUseCaseData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Recv: d:_i:47859_Elli-Wallbox-22ELLIF:NodeManagement to NodeManagement notify 5506 NodeManagementDetailedDiscoveryData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 ski 055_SKI_ELLI_e73 event cem-evcc-UseCaseSupportUpdate
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF call 3932 NodeManagementSubscriptionRequestCall
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 ski 055_SKI_ELLI_e73 event cem-opev-UseCaseSupportUpdate
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF call 3933 NodeManagementSubscriptionRequestCall
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 ski 055_SKI_ELLI_e73 event cem-evcem-UseCaseSupportUpdate
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 ski 055_SKI_ELLI_e73 event cem-oscev-UseCaseSupportUpdate
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF read 3935 DeviceClassificationManufacturerData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF call 3934 NodeManagementSubscriptionRequestCall
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF call 3936 NodeManagementBindingRequestCall
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF call 3937 NodeManagementSubscriptionRequestCall
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF call 3938 NodeManagementSubscriptionRequestCall
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF read 3939 ElectricalConnectionDescriptionListData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF read 3940 LoadControlLimitDescriptionListData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF read 3941 MeasurementDescriptionListData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF read 3942 DeviceConfigurationKeyValueDescriptionListData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF read 3943 MeasurementConstraintsListData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF read 3944 ElectricalConnectionParameterDescriptionListData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF call 3945 NodeManagementSubscriptionRequestCall
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 operation is not supported on function loadControlLimitConstraintsListData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF call 3946 NodeManagementSubscriptionRequestCall
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF read 3947 DeviceDiagnosisStateData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF read 3948 MeasurementDescriptionListData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF call 3949 NodeManagementSubscriptionRequestCall
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF read 3950 MeasurementConstraintsListData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF read 3951 ElectricalConnectionParameterDescriptionListData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF read 3952 ElectricalConnectionPermittedValueSetListData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF call 3953 NodeManagementSubscriptionRequestCall
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Recv: d:_i:47859_Elli-Wallbox-22ELLIF:NodeManagement to NodeManagement result 5508 3932 ResultData 0
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF read 3954 IdentificationListData
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 ski 055_SKI_ELLI_e73 event cem-evcc-EvConnected
[... cut ...]
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Recv: d:_i:47859_Elli-Wallbox-22ELLIF:Generic to NodeManagement call 5524 NodeManagementSubscriptionRequestCall
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Send: d:_i:47859_Elli-Wallbox-22ELLIF result 3958 5524 ResultData 0
Aug 07 23:55:25 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 23:55:25 Recv: d:_i:47859_Elli-Wallbox-22ELLIF:NodeManagement to NodeManagement result 5525 3936 ResultData 0

@andig: Das Loging-Flag "Lmicroseconds" klingt vielversprechend ... wäre es einen Request wert, das zu setzen und die Auflösung zu erhöhen/per Schalter erhöhen zu können?

Sparlader, 2024-08-08:
Hi zusammen, ein gutes 3/4 Jahr benutze ich evcc mit e-golf und elli connect. Und plötzlich ist der Wurm drin, genau die oben beschrieben Fehler sind bei mir auch, allerdings geht es zwischendurch dann doch mal wieder ohne für mich ersichtlichen Grund. Ich möchte euch nur bitte an dem Thema dranzubleiben, denn bisher war ich begeisterter Nutzer von EVCC und möchte es nartürlich weiterhin sein.

DerAndereAndi, 2024-08-08:
- Die Lampe ist immer gelb, wenn nicht der maximal mögliche Ladestrom erlaubt ist. Warum auch immer das der Fall sein mag.
- Leer bedeutet: Es gibt keine Infos, selbst wenn es ein tatsächlich von der Wallbox gesetztes Limit gibt sind keine Informationen darüber vorhanden.
- Wenn es aufgrund der Kommunikation mit evcc ein Limit reinhaut, dann ist es ein Bug. Denn es gibt keinen (ersichtlichen) Grund das zu tun.

Das Problem ist bei genau diesem Problem imho nicht die EEBUS Implementierung. Wie oben geschrieben sollte evcc bei jedem Anstecken das von sich aus erwartete Limit schreiben. Bei den meisten anderen Wallboxen (ohne EEBUS) ist ein Limit global in der Wallbox gesetzt, das ist mit EEBUS eben nicht so. Das gilt für den aktiven Ladevorgang. Und das Problem gibt es erst ab dem zweiten Ladevorgang. Würde mann evcc nach jedem abstecken neu starten, würde es immer funktionieren.

aaggog, 2024-08-08:
I have the same issue with Elli Pro.

andig, 2024-08-08:
@DerAndereAndi ich beziehe mich auf das Log aus https://github.com/evcc-io/evcc/issues/15174#issuecomment-2273862587. 

Ich sehe Fehler, kann aber sicher unrelated sein:

```
Aug 07 16:40:01 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 16:40:01 Send: d:_i:47859_Elli-Wallbox-22ELLIF result 111 135540 ResultData 7
Aug 07 16:40:01 kammerpi evcc[26471]: [eebus ] TRACE 2024/08/07 16:40:01 Error 7
Aug 07 16:40:01 kammerpi evcc[26471]: [eebus ] TRACE 2024/08/07 16:40:01 Send: 055_SKI_ELLI_e73 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_n:EVCC_HEMS-evcc-01.saw.snueq"},{"entity":[1]},{"feature":2}]},{"addressDestination":[{"device":"d:_i:47859_Elli-Wallbox-22ELLIF"},{"entity":[1,1]},{"feature":1001}]},{"msgCounter":111},{"msgCounterReference":135540},{"cmdClassifier":"result"}]},{"payload":[{"cmd":[[{"resultData":[{"errorNumber":7}]}]]}]}]}}]}
Aug 07 16:40:01 kammerpi evcc[26471]: [eebus ] TRACE 2024/08/07 16:40:01 Recv: 055_SKI_ELLI_e73 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-22ELLIF"},{"entity":[1,1]},{"feature":1001}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-evcc-01.saw.snueq"},{"entity":[0]},{"feature":0}]},{"msgCounter":135541},{"cmdClassifier":"call"},{"ackRequest":true}]},{"payload":[{"cmd":[[{"nodeManagementSubscriptionRequestCall":[{"subscriptionRequest":[{"clientAddress":[{"device":"d:_i:47859_Elli-Wallbox-22ELLIF"},{"entity":[1,1]},{"feature":1001}]},{"serverAddress":[{"device":"d:_n:EVCC_HEMS-evcc-01.saw.snueq"},{"entity":[1]},{"feature":2}]},{"serverFeatureType":"DeviceDiagnosis"}]}]}]]}]}]}}]}
Aug 07 16:40:01 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 16:40:01 Recv: d:_i:47859_Elli-Wallbox-22ELLIF:Generic to NodeManagement call 135541 NodeManagementSubscriptionRequestCall
Aug 07 16:40:01 kammerpi evcc[26471]: [eebus ] DEBUG 2024/08/07 16:40:01 Send: d:_i:47859_Elli-Wallbox-22ELLIF result 112 135541 ResultData 1
Aug 07 16:40:01 kammerpi evcc[26471]: [eebus ] TRACE 2024/08/07 16:40:01 Error 1: found feature DeviceDiagnosis is not matching required role server
```

> @andig Bei EEBUS scheint es daher notwendig zu sein, dass evcc das Ladelimit nach jedem Ansteckvorgang auch durchführt.

Das können wir machen. Die spannende Frage wäre wo:
1. Im Ladepunkt- der braucht das aber eigentlich nicht
2. Gar nicht- stattdessen über `GetMaxCurrent` von der Wallbox nachlesen und durch den Ladepunkt korrigieren lassen. Das geht aber bei EEbus nicht wenn ich es richtig verstehe?
3. Im EEbus Charger? Das wäre gar nicht schwierig weil wir den "alten" Wunschwert ja noch kennen. PR folgt.

Was ich aber nicht verstehe: warum ging das "früher"? Oder ging das noch nie? Oder hat der alte CEM das mal als hard-kodiertes Verhalten gehabt?

DerAndereAndi, 2024-08-08:
@andig diese Fehler sind nicht relevant, die kommen bei jeder Wallbox die mit dem kommerziellen Stack arbeitet.

Es ging früher weil ein Write Befehl immer ausgeführt wurde, und dieser Befehl kommt nicht mehr vom Loadpoint. So ist es in allen meinen Logs die ich hier noch habe. Kann es sein dass es da evtl. eine Änderung mit dem Lastmanagement gab die das verursachen könnte?

andig, 2024-08-08:
> Es ging früher weil ein Write Befehl immer ausgeführt wurde, und dieser Befehl kommt nicht mehr vom Loadpoint. So ist es in allen meinen Logs die ich hier noch habe. Kann es sein dass es da evtl. eine Änderung mit dem Lastmanagement gab die das verursachen könnte?

Nicht bewusst, aber ausschließen kann ich es natürlich nicht. Eigentlich haben wir immer versucht, Kommunikation nur zu machen wenn sie notwendig war.

andig, 2024-08-08:
Neues Nightly in 15min. Danke @DerAndereAndi für die Analyse. Ich hätte es nicht aus den Logs raus lesen können...

Sumpfdotter, 2024-08-08:
> * Die Lampe ist immer gelb, wenn nicht der maximal mögliche Ladestrom erlaubt ist. Warum auch immer das der Fall sein mag.

Du hast die Box vermutlich schon genauer analysiert und kennst das daher. Ich beziehe mich auf das Handbuch. Das Leuchtmuster nach dem Anstecken ist "grün/aus/gelb", und laut Handbuch Seite 42 bedeutet das:

Verlust der Kommunikation mit HEMS
Laden möglich
Die Wallbox kann nicht mit dem HEMS-Netz kommunizieren.
— Überprüfen Sie Ihre Netzwerkkonfiguration mit dem Configuration Manager.
— Überprüfen Sie Ihre HEMS-Konfiguration mit dem Configuration Manager.

Ich hielt es auch für fast gesichert, dass die Wallbox nicht aufgrund eines EEBUS-Kommandos die gelbe Lampe wirft, aber ich wollte es trotzdem ausschließen. Ich habe mir ein Delay in die EEBUS-Kommunikation eingebaut und kann nun bestätigen: Die Wallbox geht beim Anstecken sofort von sich aus auf gelb, d.h. sie wirft vermutlich sofort das Ladelimit rein noch bevor sie die erste EEBUS-Reaktion auf das Anstecken empfängt. Also wie von Dir vermutet.

> * Leer bedeutet: Es gibt keine Infos, selbst wenn es ein tatsächlich von der Wallbox gesetztes Limit gibt sind keine Informationen darüber vorhanden.

Okay. Heißt das also: Wir senden ein read an die Wallbox und sie antwortet mit leerem Datensatz "Ich weiß es zwar, aber sag es Dir nicht."? Aus Neugier: Ist das ein Bug in der Box oder noch spezifikationsgemäß?

> * Wenn es aufgrund der Kommunikation mit evcc ein Limit reinhaut, dann ist es ein Bug. Denn es gibt keinen (ersichtlichen) Grund das zu tun.

Sehe ich auch so. Aber s.o.: Das tut sie nicht. Sie macht das bei meinen Tests immer.
 

> Neues Nightly in 15min. Danke @DerAndereAndi für die Analyse. Ich hätte es nicht aus den Logs raus lesen können...

Jepp. Geht wieder. Danke!

closed 2024-08-09 by andig

DerAndereAndi, 2024-08-09:
Wenn die Box ein Limit setzt, muss es dieses auch melden -> Bug. In der Spec steht so ein Satz nicht drin, aber dann kann ich auch wild irgendwas machen und es wird am Ende nichts bringen weil es nicht funktioniert. Wie eben hier.

Sparlader, 2024-08-11:
bei mir geht es auch nur wenn EVCC neu gestartet wurde, kann also die Aussage von "der andere Andi" bestätigen. Es wäre schade wenn sich da keine Lösung finden würde das es wieder so geht wie in den "alten" Versionen, die bei mir immer gut funktionierten. Meine Frau muss Auto einstecken können und dann gehts....

DerAndereAndi, 2024-08-11:
@Sparlader Mit welcher Version von evcc?

Sparlader, 2024-08-12:
0.129.0

andig, 2024-08-18:
https://github.com/evcc-io/evcc/releases/tag/0.130.0 contains a refactor of the EEbus charger that might or might not help here.