evcc-io/evcc #15178: WARN smartCostActive: no matching rate

tobsen101, 2024-08-01:
### Describe the bug

smart cost: no matching rate
smartCostActive: no matching rate

ähnlich/gleich zu https://github.com/evcc-io/evcc/issues/14782 
allerdings habe ich bereits die Version 0.129.0

### Steps to reproduce

1. evcc starten
2. warten
3. Fehlermeldung taucht sporadisch, aber dann gehäuft auf
...

### Configuration details

```yaml
network:
  schema: http
  host: *****
  port: 7070
log: debug
levels:
  cache: error
plant: 400af4d61c430083340f9211bbcbe3c9a745bb5db46f00a53b6c5cbcfee454c6
interval: 10s # control cycle interval
sponsortoken: *****
telemetry: false
meters:
- type: template
  template: victron-energy
  usage: grid
  host: *****
  port: 502
  name: grid-victron
- type: template
  template: victron-energy
  usage: pv
  host: *****
  port: 502
  name: pv-victron
- type: template
  template: victron-energy
  usage: battery
  host: *****
  port: 502
  name: battery-victron
chargers:
- type: template
  template: victron-evcs
  id: 1
  host: *****
  port: 502
  modbus: tcpip
  name: wallbox-victron-links
vehicles:
- type: template
  template: offline
  title: BYD Atto 3
  capacity: 62
  name: byd
loadpoints:
- title: EVCS Links
  charger: wallbox-victron-links
  vehicle: byd
  mode: pv
tariffs:
  grid:
    type: fixed
    price: 0.29 # [currency]/kWh
  feedin:
    type: fixed
    price: 0.0839 # [currency]/kWh
  co2:
    type: grünstromindex # Beispiel GrünstromIndex (nur Deutschland)
    zip: ***** # Deine Postleitzahl

influx:
  url: *****
  database: iobroker: *****
  user: *****
  password: *****
site:
  title: VerTob EVCC
  meters:
    grid: grid-victron
    pv:
    - pv-victron
    battery:
    - battery-victron
```

### Log details

```text
[lp-1  ] DEBUG 2024/08/01 14:57:18 charge power: 0W
[site  ] WARN 2024/08/01 14:57:18 smartCostActive: no matching rate
[site  ] DEBUG 2024/08/01 14:57:18 pv power: 3540W
[site  ] DEBUG 2024/08/01 14:57:18 battery soc: 100%
[site  ] DEBUG 2024/08/01 14:57:18 battery power: 0W
[site  ] DEBUG 2024/08/01 14:57:18 grid meter: -2836W
[site  ] DEBUG 2024/08/01 14:57:18 site power: -2836W
[lp-1  ] DEBUG 2024/08/01 14:57:18 !! session: chargeRater.chargedEnergy=14.6 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/01 14:57:18 charger status: A
[site  ] WARN 2024/08/01 14:57:18 smart cost: no matching rate
[site  ] DEBUG 2024/08/01 14:57:28 ----
[lp-1  ] DEBUG 2024/08/01 14:57:28 charge power: 0W
[site  ] WARN 2024/08/01 14:57:28 smartCostActive: no matching rate
[site  ] DEBUG 2024/08/01 14:57:28 pv power: 3549W
[site  ] DEBUG 2024/08/01 14:57:28 battery soc: 100%
[site  ] DEBUG 2024/08/01 14:57:28 battery power: 0W
[site  ] DEBUG 2024/08/01 14:57:28 grid meter: -2834W
[site  ] DEBUG 2024/08/01 14:57:28 site power: -2834W
[lp-1  ] DEBUG 2024/08/01 14:57:28 !! session: chargeRater.chargedEnergy=14.6 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/01 14:57:28 charger status: A
[site  ] WARN 2024/08/01 14:57:28 smart cost: no matching rate
[site  ] DEBUG 2024/08/01 14:57:38 ----
[lp-1  ] DEBUG 2024/08/01 14:57:38 charge power: 0W
[site  ] WARN 2024/08/01 14:57:38 smartCostActive: no matching rate
[site  ] DEBUG 2024/08/01 14:57:38 pv power: 3546W
[site  ] DEBUG 2024/08/01 14:57:38 battery soc: 100%
[site  ] DEBUG 2024/08/01 14:57:38 battery power: 0W
[site  ] DEBUG 2024/08/01 14:57:38 grid meter: -2524W
[site  ] DEBUG 2024/08/01 14:57:38 site power: -2524W
[lp-1  ] DEBUG 2024/08/01 14:57:38 !! session: chargeRater.chargedEnergy=14.6 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/01 14:57:38 charger status: A
[site  ] WARN 2024/08/01 14:57:38 smart cost: no matching rate
[site  ] DEBUG 2024/08/01 14:57:48 ----
[lp-1  ] DEBUG 2024/08/01 14:57:48 charge power: 0W
[site  ] WARN 2024/08/01 14:57:48 smartCostActive: no matching rate
[site  ] DEBUG 2024/08/01 14:57:48 pv power: 3549W
[site  ] DEBUG 2024/08/01 14:57:48 battery soc: 100%
[site  ] DEBUG 2024/08/01 14:57:48 battery power: 0W
[site  ] DEBUG 2024/08/01 14:57:48 grid meter: -2588W
[site  ] DEBUG 2024/08/01 14:57:48 site power: -2588W
[lp-1  ] DEBUG 2024/08/01 14:57:48 !! session: chargeRater.chargedEnergy=14.6 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/01 14:57:48 charger status: A
[site  ] WARN 2024/08/01 14:57:48 smart cost: no matching rate
[site  ] DEBUG 2024/08/01 14:57:58 ----
[lp-1  ] DEBUG 2024/08/01 14:57:58 charge power: 0W
[site  ] WARN 2024/08/01 14:57:58 smartCostActive: no matching rate
[site  ] DEBUG 2024/08/01 14:57:58 pv power: 3547W
[site  ] DEBUG 2024/08/01 14:57:58 battery soc: 100%
[site  ] DEBUG 2024/08/01 14:57:58 battery power: 0W
[site  ] DEBUG 2024/08/01 14:57:58 grid meter: -2870W
[site  ] DEBUG 2024/08/01 14:57:58 site power: -2870W
[lp-1  ] DEBUG 2024/08/01 14:57:58 !! session: chargeRater.chargedEnergy=14.6 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/01 14:57:58 charger status: A
[site  ] WARN 2024/08/01 14:57:58 smart cost: no matching rate
[site  ] DEBUG 2024/08/01 14:58:08 ----
[lp-1  ] DEBUG 2024/08/01 14:58:08 charge power: 0W
[site  ] WARN 2024/08/01 14:58:08 smartCostActive: no matching rate
[site  ] DEBUG 2024/08/01 14:58:08 pv power: 3553W
[site  ] DEBUG 2024/08/01 14:58:08 battery soc: 100%
[site  ] DEBUG 2024/08/01 14:58:08 battery power: 0W
[site  ] DEBUG 2024/08/01 14:58:08 grid meter: -2860W
[site  ] DEBUG 2024/08/01 14:58:08 site power: -2860W
[lp-1  ] DEBUG 2024/08/01 14:58:08 !! session: chargeRater.chargedEnergy=14.6 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/01 14:58:08 charger status: A
[site  ] WARN 2024/08/01 14:58:08 smart cost: no matching rate
[site  ] DEBUG 2024/08/01 14:58:18 ----
[lp-1  ] DEBUG 2024/08/01 14:58:18 charge power: 0W
[site  ] WARN 2024/08/01 14:58:18 smartCostActive: no matching rate
[site  ] DEBUG 2024/08/01 14:58:18 pv power: 3552W
[site  ] DEBUG 2024/08/01 14:58:18 battery soc: 100%
[site  ] DEBUG 2024/08/01 14:58:18 battery power: 0W
[site  ] DEBUG 2024/08/01 14:58:18 grid meter: -2835W
[site  ] DEBUG 2024/08/01 14:58:18 site power: -2835W
[lp-1  ] DEBUG 2024/08/01 14:58:18 !! session: chargeRater.chargedEnergy=14.6 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/01 14:58:18 charger status: A
[site  ] WARN 2024/08/01 14:58:18 smart cost: no matching rate
[site  ] DEBUG 2024/08/01 14:58:28 ----
[lp-1  ] DEBUG 2024/08/01 14:58:28 charge power: 0W
[site  ] WARN 2024/08/01 14:58:28 smartCostActive: no matching rate
[site  ] DEBUG 2024/08/01 14:58:28 pv power: 3563W
[site  ] DEBUG 2024/08/01 14:58:28 battery soc: 100%
[site  ] DEBUG 2024/08/01 14:58:28 battery power: 0W
[site  ] DEBUG 2024/08/01 14:58:28 grid meter: -2608W
[site  ] DEBUG 2024/08/01 14:58:28 site power: -2608W
[lp-1  ] DEBUG 2024/08/01 14:58:28 !! session: chargeRater.chargedEnergy=14.6 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/01 14:58:28 charger status: A
[site  ] WARN 2024/08/01 14:58:28 smart cost: no matching rate
[site  ] DEBUG 2024/08/01 14:58:38 ----
[lp-1  ] DEBUG 2024/08/01 14:58:38 charge power: 0W
[site  ] WARN 2024/08/01 14:58:38 smartCostActive: no matching rate
[site  ] DEBUG 2024/08/01 14:58:38 pv power: 3571W
[site  ] DEBUG 2024/08/01 14:58:38 battery soc: 100%
[site  ] DEBUG 2024/08/01 14:58:38 battery power: 0W
[site  ] DEBUG 2024/08/01 14:58:38 grid meter: -2877W
[site  ] DEBUG 2024/08/01 14:58:38 site power: -2877W
[lp-1  ] DEBUG 2024/08/01 14:58:38 !! session: chargeRater.chargedEnergy=14.6 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/01 14:58:38 charger status: A
[site  ] WARN 2024/08/01 14:58:38 smart cost: no matching rate
[site  ] DEBUG 2024/08/01 14:58:48 ----
[lp-1  ] DEBUG 2024/08/01 14:58:48 charge power: 0W
[site  ] WARN 2024/08/01 14:58:48 smartCostActive: no matching rate
[site  ] DEBUG 2024/08/01 14:58:48 pv power: 3581W
[site  ] DEBUG 2024/08/01 14:58:48 battery soc: 100%
[site  ] DEBUG 2024/08/01 14:58:48 battery power: 0W
[site  ] DEBUG 2024/08/01 14:58:48 grid meter: -2882W
[site  ] DEBUG 2024/08/01 14:58:48 site power: -2882W
[lp-1  ] DEBUG 2024/08/01 14:58:48 !! session: chargeRater.chargedEnergy=14.6 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/01 14:58:48 charger status: A
[site  ] WARN 2024/08/01 14:58:48 smart cost: no matching rate
[site  ] DEBUG 2024/08/01 14:58:58 ----
```

### What type of operating system are you running?

Docker container

### Version

0.129.0

naltatis, 2024-08-02:
@tobsen101 ich kann das leider nicht reproduzieren. Die Meldung kommt, wenn die Tarifvorhersage (in deinem Fall co2 über Grünstromindex) keine aktuellen Daten hat. Kannst du im Log mal schauen, ob du Ausgaben dazu im Log findest? Bei zu vielen Aufrufen (viele evcc Neustarts?) setzten sie auch Rate-Limiting ein. Der entsprechende Request ist https://api.corrently.io/v2.0/gsi/prediction?zip=[DEINE_PLZ]

tobsen101, 2024-08-02:
leider gibt das log nicht mehr her. Auch vor dem ersten WARN kommt keine Fehlermeldung oder ein Hinweis.
Allerdings sieht man, dass der Fehler genau alle 4 Stunden auftaucht und knappe 3 Minuten anhält. dann verschwindet er wieder und 4h später das gleiche Spiel.
so war es laut den Logs zumindest gestern. heute kam der Fehler noch nicht. Das könnte natürlich mit den Rate Limits zusammenhängen. also gut möglich das es heute Mittag wieder losgeht.

zum Thema öftere Neustarts: Es läuft super stabil und es sind keine Neustarts notwendig. Ausnahme ist in dem Fall nur ein Neustart bzgl. Update des EVCC Docker images.

```
docker logs evcc-compose | grep WARN

[site  ] WARN 2024/08/01 10:57:07 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] WARN 2024/08/01 10:57:08 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] WARN 2024/08/01 11:57:18 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:57:18 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:57:28 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:57:28 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:57:38 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:57:38 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:57:48 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:57:48 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:57:58 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:57:58 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:58:08 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:58:08 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:58:18 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:58:18 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:58:28 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:58:28 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:58:38 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:58:38 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:58:48 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:58:48 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:58:58 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:58:58 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:59:08 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:59:08 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:59:18 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:59:18 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:59:28 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:59:28 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:59:38 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:59:38 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:59:48 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:59:48 smart cost: no matching rate
[site  ] WARN 2024/08/01 11:59:58 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 11:59:58 smart cost: no matching rate
[site  ] WARN 2024/08/01 14:57:18 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 14:57:18 smart cost: no matching rate
[site  ] WARN 2024/08/01 14:57:28 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 14:57:28 smart cost: no matching rate
[site  ] WARN 2024/08/01 14:57:38 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 14:57:38 smart cost: no matching rate
[site  ] WARN 2024/08/01 14:57:48 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 14:57:48 smart cost: no matching rate
[site  ] WARN 2024/08/01 14:57:58 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 14:57:58 smart cost: no matching rate
[site  ] WARN 2024/08/01 14:58:08 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 14:58:08 smart cost: no matching rate
[site  ] WARN 2024/08/01 14:58:18 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 14:58:18 smart cost: no matching rate
[site  ] WARN 2024/08/01 14:58:28 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 14:58:28 smart cost: no matching rate
[site  ] WARN 2024/08/01 14:58:38 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 14:58:38 smart cost: no matching rate
[site  ] WARN 2024/08/01 14:58:48 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 14:58:48 smart cost: no matching rate
[site  ] WARN 2024/08/01 14:58:58 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 14:58:58 smart cost: no matching rate
[site  ] WARN 2024/08/01 14:59:08 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 14:59:08 smart cost: no matching rate
[site  ] WARN 2024/08/01 14:59:18 smartCostActive: no matching rate
[... 87 lines cut ...]
[site  ] WARN 2024/08/01 23:58:08 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 23:58:08 smart cost: no matching rate
[site  ] WARN 2024/08/01 23:58:18 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 23:58:18 smart cost: no matching rate
[site  ] WARN 2024/08/01 23:58:28 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 23:58:28 smart cost: no matching rate
[site  ] WARN 2024/08/01 23:58:38 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 23:58:38 smart cost: no matching rate
[site  ] WARN 2024/08/01 23:58:48 smartCostActive: no matching rate
[... cut ...]
[site  ] WARN 2024/08/01 23:59:28 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 23:59:28 smart cost: no matching rate
[site  ] WARN 2024/08/01 23:59:38 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 23:59:38 smart cost: no matching rate
[site  ] WARN 2024/08/01 23:59:48 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 23:59:48 smart cost: no matching rate
[site  ] WARN 2024/08/01 23:59:58 smartCostActive: no matching rate
[site  ] WARN 2024/08/01 23:59:58 smart cost: no matching rate
```

naltatis, 2024-08-02:
Kannst du zum Zeitpunkt des Fehler beim nächsten Mal mal das hier aufrufen und die Antwort teilen?

http://[evcc]:7070/api/tariff/planner

naltatis, 2024-08-02:
Ist ja spannend zu sehen, dass das dann fast 3 Minuten vor der vollen Stunde beginnt und direkt danach aufhört. Aber dann auch nicht jede Stunde.

nanuaber, 2024-08-05:
Ich hatte gerade die gleiche Fehlermeldung. Ich habe EVCC (V0.129.0) als Docker Container auf einer kleinen Synology laufen. Ich hatte jetzt den Vierenscanner aktiviert der enorm viele Resourcen gezogen hat. Die Plattenauslastung war dauerhaft auf hohen Niveau. Der WebSeiten aufbau war  auch sehr langsam. Nachdem ich den Scanner pausiert habe ist die Fehlermeldung verschwunden. Ich vermute das er bei der Abfrage der API auf einen Timeout gelaufen ist. Aber nur eine Vermutung ohne die genauen Hintergründe zu kennen.

tobsen101, 2024-08-05:
Seit dem letzten auftreten am 01.08. hatte ich den Fehler nicht mehr... 
@nanuaber hast du eine ähnliche Config bzgl. co2 über Grünstromindex? oder frägst du eine andere tariffs Quelle ab?

nanuaber, 2024-08-05:
Die tariffs Section ist exakt gleich wie bei dir nur mit anderen Werten. Interessanterweise war die Auslastung wohl doch nicht das Problem. Die FehlerMeldung ist jetzt wieder da.
smartCostActive: no matching rate   .............

nanuaber, 2024-08-05:
ich habden Link von naltatis mal ausprobiert. Dies kam zurück:

{"result":{"rates":[{"start":"2024-08-05T15:00:00+02:00","end":"2024-08-05T16:00:00+02:00","price":83},{"start":"2024-08-05T16:00:00+02:00","end":"2024-08-05T17:00:00+02:00","price":74},{"start":"2024-08-05T17:00:00+02:00","end":"2024-08-05T18:00:00+02:00","price":71},{"start":"2024-08-05T18:00:00+02:00","end":"2024-08-05T19:00:00+02:00","price":59},{"start":"2024-08-05T19:00:00+02:00","end":"2024-08-05T20:00:00+02:00","price":56},{"start":"2024-08-05T20:00:00+02:00","end":"2024-08-05T21:00:00+02:00","price":235},{"start":"2024-08-05T21:00:00+02:00","end":"2024-08-05T22:00:00+02:00","price":238},{"start":"2024-08-05T22:00:00+02:00","end":"2024-08-05T23:00:00+02:00","price":238},{"start":"2024-08-05T23:00:00+02:00","end":"2024-08-06T00:00:00+02:00","price":238},{"start":"2024-08-06T00:00:00+02:00","end":"2024-08-06T01:00:00+02:00","price":241},{"start":"2024-08-06T01:00:00+02:00","end":"2024-08-06T02:00:00+02:00","price":241},{"start":"2024-08-06T02:00:00+02:00","end":"2024-08-06T03:00:00+02:00","price":241},{"start":"2024-08-06T03:00:00+02:00","end":"2024-08-06T04:00:00+02:00","price":241},{"start":"2024-08-06T04:00:00+02:00","end":"2024-08-06T05:00:00+02:00","price":244},{"start":"2024-08-06T05:00:00+02:00","end":"2024-08-06T06:00:00+02:00","price":244},{"start":"2024-08-06T06:00:00+02:00","end":"2024-08-06T07:00:00+02:00","price":241},{"start":"2024-08-06T07:00:00+02:00","end":"2024-08-06T08:00:00+02:00","price":244},{"start":"2024-08-06T08:00:00+02:00","end":"2024-08-06T09:00:00+02:00","price":241},{"start":"2024-08-06T09:00:00+02:00","end":"2024-08-06T10:00:00+02:00","price":241},{"start":"2024-08-06T10:00:00+02:00","end":"2024-08-06T11:00:00+02:00","price":223},{"start":"2024-08-06T11:00:00+02:00","end":"2024-08-06T12:00:00+02:00","price":217},{"start":"2024-08-06T12:00:00+02:00","end":"2024-08-06T13:00:00+02:00","price":211},{"start":"2024-08-06T13:00:00+02:00","end":"2024-08-06T14:00:00+02:00","price":196},{"start":"2024-08-06T14:00:00+02:00","end":"2024-08-06T15:00:00+02:00","price":146},{"start":"2024-08-06T15:00:00+02:00","end":"2024-08-06T16:00:00+02:00","price":122},{"start":"2024-08-06T16:00:00+02:00","end":"2024-08-06T17:00:00+02:00","price":128},{"start":"2024-08-06T17:00:00+02:00","end":"2024-08-06T18:00:00+02:00","price":181},{"start":"2024-08-06T18:00:00+02:00","end":"2024-08-06T19:00:00+02:00","price":202},{"start":"2024-08-06T19:00:00+02:00","end":"2024-08-06T20:00:00+02:00","price":211},{"start":"2024-08-06T20:00:00+02:00","end":"2024-08-06T21:00:00+02:00","price":235},{"start":"2024-08-06T21:00:00+02:00","end":"2024-08-06T22:00:00+02:00","price":229},{"start":"2024-08-06T22:00:00+02:00","end":"2024-08-06T23:00:00+02:00","price":232},{"start":"2024-08-06T23:00:00+02:00","end":"2024-08-07T00:00:00+02:00","price":235},{"start":"2024-08-07T00:00:00+02:00","end":"2024-08-07T01:00:00+02:00","price":217},{"start":"2024-08-07T01:00:00+02:00","end":"2024-08-07T02:00:00+02:00","price":217},{"start":"2024-08-07T02:00:00+02:00","end":"2024-08-07T03:00:00+02:00","price":220},{"start":"2024-08-07T03:00:00+02:00","end":"2024-08-07T04:00:00+02:00","price":223},{"start":"2024-08-07T04:00:00+02:00","end":"2024-08-07T05:00:00+02:00","price":223},{"start":"2024-08-07T05:00:00+02:00","end":"2024-08-07T06:00:00+02:00","price":229},{"start":"2024-08-07T06:00:00+02:00","end":"2024-08-07T07:00:00+02:00","price":229},{"start":"2024-08-07T07:00:00+02:00","end":"2024-08-07T08:00:00+02:00","price":232},{"start":"2024-08-07T08:00:00+02:00","end":"2024-08-07T09:00:00+02:00","price":229},{"start":"2024-08-07T09:00:00+02:00","end":"2024-08-07T10:00:00+02:00","price":223},{"start":"2024-08-07T10:00:00+02:00","end":"2024-08-07T11:00:00+02:00","price":196},{"start":"2024-08-07T11:00:00+02:00","end":"2024-08-07T12:00:00+02:00","price":187},{"start":"2024-08-07T12:00:00+02:00","end":"2024-08-07T13:00:00+02:00","price":175},{"start":"2024-08-07T13:00:00+02:00","end":"2024-08-07T14:00:00+02:00","price":157},{"start":"2024-08-07T14:00:00+02:00","end":"2024-08-07T15:00:00+02:00","price":30},{"start":"2024-08-07T15:00:00+02:00","end":"2024-08-07T16:00:00+02:00","price":30},{"start":"2024-08-07T16:00:00+02:00","end":"2024-08-07T17:00:00+02:00","price":30},{"start":"2024-08-07T17:00:00+02:00","end":"2024-08-07T18:00:00+02:00","price":30},{"start":"2024-08-07T18:00:00+02:00","end":"2024-08-07T19:00:00+02:00","price":30},{"start":"2024-08-07T19:00:00+02:00","end":"2024-08-07T20:00:00+02:00","price":30},{"start":"2024-08-07T20:00:00+02:00","end":"2024-08-07T21:00:00+02:00","price":220},{"start":"2024-08-07T21:00:00+02:00","end":"2024-08-07T22:00:00+02:00","price":217},{"start":"2024-08-07T22:00:00+02:00","end":"2024-08-07T23:00:00+02:00","price":217},{"start":"2024-08-07T23:00:00+02:00","end":"2024-08-08T00:00:00+02:00","price":217},{"start":"2024-08-08T00:00:00+02:00","end":"2024-08-08T01:00:00+02:00","price":220},{"start":"2024-08-08T01:00:00+02:00","end":"2024-08-08T02:00:00+02:00","price":220},{"start":"2024-08-08T02:00:00+02:00","end":"2024-08-08T03:00:00+02:00","price":220},{"start":"2024-08-08T03:00:00+02:00","end":"2024-08-08T04:00:00+02:00","price":223},{"start":"2024-08-08T04:00:00+02:00","end":"2024-08-08T05:00:00+02:00","price":223},{"start":"2024-08-08T05:00:00+02:00","end":"2024-08-08T06:00:00+02:00","price":229},{"start":"2024-08-08T06:00:00+02:00","end":"2024-08-08T07:00:00+02:00","price":232},{"start":"2024-08-08T07:00:00+02:00","end":"2024-08 [... cut ...] rt":"2024-08-08T19:00:00+02:00","end":"2024-08-08T20:00:00+02:00","price":30},{"start":"2024-08-08T20:00:00+02:00","end":"2024-08-08T21:00:00+02:00","price":223},{"start":"2024-08-08T21:00:00+02:00","end":"2024-08-08T22:00:00+02:00","price":223},{"start":"2024-08-08T22:00:00+02:00","end":"2024-08-08T23:00:00+02:00","price":223},{"start":"2024-08-08T23:00:00+02:00","end":"2024-08-09T00:00:00+02:00","price":223},{"start":"2024-08-09T00:00:00+02:00","end":"2024-08-09T01:00:00+02:00","price":223}]}}

naltatis, 2024-08-05:
@nanuaber hast du das passende Log dazu wann genau die Warnungen bei dir aufgetauch sind?

nanuaber, 2024-08-05:
nein das Log hab ich leider nicht parallel gezogen. Jetzt sind die Meldungen auch wieder weg. Scheint irgendwie zeitabhängig zu sein. Wenn sie wieder auftauchen ziehe ich parallel das Log

nanuaber, 2024-08-07:
jetzt war es einen Tag ruhig. Jetzt kam die Meldung wieder:

 WARN 2024/08/07 15:57:49 smartCostActive: no matching rate
[site ] DEBUG 2024/08/07 15:57:50 pv power: 2773W
[site ] DEBUG 2024/08/07 15:57:50 grid meter: -2699W
[site ] DEBUG 2024/08/07 15:57:50 grid powers: [-954 -915 -829]W
[site ] DEBUG 2024/08/07 15:57:50 grid currents: [-3.7 -3.7 -3.6]A
[site ] DEBUG 2024/08/07 15:57:50 site power: -2699W
[lp-1 ] DEBUG 2024/08/07 15:57:51 charge total import: 97.324kWh
[lp-1 ] DEBUG 2024/08/07 15:57:51 charger status: A
[site ] DEBUG 2024/08/07 15:57:59 ----
[lp-1 ] DEBUG 2024/08/07 15:57:59 charge power: 0W
[lp-1 ] DEBUG 2024/08/07 15:57:59 charge currents: [0 0 0]A
[site ] WARN 2024/08/07 15:57:59 smartCostActive: no matching rate
[site ] DEBUG 2024/08/07 15:58:00 pv power: 2964W
[site ] DEBUG 2024/08/07 15:58:00 grid meter: -2868W
[site ] DEBUG 2024/08/07 15:58:05 grid powers: [-1014 -983 -760]W
[site ] DEBUG 2024/08/07 15:58:05 grid currents: [-4.3 -4.1 -3.7]A
[site ] DEBUG 2024/08/07 15:58:05 site power: -2868W
[lp-1 ] DEBUG 2024/08/07 15:58:05 charge total import: 97.324kWh
[lp-1 ] DEBUG 2024/08/07 15:58:05 charger status: A
[site ] DEBUG 2024/08/07 15:58:09 ----
[lp-1 ] DEBUG 2024/08/07 15:58:09 charge power: 0W
[lp-1 ] DEBUG 2024/08/07 15:58:09 charge currents: [0 0 0]A
[site ] WARN 2024/08/07 15:58:11 smartCostActive: no matching rate
[site ] DEBUG 2024/08/07 15:58:11 pv power: 2402W
[site ] DEBUG 2024/08/07 15:58:11 grid meter: -2324W
[site ] DEBUG 2024/08/07 15:58:12 grid powers: [-832 -793 -705]W
[site ] DEBUG 2024/08/07 15:58:12 grid currents: [-3.2 -3.1 -2.7]A
[site ] DEBUG 2024/08/07 15:58:12 site power: -2324W
[lp-1 ] DEBUG 2024/08/07 15:58:12 charge total import: 97.324kWh
[lp-1 ] DEBUG 2024/08/07 15:58:12 charger status: A
[site ] DEBUG 2024/08/07 15:58:19 ----
[lp-1 ] DEBUG 2024/08/07 15:58:19 charge power: 0W
[lp-1 ] DEBUG 2024/08/07 15:58:20 charge currents: [0 0 0]A
[site ] WARN 2024/08/07 15:58:20 smartCostActive: no matching rate
[site ] DEBUG 2024/08/07 15:58:20 pv power: 2790W
[site ] DEBUG 2024/08/07 15:58:20 grid meter: -2706W
[site ] DEBUG 2024/08/07 15:58:21 grid powers: [-963 -920 -836]W
[site ] DEBUG 2024/08/07 15:58:21 grid currents: [-3.9 -3.8 -3.4]A
[site ] DEBUG 2024/08/07 15:58:22 site power: -2706W
[lp-1 ] DEBUG 2024/08/07 15:58:22 charge total import: 97.324kWh
[lp-1 ] DEBUG 2024/08/07 15:58:22 charger status: A
[site ] DEBUG 2024/08/07 15:58:29 ----
[lp-1 ] DEBUG 2024/08/07 15:58:29 charge power: 0W
[lp-1 ] DEBUG 2024/08/07 15:58:29 charge currents: [0 0 0]A
[site ] WARN 2024/08/07 15:58:29 smartCostActive: no matching rate
[site ] DEBUG 2024/08/07 15:58:30 pv power: 3174W
[site ] DEBUG 2024/08/07 15:58:30 grid meter: -3110W
[site ] DEBUG 2024/08/07 15:58:31 grid powers: [-1105 -1056 -974]W
[site ] DEBUG 2024/08/07 15:58:31 grid currents: [-4.5 -4.4 -4]A
[site ] DEBUG 2024/08/07 15:58:31 site power: -3110W
[lp-1 ] DEBUG 2024/08/07 15:58:32 charge total import: 97.324kWh
[lp-1 ] DEBUG 2024/08/07 15:58:32 charger status: A
[site ] DEBUG 2024/08/07 15:58:39 ----
[lp-1 ] DEBUG 2024/08/07 15:58:39 charge power: 0W
[lp-1 ] DEBUG 2024/08/07 15:58:39 charge currents: [0 0 0]A
[site ] WARN 2024/08/07 15:58:39 smartCostActive: no matching rate
[site ] DEBUG 2024/08/07 15:58:40 pv power: 3467W
[site ] DEBUG 2024/08/07 15:58:40 grid meter: -3360W
[site ] DEBUG 2024/08/07 15:58:40 grid powers: [-1182 -1130 -1046]W
[site ] DEBUG 2024/08/07 15:58:42 grid currents: [-4.8 -4.7 -4.3]A
[site ] DEBUG 2024/08/07 15:58:43 site power: -3360W
[lp-1 ] DEBUG 2024/08/07 15:58:43 charge total import: 97.324kWh
[lp-1 ] DEBUG 2024/08/07 15:58:43 charger status: A

naltatis, 2024-08-07:
Das Log alleine hilft mir leider nicht. Hilfreich wäre zum "gleiche Zeipunkt" wo der Fehler auftritt, den Zustand des Planner Tarifs (co2) zu haben: `http://[evcc]:7070/api/tariff/planner`

Aktuelle Hypothese: Die Grünstromindex API liefert schon vor Ablauf der aktuellen Stunde eine Antwort, in der die noch laufende Stunde nicht mehr drin ist.

nanuaber, 2024-08-07:
das war der Planner tarif zu dem Zeipunkt:

{
  "result": {
    "rates": [
      {
        "start": "2024-08-07T16:00:00+02:00",
        "end": "2024-08-07T17:00:00+02:00",
        "price": 109
      },
      {
        "start": "2024-08-07T17:00:00+02:00",
        "end": "2024-08-07T18:00:00+02:00",
        "price": 109
      },
      {
        "start": "2024-08-07T18:00:00+02:00",
        "end": "2024-08-07T19:00:00+02:00",
        "price": 109
      },
      {
        "start": "2024-08-07T19:00:00+02:00",
        "end": "2024-08-07T20:00:00+02:00",
        "price": 109
      },
      {
        "start": "2024-08-07T20:00:00+02:00",
        "end": "2024-08-07T21:00:00+02:00",
        "price": 180
      },
      {
        "start": "2024-08-07T21:00:00+02:00",
        "end": "2024-08-07T22:00:00+02:00",
        "price": 180
      },
      {
        "start": "2024-08-07T22:00:00+02:00",
        "end": "2024-08-07T23:00:00+02:00",
        "price": 180
      },
      {
        "start": "2024-08-07T23:00:00+02:00",
        "end": "2024-08-08T00:00:00+02:00",
        "price": 180
      },
      {
        "start": "2024-08-08T00:00:00+02:00",
        "end": "2024-08-08T01:00:00+02:00",
        "price": 183
      },
      {
        "start": "2024-08-08T01:00:00+02:00",
        "end": "2024-08-08T02:00:00+02:00",
        "price": 183
      },
      {
        "start": "2024-08-08T02:00:00+02:00",
        "end": "2024-08-08T03:00:00+02:00",
        "price": 183
      },
      {
        "start": "2024-08-08T03:00:00+02:00",
        "end": "2024-08-08T04:00:00+02:00",
        "price": 183
      },
      {
        "start": "2024-08-08T04:00:00+02:00",
        "end": "2024-08-08T05:00:00+02:00",
        "price": 183
      },
[... 318 lines cut ...]
        "price": 186
      },
      {
        "start": "2024-08-10T21:00:00+02:00",
        "end": "2024-08-10T22:00:00+02:00",
        "price": 183
      },
      {
        "start": "2024-08-10T22:00:00+02:00",
        "end": "2024-08-10T23:00:00+02:00",
        "price": 183
      },
      {
        "start": "2024-08-10T23:00:00+02:00",
        "end": "2024-08-11T00:00:00+02:00",
        "price": 180
      },
      {
        "start": "2024-08-11T00:00:00+02:00",
        "end": "2024-08-11T01:00:00+02:00",
        "price": 180
      }
    ]
  }
}

naltatis, 2024-08-07:
@nanuaber Super!

Das bestätigt die Hypothese. Wir updaten Grünstromindex. Und die liefern den Slot von 15-16 Uhr schon ein paar Minuten vor 16 Uhr nicht mehr zurück. Daher die Meldung, dass es keine aktuellen Preise gibt.

@andig hast du ne gute Lösungsidee? Mergen anstatt überschreiben oder zumindest den aktuellen stehen lassen?

andig, 2024-08-08:
Merge würde heissen, alle Slots zu behalten, die a) nach now() enden und b) vor dem frühesten Slot der neuen Daten starten?

naltatis, 2024-08-08:
> Merge würde heissen, alle Slots zu behalten, die a) nach now() enden und b) vor dem frühesten Slot der neuen Daten starten?

Ja.

closed 2024-08-08 by andig