evcc-io/evcc #15186: 2 phase charging not shown properly in the display

typxxi, 2024-08-01:
### Describe the bug

I have setup a 3 phase charger with a max. current of 16 A on 2 phases (the Pulsar Plus from Wallbox is also connected only with 2 phases not 3 phases

[image]

As you can see the charging of the golf is shown with 3 phases in my opinion even though I have only 2 setup (during setup I was asked for max. kW and 11 or 22  kW were offered so I had chosen "different" and then was asked for CURRENT (16A) and number of phases.

So the charging state shows 3 phases and 11 kW while the top lines are showing the real meter consumption of about 7,x kW if I remember right. That figure is most likely right for the moment, could be a bit more but I took the screenshot during start of the charging session. 
Consumption 7,2 or what so ever would fit the 2x 16A x 230 V charging I had chosen during initial setup.

But it might also be true that I had meanwhile changed the setup via UI here
[image]
from 3 to 1 phases back cause if I remember right I was wondering why I was shown this page after my first start with 3 or 1 phases and therefore wondering what had happened to my initial setup where I had clearly chosen 2 phases and 16A cause 3,6 kW and 11 kW or 22 kW did not fit.

There is an issue for sure for all those 2 phase charging cars like e-Up and e-GOLF cause  I missed the logic why I am asked for 3 or 1 phase charging if I set it up for 2 phases and only 2 phases are connected. I also intend to "turn off" the second phase via sonoff switch in case there is not much sun shine anymore and then the charger would only recognice 1 phase instead of 2 phases

### Steps to reproduce

1.connect the car
2.start charging
3.and then you will watch that 3 phases are appearing.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

# log: debug
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# unique installation id
plant: 12345678

interval: 20s # control cycle interval

meters:
- type: template
  template: growatt-hybrid 
  id: 1  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: grid  
  modbus: rs485serial  
  name: grid1
- type: template
  template: growatt-hybrid 
  id: 1  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: pv  
  modbus: rs485serial  
  name: pv2
- type: template
  template: growatt-hybrid 
  id: 1  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: battery  
  modbus: rs485serial  
  name: battery3
  capacity: 14.0

chargers:
- type: template
  template: pulsarplus 
  timeout: 2m  
  name: wallbox4

vehicles:
- name: vw
  title: GOLF          
  type: template
  template: offline
  capacity: 32.5 # in kWh

loadpoints:
- title: Garage
  charger: wallbox4
  mode: minpv  # charge mode (off, now, minpv, pv)
  vehicle: vw # set default vehicle (disables vehicle detection)
  # resetOnDisconnect: true # set defaults when vehicle disconnects
  phases: 1 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
  minCurrent: 6 # minimum charge current (default 6A)
  maxCurrent: 16 # maximum charge current (default 16A)

site:
  title: Zuhaus
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 200

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either
    # type: tibber
    # token: "12345678"
    # or
    type: fixed
    price: 0.272 # 0.272 [currency]/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.081

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
```

### Log details

```text
there is no error I guess that I could show except that screenshot
```

### What type of operating system are you running?

Linux

### Version

pi@PiOS64bit:~ $ evcc -v evcc version 0.129.0

typxxi, 2024-08-01:
there might be even a second issue cause when I choose fast charging with just 1 phase as setup like this

[image]

then I get this result, where the 3,6 kW charging indicator of just 1 phase would be right

[image]

But the truth is that the charger is running 100% but on 2 phases and therefore shows a 7,7 kW consumption figure.
My charger can not switch dynamicly from 3 to 2 or 1 phase so I have to set that up properly and maybe turn of the 2nd phase during winter.

[image]

If I choose 1 phase I should only get a fast charging based on 1 phase or 3,7 kW.
If I choose 3 phases in the UI then I can expect a fastest fast charging based on all phases available / connected in my case 2 with 7,4 kW or so. 

But I get only 3,6 kW charging or 11 kW charging. Of cause this is not the same bug as above but I guess they belong more or less together.

VolkerK62, 2024-08-02:
I guess your wallbox do not have a meter. So the power is calculated based on the configured phases and the released current.
If your car can only load with 2 phases, you must add `phases: 2` in the vehicle part of your config.