evcc-io/evcc #15196: wrong charge power for inactive loadpoint

n0cturne, 2024-08-02:
### Describe the bug

It seems that in some cases the chargepower of the active loadpoint will also be used for the second inactive wallbox which result in over power in the circuit:

At first lp-1 is active loading and lp-2 no car is connected and not loading
_[lp-1  ] DEBUG 2024/08/02 10:20:32 charge power: 9051W
[lp-1  ] DEBUG 2024/08/02 10:20:32 charge currents: [12.9 12.5 12.6]A
[lp-2  ] DEBUG 2024/08/02 10:20:32 charge power: 0W
[lp-2  ] DEBUG 2024/08/02 10:20:32 charge currents: [0 0 0]A
[circuit-wallboxen] DEBUG 2024/08/02 10:20:32 power: 9051W
[circuit-wallboxen] DEBUG 2024/08/02 10:20:32 current: 12.9A
[circuit-wallboxen] TRACE 2024/08/02 10:20:32 validate current: 12.9A + (0A -> 0A) <= 16A ok
[circuit-wallboxen] TRACE 2024/08/02 10:20:32 validate power: 9051W + (0W -> 0W) <= 12000W ok_

But afterwards the lp-2 has the same power then lp-1 but with currents 0A
_[lp-1  ] DEBUG 2024/08/02 10:20:42 charge power: 9040W
[lp-1  ] DEBUG 2024/08/02 10:20:42 charge currents: [12.8 12.5 12.5]A
[lp-2  ] DEBUG 2024/08/02 10:20:42 charge power: 9040W
[lp-2  ] DEBUG 2024/08/02 10:20:42 charge currents: [0 0 0]A
[circuit-wallboxen] WARN 2024/08/02 10:20:42 over power detected: 18080W > 12000W
[circuit-wallboxen] DEBUG 2024/08/02 10:20:42 current: 12.8A
[circuit-wallboxen] TRACE 2024/08/02 10:20:43 validate current: 12.8A + (12.6A -> 11.8A) <= 16A ok
[circuit-wallboxen] DEBUG 2024/08/02 10:20:43 validate power: 18080W + (9040W -> 8108.9W) > 12000W capped at 2960W_

And this leads to the stop of the charging:
_[lp-1  ] INFO 2024/08/02 10:21:02 stop charging <-_

And later starts again:
_[lp-1  ] INFO 2024/08/02 10:22:42 start charging ->_

This occurs several times:
[image]

### Steps to reproduce

1. start charging on LP1 and LP2 is unused
[evcc-20240802-102812-trace.log](https://github.com/user-attachments/files/16468260/evcc-20240802-102812-trace.log)

### Configuration details

```yaml
chargers:
- type: template
  template: heidelberg 
  id: 2  
  device: /dev/ttyUSB_Wallbox  
  baudrate: 19200  
  comset: 8E1  
  modbus: rs485serial  
  name: wallbox2
- type: template
  template: heidelberg 
  id: 1  
  device: /dev/ttyUSB_Wallbox 
  baudrate: 19200  
  comset: 8E1  
  modbus: rs485serial  
  name: wallbox1

circuits:
- name: wallboxen # eindeutiger Name, wird als Referenz für andere Circuits verwendet
  title: Wallboxen # Anzeige in der UI (später)
  maxCurrent: 16 # 16A (optional)
  maxPower: 12000 # 12kW (optional)

loadpoints:
- title: Garage Jochen
  circuit: wallboxen
  charger: wallbox2
  mode: off
#  meter: my_charger
- title: Garage Laura
  circuit: wallboxen
  charger: wallbox1
  mode: off
```

### Log details

```text
attached the trace log
```

### What type of operating system are you running?

Linux

### Version

0.129.0

andig, 2024-08-02:
Da passt tatsächlich etwas nicht:

```
lp-1: id 2
lp-2: id 1
```

Richtig:

```
[site  ] DEBUG 2024/08/02 10:20:32 ----
[heidel] TRACE 2024/08/02 10:20:32 modbus: send 02 04 00 0e 00 01 50 3a
[heidel] TRACE 2024/08/02 10:20:32 modbus: recv 02 04 02 23 5b a5 fb
[lp-1  ] DEBUG 2024/08/02 10:20:32 charge power: 9051W
[heidel] TRACE 2024/08/02 10:20:32 modbus: send 02 04 00 06 00 03 50 39
[heidel] TRACE 2024/08/02 10:20:32 modbus: recv 02 04 06 00 81 00 7d 00 7e 58 45
[lp-1  ] DEBUG 2024/08/02 10:20:32 charge currents: [12.9 12.5 12.6]A

[heidel] TRACE 2024/08/02 10:20:32 modbus: send 01 04 00 0e 00 01 50 09
[heidel] TRACE 2024/08/02 10:20:32 modbus: recv 01 04 02 00 00 b9 30
[lp-2  ] DEBUG 2024/08/02 10:20:32 charge power: 0W
[heidel] TRACE 2024/08/02 10:20:32 modbus: send 01 04 00 06 00 03 50 0a
[heidel] TRACE 2024/08/02 10:20:32 modbus: recv 01 04 06 00 00 00 00 00 00 60 93
[lp-2  ] DEBUG 2024/08/02 10:20:32 charge currents: [0 0 0]A
```

lp-2 falsch:

```
[site  ] DEBUG 2024/08/02 10:20:42 ----
[heidel] TRACE 2024/08/02 10:20:42 modbus: send 02 04 00 05 00 01 21 f8
[heidel] TRACE 2024/08/02 10:20:42 modbus: recv 02 04 02 00 07 bc f2
[heidel] TRACE 2024/08/02 10:20:42 modbus: send 02 04 00 0e 00 01 50 3a
[heidel] TRACE 2024/08/02 10:20:42 modbus: recv 02 04 02 23 50 e4 3c
[lp-1  ] DEBUG 2024/08/02 10:20:42 charge power: 9040W
[heidel] TRACE 2024/08/02 10:20:42 modbus: send 02 04 00 06 00 03 50 39
[heidel] TRACE 2024/08/02 10:20:42 modbus: recv 02 04 06 00 80 00 7d 00 7d 25 84
[lp-1  ] DEBUG 2024/08/02 10:20:42 charge currents: [12.8 12.5 12.5]A

[heidel] TRACE 2024/08/02 10:20:42 modbus: send 02 04 00 05 00 01 21 f8
[heidel] TRACE 2024/08/02 10:20:42 modbus: recv 02 04 02 00 07 bc f2
[heidel] TRACE 2024/08/02 10:20:42 modbus: send 02 04 00 0e 00 01 50 3a
[heidel] TRACE 2024/08/02 10:20:42 modbus: recv 02 04 02 23 50 e4 3c
[lp-2  ] DEBUG 2024/08/02 10:20:42 charge power: 9040W
[heidel] TRACE 2024/08/02 10:20:42 modbus: send 01 04 00 06 00 03 50 0a
[heidel] TRACE 2024/08/02 10:20:42 modbus: recv 01 04 06 00 00 00 00 00 00 60 93
[lp-2  ] DEBUG 2024/08/02 10:20:42 charge currents: [0 0 0]A
```

    modbus: send 02

bedeutet Abfrage von id 2- das ist nicht korrekt. Ursache allerdings völlig unklar.

andig, 2024-08-02:
@n0cturne das scheint ja nur manchmal zu passieren. Gibt es dafür ein Muster? Könntest Du (evtl. über Docker?) mal eine Testversion starten die ich Dir zur Verfügung stellen würde? Ich habe die böse Vermutung, dass es sich hier um eine Race Condition im Code aufgrund des Heartbeats der Heidelberg handelt. Die Lösung wäre https://github.com/evcc-io/evcc/pull/13676, es wäre aber schön, das mal zu validieren.

n0cturne, 2024-08-02:
Hi @andig 
Problem tritt erst seit kurzem auf, da ich erst mit Version 0.129.0 auf cuircits umgestellt habe. Problem war eventuell auch schon vorher vorhanden, ist nur nie aufgefallen.
Hab leider EVCC nativ auf meinem Raspberry laufen, wenn du das in ein Nightly bekommst, könnte ich es einmal ausprobieren.

RTTTC, 2024-08-03:
@n0cturne When I look at your log, it seems to me that your interval is a bit too shirt for your devices? It barelly finishes all communication before new interval cycle starts..
Try 14s ? 15s?
(Have you changed default Heidelberg timeout of 15s ?) 
This may be unrelated to your problem but still worth checking out.

n0cturne, 2024-08-03:
Hi @RTTTC 
Which timeout do you mean for the wallbox? Is there a possibility to set different Timer for each device? In other Apps or on the webpage of my inverter, they actualize the data each 5 seconds. Ist there the possibility to set 5 seconds for the SMA invertiert and 15 seconds for the wallbox?  Otherwise for sure. I can try to increase the Timer to 15 seconds. Thanks for the Hint!

RTTTC, 2024-08-03:
There is no need for a fast `interval` in evcc
Often it is counterproductive.

I am not sure how would Heidelberg wallbox behave if you set evcc`interval` say 30s ?
(Wallbox has default **WatchDog TimeOut** of 15s)

I have configured my Heidelberg to have timeout of 65s , my evcc `interval: 15s` and haven’t noticed any problems myself.
But you have two Heidelbergs, so it is different.

(Register 257, WatchDog TimeOut , page 13) https://www.amperfied.de/wp-content/uploads/2023/03/20220809_AMP-Erweiterte-ModBus-Registerbeschreibung.pdf#page13

@premultiply any feedback from you?

andig, 2024-08-03:
Anyway, issue fixed in nightly hopefully.

n0cturne, 2024-08-03:
Hi @andig,
The nightly version doesn't work at all on my side... :-( 
I have update to the latest Nightly (unstable) version (evcc 0.129.0 (03f62e55)) and the system doesnt start and hangs with _heartbeat: serial: timeout_:
_Aug 03 12:08:04 raspberrypi evcc[21025]: [main  ] INFO 2024/08/03 12:08:04 evcc 0.129.0 (03f62e55)
Aug 03 12:08:04 raspberrypi evcc[21025]: [main  ] INFO 2024/08/03 12:08:04 using config file: /etc/evcc.yaml
Aug 03 12:08:04 raspberrypi evcc[21025]: [db    ] INFO 2024/08/03 12:08:04 using sqlite database: /var/lib/evcc/evcc.db
Aug 03 12:08:05 raspberrypi evcc[21025]: [main  ] INFO 2024/08/03 12:08:05 listening at :7070
Aug 03 12:08:11 raspberrypi evcc[21025]: [main  ] FATAL 2024/08/03 12:08:11 cannot create charger 'wallbox2': cannot create charger type 'template': cannot create charger type 'heidelberg': serial: timeout
Aug 03 12:08:11 raspberrypi evcc[21025]: [main  ] FATAL 2024/08/03 12:08:11 will attempt restart in: 15m0s
Aug 03 12:10:04 raspberrypi evcc[21025]: [heidel] ERROR 2024/08/03 12:10:04 heartbeat: serial: timeout
Aug 03 12:10:09 raspberrypi evcc[21025]: [heidel] ERROR 2024/08/03 12:10:09 heartbeat: serial: timeout
Aug 03 12:10:14 raspberrypi evcc[21025]: [heidel] ERROR 2024/08/03 12:10:14 heartbeat: serial: timeout_

To check that nothing is wrong with my setup, I switched back to the stable version (evcc 0.129.0) and everthing works fine again:
_Aug 03 12:17:39 raspberrypi evcc[23194]: [main  ] INFO 2024/08/03 12:17:39 evcc 0.129.0
Aug 03 12:17:39 raspberrypi evcc[23194]: [main  ] INFO 2024/08/03 12:17:39 using config file: /etc/evcc.yaml
Aug 03 12:17:39 raspberrypi evcc[23194]: [db    ] INFO 2024/08/03 12:17:39 using sqlite database: /var/lib/evcc/evcc.db
Aug 03 12:17:40 raspberrypi evcc[23194]: [main  ] INFO 2024/08/03 12:17:40 listening at :7070
Aug 03 12:17:45 raspberrypi evcc[23194]: [lp-1  ] DEBUG 2024/08/03 12:17:45 charge total import: 4114.767kWh
Aug 03 12:17:45 raspberrypi evcc[23194]: [lp-2  ] DEBUG 2024/08/03 12:17:45 charge total import: 83.997kWh
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] WARN 2024/08/03 12:17:45 vehicle 'MG MG5 (Benjamin)' does not support automatic detection
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] WARN 2024/08/03 12:17:45 vehicle 'MG MG4 (Alexander)' does not support automatic detection
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] WARN 2024/08/03 12:17:45 vehicle 'Skoda Enyaq (Christian)' does not support automatic detection
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] WARN 2024/08/03 12:17:45 vehicle 'Hyundai Kona (Margit)' does not support automatic detection
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] INFO 2024/08/03 12:17:45 site config:
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] INFO 2024/08/03 12:17:45   meters:      grid ✓ pv ✓ battery ✓
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] INFO 2024/08/03 12:17:45     grid:      power ✓ energy ✓ currents ✓
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] INFO 2024/08/03 12:17:45     pv 1:      power ✓ energy ✓ currents ✗
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] INFO 2024/08/03 12:17:45     pv 2:      power ✓ energy ✓ currents ✓
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] INFO 2024/08/03 12:17:45     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] INFO 2024/08/03 12:17:45   vehicles:
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] INFO 2024/08/03 12:17:45     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] INFO 2024/08/03 12:17:45     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] INFO 2024/08/03 12:17:45     vehicle 3: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] INFO 2024/08/03 12:17:45     vehicle 4: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] INFO 2024/08/03 12:17:45     vehicle 5: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
Aug 03 12:17:45 raspberrypi evcc[23194]: [lp-1  ] INFO 2024/08/03 12:17:45 loadpoint 1:
Aug 03 12:17:45 raspberrypi evcc[23194]: [lp-1  ] INFO 2024/08/03 12:17:45   mode:        off
Aug 03 12:17:45 raspberrypi evcc[23194]: [lp-1  ] INFO 2024/08/03 12:17:45   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✓
Aug 03 12:17:45 raspberrypi evcc[23194]: [lp-1  ] INFO 2024/08/03 12:17:45   meters:      charge ✓
Aug 03 12:17:45 raspberrypi evcc[23194]: [lp-1  ] INFO 2024/08/03 12:17:45     charge:    power ✓ energy ✓ currents ✓
Aug 03 12:17:45 raspberrypi evcc[23194]: [lp-2  ] INFO 2024/08/03 12:17:45 loadpoint 2:
Aug 03 12:17:45 raspberrypi evcc[23194]: [lp-2  ] INFO 2024/08/03 12:17:45   mode:        off
Aug 03 12:17:45 raspberrypi evcc[23194]: [lp-2  ] INFO 2024/08/03 12:17:45   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✓
Aug 03 12:17:45 raspberrypi evcc[23194]: [lp-2  ] INFO 2024/08/03 12:17:45   meters:      charge ✓
Aug 03 12:17:45 raspberrypi evcc[23194]: [lp-2  ] INFO 2024/08/03 12:17:45     charge:    power ✓ energy ✓ currents ✓
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] DEBUG 2024/08/03 12:17:45 set buffer soc: 95
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] DEBUG 2024/08/03 12:17:45 set priority soc: 80
Aug 03 12:17:45 raspberrypi evcc[23194]: [site  ] DEBUG 2024/08/03 12:17:45 set battery discharge control: false
[... cut ...]
Aug 03 12:20:04 raspberrypi evcc[24198]: [main  ] INFO 2024/08/03 12:20:04 listening at :7070
Aug 03 12:20:10 raspberrypi evcc[24198]: [main  ] FATAL 2024/08/03 12:20:10 cannot create charger 'wallbox1': cannot create charger type 'template': cannot create charger type 'heidelberg': serial: timeout
Aug 03 12:20:10 raspberrypi evcc[24198]: [main  ] FATAL 2024/08/03 12:20:10 will attempt restart in: 15m0s_

Attached you can find a trace log.
[evcc-20240803-122144-trace.log](https://github.com/user-attachments/files/16479145/evcc-20240803-122144-trace.log)

andig, 2024-08-03:
Mhhm- das Log zeigt gar kein Modbus mehr:

```
[main  ] INFO 2024/08/03 12:21:15 evcc 0.129.0 (03f62e55)
[main  ] INFO 2024/08/03 12:21:15 using config file: /etc/evcc.yaml
[main  ] INFO 2024/08/03 12:21:16 listening at :7070
[main  ] FATAL 2024/08/03 12:21:22 cannot create charger 'wallbox2': cannot create charger type 'template': cannot create charger type 'heidelberg': serial: timeout
[main  ] TRACE 2024/08/03 12:21:22 GET https://api.github.com/repos/evcc-io/evcc/releases/latest
[main  ] FATAL 2024/08/03 12:21:22 will attempt restart in: 15m0s
```

Da kann einfach das Device nicht geöffnet werden? Warum ist mir unklar. Ohne Debugger kommen wir da wohl nicht weiter :(

n0cturne, 2024-08-03:
mehr als das Tracelog kann ich leider auch nicht machen... :-(

n0cturne, 2024-08-04:
gerade noch einmal mit der neusten Nightly probiert, startet leider immer noch nicht:
_[main ] INFO 2024/08/04 18:58:07 evcc 0.129.0 (f3be63f2)
[main ] INFO 2024/08/04 18:58:07 using config file: /etc/evcc.yaml
[db ] INFO 2024/08/04 18:58:07 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2024/08/04 18:58:09 listening at :7070
[main ] FATAL 2024/08/04 18:58:16 cannot create charger 'wallbox1': cannot create charger type 'template': cannot create charger type 'heidelberg': serial: timeout
[main ] FATAL 2024/08/04 18:58:16 will attempt restart in: 15m0s_

RTTTC, 2024-08-04:
So, have you tried longer interval with the evcc version that works, to see if it fixes your original problem?

n0cturne, 2024-08-04:
Hi @RTTTC 

Andig mentioned that the original Problem would be fixed with the linked changelist and I should Test this in a Nightly build. Therefore I Gabert Test it with the Adaption of the Timer.
But my car is now empty enough so that I can also try with increased Timer. What time would you suggest? 14s? 
also for my unterstanding, the Wallibox needs with the default values each 15 seconds messages from the Master (EVCC), therefore more then 14s would lead to timeouts?

RTTTC, 2024-08-04:
Try the default 30s 
Also - observe the log to see how long it takes to communicate for all of your devices. Interval must always be longer than that, with some healthy reserve.
If 30s does not work - try 14s

andig, 2024-08-06:
@n0cturne was heisst das jetzt? Timeout behoben? Dann würde ich nochmal neues Nightly mit weiterer Diagnose zur Verfügung stellen.

n0cturne, 2024-08-06:
Hallo @andig,
nein, die erhöhung des timers (interval: 30s # control cycle interval) von 10s auf 30s brachte keine Besserung. lp-2 wird immer noch als ladend angezeigt, 
_[lp-1 ] DEBUG 2024/08/06 09:25:20 charge power: 5829W
[lp-1 ] DEBUG 2024/08/06 09:25:20 charge currents: [8.1 7.9 8]A
[lp-2 ] DEBUG 2024/08/06 09:25:20 charge power: 5826W
[lp-2 ] DEBUG 2024/08/06 09:25:20 charge currents: [0 0 0]A_
auch in der GUI:
[image]
Hab dir bereits eine Mail gesendet

andig, 2024-08-06:
> nein, die Erhöhung des timers (interval: 30s # control cycle interval) von 10s auf 30s brachte keine Besserung.

Bzgl. der Messwerte?

> Timeout behoben?

Der scheint weg zu sein? Hast du noch was andere geändert?

Die Sache mit den Messwerten sieht jetzt aber auch anders aus:

    [lp-1 ] DEBUG 2024/08/06 09:25:20 charge power: 5829W
    [lp-2 ] DEBUG 2024/08/06 09:25:20 charge power: 5826W

Das ist nicht mehr- wie vorher- der identische Wert. Also völlig unklar, wo der her kommt.

> Hab dir auch bereits eine Mail bzgl debug session gesendet

Ist nicht angekommen :(

n0cturne, 2024-08-06:
Hallo @andig,

> Bzgl. der Messwerte?

RTTC hatte ja vorgeschlagen, dass ich den Timer anpassen soll und damit das Problem eventuell behoben ist. Leider hat das keine Besserung gebracht. 

> Das ist nicht mehr- wie vorher- der identische Wert. Also völlig unklar, wo der her kommt.

hatte mich auch ein wenig stutzig gemacht, warum es jetzt auch in der GUI doppelt angezeigt wird. Hab den PI einmal neu gestartet, danach hat es wieder funktioniert. Wenn ich mir jedoch den Trace [evcc-20240806-103349-trace.log](https://github.com/user-attachments/files/16509432/evcc-20240806-103349-trace.log) ansehe, dann fragt hier EVCC immer die gleiche ID (02) an, ordnet sie dann aber unterschiedlichen LPs zu:

```
Zeile  2046: [site  ] DEBUG 2024/08/06 10:24:50 ----
Zeile  2047: [heidel] TRACE 2024/08/06 10:24:50 modbus: send 02 04 00 0e 00 01 50 3a
Zeile  2048: [heidel] TRACE 2024/08/06 10:24:50 modbus: recv 02 04 02 16 a9 33 2e
Zeile  2049: [lp-1  ] DEBUG 2024/08/06 10:24:50 charge power: 5801W
Zeile  2053: [heidel] TRACE 2024/08/06 10:24:50 modbus: send 02 04 00 0e 00 01 50 3a
Zeile  2054: [heidel] TRACE 2024/08/06 10:24:50 modbus: recv 02 04 02 16 a9 33 2e
Zeile  2055: [lp-2  ] DEBUG 2024/08/06 10:24:50 charge power: 5801W
```

Das ist jedoch das gleiche verhalten wie im Ursprungstrace (evcc-20240802-102812-trace.log), auch hier fragt EVCC die gleiche ID an, weist sie dann aber verschiedenen LPs zu
```
Zeile  1618: [heidel] TRACE 2024/08/02 10:20:42 modbus: send 02 04 00 0e 00 01 50 3a
Zeile  1619: [heidel] TRACE 2024/08/02 10:20:42 modbus: recv 02 04 02 23 50 e4 3c
Zeile  1620: [lp-1  ] DEBUG 2024/08/02 10:20:42 charge power: 9040W
Zeile  1626: [heidel] TRACE 2024/08/02 10:20:42 modbus: send 02 04 00 0e 00 01 50 3a
Zeile  1627: [heidel] TRACE 2024/08/02 10:20:42 modbus: recv 02 04 02 23 50 e4 3c
Zeile  1628: [lp-2  ] DEBUG 2024/08/02 10:20:42 charge power: 9040W
Zeile  1632: [circuit-wallboxen] WARN 2024/08/02 10:20:42 over power detected: 18080W > 12000W
```

Im Gutfall ist diese dann korrekt (ID 02 -> LP-1 und ID 01 -> LP-2):
```
Zeile  1439: [site  ] DEBUG 2024/08/02 10:20:32 ----
Zeile  1440: [heidel] TRACE 2024/08/02 10:20:32 modbus: send 02 04 00 0e 00 01 50 3a
Zeile  1442: [heidel] TRACE 2024/08/02 10:20:32 modbus: recv 02 04 02 23 5b a5 fb
Zeile  1443: [lp-1  ] DEBUG 2024/08/02 10:20:32 charge power: 9051W
Zeile  1447: [heidel] TRACE 2024/08/02 10:20:32 modbus: send 01 04 00 0e 00 01 50 09
Zeile  1448: [heidel] TRACE 2024/08/02 10:20:32 modbus: recv 01 04 02 00 00 b9 30
Zeile  1449: [lp-2  ] DEBUG 2024/08/02 10:20:32 charge power: 0W
```

Bzgl. meinem weiteren Problem, dass das Nightly nicht startet, bin ich leider nicht weitergekommen. Das log sagt hier einfach wenig aus. Eine weitere Mail hab ich dir gerade gesendet, hoffentlich kommt diese bei dir an.

n0cturne, 2024-08-06:
@andig 
Daten für die debug session hab ich dir geschickt.

andig, 2024-08-09:
@n0cturne bitte nichmal aktuelles nightly probieren.

n0cturne, 2024-08-13:
@andig ,
Sieht leider nicht besser aus… :-(

`[main ] INFO 2024/08/13 12:43:11 evcc 0.129.0 (9bf8dc7b)
[main ] INFO 2024/08/13 12:43:11 using config file: /etc/evcc.yaml
[db ] INFO 2024/08/13 12:43:11 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2024/08/13 12:43:13 listening at :7070
[main ] FATAL 2024/08/13 12:43:20 cannot create charger 'wallbox2': cannot create charger type 'template': cannot create charger type 'heidelberg': serial: timeout
[main ] FATAL 2024/08/13 12:43:20 will attempt restart in: 15m0s `

andig, 2024-08-17:
Meanwhile duplicate of https://github.com/evcc-io/evcc/issues/15437

closed 2024-08-17 by andig

n0cturne, 2024-08-18:
> @andig , Sieht leider nicht besser aus… :-(
>
> `[main ] INFO 2024/08/13 12:43:11 evcc 0.129.0 (9bf8dc7b) [main ] INFO 2024/08/13 12:43:11 using config file: /etc/evcc.yaml [db ] INFO 2024/08/13 [...]

Problem tritt mit [Änderung ](https://github.com/evcc-io/evcc/commit/9ad7399b1d738605964cbb8d483ed83c356f6f4b) nicht mehr auf, EVCC startet ohne Probleme (getestet mit Nightly evcc 0.129.0 (9ad7399b))!

Das ürsprüngliche Problem werde ich bei Gelegenheit testen und schauen ob es mit dem redesign verbessert wurde!

Vielen herzlichen Danke!

andig, 2024-08-18:
KLASSE!!!