evcc-io/evcc #15200: Keba P30 (DE Edition)

asterix555, 2024-08-02:
### Describe the bug

Hi,
we have a Keba KC-P30-EC220112-000-DE ("DE 440 Edition") running in Modbus-TCP including phase-switch option.
Unfortunately, the current charge is not reported and not used for pv optimized charging therefore:

[site ] DEBUG 2024/08/02 14:55:39 ----
[lp-1 ] DEBUG 2024/08/02 14:55:39 charge power: 0W
[site ] DEBUG 2024/08/02 14:55:40 pv power: 5141W
[site ] DEBUG 2024/08/02 14:55:40 grid meter: -518W
[site ] DEBUG 2024/08/02 14:55:40 grid currents: [0 0 0]A
[site ] DEBUG 2024/08/02 14:55:40 site power: -518W
[lp-1 ] DEBUG 2024/08/02 14:55:40 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/02 14:55:40 charger status: B
[lp-1 ] DEBUG 2024/08/02 14:55:40 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/02 14:55:40 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/02 14:55:40 pv charge current: 0.751A = 0A + 0.751A (-518W @ 3p)
[site ] DEBUG 2024/08/02 14:56:09 ---- 

When changing protocol from Modbus-TCP to UDP, the charge power gets reported correctly. As 1p/3p switch doesn't work with in Keba UDP (in EVCC), this is not a recommended workaround....

regards

### Steps to reproduce

1. setup EVCC with keba-modbus template and fronius smartmeter and PV
2. start charging

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070
interval: 30s # control cycle interval

log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

meters:
# AMIS Reader (Smartmeter) in GEN24-emulation mode
  - name: amis
    type: template
    template: fronius-gen24
    usage: grid
    host: 10.10.110.239 # IP-Adresse oder Hostname
    port: 502 # Port (optional)

  - name: fronius-gen24
    type: template
    template: fronius-gen24
    usage: pv
    host: 10.10.110.251 # IP-Adresse oder Hostname
    port: 502 # Port (optional)

  - name: fronius-symo
    type: template
    template: fronius-gen24
    usage: pv
    host: 10.10.110.252 # IP-Adresse oder Hostname
    port: 502 # Port (optional)

chargers:
# UDP - works fine but doesnt support 1p/3p switch
#  - name: wallbox
#    type: template
#    template: keba
#    host: 10.10.110.209

# Modbus TCP - doesnt report charge power
  - name: wallbox
    type: template
    template: keba-modbus
    host: 10.10.110.209
    modbus: tcpip
    id: 255
    port: 502

site:
  title: Home # display name for UI
  meters:
    grid: amis
    pv:
      - fronius-gen24
      - fronius-symo
  residualPower: 0 # additional household usage margin
  maxGridSupplyWhileBatteryCharging: 0 # ignore battery charging if AC consumpt>

loadpoints:
  - title: Garage # display name for UI
    charger: wallbox # charger
    mode: "pv" # set default charge mode, use "off" to disable by default if ch>
    vehicle: cupra_born # set default vehicle (disables vehicle detection)
    phases: 0 # electrical connection (normal charger: default 3 for 3 phase, 1>
    priority: 0 # relative priority for concurrent charging in PV mode with mul>
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your >
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR O>
      poll:
        mode: charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (>
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero,>
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (defa>

sponsortoken: "well-known"
```

### Log details

```text
[site ] DEBUG 2024/08/02 15:05:39 ----
[lp-1 ] DEBUG 2024/08/02 15:05:39 charge power: 0W
[site ] DEBUG 2024/08/02 15:05:40 pv power: 6499W
[site ] DEBUG 2024/08/02 15:05:40 grid meter: -1867W
[site ] DEBUG 2024/08/02 15:05:40 grid currents: [0 0 0]A
[site ] DEBUG 2024/08/02 15:05:40 site power: -1867W
[lp-1 ] DEBUG 2024/08/02 15:05:40 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/02 15:05:40 charger status: B
[lp-1 ] DEBUG 2024/08/02 15:05:40 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/02 15:05:40 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/02 15:05:40 pv charge current: 2.71A = 0A + 2.71A (-1867W @ 3p)
[site ] DEBUG 2024/08/02 15:06:09 ----
```

### What type of operating system are you running?

Docker container

### Version

0.129.0

VolkerK62, 2024-08-02:
Ich glaube in einer Diskussion kam mal heraus, dass die DE-Edition nicht (vollständig) per Modbus gesteuert werden kann.

asterix555, 2024-08-02:
> > Unfortunately, the current charge is not reported and not used for pv optimized charging therefore:
> 
> Was meint das?

Laut Logs werden 0 Watt geladen, während jedoch zB 8kW ins Auto fließen. Die 8KW fehlen dann natürlich bei der Überschuss-Berechnung da 0 Watt reportet werden l. EVCC regelt entsprechend runter. Entsprechend ist beim nächsten Regelzyklus natürlich wieder mehr Überschuss verfügbar...

premultiply, 2024-08-02:
> [lp-1 ] DEBUG 2024/08/02 14:55:40 charger status: B

Hier wird nicht geladen.

asterix555, 2024-08-04:
> > [lp-1 ] DEBUG 2024/08/02 14:55:40 charger status: B
> 
> Hier wird nicht geladen.

[image]
[image]

Nach Umstellen auf UDP funktioniert alles wie gewünscht:
[image]

Logfile (Modbus-TCP):
[evcc-20240804-105412-debug.log](https://github.com/user-attachments/files/16486956/evcc-20240804-105412-debug.log)

premultiply, 2024-08-04:
Waren bei der Keba nicht irgendwo DIP-Schalter die man entsprechend der Steuerungsart einstellen muss?

asterix555, 2024-08-04:
Dort kann man die nur die Netzwerksteuerung generell ein- oder ausschalten. Über UDP bzw. Modbus-TCP muss man dann auch noch festlegen ob die Ansteuerung für die 1/3p-Umschaltung zB via Modbus TCP erfolgen soll.

andig, 2024-08-05:
> Laut Logs werden 0 Watt geladen, während jedoch zB 8kW ins Auto fließen. 

Mal bitte

    evcc charger --log trace,db:error

während das Auto lädt.

asterix555, 2024-08-05:
okay, da dürfte irgendwo der Fehler liegen. Es wird sowohl für 1020 als auch für 1008 (Current L1 in mA) nur nullen retourniert (schnell getestet mit Node-Red). Habe dann auch mit UDP abgefragt, auch hier kommt für L1, L2, L3 und P jeweils 0.

Der "DE"-Edition fehlt ja angeblich der Stromzähler - wobei ja im UDP-Modus sehr wohl ein realistischer Wert angezeigt (geschätzt?) wird.

```
[lp-1 ] DEBUG 2024/08/05 20:16:35 charge power: 0W
[sunspec] TRACE 2024/08/05 20:16:35 modbus: send 00 33 00 00 00 06 01 03 9d 3f 00 08
[sunspec] TRACE 2024/08/05 20:16:35 modbus: recv 00 33 00 00 00 13 01 03 10 ff fb ff fe ff fe ff fe ff ff ff ff 00 04 ff ff
[sunspec] TRACE 2024/08/05 20:16:35 modbus: send 00 34 00 00 00 06 01 03 9d 47 00 14
[sunspec] TRACE 2024/08/05 20:16:35 modbus: recv 00 34 00 00 00 2b 01 03 28 00 01 4d 50 50 54 20 31 00 00 00 00 00 00 00 00 00 00 50 0c cb d3 29 c4 1d 08 05 d9 2e 43 d3 6a 80 00 ff ff ff ff ff ff
[sunspec] TRACE 2024/08/05 20:16:35 model 160 block 1 point DCW: 106.92
[sunspec] TRACE 2024/08/05 20:16:35 modbus: send 00 35 00 00 00 06 01 03 9d 3f 00 08
[sunspec] TRACE 2024/08/05 20:16:35 modbus: recv 00 35 00 00 00 13 01 03 10 ff fb ff fe ff fe ff fe ff ff ff ff 00 04 ff ff
[sunspec] TRACE 2024/08/05 20:16:35 modbus: send 00 36 00 00 00 06 01 03 9d 5b 00 14
[sunspec] TRACE 2024/08/05 20:16:35 modbus: recv 00 36 00 00 00 2b 01 03 28 00 02 4d 50 50 54 20 32 00 00 00 00 00 00 00 00 00 00 2f f8 8f 90 11 a1 0f 53 05 8d 2e 43 d3 6a 80 00 ff ff ff ff ff ff
[sunspec] TRACE 2024/08/05 20:16:35 model 160 block 2 point DCW: 45.13
[sunspec] TRACE 2024/08/05 20:16:35 modbus: send 00 31 00 00 00 06 01 03 9d 49 00 08
[sunspec] TRACE 2024/08/05 20:16:36 modbus: recv 00 31 00 00 00 13 01 03 10 ff fe ff fe ff fe 00 00 00 00 00 00 00 02 ff ff
[sunspec] TRACE 2024/08/05 20:16:36 modbus: send 00 32 00 00 00 06 01 03 9d 51 00 14
[sunspec] TRACE 2024/08/05 20:16:36 modbus: recv 00 32 00 00 00 2b 01 03 28 00 01 53 74 72 69 6e 67 20 31 00 00 00 00 00 00 00 00 00 10 b4 a0 1c e6 00 47 d1 a0 2e 43 e1 7b 80 00 00 04 ff ff ff ff
[sunspec] TRACE 2024/08/05 20:16:36 model 160 block 1 point DCW: 73.98
[sunspec] TRACE 2024/08/05 20:16:36 modbus: send 00 33 00 00 00 06 01 03 9d 49 00 08
[sunspec] TRACE 2024/08/05 20:16:36 modbus: recv 00 33 00 00 00 13 01 03 10 ff fe ff fe ff fe 00 00 00 00 00 00 00 02 ff ff
[sunspec] TRACE 2024/08/05 20:16:36 modbus: send 00 34 00 00 00 06 01 03 9d 65 00 14
[sunspec] TRACE 2024/08/05 20:16:36 modbus: recv 00 34 00 00 00 2b 01 03 28 00 02 53 74 72 69 6e 67 20 32 00 00 00 00 00 00 00 00 00 00 2c ec 00 00 00 00 15 88 2e 43 e1 7c 80 00 00 04 ff ff ff ff
[sunspec] TRACE 2024/08/05 20:16:36 model 160 block 2 point DCW: 0
[site ] DEBUG 2024/08/05 20:16:36 pv power: 226W
[modbus] TRACE 2024/08/05 20:16:36 modbus: send 00 24 00 00 00 06 c8 03 9c 87 00 7c
[modbus] TRACE 2024/08/05 20:16:36 modbus: recv 00 24 00 00 00 fb c8 03 f8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 42 48 00 00 44 c5 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 20 19 ff 3f 84 49 ff 3f 00 00 00 00 e8 61 20 40 54 61 20 40 02 00 00 00 8c 39 ff 3f 64 22 da 95 b4 c4 04 00 58 85 22 40 00 19 ff 3f f4 4e ff 3f 40 00 00 00 00 00 00 00
[site ] DEBUG 2024/08/05 20:16:36 grid meter: 1580W
[modbus] TRACE 2024/08/05 20:16:36 modbus: send 00 25 00 00 00 06 c8 03 9c 87 00 7c
[modbus] TRACE 2024/08/05 20:16:36 modbus: recv 00 25 00 00 00 fb c8 03 f8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 42 48 00 00 44 c5 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 20 19 ff 3f 84 49 ff 3f 00 00 00 00 e8 61 20 40 54 61 20 40 02 00 00 00 8c 39 ff 3f 64 22 da 95 b4 c4 04 00 58 85 22 40 00 19 ff 3f f4 4e ff 3f 40 00 00 00 00 00 00 00
[modbus] TRACE 2024/08/05 20:16:36 modbus: send 00 26 00 00 00 06 c8 03 9c 87 00 7c
[modbus] TRACE 2024/08/05 20:16:36 modbus: recv 00 26 00 00 00 fb c8 03 f8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 42 48 00 00 44 c5 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 20 19 ff 3f 84 49 ff 3f 00 00 00 00 e8 61 20 40 54 61 20 40 02 00 00 00 8c 39 ff 3f 64 22 da 95 b4 c4 04 00 58 85 22 40 00 19 ff 3f f4 4e ff 3f 40 00 00 00 00 00 00 00
[... cut ...]
[lp-1 ] DEBUG 2024/08/05 20:16:36 charger status: B
[keba ] TRACE 2024/08/05 20:16:36 modbus: send 00 11 00 00 00 06 ff 03 03 e8 00 02
[keba ] TRACE 2024/08/05 20:16:36 modbus: recv 00 11 00 00 00 07 ff 03 04 00 00 00 03
[lp-1 ] DEBUG 2024/08/05 20:16:36 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/08/05 20:16:36 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/08/05 20:16:36 pv charge current: min 6A > 0A (1580W @ 1p, battery: false) 
```

andig, 2024-08-05:
Na wenn die keinen Zähler hat kann da auch nix kommen. Bleibt die Frage, wie sich das erkennen lässt?

andig, 2024-08-05:
Mach mal bitte ein

    evcc charger —log trace,db:error —diagnose

andig, 2024-08-06:
Bitte https://github.com/evcc-io/evcc/issues/15200#issuecomment-2269669888. Genau das versuchen ir ja schon zu erkennen...

asterix555, 2024-08-09:
> Mach mal bitte ein
> 
> ```
> evcc charger --log trace,db:error --diagnose
> ```

Bitte entschuldige die Verspätung, hier die Logs:

`[keba  ] TRACE 2024/08/09 14:05:17 modbus: send 00 04 00 00 00 06 ff 03 03 ec 00 02
[keba  ] TRACE 2024/08/09 14:05:17 modbus: recv 00 04 00 00 00 07 ff 03 04 00 00 00 05
[keba  ] TRACE 2024/08/09 14:05:17 modbus: send 00 05 00 00 00 06 ff 03 03 e8 00 02
[keba  ] TRACE 2024/08/09 14:05:17 modbus: recv 00 05 00 00 00 07 ff 03 04 00 00 00 03
[keba  ] TRACE 2024/08/09 14:05:17 modbus: send 00 06 00 00 00 06 ff 03 06 10 00 02
[keba  ] TRACE 2024/08/09 14:05:17 modbus: recv 00 06 00 00 00 07 ff 03 04 00 00 00 01
Charge status: B
Enabled:       true
Phases:        3
Diagnostic dump:
[keba  ] TRACE 2024/08/09 14:05:17 modbus: send 00 07 00 00 00 06 ff 03 03 f6 00 02
[keba  ] TRACE 2024/08/09 14:05:17 modbus: recv 00 07 00 00 00 07 ff 03 04 01 66 dd e4
	Serial:	23518692
[keba  ] TRACE 2024/08/09 14:05:17 modbus: send 00 08 00 00 00 06 ff 03 03 fa 00 02
[keba  ] TRACE 2024/08/09 14:05:17 modbus: recv 00 08 00 00 00 07 ff 03 04 03 0a 33 00
	Firmware:	3.10.51
[keba  ] TRACE 2024/08/09 14:05:17 modbus: send 00 09 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2024/08/09 14:05:17 modbus: recv 00 09 00 00 00 07 ff 03 04 00 04 c3 88
	Product:	312200
[keba  ] TRACE 2024/08/09 14:05:17 modbus: send 00 0a 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2024/08/09 14:05:17 modbus: recv 00 0a 00 00 00 07 ff 03 04 00 00 00 03
	Phases source:	3
[keba  ] TRACE 2024/08/09 14:05:17 modbus: send 00 0b 00 00 00 06 ff 03 06 10 00 02
[keba  ] TRACE 2024/08/09 14:05:17 modbus: recv 00 0b 00 00 00 07 ff 03 04 00 00 00 01
	Phases state:	1
[keba  ] TRACE 2024/08/09 14:05:17 modbus: send 00 0c 00 00 00 06 ff 03 06 42 00 02
[keba  ] TRACE 2024/08/09 14:05:17 modbus: recv 00 0c 00 00 00 07 ff 03 04 00 00 00 00
	Failsafe timeout:	0s
[main  ] INFO 2024/08/09 14:05:23 evcc 0.129.0
[main  ] INFO 2024/08/09 14:05:23 using config file: /etc/evcc.yaml
[keba  ] TRACE 2024/08/09 14:05:23 modbus: send 00 01 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2024/08/09 14:05:23 modbus: recv 00 01 00 00 00 07 ff 03 04 00 04 c3 88
[keba  ] TRACE 2024/08/09 14:05:23 modbus: send 00 02 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2024/08/09 14:05:23 modbus: recv 00 02 00 00 00 07 ff 03 04 00 00 00 03
[keba  ] TRACE 2024/08/09 14:05:23 modbus: send 00 03 00 00 00 06 ff 03 06 42 00 02
[keba  ] TRACE 2024/08/09 14:05:23 modbus: recv 00 03 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/08/09 14:05:23 modbus: send 00 04 00 00 00 06 ff 03 03 ec 00 02
[keba  ] TRACE 2024/08/09 14:05:23 modbus: recv 00 04 00 00 00 07 ff 03 04 00 00 00 05
[keba  ] TRACE 2024/08/09 14:05:23 modbus: send 00 05 00 00 00 06 ff 03 03 e8 00 02
[keba  ] TRACE 2024/08/09 14:05:23 modbus: recv 00 05 00 00 00 07 ff 03 04 00 00 00 03
[keba  ] TRACE 2024/08/09 14:05:23 modbus: send 00 06 00 00 00 06 ff 03 06 10 00 02
[keba  ] TRACE 2024/08/09 14:05:23 modbus: recv 00 06 00 00 00 07 ff 03 04 00 00 00 01
Charge status: B
Enabled:       true
Phases:        3
Diagnostic dump:
[keba  ] TRACE 2024/08/09 14:05:23 modbus: send 00 07 00 00 00 06 ff 03 03 f6 00 02
[keba  ] TRACE 2024/08/09 14:05:23 modbus: recv 00 07 00 00 00 07 ff 03 04 01 66 dd e4
	Serial:	23518692
[keba  ] TRACE 2024/08/09 14:05:23 modbus: send 00 08 00 00 00 06 ff 03 03 fa 00 02
[keba  ] TRACE 2024/08/09 14:05:23 modbus: recv 00 08 00 00 00 07 ff 03 04 03 0a 33 00
	Firmware:	3.10.51
[keba  ] TRACE 2024/08/09 14:05:23 modbus: send 00 09 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2024/08/09 14:05:23 modbus: recv 00 09 00 00 00 07 ff 03 04 00 04 c3 88
	Product:	312200
[keba  ] TRACE 2024/08/09 14:05:23 modbus: send 00 0a 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2024/08/09 14:05:23 modbus: recv 00 0a 00 00 00 07 ff 03 04 00 00 00 03
	Phases source:	3
[keba  ] TRACE 2024/08/09 14:05:23 modbus: send 00 0b 00 00 00 06 ff 03 06 10 00 02
[keba  ] TRACE 2024/08/09 14:05:23 modbus: recv 00 0b 00 00 00 07 ff 03 04 00 00 00 01
	Phases state:	1
[keba  ] TRACE 2024/08/09 14:05:23 modbus: send 00 0c 00 00 00 06 ff 03 06 42 00 02
[... 13 lines cut ...]
[keba  ] TRACE 2024/08/09 14:05:37 modbus: recv 00 05 00 00 00 07 ff 03 04 00 00 00 03
[keba  ] TRACE 2024/08/09 14:05:37 modbus: send 00 06 00 00 00 06 ff 03 06 10 00 02
[keba  ] TRACE 2024/08/09 14:05:37 modbus: recv 00 06 00 00 00 07 ff 03 04 00 00 00 01
Charge status: B
Enabled:       true
Phases:        3
Diagnostic dump:
[keba  ] TRACE 2024/08/09 14:05:37 modbus: send 00 07 00 00 00 06 ff 03 03 f6 00 02
[keba  ] TRACE 2024/08/09 14:05:37 modbus: recv 00 07 00 00 00 07 ff 03 04 01 66 dd e4
	Serial:	23518692
[keba  ] TRACE 2024/08/09 14:05:37 modbus: send 00 08 00 00 00 06 ff 03 03 fa 00 02
[keba  ] TRACE 2024/08/09 14:05:37 modbus: recv 00 08 00 00 00 07 ff 03 04 03 0a 33 00
	Firmware:	3.10.51
[keba  ] TRACE 2024/08/09 14:05:37 modbus: send 00 09 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2024/08/09 14:05:37 modbus: recv 00 09 00 00 00 07 ff 03 04 00 04 c3 88
	Product:	312200
[keba  ] TRACE 2024/08/09 14:05:37 modbus: send 00 0a 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2024/08/09 14:05:37 modbus: recv 00 0a 00 00 00 07 ff 03 04 00 00 00 03
	Phases source:	3
[keba  ] TRACE 2024/08/09 14:05:37 modbus: send 00 0b 00 00 00 06 ff 03 06 10 00 02
[keba  ] TRACE 2024/08/09 14:05:37 modbus: recv 00 0b 00 00 00 07 ff 03 04 00 00 00 01
	Phases state:	1
[keba  ] TRACE 2024/08/09 14:05:37 modbus: send 00 0c 00 00 00 06 ff 03 06 42 00 02
[keba  ] TRACE 2024/08/09 14:05:37 modbus: recv 00 0c 00 00 00 07 ff 03 04 00 00 00 00
	Failsafe timeout:	0s`

andig, 2024-08-09:
```
[keba ] TRACE 2024/08/09 14:05:17 modbus: send 00 09 00 00 00 06 ff 03 03 f8 00 02
[keba ] TRACE 2024/08/09 14:05:17 modbus: recv 00 09 00 00 00 07 ff 03 04 00 04 c3 88
Product: 312200 (04C388)
```

Ich bin verwirrt. 312200 bedeutet "kein Energiezähler". Das Dump zeigt auch, dass da kein Zähler abgefragt wird. In Deinem Log oben passiert aber genau das. Warum?

TobiasHuber1980, 2024-08-10:
Firmware-Update auf 3.10.57 bringt vielleicht abhilfe?
https://dataspace.keba.com/public/download-shares/SW4Vh8OggeKbgue23njkkEgr93EAyAJx

andig, 2024-08-10:
Erstmal passt das Log nicht zu dem was passiert.

andig, 2024-08-17:
Schade um die Mühe, die wir hier in die Analyse gesteckt haben :(

closed 2024-08-17 by andig

asterix555, 2024-08-21:
Bitte entschuldigt die späte Antwort. Hier nochmal Logs von gerade eben, laut App hat das Auto mit 1kW geladen:

[evcc-20240821-162607-site.log](https://github.com/user-attachments/files/16693233/evcc-20240821-162607-site.log)
[evcc-20240821-162508-trace-keba.log](https://github.com/user-attachments/files/16693237/evcc-20240821-162508-trace-keba.log)

Bei einer zweiten Wallbox (selbe Type, Citroen eC4) habe ich das Problem nicht:
[evcc-20240821-163507-trace-e-site.log](https://github.com/user-attachments/files/16693241/evcc-20240821-163507-trace-e-site.log)
[evcc-20240821-163432-trace-e-keba.log](https://github.com/user-attachments/files/16693244/evcc-20240821-163432-trace-e-keba.log)

andig, 2024-08-24:
Ich hab den Überblick verloren. Die These war, dass die WB lädt (https://github.com/evcc-io/evcc/issues/15200#issuecomment-2267447244) , evcc das aber nicht sieht (https://github.com/evcc-io/evcc/issues/15200#issuecomment-2266136380).

Ist das das Problem? Dann wäre es super _während_ dieses Zustands nochmal die Ausgabe von 

    evcc charger --log trace,db:error --diagnose

zu bekommen. Oder stellt https://github.com/evcc-io/evcc/issues/15200#issuecomment-2277797189 schon diesen Zustand dar?

Die Diskussion um Strom/Leistungsregister ist eigentlich überflüssig- die Box hat keinen Zähler und evcc sucht auch nicht danach.

asterix555, 2024-08-25:
genau das ist noch immer der Fall.
In allen Fällen habe ich das laden gestartet, laut Gesamtstromverbrauch des Hauses bzw. auch laut App des Autos lädt das Auto, EVCC zeigt dies aber nicht an.

In meinem letzten Kommentar habe ich noch erwähnt dass eine komplett idente Installation (bei mir zuhause) diesen Fehler nicht zeigt.

Auf beiden Wallboxen läuft mittlerweile die Firmware 3.10.57, auch evcc habe ich auf 0.130.2 aktualisiert, der Fehler bleibt aber an der einen Installation.

andig, 2024-08-25:
Ändert ja nix. Was zu tun ist steht in https://github.com/evcc-io/evcc/issues/15200#issuecomment-2308360103