evcc-io/evcc #15227: Kachel Lastmanagement fehlt im Config UI

VolkerK62, 2024-08-03:
### Describe the bug

Wegen #15110 habe ich auf das aktuelle Nightly aktualisiert.
Seitdem wird im Config UI die Kachel "Lastmanagement" nicht mehr angezeigt.
[image]

Die circuit Einträge im Log sind weiterhin vorhanden. Die Funktionalität sollte also vorhanden sein.

### Steps to reproduce

.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db

interval: 10s

sponsortoken: xxx

telemetry: true

log: debug
levels:
  #site: info
  #lp-1: info
  #lp-2: info
  #lp-3: info
  #lp-4: info
  #lp-5: info
  #lp-6: info
  #daheimladen-mb: trace
  #hyundai: trace
  #main: trace
  #cache: trace
  #db: trace
  #http: trace
  #influx: trace
  #mqtt: trace
  #proxy-5200: trace
  #gsi: trace
  
########################################################################
meters:
- name: grid
#  type: movingaverage
#  decay: 0.25
#  meter:
  type: template
  template: shelly-3em
  usage: grid
  host: 192.168.178.28

- name: pv
#  type: movingaverage
#  decay: 0.25
#  meter:
  type: template
  template: solarlog
  usage: pv
  host: 192.168.178.23
  port: 502
[... 582 lines cut ...]
[site  ] DEBUG 2024/08/03 21:25:33 grid powers: [-136 103 105]W
[site  ] DEBUG 2024/08/03 21:25:34 grid currents: [-0.87 0.95 0.67]A
[site  ] DEBUG 2024/08/03 21:25:34 aux power 1: 0W
[site  ] DEBUG 2024/08/03 21:25:34 aux power: 0W
[site  ] DEBUG 2024/08/03 21:25:34 site power: 396W
[lp-1  ] DEBUG 2024/08/03 21:25:34 charge voltages: [232 0 0]V
[lp-1  ] DEBUG 2024/08/03 21:25:34 detected connected phases: 1p
[lp-1  ] DEBUG 2024/08/03 21:25:34 charge total import: 1983.300kWh
[lp-1  ] DEBUG 2024/08/03 21:25:34 charger status: A
[lp-1  ] INFO 2024/08/03 21:25:34 car disconnected
[lp-1  ] DEBUG 2024/08/03 21:25:34 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/08/03 21:25:34 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/08/03 21:25:34 set session soc limit: 0
[lp-1  ] DEBUG 2024/08/03 21:25:34 set session energy limit: 0
[lp-1  ] DEBUG 2024/08/03 21:25:34 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[site  ] DEBUG 2024/08/03 21:25:43 ----
```

### What type of operating system are you running?

Linux

### Version

0.129.0 (bbf989c3)

andig, 2024-08-03:
Vmtl. ähnliches Problem: HEMS zeigt "nicht konfigurert". Zu prüfen mit dieser demo.yaml Anpassung:

```yaml
loadpoints:
  - title: Carport
    circuit: root
    charger: charger_1
    mode: pv
    meter: meter_charger_1
    vehicle: vehicle_1
  - title: Garage
    circuit: root
    charger: charger_2
    mode: "off"
    meter: meter_charger_2
    vehicle: vehicle_2

circuits:
  - name: root

hems:
  type: relay
  maxPower: 4200 # single SteuVE
  limit:
    source: go
    script: |
      true
```

andig, 2024-08-03:
http://localhost:7070/api/config/devices/circuit
```json
{
  "result": [
    {
      "name": "root",
      "type": ""
    }
  ]
}
```

naltatis, 2024-08-03:
@andig @VolkerK62 das Problem tritt auf, wenn der Circuit kein Config-Object hat. In eurem konkreten Fall weil habt ihr jeweils mind. einen Circuit ohne Title. Die UI ist jetzt angepasst, dass sie damit klarkommt. ☝️

andig, 2024-08-04:
@naltatis das ist nur halb behoben: der LM Circuit wird nicht angezeigt und HEMS erscheint weiterhin als "configured: no".

andig, 2024-08-04:
> Wenn du HEMS noch per yaml konfiguriert hast, ist das das erwartete Verhalten. 

Ahh, das hatte ich nicht auf dem Radar. Ich dachte "yaml" würde auch angezeigt werden.

> somit wird der lpc circuit nicht angezeigt und die config in der HEMS Kachel nicht.

Das liegt vmtl. daran, dass der lpc Circuit dynamisch erzeugt wird und kein echtes "Geräte" ist. PR folgt.

Hier damit kein weiterer Fehler.

closed 2024-08-04 by andig