evcc-io/evcc #15233: `battery mode` wechselt in jedem Interval zwischen `hold` und `normal`

VolkerK62, 2024-08-04:
### Describe the bug

Seit kurzem (irgendein Nightly) nach 0.129.0 tritt diese Eigenart auf.
Da mindestens ein Ladepunkt im Modus "schnell" aktiv ist, wäre `hold` richtig.  

### Steps to reproduce

.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db

interval: 10s

sponsortoken: xxx

telemetry: true

log: debug
levels:
  #site: info
  #lp-1: info
  #lp-2: info
  #lp-3: info
  #lp-4: info
  #lp-5: info
  #lp-6: info
  #daheimladen-mb: trace
  #hyundai: trace
  #main: trace
  #cache: trace
  #db: trace
  #http: trace
  #influx: trace
  #mqtt: trace
  #proxy-5200: trace
  #gsi: trace
  
########################################################################
meters:
- name: grid
#  type: movingaverage
#  decay: 0.25
#  meter:
  type: template
  template: shelly-3em
  usage: grid
  host: 192.168.178.28

- name: pv
#  type: movingaverage
#  decay: 0.25
#  meter:
  type: template
  template: solarlog
  usage: pv
  host: 192.168.178.23
  port: 502

- name: battery
#  type: movingaverage
[... 607 lines cut ...]
[site  ] DEBUG 2024/08/04 08:57:51 set battery mode: normal
[site  ] DEBUG 2024/08/04 08:57:51 pv power: 3406W
[site  ] DEBUG 2024/08/04 08:57:51 battery soc: 42%
[site  ] DEBUG 2024/08/04 08:57:51 battery power: 175W
[site  ] DEBUG 2024/08/04 08:57:51 grid meter: -850W
[site  ] DEBUG 2024/08/04 08:57:51 grid powers: [-876 1062 -1037]W
[site  ] DEBUG 2024/08/04 08:57:51 grid currents: [-4.8 4.71 -4.49]A
[site  ] DEBUG 2024/08/04 08:57:51 aux power 1: 0W
[site  ] DEBUG 2024/08/04 08:57:51 aux power: 0W
[site  ] DEBUG 2024/08/04 08:57:51 site power: -576W
[lp-6  ] DEBUG 2024/08/04 08:57:51 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.0
[lp-6  ] DEBUG 2024/08/04 08:57:52 charge total import: 53.464kWh
[lp-6  ] DEBUG 2024/08/04 08:57:52 charger status: C
[lp-6  ] DEBUG 2024/08/04 08:57:52 soc estimated: 2.07% (vehicle: 0.10%)
[lp-6  ] DEBUG 2024/08/04 08:57:52 vehicle soc: 2%
[site  ] DEBUG 2024/08/04 08:58:00 ----
```

### What type of operating system are you running?

Linux

### Version

0.129.0 (f3be63f2)

VolkerK62, 2024-08-04:
Wenn kein Ladepunkt aktiv ist, bleibt es dauerhaft auf `normal`.

andig, 2024-08-04:
@VolkerK62 hat https://github.com/evcc-io/evcc/commit/bbf989c3c08de8cffb766a8f6a943d29c51647b0, der letzte vor https://github.com/evcc-io/evcc/pull/14855 das Problem auch?

VolkerK62, 2024-08-04:
guter Riecher ;-)
nein, damit gibt es das Problem nicht.

closed 2024-08-05 by andig