evcc-io/evcc #15234: Notifications via MQTT

Diondk, 2024-08-04:
**Is your feature request related to a problem? Please describe.**
No its not related to a problem. 

**Describe the solution you'd like**
Currently there is a way to setup notifications in evcc configuration. I want to push the same notifications using mqtt to my home assistant. 

would be great if we can get the same events published to MQTT:

`# events:
#   start:
#     title: Charge started
#     msg: Started charging in "${mode}" mode
#   stop:
#     title: Charge finished
#     msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
#   connect:
#     title: Car connected
#     msg: "Car connected at ${pvPower:%.1fk}kW PV"
#   disconnect:
#     title: Car disconnected
#     msg: Car disconnected after ${connectedDuration}
#   soc:
#     title: Soc updated
#     msg: Battery charged to ${vehicleSoc:%.0f}%
#   guest:
#     title: Unknown vehicle
#     msg: Unknown vehicle, guest connected?`

**Describe alternatives you've considered**
I used telegram but i would like to receive all my notifications through Home Assistant.

andig, 2024-08-04:
Happy to take a PR

goebelmeier, 2024-08-04:
Why not just simply install the HACS evcc integration and create the notifications directly inside Home Assistant using native HA Automations?

closed 2024-08-16 by github-actions[bot]