evcc-io/evcc #15237: EvBox Elvi startet das Laden in der Version 0.129.0 (f3be63f2) nicht mehr

andig, 2024-08-04:
### Discussed in https://github.com/evcc-io/evcc/discussions/15236

<div type='discussions-op-text'>

<sup>Originally posted by **e-fahrer-jochen** August  4, 2024</sup>
in der Version 0.128.4 (c2b29a86) klappt das Laden noch einwandfrei.
[evcc-0.128.4 (c2b29a86)-20240804-122147-trace.log](https://github.com/user-attachments/files/16487206/evcc-0.128.4.c2b29a86.-20240804-122147-trace.log)

Auch wenn ich im "now" Modus das Auto anstecke wird nicht geladen.
Hier die Logdatei der neuen Version 0.129.0 (f3be63f2):
[evcc-0.129.0 (f3be63f2)-20240804-121244-trace.log](https://github.com/user-attachments/files/16487212/evcc-0.129.0.f3be63f2.-20240804-121244-trace.log)

chargers:
- name: evbox
  type: template
  template: elvi
  timeout: 2m
  meter: true

vehicles:
- name: Smart
  type: custom
  title: Smart
  capacity: 17.6
  icon: car
  soc:
    source: const
    value: 0
  features: ["welcomecharge"]
  phases: 1

loadpoints:
- title: Hof
  charger: evbox
  mode: minpv
  vehicle: Smart
  priority: 10</div>

e-fahrer-jochen, 2024-08-04:
Ok mit "Auto Start" wird wieder geladen.
Wenn diese Information zu lesen wäre hätte ich mein Problem eventuell selbst erkannt.

https://github.com/evcc-io/evcc/pull/14888#issuecomment-2244965252

andig, 2024-08-04:
Fehlt da noch ein Stück Doku?

e-fahrer-jochen, 2024-08-04:
da die 0.128.4 noch problemlos geladen hat und ich im Change log der Version 0.129.0 nichts über meine Hardware finden konnte
dachte ich ich muss mein Problem anzeigen und habe auch nicht nach möglichen früheren Änderungen gesucht.

Entschuldigt die Umstände die ich ausgelöst habe.

[image]

andig, 2024-08-04:
Genau das versuche ich ja rauszufinden. Deshalb die Frage:

> Fehlt da noch ein Stück Doku?

@premultiply ?

e-fahrer-jochen, 2024-08-04:
Wenn man weis wonach man suchen soll findet man einen Hinweis. Im Elvi Template leider nicht.

[image]

closed 2024-08-06 by andig