evcc-io/evcc #15239: SteuVE Improvements

andig, 2024-08-04:
Currently, evcc tries to adjust grid consumption for the LPC use case. Regulation seems to require tuning down new SteuVEs, but not a) all chargers and b) not tune down grid consumption but consumption of those SteuVEs. Also, the dynamically created `lpc` circuit is not visible in the UI (https://github.com/evcc-io/evcc/issues/15227#issuecomment-2267588531). It seems this requires these changes:

- make it user responsibility to collect all affected SteuVEs in a single circuit
- make that circuit configurable for usage by HEMS

Following this, HEMS configuration needs to change as follows, compared to https://github.com/evcc-io/evcc/discussions/15139:

Digital:

```yaml
hems:
    type: eebus
    circuit: lpc # lpc circuit
    ski: <SKI der Steuerbox>
```

Analog:

```yaml
hems:
    type: relay
    circuit: lpc # lpc circuit
    maxPower: 4200 # single SteuVE
    limit:
      source: mqtt
      topic: hems/limit/bool
```

Given the following circuits:

```yaml
circuits:
- name: root
- name: lpc
  parent: root
```

All SteuVE-affected chargers MUST be assigned to the `lpc` circuit.

VolkerK62, 2024-08-04:
>All SteuVE-affected chargers MUST be assigned to the lpc circuit.

Meinst du "chargers" oder "loadpoints"?
Wenn du loadpoint meinst, kann man dann einen loadpoint mehreren circuits zuordnen?

goebelmeier, 2024-08-04:
Bei mehreren Wallboxen kann ich das ja verstehen. Aber ist es in jedem Fall realistisch, dass man alle SteuVE hinter den gleichen Circuit bekommt? Wallboxen + Heimspeicher + Wärmepumpe + Heizstäbe + Klimaanlagen

premultiply, 2024-08-04:
Theoretisch muss ein (virtueller) SteuVE-Circuit ja nicht zwingend mit der realen Installation übereinstimmen sofern bestimmte Voraussetzungen (alle Verbraucher mit eigener Leistungsmessung) vorliegen. Letztendlich bedarf es ja nur der korrekten Summenleistung der SteuVE.

Diese darf bei aktiver maximaler Dimmung die nach Formel berechnete Maximalleistung zzgl. Eigenproduktion nicht überschreiten.

andig, 2024-08-04:
Das wäre ein Argument dafür, die Steuerung via HEMS komplett vom Lastmanagement zu trennen. Dann bräuchte es mehrere Root Circuits und ein Ladepunkt müsste sich zu mehreren Circuits zuordnen lassen?

andig, 2024-08-05:
@premultiply wie wäre es, wenn wir mehrere root circuits erlauben i.s. dass der lpc circuit stand-alone existieren darf? Für Ladepunkte würde das dann bedeuten, dass sie mehreren (i.e. 2 Circuits) zugeordnet sein dürfen. Das fände ich charmanter als noch einen anderen Mechanismus zu verwenden ("Tags") da die Leistungsredzierung dann additiv erfolgen können, circuit für circuit?

premultiply, 2024-08-05:
Da spricht für mich nichts dagegen.
Meine Beschreibung war nur funktional gemeint.

VolkerK62, 2024-09-29:
ist das absichtlich (noch) geschlossen?

bendel80, 2024-12-29:
Kann man curcuit denn schon unter hems einfügen?

Bei mir funktioniert das testweise so, aber nicht wenn ich "circuit: lpc" einfüge, den ich vorher definiert hatte und in dem sich die steuVE befindet.

Das geht:

hems:
  type: relay
  maxPower: 4200 # single SteuVE
  limit:
    source: go
    script: |
      true

Wird dann nur aber nur lpc auf 4,2 kw gesetzt oder auch der Hauptkreis?

smartlader-de, 2025-01-08:
> @premultiply wie wäre es, wenn wir mehrere root circuits erlauben i.s. dass der lpc circuit stand-alone existieren darf? Für Ladepunkte würde das dann bedeuten, dass sie mehreren (i.e. 2 Circuits) [...]

+1 für lpc circuit stand-alone und "virtuell"

Könnten wir dann nicht einfach meters.pv addieren?

SteuVE.maxImportPower = 4200 # SteuVE sagt wir müssen dimmen auf 4,2kW
meters.pv = 7000 # Aktuelle Solarproduktion liegt bei 7kW
lpc.maxPower =SteuVE.maxImportPower + meters.pv # Die Steuerbaren Verbrauchseinrichtungen sind auf 11,2kW zu limitieren, danns sollte die Priorisierung greifen und dies umsetzen

Bin kein Coder, ich probiere mich so auszudrücken damit es verständlich ist ;)

heinemannj, 2025-01-19:
Ich habe grundsätzliche Verständnisprobleme mit der Berechnung

`[lpc ] DEBUG 2025/01/19 06:53:05 validate power: 1804W + (1286.5W -> 3680W) > 1000W capped at 482.5W`

```
[site ] DEBUG 2025/01/19 06:53:04 grid power: 1804W
[site ] DEBUG 2025/01/19 06:53:04 pv 1 power: 0W
[site ] DEBUG 2025/01/19 06:53:05 battery power: -74W
[site ] DEBUG 2025/01/19 06:53:04 aux 3 power: 1316W
```

```
hems:
  type: relay
  maxPower: 1000 # single SteuVE
  # circuit: lpc # lpc circuit
  limit:
    # true or false
    source: mqtt
    topic: hems/limit/state
```

```
chargers:
  - name: WP-Aussengeraet
    type: custom
    status:
      source: const
      value: C
    enabled:
      # true or false
      source: mqtt
      topic: hems/steuVE/state
    enable:
      # true or false
      source: mqtt
      topic: hems/steuVE/state
    maxcurrent:
      source: js
      script: |
        16;
    features:
      # no vehicle
      - integrateddevice
      - heating
    icon: waterheater
```

```
loadpoints:
  - title: WP-Aussengeraet
    circuit: lpc
    charger: WP-Aussengeraet
    mode: now
    meter: em_1_c
```

Als meters nutze ich 1 x `shelly-pro-3em` und 6 x `shelly-1pm` (Shelly Pro 3EM in monophase mode)

```
circuits:
  - name: grid
    title: Hauptstromkreis
    maxCurrent: 35 # 35A pro Phase
    maxPower: 24150 # 24.15kW = 3 x 230V x 35A
    meter: grid_1
    parent: # no parent, this is the root circuit
  - name: lpc
    title: SteuVE
    parent: grid
    maxCurrent: 35 # 35A pro Phase
  - name: pv_1
    title: pv_1
    parent: lpc
    meter: pv_1
  - name: battery_1
    title: battery_1
    parent: lpc
    meter: battery_1
  - name: battery_2
    title: battery_2
    parent: lpc
    meter: battery_2
  - name: house
    title: Hausverbrauch
    parent: grid
    maxCurrent: 35 # 35A pro Phase
    meter: em_2_a
  - name: em_1_a
    title: Waschen
    parent: house
    maxCurrent: 16 # 16A L1
    meter: em_1_a
  - name: em_1_b
    title: Trocknen
    parent: house
    maxCurrent: 16 # 16A L2
    meter: em_1_b
  - name: em_1_c
    title: WP-Aussengeraet
    parent: lpc
    meter: em_1_c
    maxCurrent: 16 # 16A L3
  - name: em_2_b
    title: Wärmepumpe (Innengerät Steuerung)
    parent: house
    maxCurrent: 16 # 16A L2
    meter: em_2_b
  - name: em_2_c
    title: Messtechnik
    parent: house
    maxCurrent: 16 # 16A L3
    meter: em_2_c

site:
  title: Energy Management System (EMS)
  meters:
    grid: grid_1
    pv:
      - pv_1
    battery:
      - battery_1
      - battery_2
    aux:
      - em_1_a # Waschen
      - em_1_b # Trocknen
      - em_1_c # Wärmepumpe (Aussengeraet Kompressor)
      - em_2_a # Frei
      - em_2_b # Wärmepumpe (Innengerät Steuerung)
      - em_2_c # Messtechnik
      - em_3   # Wärmepumpe (Innengerät Heizstab)
```

```
[site ] DEBUG 2025/01/19 06:53:01 ----
[lp-1 ] DEBUG 2025/01/19 06:53:01 charge power: 1286W
[lp-1 ] DEBUG 2025/01/19 06:53:01 charge currents: [0 0 5.49]A
[circuit-pv_1] DEBUG 2025/01/19 06:53:01 power: 0W
[circuit-pv_1] DEBUG 2025/01/19 06:53:01 current: 0A
[circuit-battery_1] DEBUG 2025/01/19 06:53:01 power: -55W
[circuit-battery_1] DEBUG 2025/01/19 06:53:01 current: 0A
[circuit-battery_2] DEBUG 2025/01/19 06:53:01 power: -13W
[circuit-battery_2] DEBUG 2025/01/19 06:53:01 current: 0A
[circuit-em_1_c] DEBUG 2025/01/19 06:53:02 power: 1286.5W
[circuit-em_1_c] DEBUG 2025/01/19 06:53:02 current: 5.49A
[circuit-lpc] DEBUG 2025/01/19 06:53:02 power: 2505W
[circuit-lpc] DEBUG 2025/01/19 06:53:02 current: 11A
[circuit-em_1_a] DEBUG 2025/01/19 06:53:02 power: 2.9W
[circuit-em_1_a] DEBUG 2025/01/19 06:53:02 current: 0.081A
[circuit-em_1_b] DEBUG 2025/01/19 06:53:02 power: 0W
[circuit-em_1_b] DEBUG 2025/01/19 06:53:02 current: 0.091A
[circuit-em_2_b] DEBUG 2025/01/19 06:53:02 power: 10W
[circuit-em_2_b] DEBUG 2025/01/19 06:53:02 current: 0.09A
[circuit-em_2_c] DEBUG 2025/01/19 06:53:02 power: 82.9W
[circuit-em_2_c] DEBUG 2025/01/19 06:53:02 current: 0.518A
[circuit-house] DEBUG 2025/01/19 06:53:03 power: 0W
[circuit-house] DEBUG 2025/01/19 06:53:03 current: 0.028A
[circuit-grid] DEBUG 2025/01/19 06:53:04 power: 1794W
[circuit-grid] DEBUG 2025/01/19 06:53:04 current: 6.53A
[lpc ] WARN 2025/01/19 06:53:04 over power detected: 1804W > 1000W
[lpc ] DEBUG 2025/01/19 06:53:04 current: 6.53A
[site ] DEBUG 2025/01/19 06:53:04 grid power: 1804W
[site ] DEBUG 2025/01/19 06:53:04 aux 7 power: 2W
[site ] DEBUG 2025/01/19 06:53:04 pv 1 power: 0W
[site ] DEBUG 2025/01/19 06:53:04 aux 1 power: 3W
[site ] DEBUG 2025/01/19 06:53:04 aux 4 power: 0W
[site ] DEBUG 2025/01/19 06:53:04 battery 2 power: -11W
[site ] DEBUG 2025/01/19 06:53:04 aux 5 power: 10W
[site ] DEBUG 2025/01/19 06:53:04 aux 2 power: 0W
[site ] DEBUG 2025/01/19 06:53:04 battery 1 power: -63W
[site ] DEBUG 2025/01/19 06:53:04 aux 3 power: 1316W
[site ] DEBUG 2025/01/19 06:53:04 aux 6 power: 85W
[site ] DEBUG 2025/01/19 06:53:04 aux power: 1416W
[site ] DEBUG 2025/01/19 06:53:05 grid currents: [1.14 0.51 6.54]A
[site ] DEBUG 2025/01/19 06:53:05 battery 1 soc: 52%
[site ] DEBUG 2025/01/19 06:53:05 battery 2 soc: 36%
[site ] DEBUG 2025/01/19 06:53:05 battery power: -74W
[site ] DEBUG 2025/01/19 06:53:05 battery soc: 47%
[site ] DEBUG 2025/01/19 06:53:05 battery has priority at soc 47% (< 100%)
[site ] DEBUG 2025/01/19 06:53:05 site power: 488W
[lp-1 ] DEBUG 2025/01/19 06:53:05 detected active phases: 1p
[lp-1 ] DEBUG 2025/01/19 06:53:05 charger status: C
[lp-1 ] WARN 2025/01/19 06:53:05 charger logic error: disabled but charging
[lp-1 ] DEBUG 2025/01/19 06:53:05 charger enable
[lpc ] DEBUG 2025/01/19 06:53:05 validate power: 1804W + (1286.5W -> 3680W) > 1000W capped at 482.5W
[lp-1 ] DEBUG 2025/01/19 06:53:05 charger disable
[lp-1 ] DEBUG 2025/01/19 06:53:05 wake-up timer: stop
```

andig, 2025-01-19:
@heinemannj mach bitte eine Discussion zu Deinen Fragen. Hier OT.

lostinbyte, 2025-02-07:
> Kann man curcuit denn schon unter hems einfügen?

Das würde mich auch interessieren, bei mir meckert er: 
failed configuring hems: decoding failed due to the following error(s): '' has invalid keys: circuit

hems:
   type: relay
   circuit: lpc   
   maxPower: 4200 # single SteuVE
   limit:
     source: mqtt
     topic: hems/limit/bool

FSXPET, 2025-02-28:
Hallo

Ist denn das Aktivieren der Begrenzung der StVE nun auch über ein GPIO Pin des Raspberrys möglich?
Im Blog steht das ja drin, ein Beispiel habe ich aber noch nicht gefunden.

CiNcH83, 2025-07-17:
Wie sieht das bei mehreren SteuVE am lpc Circuit aus? Wird da bei 2 SteuVE die Leistung auf beide aufgeteilt? Was ist, wenn eines davon eine Wärmepumpe ist? Im Winter will man evtl. mehr Leistung in der WP und im Sommer eher in der Wallbox?

Und was ist, wenn die Wärmepumpe nicht alles abnimmt, was sie zugewiesen bekommt? Da bedarf es dann wahrscheinlich eines Meters, der zurück misst, was die Wärmepumpe zieht? Oder können das die Wärmepumpen via API, sofern diese auch gedimmt werden können? Vaillant unterstützt neben EEBUS LPC auch EEBUS MPC...

chhtz, 2025-09-29:
Hi, ist es eigentlich prinzipiell möglich das Lastmanagement über das Webinterface einzurichten oder geht das aktuell nur über die evcc.yaml? In der Doku wird es nur über die yaml beschrieben, im Webinterface finde ich keine Möglichkeit einen loadpoint einem circuit hinzuzufügen.

andig, 2025-09-29:
Ja- experimental einschalten.