evcc-io/evcc #15246: Laden startet und stoppt immer wieder im PV Modus ohne PV Leistung

CKlapper, 2024-08-04:
### Describe the bug

Auto (Cupra Formentor PluginHybrid)

Das Verhalten tritt auf sobald die Entladegrenze vom Hausakku unterschritten wird.

Das laden stoppt dann, startet aber wieder für ca. 3 Minuten, stoppt kurz, startet wieder und immer so weiter.

Auto per UI angelegt. 

enable / disable treshold habe ich in der evcc.yaml mal deaktivert. Das hat nichts verändert.

### Steps to reproduce

Ladestecker ins Auto

Grenzen vom Hausakku 80% & 75%

Vorgang zum log:
23:13 von Modus "AUS" auf "PV" gewechselt
Ladebereit steht an
23:14 Auto lädt
Timer läuft von 3min runter zum abschalten
23:16 laden stoppt
23:17 laden startet wieder
Timer läuft von 3min runter zum abschalten

Modus wieder auf "Aus" 

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug

# unique installation id
plant: 6a9108443f558ede6279a487f79bde8d52cf63de0139ce3554c59e10b17a0bd7

interval: 15s

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3ODg2MzAxMTAsImlhdCI6MTY5NDAyMjExMCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IkNLbGFwcGVyIn0.usuwxlsQZ8XD7WrOqzAA0WCUiTC7hfL5P8pcD6JgKE0

meters:
- name: my_grid
  type: template
  template: goodwe-hybrid 
  id: 247  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: grid
  modbus: rs485serial  
- name: my_pv 
  type: template
  template: goodwe-hybrid 
  id: 247  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: pv  
  modbus: rs485serial  
- name: my_battery 
  type: template
  template: goodwe-hybrid 
  id: 247  
[... 587 lines cut ...]
[lp-1  ] DEBUG 2024/08/04 23:17:51 charge power: 0W
[lp-1  ] DEBUG 2024/08/04 23:17:51 charge currents: [0 0 0]A
[site  ] DEBUG 2024/08/04 23:17:51 pv power: 0W
[site  ] DEBUG 2024/08/04 23:17:51 battery soc: 58%
[site  ] DEBUG 2024/08/04 23:17:51 battery power: 614W
[site  ] DEBUG 2024/08/04 23:17:51 grid meter: -11W
[site  ] DEBUG 2024/08/04 23:17:51 site power: 703W
[lp-1  ] DEBUG 2024/08/04 23:17:51 !! session: chargeRater.chargedEnergy=3.1 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/04 23:17:51 charger status: B
[lp-1  ] WARN 2024/08/04 23:17:51 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2024/08/04 23:17:51 charger enable
[lp-1  ] DEBUG 2024/08/04 23:17:51 !! active phases: 1p = min(1p measured 1p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/08/04 23:17:51 pv charge current: 0A = 0A + -3.06A (703W @ 1p)
[lp-1  ] DEBUG 2024/08/04 23:17:51 projected site power 703W >= 0W disable threshold
[lp-1  ] DEBUG 2024/08/04 23:17:51 pv disable timer start: 3m0s
[lp-1  ] DEBUG 2024/08/04 23:17:51 pv disable in 3m0s
```

### What type of operating system are you running?

Linux

### Version

0.129.0

StefanSchoof, 2024-08-05:
Ist das https://github.com/evcc-io/evcc/issues/14839?

CKlapper, 2024-08-05:
[evcc-20240804-231927-trace.log](https://github.com/user-attachments/files/16492220/evcc-20240804-231927-trace.log)

Hab da mal den TRACE Log angehangen.

Hallo Stefan,
bin mir nicht sicher ob es exakt das gleiche Problem wie #14839 ist. Es ist auf jedenfall ein ähnliches Verhalten. Gemeinsamkeit ist wohl die Nutzung von EEBus. Meine Walbox ist jedoch keine Elli sondern Mennekes Charge Control.

Das Problem tritt auch erst seit ca. 3-4 Wochen auf, was zeitgleich mit der Anschaffung des Cupras ist.
Ob ich dieses Verhalten jetzt auch mit meinem anderen Auto habe (Peugot e208, den habe ich schon 2 Jahren, meiste Zeit mit EVCC) ,werde ich es heute Spät Abend mal probieren.

Es gibt nur halt überhaupt keinen Sinn warum EVCC den Ladevorgang immer wieder startet.

Denn:
Modus = "PV"
PV Erzeugung = "0"
Ladestand Hausakku = kleiner des eingestellten Wertes "Haus hat Priorität, nicht zum laden verwenden)

StefanSchoof, 2024-08-05:
Probiere mal das nightly. Da sind schon eebus Fixes drin und mit ein bisschen Glück ist das auch bei dir dann schon gelöst.

DerAndereAndi, 2024-08-05:
Ich sehe hier keinen EEBUS Fehler:

```
[lp-1  ] DEBUG 2024/08/04 23:18:06 charge power: 0W
[lp-1  ] DEBUG 2024/08/04 23:18:06 charge currents: [0 0 0]A
[site  ] DEBUG 2024/08/04 23:18:06 pv power: 0W
[site  ] DEBUG 2024/08/04 23:18:06 battery soc: 58%
[site  ] DEBUG 2024/08/04 23:18:06 battery power: 608W
[site  ] DEBUG 2024/08/04 23:18:06 grid meter: 17W
[eebus ] TRACE 2024/08/04 23:18:06 Recv: 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.2.0"},{"addressSource":[{"device":"d:_i:4711_CC612_2S0R_CC"},{"entity":[1,1]},{"feature":11}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-EVCC"},{"entity":[1]},{"feature":6}]},{"msgCounter":2638879},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"function":"measurementListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"measurementListData":[{"measurementData":[[{"measurementId":0},{"valueType":"value"},{"value":[{"number":0},{"scale":-3}]}],[{"measurementId":1},{"valueType":"value"},{"value":[{"number":0},{"scale":-3}]}],[{"measurementId":2},{"valueType":"value"},{"value":[{"number":0},{"scale":-3}]}],[{"measurementId":3},{"valueType":"value"},{"value":[{"number":0}]}],[{"measurementId":4},{"valueType":"value"},{"value":[{"number":0}]}],[{"measurementId":5},{"valueType":"value"},{"value":[{"number":0}]}],[{"measurementId":6},{"valueType":"value"},{"value":[{"number":3111}]}]]}]}]]}]}]}}]}
[eebus ] DEBUG 2024/08/04 23:18:06 Recv: d:_i:4711_CC612_2S0R_CC:Measurement to Measurement notify 2638879 MeasurementListData
[eebus ] DEBUG 2024/08/04 23:18:06 ski 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 event cem-evcem-DataUpdateCurrentPerPhase
[eebus ] DEBUG 2024/08/04 23:18:06 ski 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 event cem-evcem-DataUpdatePowerPerPhase
[eebus ] DEBUG 2024/08/04 23:18:06 ski 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 event cem-evcem-DataUpdateEnergyCharged
[site  ] DEBUG 2024/08/04 23:18:06 site power: 725W
[lp-1  ] DEBUG 2024/08/04 23:18:06 !! session: chargeRater.chargedEnergy=3.1 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/04 23:18:06 charger status: B
[lp-1  ] WARN 2024/08/04 23:18:06 charger logic error: current mismatch (got 0A, expected 12A)
[lp-1  ] DEBUG 2024/08/04 23:18:06 !! active phases: 1p = min(1p measured 1p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/08/04 23:18:06 pv charge current: 0A = 0A + -3.15A (725W @ 1p)
[lp-1  ] DEBUG 2024/08/04 23:18:06 projected site power 725W >= 0W disable threshold
[lp-1  ] DEBUG 2024/08/04 23:18:06 pv disable in 2m45s
[lp-1  ] DEBUG 2024/08/04 23:18:06 pv disable timer remaining: 2m45s
[eebus ] DEBUG 2024/08/04 23:18:06 Send: d:_i:4711_CC612_2S0R_CC write 67848 LoadControlLimitListData
[eebus ] TRACE 2024/08/04 23:18:06 Send: 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_n:EVCC_HEMS-EVCC"},{"entity":[1]},{"feature":7}]},{"addressDestination":[{"device":"d:_i:4711_CC612_2S0R_CC"},{"entity":[1,1]},{"feature":10}]},{"msgCounter":67848},{"cmdClassifier":"write"},{"ackRequest":true}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":12},{"scale":0}]}]]}]}]]}]}]}}]}
[lp-1  ] DEBUG 2024/08/04 23:18:06 max charge current: 12A
[eebus ] TRACE 2024/08/04 23:18:06 Recv: 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.2.0"},{"addressSource":[{"device":"d:_i:4711_CC612_2S0R_CC"},{"entity":[1,1]},{"feature":10}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-EVCC"},{"entity":[1]},{"feature":7}]},{"msgCounter":2638883},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":12},{"scale":0}]}]]}]}]]}]}]}}]}
[eebus ] DEBUG 2024/08/04 23:18:06 Recv: d:_i:4711_CC612_2S0R_CC:LoadControl to LoadControl notify 2638883 LoadControlLimitListData
```

evcc sagt der Wallbox, dass mit 12A geladen werden kann.

andig, 2024-08-05:
Ich bin immer dankbar, wenn @StefanSchoof einen Erstaufschlag macht. Klappt halt nicht immer ;)

premultiply, 2024-08-05:
> Nutzung von EEBus. Meine Walbox ist jedoch keine Elli sondern Mennekes Charge Control.

Oho, Bender CC613 via EEBus!
Braucht das auch ein Template?

andig, 2024-08-05:
Also _was_ hier passiert scheint klar, warum weniger:

```
[site  ] DEBUG 2024/08/04 23:06:21 ----
[lp-1  ] DEBUG 2024/08/04 23:06:21 charge power: 0W
[lp-1  ] DEBUG 2024/08/04 23:06:21 charge currents: [0 0 0]A
[site  ] DEBUG 2024/08/04 23:06:21 pv power: 0W
[eebus ] TRACE 2024/08/04 23:06:21 Recv: 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.2.0"},{"addressSource":[{"device":"d:_i:4711_CC612_2S0R_CC"},{"entity":[1,1]},{"feature":11}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-EVCC"},{"entity":[1]},{"feature":6}]},{"msgCounter":2635659},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"function":"measurementListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"measurementListData":[{"measurementData":[[{"measurementId":0},{"valueType":"value"},{"value":[{"number":0},{"scale":-3}]}],[{"measurementId":1},{"valueType":"value"},{"value":[{"number":0},{"scale":-3}]}],[{"measurementId":2},{"valueType":"value"},{"value":[{"number":0},{"scale":-3}]}],[{"measurementId":3},{"valueType":"value"},{"value":[{"number":0}]}],[{"measurementId":4},{"valueType":"value"},{"value":[{"number":0}]}],[{"measurementId":5},{"valueType":"value"},{"value":[{"number":0}]}],[{"measurementId":6},{"valueType":"value"},{"value":[{"number":2980}]}]]}]}]]}]}]}}]}
[lp-1  ] DEBUG 2024/08/04 23:06:21 charger status: B
[lp-1  ] WARN 2024/08/04 23:06:21 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2024/08/04 23:06:21 charger enable
[lp-1  ] DEBUG 2024/08/04 23:06:21 charger disable
```

evcc sieht einen unerwarteten `enabled` Status, korrigiert sich intern und schaltet dann aufgrund Netzbezug wieder ab.

Das passiert dann immer mal wieder:

```
[site  ] DEBUG 2024/08/04 23:07:36 ----
[lp-1  ] WARN 2024/08/04 23:07:36 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2024/08/04 23:07:36 charger enable
[lp-1  ] DEBUG 2024/08/04 23:07:36 charger disable
```

Mir ist allerdings nicht klar, warum wir hier "enabled" sehen.

@DerAndereAndi eine Nebenwirkung von https://github.com/evcc-io/evcc/pull/15023? Das sollte sich allerdings über `isCharging()` allenfalls auf `Status` auswirken, aber nicht auf `Enabled()`?

Der verbleibende Kern von `Enabled()`- was m.E. jetzt die Ursache darstellen sollte- sieht so aus:

```go
	// if the VW VAS PV mode is active, use PV limits
	if c.hasActiveVASVW() {
		limits, err := c.uc.OscEV.LoadControlLimits(evEntity)
		if err != nil {
			// there are no limits available, e.g. because the data was not received yet
			return true, nil
		}

		for _, limit := range limits {
			// check if there is an active limit set
			if limit.IsActive && limit.Value >= 1 {
				return true, nil
			}
		}

		return false, nil
	}

	limits, err := c.uc.OpEV.LoadControlLimits(evEntity)
	if err != nil {
		// there are limits available, e.g. because the data was not received yet
		return true, nil
	}

	for _, limit := range limits {
		// for IEC61851 the pause limit is 0A, for ISO15118-2 it is 0.1A
		// instead of checking for the actual data, hardcode this, so we might run into less
		// timing issues as the data might not be received yet
		// if the limit is not active, then the maximum possible current is permitted
		if (limit.IsActive && limit.Value >= 1) ||
			!limit.IsActive {
			return true, nil
		}
	}
```

Wir können da gerne noch irgendwo ein Logging mein rein stecken?

andig, 2024-08-05:
Interessant ist auch der Ablauf zwischen erstem und zweitem Out of Sync. ich habe mal alle `measurementListData` gelöscht. Dann verbleibt nur noch:

```
[site  ] DEBUG 2024/08/04 23:06:21 ----
[lp-1  ] DEBUG 2024/08/04 23:06:21 charge power: 0W
[lp-1  ] DEBUG 2024/08/04 23:06:21 charge currents: [0 0 0]A
[site  ] DEBUG 2024/08/04 23:06:21 pv power: 0W
[site  ] DEBUG 2024/08/04 23:06:21 battery soc: 60%
[site  ] DEBUG 2024/08/04 23:06:21 battery power: 535W
[site  ] DEBUG 2024/08/04 23:06:21 grid meter: 8W
[site  ] DEBUG 2024/08/04 23:06:21 site power: 643W
[lp-1  ] DEBUG 2024/08/04 23:06:21 !! session: chargeRater.chargedEnergy=3.0 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/04 23:06:21 charger status: B
[lp-1  ] WARN 2024/08/04 23:06:21 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2024/08/04 23:06:21 charger enable
[lp-1  ] DEBUG 2024/08/04 23:06:21 charger disable
[eebus ] TRACE 2024/08/04 23:06:21 Send: 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_n:EVCC_HEMS-EVCC"},{"entity":[1]},{"feature":7}]},{"addressDestination":[{"device":"d:_i:4711_CC612_2S0R_CC"},{"entity":[1,1]},{"feature":10}]},{"msgCounter":67488},{"cmdClassifier":"write"},{"ackRequest":true}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":0}]}]]}]}]]}]}]}}]}
[lp-1  ] DEBUG 2024/08/04 23:06:21 wake-up timer: stop
[eebus ] TRACE 2024/08/04 23:06:21 Recv: 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.2.0"},{"addressSource":[{"device":"d:_i:4711_CC612_2S0R_CC"},{"entity":[1,1]},{"feature":10}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-EVCC"},{"entity":[1]},{"feature":7}]},{"msgCounter":2635663},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":0}]}]]}]}]]}]}]}}]}
[eebus ] TRACE 2024/08/04 23:06:21 Recv: 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.2.0"},{"addressSource":[{"device":"d:_i:4711_CC612_2S0R_CC"},{"entity":[1,1]},{"feature":10}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-EVCC"},{"entity":[1]},{"feature":7}]},{"msgCounter":2635665},{"msgCounterReference":67488},{"cmdClassifier":"result"}]},{"payload":[{"cmd":[[{"resultData":[{"errorNumber":0}]}]]}]}]}}]}

[site  ] DEBUG 2024/08/04 23:06:36 ----
[lp-1  ] DEBUG 2024/08/04 23:06:36 charge power: 0W
[lp-1  ] DEBUG 2024/08/04 23:06:36 charge currents: [0 0 0]A
[site  ] DEBUG 2024/08/04 23:06:36 pv power: 0W
[site  ] DEBUG 2024/08/04 23:06:36 battery soc: 60%
[site  ] DEBUG 2024/08/04 23:06:36 battery power: 534W
[site  ] DEBUG 2024/08/04 23:06:36 grid meter: 3W
[site  ] DEBUG 2024/08/04 23:06:36 site power: 637W
[lp-1  ] DEBUG 2024/08/04 23:06:36 !! session: chargeRater.chargedEnergy=3.0 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/04 23:06:36 charger status: B

[site  ] DEBUG 2024/08/04 23:06:51 ----
[lp-1  ] DEBUG 2024/08/04 23:06:51 charge power: 0W
[lp-1  ] DEBUG 2024/08/04 23:06:51 charge currents: [0 0 0]A
[site  ] DEBUG 2024/08/04 23:06:51 pv power: 0W
[site  ] DEBUG 2024/08/04 23:06:51 battery soc: 60%
[site  ] DEBUG 2024/08/04 23:06:51 battery power: 532W
[site  ] DEBUG 2024/08/04 23:06:51 grid meter: 2W
[site  ] DEBUG 2024/08/04 23:06:51 site power: 634W
[lp-1  ] DEBUG 2024/08/04 23:06:51 !! session: chargeRater.chargedEnergy=3.0 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/04 23:06:51 charger status: B

[site  ] DEBUG 2024/08/04 23:07:06 ----
[lp-1  ] DEBUG 2024/08/04 23:07:06 charge power: 0W
[lp-1  ] DEBUG 2024/08/04 23:07:06 charge currents: [0 0 0]A
[site  ] DEBUG 2024/08/04 23:07:06 pv power: 0W
[site  ] DEBUG 2024/08/04 23:07:06 battery soc: 60%
[site  ] DEBUG 2024/08/04 23:07:06 battery power: 527W
[site  ] DEBUG 2024/08/04 23:07:06 grid meter: 10W
[site  ] DEBUG 2024/08/04 23:07:06 site power: 637W
[lp-1  ] DEBUG 2024/08/04 23:07:06 !! session: chargeRater.chargedEnergy=3.0 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/04 23:07:06 charger status: B

[site  ] DEBUG 2024/08/04 23:07:21 ----
[lp-1  ] DEBUG 2024/08/04 23:07:21 charge power: 0W
[lp-1  ] DEBUG 2024/08/04 23:07:21 charge currents: [0 0 0]A
[site  ] DEBUG 2024/08/04 23:07:21 pv power: 0W
[site  ] DEBUG 2024/08/04 23:07:21 battery soc: 60%
[site  ] DEBUG 2024/08/04 23:07:21 battery power: 651W
[site  ] DEBUG 2024/08/04 23:07:21 grid meter: -85W
[site  ] DEBUG 2024/08/04 23:07:21 site power: 666W
[lp-1  ] DEBUG 2024/08/04 23:07:21 !! session: chargeRater.chargedEnergy=3.0 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/04 23:07:21 charger status: B

[site  ] DEBUG 2024/08/04 23:07:36 ----
[lp-1  ] DEBUG 2024/08/04 23:07:36 charge power: 0W
[lp-1  ] DEBUG 2024/08/04 23:07:36 charge currents: [0 0 0]A
[site  ] DEBUG 2024/08/04 23:07:36 pv power: 0W
[site  ] DEBUG 2024/08/04 23:07:36 battery soc: 60%
[site  ] DEBUG 2024/08/04 23:07:36 battery power: 546W
[site  ] DEBUG 2024/08/04 23:07:36 grid meter: 16W
[site  ] DEBUG 2024/08/04 23:07:36 site power: 662W
[... cut ...]
[eebus ] TRACE 2024/08/04 23:07:36 Recv: 4b293b3bfa2e36f728b0f89a006dbe4b567be8e3 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.2.0"},{"addressSource":[{"device":"d:_i:4711_CC612_2S0R_CC"},{"entity":[1,1]},{"feature":10}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-EVCC"},{"entity":[1]},{"feature":7}]},{"msgCounter":2636007},{"msgCounterReference":67526},{"cmdClassifier":"result"}]},{"payload":[{"cmd":[[{"resultData":[{"errorNumber":0}]}]]}]}]}}]}
```

Was also hat sich zwischendurch geändert?

andig, 2024-08-05:
@CKlapper falls Andi nix Besseres einfällt bitte morgen nochmal mit dem Nightly probieren. Es gibt zusätzliches Logging das uns vielleicht Aufschluss darüber gibt, ob/was hier zu der falschen "enabled" Meldung führen könnte.

DerAndereAndi, 2024-08-05:
Mit vwVAS kann das nichts zu tun haben, nur bei der PMCC mit Taycan kommt er da überhaupt rein.

andig, 2024-08-05:
Weitere Erkenntnis: 

    charger out of sync: expected disabled, got enabled

wird nach einem Schaltvorgang für 1m unterdrückt. Genau so lange taucht die Meldung oben im gekürzten Log nicht auf. Es scheint also, als würde das `disable` gar nicht erst greifen. Auch das sollte uns das Logging jetzt mitteilen können.

DerAndereAndi, 2024-08-05:
Ein Logfile das auch den Zeitpunkt des Ansteckens beinhaltet, würde sehr helfen.

CKlapper, 2024-08-05:
[evcc-20240805-230958-trace.log](https://github.com/user-attachments/files/16502829/evcc-20240805-230958-trace.log)

Habe dann mal das Nightly installiert. Denke es hat geklappt, da so ein Mondsymbol hinter der Versionsnummer zu sehen ist. (War mein erstes mal. :) )

Logdatei sollte das einstecken des Fahrzeug um ca. 23:03 Uhr anzeigen und danach das übliche hin und her:

23:03 Auto angesteckt
23:04 Ladebereit. Warte auf Fahrzeug
Ladevorgang startet
Timer für Ladestopp läuft ab
23:08 Laden stoppt
23:09 Ladebereit. Warte auf Fahrzeug
Ladevorgang startet

Vielen Dank für euer Bemühen.

CKlapper, 2024-08-05:
[evcc-20240805-233123-trace.log](https://github.com/user-attachments/files/16502942/evcc-20240805-233123-trace.log)

Und jetzt noch ein anderes Problem.

Habe den Ladevorgang dann auf "AUS" gesetzt
Das Entladen der Hausbatterie gesperrt und das Laden auf "Schnell" wieder gestartet.
Dann erstmal nicht weiter hingeguckt, da das immer soweit funktioniert hat.
Dann sehe ich wie nach jeden (15s) aktualisieren der Daten das Sperren der Hausbatterie aktiviert wird, dann wieder 15s deaktivert, dann wieder aktiviert usw.

nochmal auf "Aus" wieder auf "Schnell" gestartet, gleiches Verhalten.

andig, 2024-08-06:
@CKlapper für das zusätzliche Logging warst Du leider zu schnell. Das "nightly" ist heute nacht erst mit dem zusätzlichen Logging gebaut worden.

CKlapper, 2024-08-06:
evcc ist schon die neueste Version (0.129.0+1722824546)

Das meldet er mir wenn ich jetzt ein Upgrade anstoße. Oder mache ich da mit der Nightly etwas was falsch ?

habe aus der Doku den Befehl genommen: ohne vorher etwas zu deinstallieren.
curl -1sLf 'https://dl.evcc.io/public/evcc/unstable/setup.deb.sh' | sudo -E bash
und dann normal das Upgrade angestoßen. Sah eigentlich gut aus.

StefanSchoof, 2024-08-06:
Die neuste nightly ist 0.129.0+1722910911, siehe https://cloudsmith.io/~evcc/packages/

Eigentlich sollte mit einem `sudo apt update && sudo apt upgrade` die neuste nightly bei dir installiert werden.

CKlapper, 2024-08-06:
@StefanSchoof 

Danke für den Hinweis.
habe das bisher immer so gemacht:

sudo apt update
sudo apt --only-upgrade install -y evcc

Probiere ich heute Abend nochmal.

StefanSchoof, 2024-08-06:
Das sollte meines Wissens auch funktionieren.

andig, 2024-08-06:
Es macht halt erst Sinn wenn die neue Version da ist. Gestern abend war sie es noch nicht.

CKlapper, 2024-08-07:
@andig 

Neue Logdatei mit hoffentlich aufschlussreichen Inhalt.

[evcc-20240807-063929-trace.log](https://github.com/user-attachments/files/16518305/evcc-20240807-063929-trace.log)

DerAndereAndi, 2024-08-07:
@CKlapper Thanks for the log, this will help I think.

DerAndereAndi, 2024-08-07:
Ich habe das Problem identifiziert und reproduziert bekommen. Nach meinem Verständnis verhält sich der Bender Controller hier nicht Spec konform. Habe dafür einen "Workaround" eingebaut. Mit dem nächsten Nightly Build sollte es funktionieren.

CKlapper, 2024-08-07:
@DerAndereAndi , ich danke dir.
Ich werde es wohl erst morgen Abend ausprobieren können. Werde dann berichten.

CKlapper, 2024-08-08:
@DerAndereAndi , sieht gut aus. Ladeverhalten ist wieder so wie es soll.

closed 2024-08-08 by andig