evcc-io/evcc #15250: EVCC and commandProxy not working reliably

swa72, 2024-08-05:
### Describe the bug

Bei mir funktioniert das Verwenden des commandProxy nicht zuverlässig.  Ähnliches Verhalten wie der ein oder andere schon unter https://github.com/evcc-io/evcc/pull/14616 beschrieben hat.

Super wäre, wenn es irgendwo eine Beschreibung von `commandProxy ` gäbe. Trotz Suchens habe ich in den Docs nichts gefunden.

### Steps to reproduce

1. Tesla R1 abstöpseln
2. Tesla T1 anschließen
3. FleetAPI wird verwendet

### Configuration details

```yaml
`evcc.yaml` siehe https://github.com/swa72/home-assistant/blob/main/evcc-sanitized.yaml
```

### Log details

```text
Start des debug logs

[main  ] INFO 2024/08/05 07:59:08 evcc 0.129.0
[main  ] INFO 2024/08/05 07:59:08 using config file: /config/evcc.yaml
[db    ] INFO 2024/08/05 07:59:08 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/08/05 07:59:08 connecting evcc-1217708430 at tcp://192.168.178.83:1883
[mqtt  ] DEBUG 2024/08/05 07:59:08 tcp://192.168.178.83:1883 connected
[main  ] INFO 2024/08/05 07:59:08 listening at :7070
[proxy-5200] DEBUG 2024/08/05 07:59:08 modbus proxy for 192.168.178.132:502 listening at :5200
[lp-2  ] WARN 2024/08/05 07:59:10 deprecated: phases setting is ignored, please remove
[lp-2  ] DEBUG 2024/08/05 07:59:10 charge total import: 55.145kWh
[site  ] INFO 2024/08/05 07:59:11 site config:
[site  ] INFO 2024/08/05 07:59:11   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/08/05 07:59:11     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/08/05 07:59:11     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/08/05 07:59:11     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/08/05 07:59:11   vehicles:
[site  ] INFO 2024/08/05 07:59:11     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[site  ] INFO 2024/08/05 07:59:11     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/08/05 07:59:11 loadpoint 1:
[lp-1  ] INFO 2024/08/05 07:59:11   mode:        pv
[lp-1  ] INFO 2024/08/05 07:59:11   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/08/05 07:59:11   meters:      charge ✓
[lp-1  ] INFO 2024/08/05 07:59:11     charge:    power ✓ energy ✗ currents ✓
[lp-2  ] INFO 2024/08/05 07:59:11 loadpoint 2:
[lp-2  ] INFO 2024/08/05 07:59:11   mode:        pv
[lp-2  ] INFO 2024/08/05 07:59:11   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2024/08/05 07:59:11   meters:      charge ✓
[lp-2  ] INFO 2024/08/05 07:59:11     charge:    power ✓ energy ✓ currents ✗
[site  ] DEBUG 2024/08/05 07:59:11 set buffer soc: 100
[site  ] DEBUG 2024/08/05 07:59:11 set buffer start soc: 100
[site  ] DEBUG 2024/08/05 07:59:11 set priority soc: 100
[site  ] DEBUG 2024/08/05 07:59:11 set battery discharge control: true
[site  ] DEBUG 2024/08/05 07:59:11 set residual power: 100
[lp-1  ] DEBUG 2024/08/05 07:59:11 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/08/05 07:59:11 phase timer inactive
[lp-1  ] DEBUG 2024/08/05 07:59:11 pv timer inactive
[lp-2  ] DEBUG 2024/08/05 07:59:11 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 1p charger)
[lp-2  ] DEBUG 2024/08/05 07:59:11 phase timer inactive
[lp-2  ] DEBUG 2024/08/05 07:59:11 pv timer inactive
[site  ] DEBUG 2024/08/05 07:59:11 ----
[lp-1  ] DEBUG 2024/08/05 07:59:11 charge power: 0W
[lp-1  ] DEBUG 2024/08/05 07:59:11 charge currents: [0.4 0.4 0.2]A
[lp-2  ] DEBUG 2024/08/05 07:59:11 charge power: 0W
[site  ] DEBUG 2024/08/05 07:59:14 pv power: 893W
[site  ] DEBUG 2024/08/05 07:59:16 battery soc: 34%
[site  ] DEBUG 2024/08/05 07:59:16 battery power: -435W
[site  ] DEBUG 2024/08/05 07:59:17 grid meter: 1W
[site  ] DEBUG 2024/08/05 07:59:21 grid currents: [-0.74 1.08 -0.31]A
[... 90 lines cut ...]
[tesla ] TRACE 2024/08/05 10:51:20 POST https://tesla.evcc.io/api/1/vehicles/**VIN-T1**/command/set_charging_amps
[tesla ] TRACE 2024/08/05 10:51:21 {"charging_amps": 2}
--
{"response":{"result":true,"reason":""}}
[tesla ] TRACE 2024/08/05 10:54:18 POST https://tesla.evcc.io/api/1/vehicles/**VIN-T1**/command/set_charging_amps
[tesla ] TRACE 2024/08/05 10:54:19 {"charging_amps": 3}
--
{"response":{"result":true,"reason":""}}
[tesla ] TRACE 2024/08/05 10:55:20 POST https://tesla.evcc.io/api/1/vehicles/**VIN-T1**/command/set_charging_amps
[tesla ] TRACE 2024/08/05 10:55:21 {"charging_amps": 4}
--
{"response":{"result":true,"reason":""}}
[tesla ] TRACE 2024/08/05 10:56:20 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/**VIN-T1**/vehicle_data
[tesla ] TRACE 2024/08/05 10:56:20 {"response":{"id":929667535129410,"user_id":1689177578561886,"vehicle_id":1689225462337603,"vin":"**VIN-T1**","color":null,"access_type":"OWNER","granular_access":{"hide_private":false},"tokens":null,"state":"online","in_service":false,"id_s":"929667535129410","calendar_enabled":true,"api_version":76,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false,"charge_state":{"battery_heater_on":false,"battery_level":31,"battery_range":86.94,"charge_amps":4,"charge_current_request":4,"charge_current_request_max":16,"charge_enable_request":true,"charge_energy_added":0.54,"charge_limit_soc":70,"charge_limit_soc_max":100,"charge_limit_soc_min":50,"charge_limit_soc_std":80,"charge_miles_added_ideal":2.0,"charge_miles_added_rated":2.0,"charge_port_cold_weather_mode":false,"charge_port_color":"\u003cinvalid\u003e","charge_port_door_open":true,"charge_port_latch":"Engaged","charge_rate":9.3,"charger_actual_current":4,"charger_phases":2,"charger_pilot_current":16,"charger_power":3,"charger_voltage":232,"charging_state":"Charging","conn_charge_cable":"IEC","est_battery_range":100.59,"fast_charger_brand":"Tesla","fast_charger_present":false,"fast_charger_type":"ACSingleWireCAN","ideal_battery_range":86.94,"max_range_charge_counter":0,"minutes_to_full_charge":680,"not_enough_power_to_heat":null,"off_peak_charging_enabled":false,"off_peak_charging_times":"all_week","off_peak_hours_end_time":360,"preconditioning_enabled":false,"preconditioning_times":"weekdays","scheduled_charging_mode":"Off","scheduled_charging_pending":false,"scheduled_charging_start_time":null,"scheduled_charging_start_time_app":0,"scheduled_departure_time":1711085400,"scheduled_departure_time_minutes":390,"supercharger_session_trip_planner":false,"time_to_full_charge":11.33,"timestamp":1722848180809,"trip_charging":false,"usable_battery_level":31,"user_charge_enable_request":true},"climate_state":{"allow_cabin_overheat_protection":true,"auto_seat_climate_left":false,"auto_seat_climate_right":false,"auto_steering_wheel_heat":false,"battery_heater":false,"battery_heater_no_power":null,"bioweapon_mode":false,"cabin_overheat_protection":"Off","cabin_overheat_protection_actively_cooling":false,"climate_keeper_mode":"off","cop_activation_temperature":"High","defrost_mode":0,"driver_temp_setting":22.0,"fan_status":0,"hvac_auto_request":"On","inside_temp":25.7,"is_auto_conditioning_on":false,"is_climate_on":false,"is_front_defroster_on":false,"is_preconditioning":false,"is_rear_defroster_on":false,"left_temp_direction":0,"max_avail_temp":28.0,"min_avail_temp":15.0,"outside_temp":21.0,"passenger_temp_setting":22.0,"remote_heater_control_enabled":false,"right_temp_direction":0,"seat_heater_left":0,"seat_heater_rear_center":0,"seat_heater_rear_left":0,"seat_heater_rear_right":0,"seat_heater_right":0,"side_mirror_heaters":false,"steering_wheel_heat_level":0,"steering_wheel_heater":false,"supports_fan_only_cabin_overheat_protection":true,"timestamp":1722848180809,"wiper_blade_heater":false},"drive_state":{"power":-2,"shift_state":null,"speed":null,"timestamp":1722848180809},"gui_settings":{"gui_24_hour_time":true,"gui_charge_rate_units":"km/hr","gui_distance_units":"km/hr","gui_range_display":"Rated","gui_temperature_units":"C","gui_tirepressure_units":"Bar","show_range_units":false,"timestamp":1722848180809},"parked_accessory":{"aux_park_lamps":"Eu","badge_version":0,"can_accept_navigation_requests":true,"can_actuate_trunks":true,"car_special_type":"base","car_type":"modely","charge_port_type":"CCS","cop_user_set_temp_supported":true,"dashcam_clip_save_supported":true,"default_charge_to_max":false,"driver_assist":"TeslaAP3","ece_restrictions":true,"efficiency_package":"MY2021","eu_vehicle":true,"exterior_color":"SolidBlack","exterior_trim":"Black","exterior_trim_override":"","has_air_suspension":false,"has_ludicrous_mode":false,"has_seat_cooling":false,"headlamp_type":"Global","interior_t [... cut ...] ms_pressure_fr":2.95,"tpms_pressure_rl":2.95,"tpms_pressure_rr":3.05,"tpms_rcp_front_value":2.9,"tpms_rcp_rear_value":2.9,"tpms_soft_warning_fl":false,"tpms_soft_warning_fr":false,"tpms_soft_warning_rl":false,"tpms_soft_warning_rr":false,"valet_mode":false,"valet_pin_needed":false,"vehicle_name":"T1","vehicle_self_test_progress":0,"vehicle_self_test_requested":false,"webcam_available":true}}}

```
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.129.0

andig, 2024-08-05:
> Bei mir funktioniert das Verwenden des commandProxy nicht zuverlässig.

Hier fehlt die Fehlerbeschreibung.

swa72, 2024-08-05:
Danke für die Kommentar. Mein Fehler lautet: Nutzung von `commandProxy` konfiguriert, dennoch wird `tesla.evcc.io` verwendet.

andig, 2024-08-05:
Was sagt

    cat /config/evcc.yaml | grep --context=3 commandProxy

swa72, 2024-08-05:
Beim `cat` Befehl mag er den context nicht.
```
➜  ~ cat /config/evcc.yaml | grep commandProxy 
    commandProxy: http://192.168.178.78:8080
    commandProxy: http://192.168.178.78:8080
```

Zur Vollständigkeit noch einmal das jüngste Log
[evcc-20240805-134653-trace.log](https://github.com/user-attachments/files/16495132/evcc-20240805-134653-trace.log)

swa72, 2024-08-05:
Ich wünschte, dass ich das Verhalten besser eingrenzen kann, aber bei einem Tesla nimmt es den Proxy, beim anderen ggw. nicht. Leider enthält der letzte Satz auch einen Weichmacher, denn es hat bereits für den anderen Tesla auch funktioniert. 
Die Konfiguration ist identisch und ich habe das HA-addon bereits häufig durchgestartet. 
```
vehicles:
  # - name: T1 # no spaces or special characters
  # type: template
  # template: tesla-command
  # accessToken: XXX
  # refreshToken: XXX
  # vin: XXX7002
  # no spaces or special characters in the next line
  - name: R1
    type: template
    template: tesla
    accessToken: XXX
    refreshToken: XXX
    vin: XXX000
    commandProxy: http://192.168.178.78:8080
  # no spaces or special characters in the next line
  - name: T1
    type: template
    template: tesla
    accessToken: XXX
    refreshToken: XXX
    vin: XXX7002
    commandProxy: http://192.168.178.78:8080
```
```

andig, 2024-08-05:
> aber bei einem Tesla nimmt es den Proxy, beim anderen ggw. nicht.

Und nochmal ein neuer Informationshappen. Es ist eigentlich fast unmöglich, dass das passiert (tm). Könntest Du deine vollständige Config bitte an info@evcc.io schicken?

andig, 2024-08-05:
Config confirmed working.

laenglea, 2024-08-08:
Hallo Miteinander,

habe eine aehnliche Konfiguration - auch mit zwei Teslas, auch beide via commandProxy und sehe im Log fehlerhafte calls via tesla.evcc.io

Falls der Fehler bei swa72 an der Config lag habe ich moeglicherweise denselben?

Da ich gerade beide Autos zur Verfuegung haette koennte ich bei bedarf auch weitere tests machen.
Sg,
Alex

`[lp-2 ] DEBUG 2024/08/08 17:18:45 charge power: 0W                                                        
[lp-2 ] DEBUG 2024/08/08 17:18:45 charge currents: [0 0.3 0.2]A
[lp-3 ] DEBUG 2024/08/08 17:18:45 charge power: 0W
[site ] DEBUG 2024/08/08 17:18:45 pv power: 6359W
[site ] DEBUG 2024/08/08 17:18:45 grid meter: -144W
[site ] DEBUG 2024/08/08 17:18:45 site power: -144W
[lp-2 ] DEBUG 2024/08/08 17:18:45 charge voltages: [4.7 0 0]V
[lp-2 ] DEBUG 2024/08/08 17:18:45 !! session: chargeRater.chargedEnergy=14.7 - chargedAtStartup=0.0
[lp-2 ] DEBUG 2024/08/08 17:18:45 charger status: B
[lp-2 ] DEBUG 2024/08/08 17:18:45 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
[lp-2 ] DEBUG 2024/08/08 17:18:45 pv charge current: 0.209A = 0A + 0.209A (-144W @ 3p)
[tesla ] TRACE 2024/08/08 17:18:45 POST https://tesla.evcc.io/api/1/vehicles/<***VIN***>/command/set_charging_amps
[tesla ] TRACE 2024/08/08 17:18:45 {"charging_amps": 3} -- Retry in 31275 seconds
[lp-2 ] ERROR 2024/08/08 17:18:45 max charge current 3A: 429 Too Many Requests
[site ] TRACE 2024/08/08 17:18:45 telemetry: charge: Δ72/72Wh @ 4347W
[site ] TRACE 2024/08/08 17:18:45 POST https://api.evcc.io/v1/charge
[site ] TRACE 2024/08/08 17:18:45 {"instanceId":"3d2348bb34f019b6bf756c117a8d0bc3b5028d8a22150d4fb5791b0b57f9587f","chargePower":4347.469999999999,"greenPower":4347.469999999999,"chargeEnergy":0.07189999999999985,"greenEnergy":0.07189999999999985} -- {"status":"ok"}
[mqtt ] TRACE 2024/08/08 17:18:50 recv pvdataevcc/sensor/pv01_power_meter_active_power: '1608.0'
[mqtt ] TRACE 2024/08/08 17:18:50 recv pvdataevcc/sensor/pv02_power_meter_active_power: '5365'
[mqtt ] TRACE 2024/08/08 17:18:50 recv pvdataevcc/sensor/grid_power_consumption_ac_power: '480' `

andig, 2024-08-08:
Log unleserlich. Die Config funktionierte im Fall oben reproduzierbar. Was sagt bei Dir

    evcc charger --log trace --enable

laenglea, 2024-08-08:
Hoffe dieses mal funktioniert das einruecken

die config:

> vehicles:
- type: template
  template: tesla
  title: Egon
  accessToken: *****
  refreshToken: *****
  vin: XP7YGCEK6P*****2
  capacity: 76.8
  name: egon
  commandProxy: http://172.16.80.77:8080

- type: template
  template: tesla
  title: Rennate
  accessToken: *****
  refreshToken: *****
  vin: XP7YGCEK6*****1
  capacity: 76.8
  name: rennate
  commandProxy: http://172.16.80.77:8080

anbei der trace:

> /app # evcc charger --log trace --enable
[main  ] INFO 2024/08/08 21:24:46 evcc 0.129.0
[main  ] INFO 2024/08/08 21:24:46 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/08 21:24:46 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/08/08 21:24:46 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT * FROM `settings` 54 <nil>
[mqtt  ] INFO 2024/08/08 21:24:46 connecting evcc-4204597 at tcp://172.16.120.245:1883
[mqtt  ] DEBUG 2024/08/08 21:24:46 tcp://172.16.120.245:1883 connected
[db    ] TRACE 2024/08/08 21:24:46 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2024/08/08 21:24:46 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[tasmota] TRACE 2024/08/08 21:24:46 GET http://172.16.90.73/cm?cmnd=Status+0&password=&user=
[tasmota] TRACE 2024/08/08 21:24:47 {"Status":{"Module":46,"DeviceName":"tm-wasserboiler","FriendlyName":["Boiler"],"Topic":"tasmota_555694","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://172.16.120.245:8080/13_tasmota-minimal.bin.gz","RestartReason":"Exception","Uptime":"2T08:46:16","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":94,"BCResetTime":"1970-01-01T00:00:00","SaveCount":1468,"SaveAddress":"F8000"},"StatusFWR":{"Version":"13.2.0.1(tasmota)","BuildDateTime":"2023-10-28T14:22:40","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"387/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SuperMQTT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":636,"Free":364,"Heap":20,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"},"StatusNET":{"Hostname":"tm-wasserboiler","IPAddress":"172.16.90.73","Gateway":"172.16.90.254","Subnetmask":"255.255.255.0","DNSServer1":"172.16.90.254","DNSServer2":"0.0 [... cut ...]
in diesem falle jetzt fuer den zweiten an die tesla api.
> [tesla ] TRACE 2024/08/08 17:18:45 POST https://tesla.evcc.io/api/1/vehicles/XP7YGCE*****2/command/set_charging_amps
[tesla ] TRACE 2024/08/08 17:18:45 {"charging_amps": 3} -- Retry in 31275 seconds
[lp-2 ] ERROR 2024/08/08 17:18:45 max charge current 3A: 429 Too Many Requests 

Manchmal ist es auch umgekehrt und calls fuer den ***2 werden lokal abgearbeitet und ***1 gehen richtung tesla api.
Manuelle calls an das gateway funktionieren fuer beide.

andig, 2024-08-08:
> [main ] ERROR 2024/08/08 21:24:47 enable: loadpoint not initialized

Schade, das kann man so nicht testen.

> fuer einen Tesla schickt er sauber an den proxy wie hier z.b.

Wenn das passiert- ab Start falsch? Oder zwischendrin? Verwendest Du HA?

laenglea, 2024-08-08:
> Wenn das passiert- ab Start falsch? Oder zwischendrin? Verwendest Du HA?

auffallen tut es am abend. Auto wird angesteckt und laedt dann los anstatt zu stoppen.
Habe gerade nochmal evcc neu gestartet.
Der eine Tesla kann problemlos gesteuert werden und die commands kommen am proxy auch an. Fuer den anderen hagelt es nur fehler von der tesla api.

Gut moeglich, dass er nach einem evcc neustart auch den ersten Wagen mit dem er interagiert richtung proxy schickt und den zweiten nicht.

evcc laeuft im Docker-container und bekommt die meisten daten (PV, Meter) via mqtt

Kann auch gerne das ganze docker trace-log zur Verfuegung stellen und per mail senden wenn benoetigt.

Sg,
Alex

swa72, 2024-08-09:
Ich hatte das add-on von HA im Einsatz und dort die kuriose Situation, dass ein Ändern der Reihenfolg in der `evcc.yaml` einen Einfluss darauf hatte, ob EVCC den Proxy nimmt oder nicht. Erklären konnte ich mir das nicht - mal funktionierte die Konfiguration, mal nicht. Ich habe dann verzweifelt aufgegeben. 

Mittlerweile habe ich EVCC von HA getrennt (separate VM auf Proxmox) und es läuft stabil (noch).

[image]

andig, 2024-08-10:
Es gibt keinen Grund, dass evcc zur Laufzeit die Config wechseln sollte. Es braucht weiter ein Log ab Start, ohne HA, das ein Problem zeigt.

laenglea, 2024-08-10:
Konnte nun einen testlauf machen.
10:21 erstes auto (Rennate) angesteckt
10:22 zweites auto (Egon) angesteckt
10:25 Hausecke (Rennate) auf off
10:26 Garage (Egon) off
dann:
Hausecke (Rennate) auf Fast (ging ueber tesla api)
Garage (Egon) auf Fast (ueber Proxy)
Hausecke (Rennate) auf Off (wieder tesla)
Garage (Egon) Off (wieder ueber Proxy)

Das Tracelog:
https://gist.github.com/laenglea/4ab47ecc5f58059d2bf3bec3911d172c

eventuell hilft das

andig, 2024-08-10:
Das sagt leider nichts über Deine Config. Einfache Diagnose: Rennate steht auf Tesla API.

laenglea, 2024-08-10:
> Das sagt leider nichts über Deine Config. Einfache Diagnose: Rennate steht auf Tesla API.

Leider nicht.
Anbei die config:
https://gist.github.com/laenglea/9c386a32b2c9bd536aef90ae6641db9f

andig, 2024-08-10:
Sagst Du ;) Ab morgen wird im nightly der Proxy ausgegeben, dann sehen wir weiter.

andig, 2024-08-10:
In 15min gibts neues Nightly, gerne auch heute testen.

laenglea, 2024-08-10:
https://gist.github.com/laenglea/08c3274203da8df8178db62858b447cf

16:27 erstes auto (Rennate) angesteckt
16:29 zweites auto (Egon) angesteckt
16:30 Hausecke (Rennate) auf off
16:31 Garage (Egon) off

dann:
Hausecke (Rennate) auf Fast 
Garage (Egon) auf Fast
Hausecke (Rennate) auf Off 
Garage (Egon) Off

andig, 2024-08-10:
…und? Kann von unterwegs nicht rein schauen, bin aber neugierig 🧐

laenglea, 2024-08-10:
oben steht jetzt:
    [tesla ] INFO 2024/08/10 16:25:25 !! tesla proxy for Rennate: http://172.16.80.77:8080
    [tesla ] INFO 2024/08/10 16:25:25 !! tesla proxy for Egon: http://172.16.80.77:8080
jedoch immer noch calls via api
    [site  ] TRACE 2024/08/10 16:33:00 POST https://api.evcc.io/v1/charge
braucht es diese calls?

laenglea, 2024-08-10:
hier noch eines von heute abend:
https://gist.github.com/laenglea/847aadce378aa8d50e79fcc40636b162

22:06 erstes auto (Rennate) angesteckt
22:08 zweites auto (Egon) angesteckt
22:09 Hausecke (Rennate) auf off
22:10 Garage (Egon) off (laedt weiter)

dann:
Hausecke (Rennate) auf Fast
Garage (Egon) auf Fast
Hausecke (Rennate) auf Off
Garage (Egon) Off

andig, 2024-08-11:
Mir fehlt leider die Zeit das alles durchzuwühlen. Gibts denn noch ein Problem?

laenglea, 2024-08-11:
> Mir fehlt leider die Zeit das alles durchzuwühlen. Gibts denn noch ein Problem?

Ja!
eines der beiden Autos geht trotz eingestelltem proxy ueber die tesla-api und hat damit probleme mit dem api-call limit.

bei den ersten beiden traces war es Rennate, beim dritten Egon.

Das fuehrt in meinem Fall (teilweise) dazu dass ein auto sich nicht sauber steuert und auch am abend die Ladung im PV Modus nicht abdreht und dann in der Nacht laedt.

Zielladungen funktionieren dadurch auch nicht zwingend - vorallem jene am abend wenn das api-limit schon erreicht wurde.

andig, 2024-08-11:
Wo sehe ich das im Log?

laenglea, 2024-08-11:
Das ist nur die Folge davon wie in #14226 beschrieben.
Da wir gerade urlaub haben und die Autos relativ voll sind und meist genug ueberschuss vorhanden ist kann ich das so nicht nachstellen.

Wuerden die commands beider Autos sauber ueber den Proxy laufen wuerde ich den Tesla fleet API error 429 ja niemals bekommen. 

Habe den Bluetooth Proxy ja genau deswegen hier.

Im log sehen wir nur post-calls welche anstatt an den proxy an die api gehen. Bei Log1 und Log2 fuer Rennate und bei Log3 fuer Egon ohne dass sich an der evcc.yaml was geaendert hat

andig, 2024-08-11:
> Wo sehe ich das im Log?

Du beschreibst einen vermeintlichen Fehler. Bei welchem Timestamp?

laenglea, 2024-08-11:
Musste gerade suchen da die Fehler nicht sofort nach start auftreten.
https://gist.github.com/laenglea/af6ab8ccd193d3319e85d655fe1dc11b
das ist das Log von gestern 10:19 nur etwas laenger

Hier sind ein paar fehler mit drauf.
z.b. [lp-1  ] ERROR 2024/08/10 13:14:18 charger disable: Post "https://tesla.evcc.io/api/1/vehicles/VIN_Rennate/command/charge_stop": net/http: request canceled (Client.Timeout exceeded while awaiting headers)

andig, 2024-08-11:
Ich frag mich manchmal, ob das so eine Art Versteckspiel ist. In dem ganzen Log sehe ich keine Ausgabe von `commandProxy`. Wo ist die hin? Eben gabs die noch...

laenglea, 2024-08-11:
die Teslaproxy ausgabe gibt es ja erst seit gestern mittag in der nightly. Das Log ist von davor. 

Du wolltest ja sehen wie er in das api-timeout rennt.
die config ist immer dieselbe.
Anbei die config:
https://gist.github.com/laenglea/9c386a32b2c9bd536aef90ae6641db9f

Bitte sag mir was ich wie testen soll und ich werde es machen.
Gerne schicke ich die logs auch ohne sie immer anonymisieren zu muessen.

Fuer mich als user schaut es aus als ob commandProxy bei zwei Fahrzeugen einfach nicht alles ueber command proxy schickt und das fuehrt zu problemen.

Wie kann ich dich bei der Fehlersuche unterstuetzen?

> Ich frag mich manchmal, ob das so eine Art Versteckspiel ist.
definitiv nicht

andig, 2024-08-11:
Das Problem ist unverändert. Es braucht ein aktuelles Log in dem nachvollziehbar wäre, dass a) die Config beim Start korrekt ist (Log Ausgabe) und dennoch b) falsche Requests gemacht werden. Bisher habe ich das noch nicht gesehen.

laenglea, 2024-08-11:
vielleicht hilft dieser.
Habe den container mehrmals neu gestartet
https://gist.github.com/laenglea/a0f51d9478780778951f3db2ec4601d4

docker-compose logs -f | grep 'tesla proxy'
 [tesla ] INFO 2024/08/11 14:52:09 !! tesla proxy for Egon: http://172.16.80.77:8080
 [tesla ] INFO 2024/08/11 14:52:09 !! tesla proxy for Rennate: https://tesla.evcc.io/
 [tesla ] INFO 2024/08/11 14:52:41 !! tesla proxy for Egon: https://tesla.evcc.io/
 [tesla ] INFO 2024/08/11 14:52:41 !! tesla proxy for Rennate: http://172.16.80.77:8080
 [tesla ] INFO 2024/08/11 14:52:57 !! tesla proxy for Egon: http://172.16.80.77:8080
 [tesla ] INFO 2024/08/11 14:52:57 !! tesla proxy for Rennate: http://172.16.80.77:8080
 [tesla ] INFO 2024/08/11 14:53:14 !! tesla proxy for Rennate: https://tesla.evcc.io/
 [tesla ] INFO 2024/08/11 14:53:14 !! tesla proxy for Egon: https://tesla.evcc.io/
 [tesla ] INFO 2024/08/11 14:53:30 !! tesla proxy for Egon: https://tesla.evcc.io/
 [tesla ] INFO 2024/08/11 14:53:30 !! tesla proxy for Rennate: https://tesla.evcc.io/
 [tesla ] INFO 2024/08/11 14:53:48 !! tesla proxy for Rennate: https://tesla.evcc.io/
 [tesla ] INFO 2024/08/11 14:53:48 !! tesla proxy for Egon: http://172.16.80.77:8080

andig, 2024-08-11:
Ohne ins Log zu schauen: die Ausgabe beweist, dass für Rennate konsequent die Config falsch ist. Keine Magie.

andig, 2024-08-11:
Hurra! Ich kann den Fehler jetzt nachvollziehen:

```yaml
vehicles:
  - name: t1
    type: template
    template: tesla
    commandproxy: http://foo
  - name: t2
    type: template
    template: tesla
    commandproxy: http://foo
  - name: t3
    type: template
    template: tesla
    commandproxy: http://foo
  - name: t4
    type: template
    template: tesla
    commandproxy: http://foo
```

gibt

```
[main  ] INFO 2024/08/11 15:34:48 evcc 0.0.0
[main  ] INFO 2024/08/11 15:34:48 using config file: cmd/demo.yaml
https://tesla.evcc.io/
[main  ] ERROR 2024/08/11 15:34:49 creating vehicle t3 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': init
http://foo
[main  ] ERROR 2024/08/11 15:34:49 creating vehicle t4 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': init
http://foo
[main  ] ERROR 2024/08/11 15:34:49 creating vehicle t2 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': init
https://tesla.evcc.io/
[main  ] ERROR 2024/08/11 15:34:49 creating vehicle t1 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': init
```

...und das ist falsch.

swa72, 2024-08-11:
Ich blicke es nicht. Liegt es an der Konfiguration (mir fehlen
accessToken, refreshToken und vin - zudem dachte ich dass es case sensitive
sei "commandProxy") oder ist es die Anzahl der Fahrzeuge oder ist es der
Code von EVCC? Oder einfach  "tesla-command " statt "tesla"  Ich würde ja
gerne selber testen, bin aber mit dem Auto für ein paar Tage unterwegs...

andig, 2024-08-11:
Lass mich mal schauen. Repro ist der erste Schritt, der Rest findet sich.

andig, 2024-08-11:
Danke für den wunderbaren, tief in evcc versteckten Fehler und Eure Geduld! Mit etwas Glück ist das Problem morgen gelöst.

closed 2024-08-11 by andig

laenglea, 2024-08-11:
Vielen Dank auch für deine Geduld und den superschnellen fix ❤️