evcc-io/evcc #15252: Draws from grid when house battery not full and 1p charge (maxGridSupplyWhileBatteryCharging)

toleabivol, 2024-08-05:
### Describe the bug

Not totally sure if bug or my situation is really an edge case.

I have : go-e 3P charger, 1p AC charged Nissan Leaf 2018, Deye 3p hybrid, 7,5kWp PV, HA
When PV produces more than the inverter can output on one phase (3kW+) and the house battery is not full and the 1p car is charing then it will pull from the grid the remaining needed charging power.
e.g. PV produces 7kW , then the inverter will supply 3kW to the charger and 4kW into the house battery and will draw 4kW from grid on the charging phase.
I though using [maxGridSupplyWhileBatteryCharging: 50](https://docs.evcc.io/en/docs/reference/configuration/site#maxgridsupplywhilebatterycharging) would fix the issue and make the car load with maximum possible power without drawing more than 50W from grid. But the result is that it switches back and forth from a low max current to a higher max current.

When the house battery if full, the inverter will send the excess current not in battery  but to the other phase of the grid and "balance" the import on the other phase => 0 costs calculated by the grid meter.

### Steps to reproduce

1. Have house battery not full
2. Have 1p charging AC car
3. Have inverter that outputs less on AC than there is DC available
4. Set evcc to charge with solar
5. set maxGridSupplyWhileBatteryCharging: 50
6. Have PV that generates more than inverter can output on one phase
7. Charge the car and watch the grid meter
8. The grid meter shows that on the charing phase it draws power from grid to match the current set by evcc
9. Charge house battery till full
10. Now watch the grid again - inverter will export current on one phase to balance the import on another phase which results the grid meter to stay at ~0 (no import and no export overall calculated and thus no grid costs)

### Configuration details

```yaml
(will update later, have to switch to another OS, can't copy from HA terminal for some reason)
```

### Log details

```text
[site ] DEBUG 2024/08/05 12:05:09 ----
[lp-1 ] DEBUG 2024/08/05 12:05:09 charge power: 1840W
[lp-1 ] DEBUG 2024/08/05 12:05:09 charge currents: [8.3 0.1 0.1]A
[site ] DEBUG 2024/08/05 12:05:09 pv power: 4718W
[site ] DEBUG 2024/08/05 12:05:09 battery soc: 91%
[site ] DEBUG 2024/08/05 12:05:09 battery power: -2095W
[site ] DEBUG 2024/08/05 12:05:09 grid meter: 6W
[site ] DEBUG 2024/08/05 12:05:09 site power: -1989W
[lp-1 ] DEBUG 2024/08/05 12:05:09 charge voltages: [219 221 222]V
[lp-1 ] DEBUG 2024/08/05 12:05:09 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/05 12:05:09 detected active phases: 1p
[lp-1 ] DEBUG 2024/08/05 12:05:09 charge total import: 1763.000kWh
[lp-1 ] DEBUG 2024/08/05 12:05:09 charger status: C
[lp-1 ] DEBUG 2024/08/05 12:05:09 soc estimated: 18.00% (vehicle: 18.00%)
[lp-1 ] DEBUG 2024/08/05 12:05:09 vehicle soc: 18%
[lp-1 ] DEBUG 2024/08/05 12:05:09 vehicle range: 42km
[lp-1 ] DEBUG 2024/08/05 12:05:09 !! active phases: 1p = min(1p measured 1p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/05 12:05:09 pv charge current: 16.6A = 8A + 8.65A (-1989W @ 1p)
[lp-1 ] DEBUG 2024/08/05 12:05:09 max charge current: 16A
[site ] DEBUG 2024/08/05 12:05:39 ----
[lp-1 ] DEBUG 2024/08/05 12:05:39 charge power: 3670W
[lp-1 ] DEBUG 2024/08/05 12:05:39 charge currents: [16.4 0 0.1]A
[site ] DEBUG 2024/08/05 12:05:39 pv power: 4785W
[site ] DEBUG 2024/08/05 12:05:39 battery soc: 91%
[site ] DEBUG 2024/08/05 12:05:39 battery power: -892W
[site ] DEBUG 2024/08/05 12:05:39 grid meter: 773W
[site ] DEBUG 2024/08/05 12:05:39 site power: 873W
[lp-1 ] DEBUG 2024/08/05 12:05:39 charge voltages: [219 221 222]V
[lp-1 ] DEBUG 2024/08/05 12:05:39 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/05 12:05:39 detected active phases: 1p
[lp-1 ] DEBUG 2024/08/05 12:05:39 charge total import: 1763.000kWh
[lp-1 ] DEBUG 2024/08/05 12:05:39 charger status: C
[lp-1 ] DEBUG 2024/08/05 12:05:39 soc estimated: 18.00% (vehicle: 18.00%)
[lp-1 ] DEBUG 2024/08/05 12:05:39 vehicle soc: 18%
[lp-1 ] DEBUG 2024/08/05 12:05:39 vehicle range: 42km
[lp-1 ] DEBUG 2024/08/05 12:05:39 !! active phases: 1p = min(1p measured 1p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/05 12:05:39 pv charge current: 12.2A = 16A + -3.8A (873W @ 1p)
[... 81 lines cut ...]
[site ] DEBUG 2024/08/05 12:08:09 battery soc: 91%
[site ] DEBUG 2024/08/05 12:08:09 battery power: -1457W
[site ] DEBUG 2024/08/05 12:08:09 grid meter: 9W
[site ] DEBUG 2024/08/05 12:08:09 site power: -1348W
[lp-1 ] DEBUG 2024/08/05 12:08:09 charge voltages: [219 221 221]V
[lp-1 ] DEBUG 2024/08/05 12:08:09 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/05 12:08:09 detected active phases: 1p
[lp-1 ] DEBUG 2024/08/05 12:08:09 charge total import: 1763.000kWh
[lp-1 ] DEBUG 2024/08/05 12:08:09 charger status: C
[lp-1 ] DEBUG 2024/08/05 12:08:09 soc estimated: 18.00% (vehicle: 18.00%)
[lp-1 ] DEBUG 2024/08/05 12:08:09 vehicle soc: 18%
[lp-1 ] DEBUG 2024/08/05 12:08:09 vehicle range: 42km
[lp-1 ] DEBUG 2024/08/05 12:08:09 !! active phases: 1p = min(1p measured 1p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/05 12:08:09 pv charge current: 17.9A = 12A + 5.86A (-1348W @ 1p)
[lp-1 ] DEBUG 2024/08/05 12:08:09 max charge current: 17A
[site ] DEBUG 2024/08/05 12:08:39 ----
```

### What type of operating system are you running?

Linux

### Version

1.129.0

premultiply, 2024-08-05:
Thank you for the report.

@andig I think we have to investigate a bit deeper what exactly happens here.

andig, 2024-08-08:
Das Problem ist das nichtlineare Verhalten. Bei Bezug wird die WB-Leistung zurück gefahren. Wird die aber zu klein- und kann die Batterie die zusätzliche Leistung nicht aufnehmen (die ist am Ladeende!)- so kommt es zu Einspeisung. Die wird dann wieder zugunsten der WB umgeshiftet. 

Das eigentliche Problem hier scheint mir aber zu sein, dass die Batterie einfach nicht liefert:

```
[site ] DEBUG 2024/08/05 12:05:39 pv power: 4785W
[site ] DEBUG 2024/08/05 12:05:39 battery soc: 91%
[site ] DEBUG 2024/08/05 12:05:39 battery power: -892W
[site ] DEBUG 2024/08/05 12:05:39 grid meter: 773W
[site ] DEBUG 2024/08/05 12:05:39 site power: 873W
```

Warum liefert hier der Akku nicht die 870W die aus dem Netz gezogen werden???

Ich sehe da kein Problem bei evcc.

toleabivol, 2024-08-09:
> Warum liefert hier der Akku nicht die 870W die aus dem Netz gezogen werden???

Because inverter cannot output more on that one phase that is used and draws from grid and the rest of PV generation goes to battery even if battery.

> PV produces more than the inverter can output on one phase (3kW+) and the house battery is not full and the 1p car is charing then it will pull from the grid the remaining needed charging power.

> Deye 3p Hybrid 8kW (3kW AC output per phase)

> When PV produces more than the inverter can output on one phase (3kW+) and the house battery is not full and the 1p car is charing then it will pull from the grid the remaining needed charging power.

And this sounds exactly what [maxGridSupplyWhileBatteryCharging](https://docs.evcc.io/en/docs/reference/configuration/site#maxgridsupplywhilebatterycharging) should do as it also answer your question :

> Normally, the current battery charging power is considered as additional available vehicle charging power (once prioritySoc is reached). In the above example, grid consumption could occur equal to the current battery charging power since the inverter cannot provide this power to the grid (and thus the vehicle). The fully utilized AC path of the hybrid inverter creates an unexpected bottleneck for the standard regulation.

>  the DC power generated in combination with a directly connected storage system can be greater than the AC output power of the inverter. This can result in grid consumption during vehicle charging, even when the battery is being charged simultaneously.

In this case evcc should do (and from docs it's exactly what it's already thought to do as I read it) :
- if battery not full : detect grid import and allow only the threshold specified in `maxGridSupplyWhileBatteryCharging`
- if battery full : ignore setting of `maxGridSupplyWhileBatteryCharging`

From my logs you can see it is detecting grid import, lowers the current BUT then it switches back. And then it switches back and forth.

> it switches back and forth from a low max current to a higher max current.

andig, 2024-08-09:
> Because inverter cannot output more on that one phase that is used and draws from grid and the rest of PV generation goes to battery even if battery.

Maybe I misunderstand. But we've said the inverter has 3kW per phase. Where is that power?

toleabivol, 2024-08-09:
Most of it goes into the car charger and some goes to appliances in the house that are on the same phase.
From the part of the log you quoted, 
PV outputs 4785
Charger sends to car 3670W
Drawn from grid 773 (this is ~what inverter cannot output on the phase so it draws from grid, and what we want to keep at ~0)
Charging house battery 892W (This is what's left from PV output - max phase output - other house appliances on other phases)

```
[lp-1 ] DEBUG 2024/08/05 12:05:09 max charge current: 16A
[site ] DEBUG 2024/08/05 12:05:39 ----
[lp-1 ] DEBUG 2024/08/05 12:05:39 charge power: **3670W**  <---Here is
[lp-1 ] DEBUG 2024/08/05 12:05:39 charge currents: [16.4 0 0.1]A
[site ] DEBUG 2024/08/05 12:05:39 pv power: 4785W
[site ] DEBUG 2024/08/05 12:05:39 battery soc: 91%
[site ] DEBUG 2024/08/05 12:05:39 battery power: -892W
[site ] DEBUG 2024/08/05 12:05:39 grid meter: 773W
[site ] DEBUG 2024/08/05 12:05:39 site power: 873W
```

toleabivol, 2024-08-09:
Not sure if it's the same issue, but it seems to be, because after adding this parameter it behave strange, here is one example when it went to full 32A (6.3kW) when PV has only ~4.5KW generation and battery SOC is not at 95%+ when it should support charging.

```
[site ] DEBUG 2024/08/09 11:51:43 ----
[lp-1 ] DEBUG 2024/08/09 11:51:43 charge power: 1320W
[lp-1 ] DEBUG 2024/08/09 11:51:43 charge currents: [5.9 0 0]A
[site ] DEBUG 2024/08/09 11:51:43 pv power: 4598W
[site ] DEBUG 2024/08/09 11:51:43 battery soc: 77%
[site ] DEBUG 2024/08/09 11:51:43 battery power: -2497W
[site ] DEBUG 2024/08/09 11:51:43 grid meter: 7W
[site ] DEBUG 2024/08/09 11:51:43 site power: -2390W
[lp-1 ] DEBUG 2024/08/09 11:51:43 charge voltages: [220 221 222]V
[lp-1 ] DEBUG 2024/08/09 11:51:43 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/09 11:51:43 detected active phases: 1p
[lp-1 ] DEBUG 2024/08/09 11:51:43 !! session: chargeRater.chargedEnergy=1.0 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/08/09 11:51:43 charge total import: 1794.000kWh
[lp-1 ] DEBUG 2024/08/09 11:51:43 charger status: C
[lp-1 ] DEBUG 2024/08/09 11:51:43 soc estimated: 54.00% (vehicle: 54.00%)
[lp-1 ] DEBUG 2024/08/09 11:51:43 vehicle soc: 54%
[lp-1 ] DEBUG 2024/08/09 11:51:43 vehicle range: 125km
[lp-1 ] DEBUG 2024/08/09 11:51:43 !! active phases: 1p = min(1p measured 1p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/09 11:51:43 pv charge current: 16.4A = 6A + 10.4A (-2390W @ 1p)
[lp-1 ] DEBUG 2024/08/09 11:51:44 max charge current: 16A
[site ] DEBUG 2024/08/09 11:52:13 ----
[lp-1 ] DEBUG 2024/08/09 11:52:13 charge power: 3600W
[lp-1 ] DEBUG 2024/08/09 11:52:13 charge currents: [16.1 0 0.1]A
[site ] DEBUG 2024/08/09 11:52:13 pv power: 4598W
[site ] DEBUG 2024/08/09 11:52:13 battery soc: 77%
[site ] DEBUG 2024/08/09 11:52:13 battery power: -2497W
[site ] DEBUG 2024/08/09 11:52:13 grid meter: 7W
[site ] DEBUG 2024/08/09 11:52:13 site power: -2390W
[lp-1 ] DEBUG 2024/08/09 11:52:13 charge voltages: [219 222 222]V
[lp-1 ] DEBUG 2024/08/09 11:52:13 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/09 11:52:13 detected active phases: 1p
[lp-1 ] DEBUG 2024/08/09 11:52:13 !! session: chargeRater.chargedEnergy=2.0 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/08/09 11:52:13 charge total import: 1795.000kWh
[lp-1 ] DEBUG 2024/08/09 11:52:13 charger status: C
[lp-1 ] DEBUG 2024/08/09 11:52:13 soc estimated: 56.25% (vehicle: 54.00%)
[lp-1 ] DEBUG 2024/08/09 11:52:13 vehicle soc: 56%
[lp-1 ] DEBUG 2024/08/09 11:52:13 vehicle range: 125km
[lp-1 ] DEBUG 2024/08/09 11:52:13 !! active phases: 1p = min(1p measured 1p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/09 11:52:13 pv charge current: 26.4A = 16A + 10.4A (-2390W @ 1p)
[lp-1 ] DEBUG 2024/08/09 11:52:14 max charge current: 26A
[site ] DEBUG 2024/08/09 11:52:43 ----
[lp-1 ] DEBUG 2024/08/09 11:52:44 charge power: 5860W
[lp-1 ] DEBUG 2024/08/09 11:52:44 charge currents: [26.2 0.1 0]A
[site ] DEBUG 2024/08/09 11:52:44 pv power: 4598W
[site ] DEBUG 2024/08/09 11:52:44 battery soc: 77%
[site ] DEBUG 2024/08/09 11:52:44 battery power: -2497W
[site ] DEBUG 2024/08/09 11:52:44 grid meter: 7W
[site ] DEBUG 2024/08/09 11:52:44 site power: -2390W
[lp-1 ] DEBUG 2024/08/09 11:52:44 charge voltages: [218 222 222]V
[lp-1 ] DEBUG 2024/08/09 11:52:44 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/09 11:52:44 detected active phases: 1p
[lp-1 ] DEBUG 2024/08/09 11:52:44 !! session: chargeRater.chargedEnergy=2.0 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/08/09 11:52:44 charge total import: 1795.000kWh
[lp-1 ] DEBUG 2024/08/09 11:52:44 charger status: C
[lp-1 ] DEBUG 2024/08/09 11:52:44 soc estimated: 56.25% (vehicle: 54.00%)
[lp-1 ] DEBUG 2024/08/09 11:52:44 vehicle soc: 56%
[lp-1 ] DEBUG 2024/08/09 11:52:44 vehicle range: 125km
[lp-1 ] DEBUG 2024/08/09 11:52:44 !! active phases: 1p = min(1p measured 1p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/09 11:52:44 pv charge current: 36.4A = 26A + 10.4A (-2390W @ 1p)
[lp-1 ] DEBUG 2024/08/09 11:52:44 max charge current: 32A
[site ] DEBUG 2024/08/09 11:53:13 ----
[lp-1 ] DEBUG 2024/08/09 11:53:13 charge power: 6370W
[lp-1 ] DEBUG 2024/08/09 11:53:13 charge currents: [28.5 0 0]A
[site ] DEBUG 2024/08/09 11:53:13 pv power: 4598W
[site ] DEBUG 2024/08/09 11:53:13 battery soc: 77%
[site ] DEBUG 2024/08/09 11:53:13 battery power: -2497W
[site ] DEBUG 2024/08/09 11:53:13 grid meter: 7W
[... 86 lines cut ...]
[lp-1 ] DEBUG 2024/08/09 11:55:13 vehicle range: 125km
[lp-1 ] DEBUG 2024/08/09 11:55:13 !! active phases: 1p = min(1p measured 1p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/09 11:55:13 pv charge current: 12A = 14A + -1.95A (449W @ 1p)
[lp-1 ] DEBUG 2024/08/09 11:55:14 max charge current: 12A
[site ] DEBUG 2024/08/09 11:55:43 ----
[lp-1 ] DEBUG 2024/08/09 11:55:43 charge power: 2700W
[lp-1 ] DEBUG 2024/08/09 11:55:43 charge currents: [12 0.1 0]A
[site ] DEBUG 2024/08/09 11:55:43 pv power: 4650W
[site ] DEBUG 2024/08/09 11:55:43 battery soc: 78%
[site ] DEBUG 2024/08/09 11:55:43 battery power: -1173W
[site ] DEBUG 2024/08/09 11:55:43 grid meter: 17W
[site ] DEBUG 2024/08/09 11:55:43 site power: -1056W
[lp-1 ] DEBUG 2024/08/09 11:55:43 charge voltages: [220 221 222]V
[lp-1 ] DEBUG 2024/08/09 11:55:43 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/09 11:55:43 detected active phases: 1p
[lp-1 ] DEBUG 2024/08/09 11:55:43 !! session: chargeRater.chargedEnergy=2.0 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/08/09 11:55:43 charge total import: 1795.000kWh
[lp-1 ] DEBUG 2024/08/09 11:55:43 charger status: C
[lp-1 ] DEBUG 2024/08/09 11:55:43 soc estimated: 56.25% (vehicle: 54.00%)
[lp-1 ] DEBUG 2024/08/09 11:55:43 vehicle soc: 56%
[lp-1 ] DEBUG 2024/08/09 11:55:43 vehicle range: 125km
[lp-1 ] DEBUG 2024/08/09 11:55:43 !! active phases: 1p = min(1p measured 1p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/09 11:55:43 pv charge current: 16.6A = 12A + 4.59A (-1056W @ 1p)
[lp-1 ] DEBUG 2024/08/09 11:55:44 max charge current: 16A
```

andig, 2024-08-09:
Wie erfolgt denn die Batterieladung? Benötigt die auch den AC Pfad oder direkt über DC?

toleabivol, 2024-08-09:
House battery is connected to DC port of the inverter. In the case above it will be charged directly from PV via inverter via the DC path.

andig, 2024-08-09:
```go
// sitePower returns the available delta power that the charger might additionally consume
// negative value: available power (grid export), positive value: grid import
func sitePower(log *util.Logger, maxGrid, grid, battery, residual float64) float64 {
	// For hybrid inverters, battery can be charged from DC power in excess of
	// inverter AC rating. This battery charge must not be counted as available for AC consumption.
	// https://github.com/evcc-io/evcc/issues/2734, https://github.com/evcc-io/evcc/issues/2986
	if maxGrid > 0 && grid > maxGrid && battery < 0 {
		log.TRACE.Printf("ignoring excess DC charging due to grid consumption: %.0fW > %.0fW", grid, maxGrid)
		battery = 0
	}

	return grid + battery + residual
}
```

Das produziert bei Über/Unterschreitung der Schwelle einen  Sprung um `battery`. Das ist hier das Problem.

toleabivol, 2024-08-16:
So it's a bug ?

jmtatsch, 2024-08-24:
I am also troubled by this issue, see https://github.com/evcc-io/evcc/discussions/15329
Losing about 10 % of solar charging share because of this bug.

WordsOfMe, 2024-08-27:
Same here. I have a 32 A wallbox and a 12 kW inverter which cannot deliver (convert from PV or battery DC to AC) more than 5 kW per phase. The automated phase switching is enabled, and occasionally evcc remains at 1 phase and approaches 22 A (5 kW) instead of switching to three phase mode. The inverter reaches its limit on one phase, and the remaining power is drawn from the grid:
[image]
As a workaround I now manually limit to 16 A when charging in PV mode.

Peter54879, 2024-09-04:
Is there a timeline when the bug will be fixed?

jmtatsch, 2024-09-08:
Today I lost around 2 kwh (20%) while only slow charging (3,5kw) with a charging brick on one phase.

[image]

andig, 2024-09-20:
@premultiply hast du eine Idee?

VolkerK62, 2024-09-20:
Aus dem code:
```
// For hybrid inverters, battery can be charged from DC power in excess of
// inverter AC rating. This battery charge must not be counted as available for AC consumption.
// https://github.com/evcc-io/evcc/issues/2734, https://github.com/evcc-io/evcc/issues/2986
```
Das Limit ist doch die Inverterleistung.
Wäre es eine Option, bei der Berechnung der sitePower, die PV-Leistung, die über dieses Limit hinausgeht, von der batteryPower abzuziehen, wenn prioritySoc überschritten ist?
Inverterleistung wäre u.U. ein neuer Parameter, aber dafür könnte dann maxGridSupply.... weg.

buennerbernd, 2024-10-07:
The bug was already described several times.
Here I have proposed a detailed solution: https://github.com/evcc-io/evcc/discussions/14565#discussioncomment-10059785
This is exactly what @VolkerK62 has proposed.

andig, 2024-10-07:
@buennerbernd die Formel verstehe ich. Müsste man aber pro WR betrachten statt global, oder? Hast Du das mal lokal mit dieser Lösung ausprobiert?

buennerbernd, 2024-10-08:
@andig Bisher habe ich mich gescheut, eine Entwicklungsumgebung aufzusetzen.
Mein Vorschlag passt für einen Hybridwechselrichter mit dort angeschlossener Batterie.
Ich war nicht so tief in die Materie eingetaucht um zu verstehen, was das im Code für mehrere Wechselrichter und Batterien bedeutet. Deshalb hatte ich auch einem Pull-Request geringe Chancen zugerechnet.
Die korrekte Lösung müsste für jeden Hybridwechselrichter seine DC angeschlossene Batterie mit seiner max AC Power verrechnen.

Es wäre super, wenn es diese Lösung in den Code findet. EVCC funktioniert so toll, nur gerade, wenn es darum geht, den Strom oberhalb der maximalen AC Leistung des WR abzuernten, dann wird es echt holperig und man kauft die Hälfte der Zeit noch etwas Strom dazu.

andig, 2024-10-08:
Technisch müssten wir das allerdings (leider) anders umsetzen da die Limitierung ja WR passier und nicht global. Deshalb bräuchte pro WR (und damit je `meter`) einen `MaxACPower` parameter der für alle Hybrid WR als Teil der `battery` Usage zu pflegen wäre.

Ich würde mir hier noch etwas mehr Feedback wünschen. Insbesondere ist mir nicht klar, warum der ursprüngliche Lösungsansatz überhaupt für einige Anwender zu funktionieren scheint.

Einen zweiten inhaltlichen Klemmer habe ich nochmal bei der Problemstellung. Wenn wir DC als "verfügbar" zählen dann würde der Ladepunkt die Leistung erhöhen (DC "verfügbar"). Dadurch steigt der Netzbezug. Die Regelung erfolgt ja nach verfügbarer `sitepower`, also `grid` plus `battery`. Da wir bei `grid` jetzt in Bezug rutschen würde ich eigentlich erwarten, dass dieser Bezug die zusätzliche DC Verfügbarkeit wieder canceled. Das scheint ja aber nicht zu passieren. Ich sehe spontan aber nicht, warum eigentlich?

buennerbernd, 2024-10-08:
Ich beschreibe noch einmal, was ich beim SMA STP 10.0 + BYD HVS 7,7 beobachtet habe. Die Batterie-Regelung im WR hat hier definitiv einen Einfluss. Das könnte begründen, warum der Parameter maxGridSupplyWhileBatteryCharging bei manchen Nutzern doch eine Verbesserung gebracht hat.

Bei mir läuft das so:
Voraussetzung ist, dass noch Platz in der Hausbatterie ist, sonst tritt der Fehler nicht auf.

Sagen wir, die Sonne scheint mit 8 kW, das Auto lädt so, dass die 8 kW abgeholt werden. Die Hausbatterie verhält sich weitestgehend ruhig.
Plötzlich scheint die Sonne mit 12 kW -> Der WR schickt 4 kW in die Hausbatterie.
EVCC erhöht nun die Ladeleistung um 4 kW. (Hier dürfte es die 10 kW AC des WR nicht überschreiten und nur um 2 kW erhöhen.)

Jetzt verbraucht das Haus + Auto 12 kW. -> 2kW werden vom Netz bezogen.
Scheinbar regelt der WR die Solarproduktion in solchen Situationen auf die 10 kW zurück und lädt erstmal nicht mehr die Hausbatterie.
EVCC korrigiert das Laden des Autos so, dass kein Bezug mehr notwendig ist: Haus + Auto 10 kW.

Wenn sich alles beruhigt hat, regelt der WR die Solarproduktion wieder auf 12 kW hoch und schickt 2 kW in die Hausbatterie.
Jetzt erhöht EVCC die Ladeleistung wieder um 2 kW. (Was es nicht tun sollte)
Es werden wieder 2 kW vom Netz bezogen. Und so fängt es an zu schwingen.

> Einen zweiten inhaltlichen Klemmer habe ich nochmal bei der Problemstellung. Wenn wir DC als "verfügbar" zählen dann würde der Ladepunkt die Leistung erhöhen (DC "verfügbar"). Dadurch steigt der [...]

Falls ich nichts übersehe, wird im Code die Batterie bei Netzbezug ignoriert und nicht verrechnet. Ich denke, deshalb hebt es sich nicht auf. Auch wenn es sich aufheben würde, käme es trotzdem zum Schwingen. Der Effekt wird umso größer, je höher die Überdimensionierung der PV ist.

andig, 2024-10-08:
> Voraussetzung ist, dass noch Platz in der Hausbatterie ist, sonst tritt der Fehler nicht auf.

Das ist klar- sonst gibts kein DC Charging.

> Jetzt verbraucht das Haus + Auto 12 kW. -> 2kW werden vom Netz bezogen.
Scheinbar regelt der WR die Solarproduktion in solchen Situationen auf die 10 kW zurück und lädt erstmal nicht mehr die Hausbatterie.

Das ist merkwürdig. Warum sollte er das tun- für den WR hat sich ja nichts geändert? Idee: kann es sein, dass der WR hier "Hausunterstützung" vor "Batterieladung" priorisiert und dabei vergisst, dass er bzgl. Hausunterstützung schon am Limit ist? Wäre das nicht eigentlich ein Supportticket wert? 

> Falls ich nichts übersehe, wird im Code die Batterie bei Netzbezug ignoriert und nicht verrechnet. Ich denke, deshalb hebt es sich nicht auf. 

Ich frag mich in der Tat warum es notwendig war das so zu bauen.

> Auch wenn es sich aufheben würde, käme es trotzdem zum Schwingen. Der Effekt wird umso größer, je höher die Überdimensionierung der PV ist.

Schwingen sehe ich jetzt nur weil Dein WR so merkwürdig reagiert (und vllt. aufgrund der aktuellen Implementierung).

WordsOfMe, 2024-10-08:
Das Problem resultiert aus der maximal möglichen Schieflast, und die ist geräteabhängig. Mein Wechselrichter kann nominal 12 kW. Wenn ich jetzt zB 8 kW Überschuss habe wird er den gleichmäßig über 3 Phasen ins Netz einspeisen. Wenn evcc nun diesen Überschuss für ein 1-phasiges Laden heranzieht, dann müsste der Wechselrichter auf nur einer Phase diese 8 kW bereitstellen - das funktioniert aber nicht, denn die maximal mögliche Leistung pro Phase beträgt (bei mir) nur 5 kW.

buennerbernd, 2024-10-08:
> Das ist merkwürdig. Warum sollte er das tun- für den WR hat sich ja nichts geändert? Idee: kann es sein, dass der WR hier "Hausunterstützung" vor "Batterieladung" priorisiert und dabei vergisst, dass er bzgl. Hausunterstützung schon am Limit ist? Wäre das nicht eigentlich ein Supportticket wert?

Das ist auch meine Theorie. Aber ich würde mich nicht daran festbeißen, weil ich denke, dass man das nur mit der max-AC-Power Konfiguration sauber und allgemeingültig lösen kann.

> > Auch wenn es sich aufheben würde, käme es trotzdem zum Schwingen. Der Effekt wird umso größer, je höher die Überdimensionierung der PV ist.
> 
> Schwingen sehe ich jetzt nur weil Dein WR so merkwürdig reagiert (und vllt. aufgrund der aktuellen Implementierung).

Angenommen, alles ist momentan im optimalen Zustand und ignorieren wir mal, dass der WR auch noch regelt:
PV produziert 12 kW, Haus + Auto bekommen 10 kW und die Hausbatterie bekommt 2 kW.
Was wird EVCC im nächsten Intervall machen, wenn es nichts von der max-AC-Power weiß? Es wird 2 kW draufpacken.
Dann läuft es mindestens ein Intervall schlecht und dann wird wieder nachgeregelt.

Ich sehe die max-AC-Power Konfiguration als einzige saubere Lösung.

Zusätzlich sehe ich bei der max-AC-Power Konfiguration noch einen weiteren Vorteil:
Eine Hausbatterie kann es gut ausgleichen, wenn EVCC mal ein wenig drüber oder drunter liegt. Sie regelt ja schneller. Im Bereich von 10 kW klappt das bei einem 10 kW Hybridwechselrichter nicht mehr. Wenn dann der Herd zuschaltet, dann kann die Hausbatterie nicht helfen und es wird Strom gekauft. Hier könnten die Nutzer sich bewusst für eine etwas kleinere max-AC-Power entscheiden, um noch etwas Spielraum für die Unterstützung durch die Hausbatterie zu haben.

buennerbernd, 2024-10-08:
> Das Problem resultiert aus der maximal möglichen Schieflast, und die ist geräteabhängig. Mein Wechselrichter kann nominal 12 kW. Wenn ich jetzt zB 8 kW Überschuss habe wird er den gleichmäßig über 3 [...]

Sorry, das Thema ist ja auch noch einphasiges Laden. Das habe ich mal komplett ignoriert.

ttjaden, 2024-10-08:
Hier zum Verständnis auch nochmal eine Zeitreihe in 1min Auflösung mit 10 kW Hybridwechselrichter und 15 kWp Photovoltaik. 

[image]

Streetscooter ist nicht steuerbar. Aber evcc könnte den MG ZS EV in der Leistung runter regeln, so dass es nicht zu Netzbezug kommt. Hinweis: Der MG ZS EV lägt einphasig, aber das sollte nicht das Problem sein.

Hier noch die Rohdaten 1min
[Ladepunkte-data-as-seriestocolumns-2024-10-08 22_17_22.csv](https://github.com/user-attachments/files/17299053/Ladepunkte-data-as-seriestocolumns-2024-10-08.22_17_22.csv)

ttjaden, 2024-10-08:
Hier nochmal besser sichtbar in 1sek Auflösung:
[image]

ttjaden, 2024-10-08:
Zoom in -> Das ist das Schwingen, was @buennerbernd  meint: 
[image]

buennerbernd, 2024-10-11:
Danke, dass du dich kümmerst. Am Wochenende werde ich verreisen. Wird also frühestens am Montag was. Kannst du mir noch Tipps geben, wie ich das bei mir einspiele?

andig, 2024-10-12:
Bzgl. https://github.com/evcc-io/evcc/pull/16609: für welche WR bräuchten wir dieses Konfigurationssetting?

Peter54879, 2024-10-12:
SMA Sunny Tripower Smart Energy

andig, 2024-10-12:
Welches Template wäre das?

Peter54879, 2024-10-12:
sma-hybrid

andig, 2024-10-13:
Die Templates sind im PR jetzt auch aktualisiert. Kann ab morgen im nightly getestet werden.

/cc @VolkerK62

buennerbernd, 2024-10-13:
> Bzgl. #16609: für welche WR bräuchten wir dieses Konfigurationssetting?

Betrifft das nicht alle Hybrid WR mit begrenzter AC Leistung?

jmtatsch, 2024-10-14:
Und fronius symo gen24 auch

andig, 2024-10-14:
Hab ich die im PR vergessen? Sonst müsste das ja jetzt "nur noch" getestet werden ;)

farcorben, 2024-10-14:
wenn ich nicht ganz schief schaue wurde [Solax](https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/solax.yaml) nicht berücksichtigt im PR.

Könntest du diese bitte nachholen?

toleabivol, 2024-10-14:
Sorry, so how do we use this `maxACPower` for the initial issue ? Is it the maximum power per phase the Inverter can output ? 
In my case it is a custom setup , not a template.

buennerbernd, 2024-10-15:
So, das Auto ist leer, das nightly von heute ist installiert und vielleicht habe ich heute Glück mit dem Wetter, wenn nicht dann morgen.

Ein paar Fragen zur Config.
Ich habe die maxacpower der Batterie zugeordnet. Ist das richtig?
Ist die Groß-Kleinschreibung wichtig?
Kann ich irgendwie überprüfen, ob die neue Configuration herangezogen wird? (API oder so?)

Bei mir sieht es jetzt so aus:

```
meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.178.29  
  name: grid1
- type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.178.168  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.178.168  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery5  
  maxacpower: 9500
```

andig, 2024-10-15:
Sieht gut aus. Jetzt bräuchte es richtig sonne > 9,5kw und dann beim laden einen Blick ins Logfile.

farcorben, 2024-10-15:
@andig , bei mir läuft ebenfalls ein "test" heute. Ich habe mir gestern den Nightly geholt, die "maxacpower" in die evcc/templates/definition/meter/solax.yaml händisch patched um testen zu können sowie ein container gebaut.

`
...
params:
  - name: usage
    choice: ["grid", "pv", "battery"]
    allinone: true
  - name: modbus
    choice: ["rs485", "tcpip"]
    baudrate: 19200
  - name: capacity
    advanced: true
  - name: maxacpower
...
  capacity: {{ .capacity }} # kWh
  maxacpower: {{ .maxacpower }} # W
  {{- end }}
`
 Hoffe das passt ^^

Meine evcc.yaml sieht so aus:
`...
- type: template
  template: solax
  id: 1
  host: 192.168.1.27
  port: 502
  usage: battery
  modbus: tcpip
  name: battery1
  capacity: "12.2"
  maxacpower: "6000" # Watt
` 

evcc ist ebenfalls ohne Fehler gestartet.
Was mir jedoch am Nightly aufgefallen ist...

[image]
Sollte da nicht "Nur bei genug PV-Überschuss" stehen?

andig, 2024-10-15:
> Ich habe mir gestern den Nightly geholt, die "maxacpower" in die evcc/templates/definition/meter/solax.yaml händisch patched um testen zu können sowie ein container gebaut.

PR wäre Klasse- dann muss ich es nicht machen ;)

> Sollte da nicht "Nur bei genug PV-Überschuss" stehen?

/cc @naltatis

farcorben, 2024-10-15:
Hallo in die runde, @andig,

an meiner Anlage kann ich die Funktionalität leider **nicht** bestätigen.
Nach meiner Ansicht nach sollte die `charge power` bei gesetzter `maxacpower ` diese nicht signifikant übersteigen, oder?

```
evcc    | [main  ] INFO 2024/10/15 10:12:24 evcc 0.130.13 (57aba0db)
evcc    | [main  ] INFO 2024/10/15 10:12:24 using config file: /etc/evcc.yaml
evcc    | [db    ] INFO 2024/10/15 10:12:24 using sqlite database: /root/.evcc/evcc.db
evcc    | [main  ] INFO 2024/10/15 10:12:30 listening at :7070
evcc    | [site  ] INFO 2024/10/15 10:13:23 site config:
evcc    | [site  ] INFO 2024/10/15 10:13:23   meters:      grid ✓ pv ✓ battery ✓
evcc    | [site  ] INFO 2024/10/15 10:13:23     grid:      power ✓ energy ✓ currents ✗
evcc    | [site  ] INFO 2024/10/15 10:13:23     pv 1:      power ✓ energy ✓ currents ✗
evcc    | [site  ] INFO 2024/10/15 10:13:23     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
evcc    | [site  ] INFO 2024/10/15 10:13:23   vehicles:
evcc    | [site  ] INFO 2024/10/15 10:13:23     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
evcc    | [site  ] INFO 2024/10/15 10:13:23     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
evcc    | [lp-1  ] INFO 2024/10/15 10:13:23 loadpoint 1:
evcc    | [lp-1  ] INFO 2024/10/15 10:13:23   mode:        off
evcc    | [lp-1  ] INFO 2024/10/15 10:13:23   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
evcc    | [lp-1  ] INFO 2024/10/15 10:13:23   meters:      charge ✓
evcc    | [lp-1  ] INFO 2024/10/15 10:13:23     charge:    power ✓ energy ✓ currents ✓
evcc    | [site  ] WARN 2024/10/15 10:13:23 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
evcc    | [lp-1  ] INFO 2024/10/15 10:13:25 car connected
evcc    | [lp-1  ] INFO 2024/10/15 10:13:33 vehicle updated: unknown -> E208

...

lp-1 ] DEBUG 2024/10/15 13:01:56 charge power: 0W
[lp-1 ] DEBUG 2024/10/15 13:01:56 charge currents: [0 0 0]A
[site ] DEBUG 2024/10/15 13:01:56 grid power: -182W
[site ] DEBUG 2024/10/15 13:01:57 battery soc: 70%
[site ] DEBUG 2024/10/15 13:01:57 battery power: -5719W
[site ] DEBUG 2024/10/15 13:01:57 pv power: 6353W
[site ] DEBUG 2024/10/15 13:01:57 site power: -5800W
[lp-1 ] DEBUG 2024/10/15 13:01:57 charge voltages: [241 243 242]V
[lp-1 ] DEBUG 2024/10/15 13:01:57 detected connected phases: 3p
[lp-1 ] DEBUG 2024/10/15 13:01:57 charge total import: 6009.500kWh
[lp-1 ] DEBUG 2024/10/15 13:01:57 charger status: B
[lp-1 ] DEBUG 2024/10/15 13:01:57 pv charge current: 8.41A = 0A + 8.41A (-5800W @ 3p)
[lp-1 ] DEBUG 2024/10/15 13:01:57 site power -5800W <= -2500W enable threshold
[lp-1 ] DEBUG 2024/10/15 13:01:57 pv enable in 0s
[lp-1 ] DEBUG 2024/10/15 13:01:57 pv enable timer elapsed
[lp-1 ] DEBUG 2024/10/15 13:01:57 max charge current: 6A
[lp-1 ] DEBUG 2024/10/15 13:01:57 charger enable
[lp-1 ] DEBUG 2024/10/15 13:01:57 wake-up timer: start
[site ] DEBUG 2024/10/15 13:02:16 ----
[lp-1 ] DEBUG 2024/10/15 13:02:16 charge power: 3796W
[lp-1 ] DEBUG 2024/10/15 13:02:16 charge currents: [5.9 5.68 5.87]A
[site ] DEBUG 2024/10/15 13:02:16 grid power: 211W
[site ] DEBUG 2024/10/15 13:02:17 battery soc: 71%
[site ] DEBUG 2024/10/15 13:02:17 battery power: -3264W
[site ] DEBUG 2024/10/15 13:02:17 pv power: 7247W
[site ] DEBUG 2024/10/15 13:02:17 site power: -2952W
[lp-1 ] DEBUG 2024/10/15 13:02:17 charge voltages: [241 243 242]V
[lp-1 ] DEBUG 2024/10/15 13:02:17 detected connected phases: 3p
[lp-1 ] DEBUG 2024/10/15 13:02:17 charge total import: 6009.500kWh
[lp-1 ] DEBUG 2024/10/15 13:02:17 charger status: C
[lp-1 ] DEBUG 2024/10/15 13:02:17 charge start energy: 6009.500kWh
[lp-1 ] INFO 2024/10/15 13:02:17 start charging ->
[lp-1 ] DEBUG 2024/10/15 13:02:17 wake-up timer: stop
[lp-1 ] DEBUG 2024/10/15 13:02:17 soc estimated: 79.00% (vehicle: 79.00%)
[lp-1 ] DEBUG 2024/10/15 13:02:17 vehicle soc: 79%
[lp-1 ] DEBUG 2024/10/15 13:02:17 vehicle range: 316km
[lp-1 ] DEBUG 2024/10/15 13:02:17 pv charge current: 10.3A = 6A + 4.28A (-2952W @ 3p)
[lp-1 ] DEBUG 2024/10/15 13:02:17 pv timer reset
[lp-1 ] DEBUG 2024/10/15 13:02:17 pv timer inactive
[lp-1 ] DEBUG 2024/10/15 13:02:17 max charge current: 10.3A
[site ] DEBUG 2024/10/15 13:02:36 ----
[... 103 lines cut ...]
[lp-1 ] DEBUG 2024/10/15 13:04:18 vehicle soc: 79%
[lp-1 ] DEBUG 2024/10/15 13:04:18 vehicle range: 316km
[lp-1 ] DEBUG 2024/10/15 13:04:18 pv charge current: 9.63A = 9.3A + 0.332A (-229W @ 3p)
[lp-1 ] DEBUG 2024/10/15 13:04:18 max charge current: 9.63A
[site ] DEBUG 2024/10/15 13:04:36 ----
[lp-1 ] DEBUG 2024/10/15 13:04:36 charge power: 6671W
[lp-1 ] DEBUG 2024/10/15 13:04:36 charge currents: [9.69 9.42 9.72]A
[site ] DEBUG 2024/10/15 13:04:36 grid power: 411W
[site ] DEBUG 2024/10/15 13:04:37 battery soc: 71%
[site ] DEBUG 2024/10/15 13:04:37 battery power: -517W
[site ] DEBUG 2024/10/15 13:04:37 pv power: 7114W
[site ] DEBUG 2024/10/15 13:04:37 site power: -5W
[lp-1 ] DEBUG 2024/10/15 13:04:37 charge voltages: [240 241 239]V
[lp-1 ] DEBUG 2024/10/15 13:04:37 detected connected phases: 3p
[lp-1 ] DEBUG 2024/10/15 13:04:37 detected active phases: 3p
[lp-1 ] DEBUG 2024/10/15 13:04:38 charge total import: 6009.800kWh
[lp-1 ] DEBUG 2024/10/15 13:04:38 charger status: C
[lp-1 ] DEBUG 2024/10/15 13:04:38 soc estimated: 79.59% (vehicle: 79.00%)
[lp-1 ] DEBUG 2024/10/15 13:04:38 vehicle soc: 80%
[lp-1 ] DEBUG 2024/10/15 13:04:38 vehicle range: 316km
[lp-1 ] DEBUG 2024/10/15 13:04:38 pv charge current: 9.64A = 9.63A + 0.00725A (-5W @ 3p)
[lp-1 ] DEBUG 2024/10/15 13:04:38 max charge current: 9.64A
```

[image]

farcorben, 2024-10-15:
> > Ich habe mir gestern den Nightly geholt, die "maxacpower" in die evcc/templates/definition/meter/solax.yaml händisch patched um testen zu können sowie ein container gebaut.
> 
> PR wäre Klasse- dann muss ich es nicht machen ;)

Hat er gemacht, habe denn fronius wie @jmtatsch wünscht auch reingepackt
https://github.com/evcc-io/evcc/pull/16662

buennerbernd, 2024-10-15:
Hallo @andig, für mich sieht es so aus, als ob etwas noch nicht stimmt:
Ich habe maxacpower: 9500 gesetzt.
In dem Log ist pv power> 10000W, aber grid power meist positiv, also Bezug.
Siehst du in den Logs Spuren, ob die neue Config überhaupt herangezogen wurde?

```
[lp-1 ] DEBUG 2024/10/15 12:52:00 charge power: 9287W
[lp-1 ] DEBUG 2024/10/15 12:52:00 charge currents: [13.5 13.5 13.4]A
[site ] DEBUG 2024/10/15 12:52:00 grid power: 526W
[site ] DEBUG 2024/10/15 12:52:00 grid powers: [150 58 319]W
[site ] DEBUG 2024/10/15 12:52:00 grid currents: [1.76 1.99 1.83]A
[site ] DEBUG 2024/10/15 12:52:00 pv power: 11003W
[site ] DEBUG 2024/10/15 12:52:00 battery soc: 92%
[site ] DEBUG 2024/10/15 12:52:00 battery power: 0W
[site ] DEBUG 2024/10/15 12:52:00 site power: 626W
[lp-1 ] DEBUG 2024/10/15 12:52:00 detected active phases: 3p
[lp-1 ] DEBUG 2024/10/15 12:52:00 charger status: C
[lp-1 ] DEBUG 2024/10/15 12:52:00 soc estimated: 18.92% (vehicle: 17.00%)
[lp-1 ] DEBUG 2024/10/15 12:52:00 vehicle soc: 19%
[lp-1 ] DEBUG 2024/10/15 12:52:00 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/10/15 12:52:00 vehicle range: 70km
[lp-1 ] DEBUG 2024/10/15 12:52:00 pv charge current: 13.1A = 14A + -0.908A (626W @ 3p)
[lp-1 ] DEBUG 2024/10/15 12:52:00 max charge current: 13A
[site ] DEBUG 2024/10/15 12:52:20 ----
[lp-1 ] DEBUG 2024/10/15 12:52:20 charge power: 8692W
[lp-1 ] DEBUG 2024/10/15 12:52:20 charge currents: [12.6 12.6 12.6]A
[site ] DEBUG 2024/10/15 12:52:20 grid power: 633W
[site ] DEBUG 2024/10/15 12:52:20 grid powers: [208 122 303]W
[site ] DEBUG 2024/10/15 12:52:20 grid currents: [2.08 2.26 1.96]A
[site ] DEBUG 2024/10/15 12:52:20 pv power: 10994W
[site ] DEBUG 2024/10/15 12:52:20 battery soc: 92%
[site ] DEBUG 2024/10/15 12:52:20 battery power: -1165W
[site ] DEBUG 2024/10/15 12:52:20 site power: -432W
[lp-1 ] DEBUG 2024/10/15 12:52:20 detected active phases: 3p
[lp-1 ] DEBUG 2024/10/15 12:52:20 charger status: C
[lp-1 ] DEBUG 2024/10/15 12:52:20 soc estimated: 18.97% (vehicle: 17.00%)
[lp-1 ] DEBUG 2024/10/15 12:52:20 vehicle soc: 19%
[lp-1 ] DEBUG 2024/10/15 12:52:20 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/10/15 12:52:20 vehicle range: 70km
[lp-1 ] DEBUG 2024/10/15 12:52:20 pv charge current: 13.6A = 13A + 0.626A (-432W @ 3p)
[site ] DEBUG 2024/10/15 12:52:40 ----
[lp-1 ] DEBUG 2024/10/15 12:52:40 charge power: 8692W
[lp-1 ] DEBUG 2024/10/15 12:52:40 charge currents: [12.6 12.6 12.6]A
[site ] DEBUG 2024/10/15 12:52:40 grid power: -4W
[site ] DEBUG 2024/10/15 12:52:40 grid powers: [-10 -92 99]W
[site ] DEBUG 2024/10/15 12:52:40 grid currents: [-1.66 -2.04 1.25]A
[site ] DEBUG 2024/10/15 12:52:40 pv power: 10941W
[site ] DEBUG 2024/10/15 12:52:40 battery soc: 92%
[site ] DEBUG 2024/10/15 12:52:40 battery power: -1235W
[site ] DEBUG 2024/10/15 12:52:40 site power: -1139W
[lp-1 ] DEBUG 2024/10/15 12:52:40 detected active phases: 3p
[lp-1 ] DEBUG 2024/10/15 12:52:40 charger status: C
[lp-1 ] DEBUG 2024/10/15 12:52:40 soc estimated: 19.03% (vehicle: 17.00%)
[lp-1 ] DEBUG 2024/10/15 12:52:40 vehicle soc: 19%
[lp-1 ] DEBUG 2024/10/15 12:52:40 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/10/15 12:52:40 vehicle range: 70km
[lp-1 ] DEBUG 2024/10/15 12:52:40 pv charge current: 14.7A = 13A + 1.65A (-1139W @ 3p)
[lp-1 ] DEBUG 2024/10/15 12:52:40 max charge current: 14A
[site ] DEBUG 2024/10/15 12:53:00 ----
[lp-1 ] DEBUG 2024/10/15 12:53:00 charge power: 9269W
[lp-1 ] DEBUG 2024/10/15 12:53:00 charge currents: [13.3 13.4 13.5]A
[site ] DEBUG 2024/10/15 12:53:00 grid power: 417W
[site ] DEBUG 2024/10/15 12:53:00 grid powers: [159 17 241]W
[site ] DEBUG 2024/10/15 12:53:00 grid currents: [1.76 2.02 1.51]A
[site ] DEBUG 2024/10/15 12:53:00 battery soc: 92%
[site ] DEBUG 2024/10/15 12:53:00 battery power: -955W
[site ] DEBUG 2024/10/15 12:53:00 pv power: 10910W
[site ] DEBUG 2024/10/15 12:53:00 site power: -438W
[lp-1 ] DEBUG 2024/10/15 12:53:00 detected active phases: 3p
[lp-1 ] DEBUG 2024/10/15 12:53:00 charger status: C
[... 271 lines cut ...]
[lp-1 ] DEBUG 2024/10/15 12:58:20 detected active phases: 3p
[lp-1 ] DEBUG 2024/10/15 12:58:20 charger status: C
[lp-1 ] DEBUG 2024/10/15 12:58:20 soc estimated: 20.76% (vehicle: 20.00%)
[lp-1 ] DEBUG 2024/10/15 12:58:20 vehicle soc: 21%
[lp-1 ] DEBUG 2024/10/15 12:58:20 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/10/15 12:58:20 vehicle range: 80km
[lp-1 ] DEBUG 2024/10/15 12:58:20 pv charge current: 13.5A = 13A + 0.487A (-336W @ 3p)
[site ] DEBUG 2024/10/15 12:58:40 ----
[lp-1 ] DEBUG 2024/10/15 12:58:40 charge power: 8712W
[lp-1 ] DEBUG 2024/10/15 12:58:40 charge currents: [12.7 12.6 12.6]A
[site ] DEBUG 2024/10/15 12:58:40 grid power: 649W
[site ] DEBUG 2024/10/15 12:58:40 grid powers: [220 85 344]W
[site ] DEBUG 2024/10/15 12:58:40 grid currents: [1.96 2.01 1.83]A
[site ] DEBUG 2024/10/15 12:58:40 pv power: 11006W
[site ] DEBUG 2024/10/15 12:58:40 battery soc: 93%
[site ] DEBUG 2024/10/15 12:58:40 battery power: -1054W
[site ] DEBUG 2024/10/15 12:58:40 site power: -305W
[lp-1 ] DEBUG 2024/10/15 12:58:40 detected active phases: 3p
[lp-1 ] DEBUG 2024/10/15 12:58:40 charger status: C
[lp-1 ] DEBUG 2024/10/15 12:58:40 soc estimated: 20.82% (vehicle: 20.00%)
[lp-1 ] DEBUG 2024/10/15 12:58:40 vehicle soc: 21%
[lp-1 ] DEBUG 2024/10/15 12:58:40 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/10/15 12:58:40 vehicle range: 80km
[lp-1 ] DEBUG 2024/10/15 12:58:40 pv charge current: 13.4A = 13A + 0.441A (-305W @ 3p)
```

andig, 2024-10-15:
Logging ist ergänzt, neues nightly in 20min.

buennerbernd, 2024-10-15:
Ich habe mir die Werte in meinem Log mal angeschaut. Wenn der WR tatsächlich 10kW AC kann, dann gibt es keinen Grund für den Grid Bezug. Es hätte reichen müssen. Für morgen stelle ich mal maxacpower: 9000 statt 9500 ein.

premultiply, 2024-10-15:
In der Regel haben diese WR dannn 10000 VA (Scheinleistung). Das sind keine 10 kW / 10000 W!
Da geht noch die Blindleistung ab. Je nach Bedingungen des Netzbetreibers und den eingestellten Parametern ist da auch schon mal bei 9000 W (Wirkleistung) das absolute Limit erreicht.

Das werden elektrotechnische Laien aber in der Regel nicht wissen und daher tendenziell falsche (zu hohe/zu optimistische) Werte konfigurieren. Daher sind solche Benutzereingaben eher kritisch.

buennerbernd, 2024-10-15:
> Logging ist ergänzt, neues nightly in 20min.

Ich sehe schon mal was in den Logs:

`[site ] DEBUG 2024/10/15 21:06:37 battery power: 1787W (includes 0W excess DC)`

andig, 2024-10-16:
> Daher sind solche Benutzereingaben eher kritisch.

@premultiply eine bessere Idee hatten wir ja nicht :( Falls es nicht funktionieren sollte (@buennerbernd ?) können wir es vor Release immer noch wieder raus werfen.

buennerbernd, 2024-10-17:
@andig 
Ich habe heute geladen und Logs gesammelt. [evcc-20241017-131107-debug.log](https://github.com/user-attachments/files/17412850/evcc-20241017-131107-debug.log)
Es waren aber zu viele Schleierwolken um die 10kW des WR zu erreichen.
Ich bin aber über 9400W PV gekommen und hatte 9000 Watt maxacpower konfiguriert. Dennoch blieb excess DC immer auf 0.
Deshalb habe ich noch einmal in den aktuellen Code geschaut und ich denke da ist noch ein Fehler oder Missverständnis.

```
// updateBatteryMeters updates battery meters
func (site *Site) updateBatteryMeters() error {
	if len(site.batteryMeters) == 0 {
		return nil
	}

	var totalCapacity, totalEnergy float64
	var eg errgroup.Group
	var mu sync.Mutex

	site.batteryPower = 0
	site.batteryExcessDC = 0
	site.batterySoc = 0

	mm := make([]batteryMeasurement, len(site.batteryMeters))

	fun := func(i int, meter api.Meter) error {
		power, err := backoff.RetryWithData(meter.CurrentPower, bo())
		if err != nil {
			// power is required- return on error
			return fmt.Errorf("battery %d power: %v", i+1, err)
		}

		mu.Lock()
		site.batteryPower += power
		mu.Unlock()

		var excessDC float64
		var excessStr string
		if m, ok := meter.(api.BatteryMaxACPower); ok {
			if dc := power + m.MaxACPower(); dc < 0 {
				mu.Lock()
				site.batteryExcessDC += dc
				mu.Unlock()

				excessDC = dc
				excessStr = fmt.Sprintf(" (includes %.0fW excess DC)", dc)
			}
		}
...
```

In dieser Zeile wird, soweit ich es verstehe, die Batterie-Power genommen:
`			if dc := power + m.MaxACPower(); dc < 0 {`
Es müsste aber die negative PV-Power dieses WR sein.

Was denkst du?

andig, 2024-10-18:
Batterieleistung ist negativ bei Ladung, analog Grid. Damit bleibt dc in der Rechnung negativ. Später dann:

	sitePower := site.gridPower + batteryPower - batteryExcessDC + residualPower - site.auxPower - flexiblePower

Also wird `sitePower` größer, spricht es steht weniger Leistung zur Verfügung. Scheint mir zu passen.

buennerbernd, 2024-10-18:
> Aber ich sehe was Du meinst (?) hier wären vermeintlich 4kW Batterieladung zusätzlich verfügbar, tatsächlich sind es aber nur weitere 2kW.

Genau darauf wollte ich mit dem Beispiel hinaus.

> Ich sehe gerade (und mal wieder) keine gute Lösung, v.a. weil wir auch keine Beziehung zwischen Speicher und PV kennen. Es sieht aus, als müssten wir dafür die Leistungen von WR/BAT differenzierter nach AC/DC kennen? Arghhh.

Ich hatte schon befürchtet, dass keine Navigation vom Batterie-Zähler zum PV-Zähler vorgesehen ist.
Was wäre, wenn du die batteryExcessDC Berechnung komplett in den PV-Zähler umziehst? Später kann sie dann in die sitePower-Berechnung mit eingehen. Für die Config passt MaxACPower auch besser zum WR, als zur Batterie.

buennerbernd, 2024-10-18:
> brauchen wir nicht Beides?

Mir fehlt die Vorstellungskraft wie man maxACPower für "Beides" anwenden soll. Hast du ein anschauliches Beispiel mit Zahlen für den Batterie-Zähler?

> Wenn in Deinem Beispiel der Akku schon voll ist gibts da auch keine excessDC mehr.

Genau.
Wenn der Accu voll ist, dann ist bei meinem WR pvDC etwa 9970 W. Man kann nicht wissen, wieviel der WR produzieren könnte.
Dann ist excessDC = 0 und EVCC regelt wie ohne excessDC und alles ist super.

Deine Formel braucht keine Abrieglung, denn pvDC übersteigt nie das, was der Wechselrichter verarbeiten kann. 
`pvDC = batDC + invAC`

Ich finde deine Idee, die ExcessDC für alle Zähler aufzusummieren super, es gehört meiner Meinung nach nur in den PV-Zähler.
Das charmante an dieser Lösung ist, dass es es gar nicht nötig ist, die AC-Auslastung explizit auszulesen oder zu berechnen. Am Ende findet sich alles in der Summe in site, wie du schreibst:

`sitePower := site.gridPower + batteryPower - batteryExcessDC + residualPower - site.auxPower - flexiblePower`

Ich würde dann batteryExcessDC in excessDC umbenennen.

andig, 2024-10-19:
@ttjaden könntest Du den Vorschlag überprüfen, dass wir statt auf Battery hier auf PV schauen sollten? Würde das bei Deinem WR funktionieren?

andig, 2024-10-19:
I'm reopening here since we should fix this before the next release.

buennerbernd, 2024-10-19:
Wenn da noch ein Nighly draus wird, dann schaffe ich es morgen noch einmal zu testen.
Das Wetter über Mittag soll gut werden und wir verreisen erst später am Nachmittag.

buennerbernd, 2024-10-20:
Ich habe einen Versuch gestartet, aber es kommen keine Logs.
Ich denke, da ist ein Vorzeichenfehler beim Logging:

exsessDCPower ist negativ, daher keine Logs

```
	var excessStr string
	if site.excessDCPower > 0 {
		excessStr = fmt.Sprintf(" (includes %.0fW excess DC)", site.excessDCPower)
	}

```

Ich beobachte aber, ob es sonst so rund läuft.

toleabivol, 2024-10-20:
Vielen Dank für die Implementierung der Fix. Wie können wir diesen Parameter `maxacpowe` verwenden, um evcc die maximale Leistung pro Phase mitzuteilen, wenn das Fahrzeug nur an einer Phase lädt?

```yaml
 type: custom
  name: pv1
  power:
    source: mqtt
    topic: SUNSYNK/status/2210249218/pv1_power
  maxacpower: 8000 # oder 8800 ?
- type: custom
  name: pv2
  power:
    source: mqtt
    topic: SUNSYNK/status/2210249218/pv2_power
  maxacpower: 8000 # oder 8800 ?
```

"Deye SUN-8k-SG04LP3-EU" - Mittlere Spalte der Tabelle. 
[image]

[9 more replies]