evcc-io/evcc #15253: charger out of sync: expected enabled, got disabled

abaginski, 2024-08-05:
### Describe the bug

Seit Version 0.129.0 ist der Charger dauerhaft in Modus "B" mit der Meldung "Ladebereit. Warte auf Fahrzeug …"

OCPP in Verwendung, absolut keine Möglichkeit den Ladevorgang zu starten.

Der Charger an sich funktioniert abseits von OCPP weiterhin fehlerfrei.

### Steps to reproduce

1.Kabel anstecken, Charger wird erkannt, produziert aber konsistent diesen Fehlercode
2.
3.
.

### Configuration details

```yaml
Keine Änderung am Configuration File.
```

### Log details

```text
[site ] DEBUG 2024/08/05 13:00:19 ----
[lp-1 ] DEBUG 2024/08/05 13:00:19 charge power: 0W
[sunspec] TRACE 2024/08/05 13:00:19 model 160 block 1 point DCW: 1134.8
[sunspec] TRACE 2024/08/05 13:00:19 model 160 block 2 point DCW: 1141.3
[site ] DEBUG 2024/08/05 13:00:19 pv power: 2276W
[sunspec] TRACE 2024/08/05 13:00:19 model 160 block 3 point DCW: 0
[sunspec] TRACE 2024/08/05 13:00:20 model 160 block 4 point DCW: 0
[sunspec] TRACE 2024/08/05 13:00:20 model 124 block 0 point ChaState: 0
[site ] DEBUG 2024/08/05 13:00:20 battery soc: 0%
[site ] DEBUG 2024/08/05 13:00:20 battery power: 0W
[site ] DEBUG 2024/08/05 13:00:20 grid meter: -495W
[site ] DEBUG 2024/08/05 13:00:20 grid currents: [1.79 2.23 2.88]A
[site ] DEBUG 2024/08/05 13:00:20 site power: -495W
[lp-1 ] DEBUG 2024/08/05 13:00:20 charge total import: 357.076kWh
[lp-1 ] DEBUG 2024/08/05 13:00:20 charger status: B
[ocpp ] TRACE 2024/08/05 13:00:20 send 9EX-ANV-AWA: [2,"3083216000","GetCompositeSchedule",{"connectorId":1,"duration":1}]
[ocpp ] TRACE 2024/08/05 13:00:20 recv 9EX-ANV-AWA: [3,"3083216000",{"status":"Accepted","connectorId":1,"scheduleStart":"2024-08-05T11:00:19.515Z","chargingSchedule":{"duration":1,"startSchedule":"2024-08-05T11:00:19.515Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":-1,"numberPhases":-1}]}}]

[lp-1 ] WARN 2024/08/05 13:00:20 charger out of sync: expected enabled, got disabled
[lp-1 ] DEBUG 2024/08/05 13:00:20 charger disable

[lp-1 ] DEBUG 2024/08/05 13:00:20 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/08/05 13:00:20 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/08/05 13:00:20 pv charge current: min 6A > 2.15A (-495W @ 1p, battery: false)
[ocpp ] TRACE 2024/08/05 13:00:20 send 9EX-ANV-AWA: [2,"3629913278","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-05T11:00:20Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6,"numberPhases":1}]}}}]
[ocpp ] TRACE 2024/08/05 13:00:20 recv 9EX-ANV-AWA: [3,"3629913278",{"status":"Accepted"}]
[lp-1 ] DEBUG 2024/08/05 13:00:20 charger enable
[lp-1 ] DEBUG 2024/08/05 13:00:20 wake-up timer: start
[ocpp ] TRACE 2024/08/05 13:00:22 recv 9EX-ANV-AWA: [2,"1000012","Heartbeat",{}]
[ocpp ] TRACE 2024/08/05 13:00:22 send 9EX-ANV-AWA: [3,"1000012",{"currentTime":"2024-08-05T11:00:22Z"}]
[site ] DEBUG 2024/08/05 13:00:49 ----
[lp-1 ] DEBUG 2024/08/05 13:00:49 charge power: 0W
[sunspec] TRACE 2024/08/05 13:00:49 model 160 block 1 point DCW: 1137.4
[sunspec] TRACE 2024/08/05 13:00:49 model 160 block 2 point DCW: 1141
[site ] DEBUG 2024/08/05 13:00:49 pv power: 2278W
[sunspec] TRACE 2024/08/05 13:00:49 model 160 block 3 point DCW: 0
[sunspec] TRACE 2024/08/05 13:00:49 model 160 block 4 point DCW: 0
[sunspec] TRACE 2024/08/05 13:00:50 model 124 block 0 point ChaState: 0
[site ] DEBUG 2024/08/05 13:00:50 battery soc: 0%
[site ] DEBUG 2024/08/05 13:00:50 battery power: 0W
[site ] DEBUG 2024/08/05 13:00:50 grid meter: -497W
[site ] DEBUG 2024/08/05 13:00:50 grid currents: [1.79 2.23 2.88]A
[site ] DEBUG 2024/08/05 13:00:50 site power: -497W
[lp-1 ] DEBUG 2024/08/05 13:00:50 charge total import: 357.076kWh
[lp-1 ] DEBUG 2024/08/05 13:00:50 charger status: B
[ocpp ] TRACE 2024/08/05 13:00:50 send 9EX-ANV-AWA: [2,"3741780040","GetCompositeSchedule",{"connectorId":1,"duration":1}]
[ocpp ] TRACE 2024/08/05 13:00:50 recv 9EX-ANV-AWA: [3,"3741780040",{"status":"Accepted","connectorId":1,"scheduleStart":"2024-08-05T11:00:49.525Z","chargingSchedule":{"duration":1,"startSchedule":"2024-08-05T11:00:49.525Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":-1,"numberPhases":-1}]}}]
[lp-1 ] DEBUG 2024/08/05 13:00:50 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/08/05 13:00:50 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/08/05 13:00:50 pv charge current: min 6A > 2.16A (-497W @ 1p, battery: false)
```

### What type of operating system are you running?

Linux

### Version

0.129.0

andig, 2024-08-05:
@premultiply hier gibt es sogar eine period:

    "chargingSchedulePeriod":[{"startPeriod":0,"limit":-1,"numberPhases":-1}]

premultiply, 2024-08-05:
Einfach mit RFID freischalten oder Alternativ die Box auf "Autostart", "Freemode", "Freies Laden", "Free Vending" oder wie auch immer der Modus in der Konfiguration einstellen.

Siehe https://github.com/evcc-io/evcc/pull/14888

premultiply, 2024-08-05:
@andig @naltatis Hier würde ein neuer Anzeigestatus "Ladepunkt wartet auf Authentifizierung" helfen.

abaginski, 2024-08-05:
@premultiply ich verstehe nicht, wie soll ich die YAML ändern, damit er lädt? Die Wallbox hat keine RFID oder Autostart-Funktionalität, einfach anstecken und es lädt.

premultiply, 2024-08-05:
Was ist es denn überhaupt für eine Box?

Bitte mal mit dem aktuellen nightly `evcc charger --diagnose` ausführen und die Ausgabe hier teilen.

Wie sieht die OCPP-Konfigurationsseite der Box aus?

closed 2024-08-05 by abaginski

premultiply, 2024-08-05:
Wie unter https://github.com/evcc-io/evcc/pull/14888 beschrieben gibt es theoretisch die Möglichkeit via `remotestart: true` durch evcc Transaktionen zu starten. Die ist ausdrücklich NICHT der bevorzugte Weg und nur für konzeptionell ganz misslungene OCPP-Boxen als letzter Strohhalm vorgesehen da damit sämtliche lokalen Authentifizierungsoption gänzlicher ausser Kraft gesetzt werden.

**Eigentlich** müssen alle OCPP-Boxen lokale Authentifizierungsoptionen haben.
Wenn sie **keinen** RFID-Reader haben (was bei OCPP eher die Ausnahme sein sollte) müssten sie entweder permanent im Autostart-Modus laufen oder zumindest eine entsprechende optionale Konfigurationsoption haben.
Denn ohne Anbindung an ein funktionsfähiges OCPP-CS (wie evcc) wäre die Box sonst ein nutzloser Briefbeschwerer.