evcc-io/evcc #15256: Telegram Messaging causing SIGSEGV

goebelmeier, 2024-08-05:
### Describe the bug

Configured telegram messaging is causing Segmentation Violations which causes evcc to crash as soon as evcc tries to send out a message to telegram

### Steps to reproduce

1. Configure telegram messaging
2. Start evcc
3. Wait for an evcc messaging event to happen
...

### Configuration details

```yaml
messaging:
  events:
    start: # charge start event
      title: "Charge of {{.vehicleTitle}} started from {{.vehicleSoc}}% to {{.effectiveLimitSoc}}% SoC"
      msg: |
        Started charging ${vehicleTitle} from ${vehicleSoc:%.0f}% to ${effectiveLimitSoc}% SoC
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    stop: # charge stop event
      title: "Charge of {{.vehicleTitle}} with {{.vehicleSoc}}% SoC ({{.vehicleRange}}km Range)"
      msg: |
        Finished charging ${vehicleTitle} with ${chargedEnergy:%.1fk}kWh in ${chargeDuration}. ${vehicleSoc:%.0f}% SoC (${vehicleRange}km Range)
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    connect: # vehicle connect event
      title: "A vehicle has been connected"
      msg: "A vehicle has been connected"
    disconnect: # vehicle connected event
      title: "A vehicle was disconnected with {{.vehicleSoc}}% SoC ({{.vehicleRange}}km Range)"
      msg: |
        A vehicle was disconnected after ${connectedDuration}.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    soc: # vehicle soc update event
      title: ${vehicleTitle} is currently charged to ${vehicleSoc:%.0f}% SoC (${vehicleRange}km Range)
      msg: ${vehicleTitle} is currently charged to ${vehicleSoc:%.0f}% SoC (${vehicleRange}km Range)
  services:
    - type: telegram
      token: ABZDEFGHIJKLMNOPQRSTUVWXYZ
      chats:
        - 1234567890
```

### Log details

```text
[lp-1  ] DEBUG 2024/08/05 13:36:22 !! session: chargeRater.chargedEnergy=3.0 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/05 13:36:22 charge total import: 18441.423kWh
[lp-1  ] DEBUG 2024/08/05 13:36:23 charger status: B
[lp-1  ] DEBUG 2024/08/05 13:36:23 limitSoc reached: 81.0% > 80%
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x20 pc=0x242ec57]

goroutine 165 [running]:
github.com/evcc-io/evcc/push.(*Telegram).trackChats(0xc0007e91e0)
	github.com/evcc-io/evcc/push/telegram.go:69 +0x177
created by github.com/evcc-io/evcc/push.NewTelegramFromConfig in goroutine 1
	github.com/evcc-io/evcc/push/telegram.go:57 +0x325
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/05 13:46:51 evcc 0.129.0
[main  ] INFO 2024/08/05 13:46:51 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/08/05 13:46:51 connecting evcc-735423241 at tcp://192.168.4.5:1883
[mqtt  ] DEBUG 2024/08/05 13:46:51 tcp://192.168.4.5:1883 connected
```

### What type of operating system are you running?

Linux

### Version

0.129.0

goebelmeier, 2024-08-05:
So far this has been a one-off event over a short period of time (< 30 minutes). My guess is that the Telegram API behaved non-standard or was not accessible. Possibly incorrect or missing error handling in the Telegram library used?

andig, 2024-08-05:
Closed in https://github.com/evcc-io/evcc/commit/314acde61ffa27d063a1a44807e3ffccedecdfa0

closed 2024-08-05 by andig