evcc-io/evcc #15261: Weidmüller: fix implementation

premultiply, 2024-08-05:
Neue Erkenntnisse:

Von einem Weidmüller-MA habe ich eine erste Antwort bekommen, die mich nicht wirklich weiter gebracht, aber beschäftigt hat.

consumed_energy_total (Register 457) wird nur beim Modell "Advanced", welches einen MID Zähler verbaut hat angezeigt.

car_state (Reg. 301) kommt nach Weidmüller Aussage vom Auto; sie haben auf das Timing keinen Einfluss. (Werde ich noch mit einem zu leihenden Oszi nachmessen).
Macht es Sinn hier das Reg 302 charger_state zu verwenden, welches aber wine etwas komplexere Aussage macht?

Reg 702 user_current-limit soll für die Regelung das richtige sein. (obwohl Nichtflüchtig??)

Ich habe jedoch per Zufall noch eine **Application-Note** gefunden, die hier interessant ist. Wir müssen die WB wohl als Satellit konfigurieren (per WEB Oberfläche oder per Modbus) um sie dann über die Register für das Lastmanagement Register 11000ff zu steuern. Hier gibt es auch das Reg 11052 LCM_ACTUAL_CURRENT_LIMIT, welches flüchtig ist.
Die WB muss dann zyklisch auf Reg. 11050 retriggert werden, was mit dem 30s Zyklus von EVCC ja gehen sollte.

Zum Testen habe ich in meinem ioBroker System etwas Test-Code in Blockly/Javascript geschrieben und mich dabei an eure Nomenklatur angelehnt.

Hier mein Code als TXT Datei. Vielleicht Ideengeber für eigene Überlegungen.
[ProgBsp.txt](https://github.com/user-attachments/files/16502087/ProgBsp.txt)

App Note: 
[Application Note Einbindung der AC Smart in Lade- Lastmanagement (DE, EN).pdf](https://github.com/user-attachments/files/16502089/Application.Note.Einbindung.der.AC.Smart.in.Lade-.Lastmanagement.DE.EN.pdf)

Modbus Register als Excel-Datei (für mich strukturierter):
[AC-SMART_Modbus-Register.xlsx](https://github.com/user-attachments/files/16502091/AC-SMART_Modbus-Register.xlsx)

Nach seiner Rückkehr aus dem Urlaub hat mir ein Weidmüller MA einen weiteren Kontakt zugesagt.
Meine Themen:
	Consumed_energy_total
	Flüchtiges register für Phasenumschaltung
	Car-State

Ich gebe die Ergebnisse dann weiter.

_Originally posted by @tobi119 in https://github.com/evcc-io/evcc/issues/15071#issuecomment-2269794334_

andig, 2024-08-10:
Was ist hier der "bug"? Fehlerbeschreibung fehlt.

tobi119, 2024-08-10:
(1) **Fehler**: zum Regeln wird ein _nichtflüchtiges_ Register verwendet mit der Folge, dass das EEProm irgendwann sterben wird.
> Vor allem verstehe ich nicht warum angeblich die externen Stromvorgaben im Flash gespeichert werden.

Wie @premultiply festgestellt hat erfolgt die Regelung derzeit über ein nichtflüchtiges Register (702).

Nach meiner Sicht ist hier Reg 11052 LCM_ACTUAL_CURRENT_LIMIT zu verwenden, auch um bei Ladepausen auf Null zu regeln.
Reg. 11050 muss dann retriggert werden und die WB auf Satellit konfiguriert sein
Reg 705 für Enable verwenden.

(2) Eine **Warning** wird derzeit beim Disable generiert, da das Auto erst nach 63 Sekunden von Status "C" in den Status "B" übergeht.

(3) Für die Phasenumschaltung wird auch ein nichtflüchtiges Register verwendet, dies wird aber nicht so oft vorkommen.
(in der Hoffnung, dass auf das Register nicht zyklisch geschrieben wird)

(4) "Geladen" (Gesamt-Energie) wird **mit Null angezeigt**, da Register 457 consumed_energy_total bei den Modellvarianten ECO und VALUE nicht gesetzt wird, sondern nur bei ADVANCED (hier ist ein externer MIDI Zähler verbaut)

Der erste Fehler muss behoben werden, die anderen Punkte machen den Adapter besser/ robuster.
Hie hoffe ich in der kommenden Woche noch eine Aussage von Weidmüller zu erhalten über deren Möglichkeiten.

Für Tests stehe ich gerne zur Verfügung.

hno2fe, 2024-08-25:
Hi gibt es schon neu Erkenntnisse ?

tobi119, 2024-08-25:
Ich habe gerade heute Weidmüller nochmal angeschrieben bezüglich Stellungnahme.
Wenn da nichts kommt müsste der Punkt (1) zumindest umgesetzt werden. Dann wäre der Adapter einsetzbar.
Für (2) und (4) kann man natürlich in EVCC ein work around programmieren. Für (2) gibt es keine Lösung in der WB.
Punkt (3) sollte unkritisch sein.

andig, 2024-09-01:
Soweit ich das verstehe funktioniert die Implementierung. Gerne PR mit den gewünschten Änderungen- die Registernummern auszutauschen ist ja trivial.

t1mb0, 2024-09-17:
Apologies in advance for swapping to English, confirmed the current integration does 'work' but it's writing to non-volatile memory locations which are not meant to be written cyclically (as per manual).  They are for seldom-changing configuration adjustments.

tobi119, 2024-10-27:
Nachdem von Weidmüller nicht viele Informationen gekommen sind möchte ich meine Versuche und Lösungsansätze zusammenfassen in anhängendem Dokument.
Für die **Stromregelung** muss das flüchtige **Reg. 11052** LCM_ACTUAL_CURRENT_LIMIT  verwendet werden. Dazu muss die WB als Satellit konfiguriert sein **und Reg. 11050 muss regelmäßig getriggert werden**.
Für die Phasenumschaltung wird, wie bereits implementiert, Reg. 318 verwendet.
Die Gesamtenergie wird von der WB nicht berechnet. Sie könnte nur von EVCC aus der "energy last charging" in Reg. 430 aufsummiert werden.
Für weitere Fragen oder Tests bin ich gern ansprechbar. Meine Implementierung habe ich mit ioBroker getestet.
Grüße  aus dem oberen Donautal.

[Weidmüller WB_ meine Implementierung.pdf](https://github.com/user-attachments/files/17535200/Weidmuller.WB_.meine.Implementierung.pdf)

Nautilus-X, 2025-01-01:
Hallo @tobi119,
bin heute durch Zufall auf diesen Thread gestoßen und danke für die ausführliche Dokumentation deiner Erkenntnisse. Die schlechte Doku der Modbus-Adressen hatte mich auch einige Zeit gekostet. Ich hatte die WB AC Value mit FW 01.06.01 als Custom Charger angelegt und war ebenfalls über das Thema Stromregelung und dem nicht flüchtigen Register gestolpert. Jetzt im Winter habe ich noch nicht das Problem der Ladung mit PV-Unterstützung und hatte für's erste 11052 als Register eingetragen - auch wenn es nicht wie gewünscht funktioniert - der Ladestrom bleibt zumindest immer bei den vorgegeben 16A und ich mache nix kaputt. Auf die Applikation mit dem Lastmanagement war ich nicht gestoßen.

Habe jetzt den Satellitenbetrieb erfolgreich ausprobiert und nun beschäftigt mich das zyklische Beschreiben des Registers 11050. Wenn man nichts macht, geht meiner Beobachtung nach, im Fehlerfall die Ladung unverändert weiter, es wird nicht auf die Stromstärke lt. Register 11054 gewechselt, lediglich die rote Leuchte geht an und der Verbindungsverlust wird protokolliert. Wenn man das Register mit einem Modbus-Tester wieder "füllt", wechselt die Kontrollleuchte wieder auf grün und das Spiel beginnt von vorn. Man kann das Herunterzählen gut beobachten.

Hast du eine Idee, wie ich es anstellen könnte, das Register durch evcc zyklisch wieder auf einen Ausgangswert zu füllen?

tobi119, 2025-01-01:
> Hast du eine Idee, wie ich es anstellen könnte, das Register durch evcc zyklisch wieder auf einen Ausgangswert zu füllen?

Ich hatte gehofft, dass @premultiply den Treiber für die Weidmüller WB umgestellt  von Reg. 702 auf Reg. 11052.
In diesem Treiber muss dann auch das zyklische Triggern erfolgen. 
Mit Custom Charger habe ich mich nicht beschäftigt und weiß daher nicht, ob hierin zyklische Aktionen möglich sind.

Ich steuere die WB derzeit über das SmartHome System ioBroker

Nautilus-X, 2025-01-02:
Okay :-) dann schaue ich mal weiter

mfuchs1984, 2025-01-02:
Hier ist ein charger mit heartbeat, es wird zyklisch in ein register geschrieben. https://github.com/evcc-io/evcc/blob/413339b1b2d24cb8953aceffa45abb7199f7a92c/charger/mennekes-compact.go#L119

Entsprechend konntet ihr die Weidmüller Implementierung anpassen https://github.com/evcc-io/evcc/blob/master/charger/weidm%C3%BCller.go

PR #18019 (2025-01-03): Weidmüller: fix register usage and check for meter
Fixes #15261
closed 2025-01-03 by andig