evcc-io/evcc #15274: Climater behaviour / Gui not correct

Lightmanster, 2024-08-06:
### Describe the bug

Display climater icon isn't show the right way.
2 different scenarios, 
1 it is not showed while it should.
2 it is showed but doesnt disappear if climate is stopped

### Steps to reproduce

Scenario 1 
1. EVCC is in PV mode not charging. (waiting for vehicle)
2. Activate pre-conditioning in Tesla app
3. EVCC shows charging with 2,0 kwh 3p 
4. 
[image]

But this is Climate, even when no climate sign is shown 
Log dont detect climate is on around 16:17. MQTT show that precondition is true

[image]

Scenario 2
1. EVCC is in PV mode not charging. (waiting for vehicle)
2. Activate pre-conditioning in Tesla app
3. EVCC detects pre-conditioning

Aug 06 15:58:06 raspberrypi evcc[31822]: [lp-1  ] DEBUG 2024/08/06 15:58:06 climater active
Aug 06 15:58:06 raspberrypi evcc[31822]: [lp-1  ] DEBUG 2024/08/06 15:58:06 !! active phases: 3p = min(3p measured 0p vehicle 3p physical 0p charger)
Aug 06 15:58:06 raspberrypi evcc[31822]: [lp-1  ] DEBUG 2024/08/06 15:58:06 charger enable

[image]

4. After a minute disable climate in tesla app.
5. Mqtt tells it is false, 

[image]

but EVCC keep showing climate icon 

[image]

### Configuration details

```yaml
vehicles:
- type: custom
  title: Tesla
  capacity: 82
  features: ["coarsecurrent"]
  soc:
    source: mqtt
    topic: teslamate/cars/1/battery_level
    timeout: 720h # 30d
  range:
    source: mqtt
    topic: teslamate/cars/1/rated_battery_range_km
  status:
    source: combined
    plugged:
      source: mqtt
      topic: teslamate/cars/1/plugged_in
    charging:
      source: mqtt
      topic: teslamate/cars/1/charger_actual_current
  climater:
    source: mqtt
    topic: teslamate/cars/1/is_climate_on
```

### Log details
[... 94 lines cut ...]
Aug 06 15:59:27 raspberrypi evcc[31822]: [lp-1  ] DEBUG 2024/08/06 15:59:27 max charge current: 6A
Aug 06 15:59:45 raspberrypi evcc[31822]: [site  ] DEBUG 2024/08/06 15:59:45 ----
Aug 06 15:59:46 raspberrypi evcc[31822]: [lp-1  ] DEBUG 2024/08/06 15:59:46 charge power: 0W
Aug 06 15:59:46 raspberrypi evcc[31822]: [lp-1  ] DEBUG 2024/08/06 15:59:46 charge currents: [0 0 0]A
Aug 06 15:59:46 raspberrypi evcc[31822]: [lp-2  ] DEBUG 2024/08/06 15:59:46 charge power: 0W
Aug 06 15:59:46 raspberrypi evcc[31822]: [circuit-main] DEBUG 2024/08/06 15:59:46 power: -4485W
Aug 06 15:59:46 raspberrypi evcc[31822]: [circuit-main] DEBUG 2024/08/06 15:59:46 current: -4A
Aug 06 15:59:46 raspberrypi evcc[31822]: [site  ] DEBUG 2024/08/06 15:59:46 pv power: 5019W
Aug 06 15:59:46 raspberrypi evcc[31822]: [site  ] DEBUG 2024/08/06 15:59:46 grid meter: -4485W
Aug 06 15:59:46 raspberrypi evcc[31822]: [site  ] DEBUG 2024/08/06 15:59:46 grid currents: [-4 -8 -5]A
Aug 06 15:59:46 raspberrypi evcc[31822]: [site  ] DEBUG 2024/08/06 15:59:46 site power: -4485W
Aug 06 15:59:46 raspberrypi evcc[31822]: [lp-2  ] DEBUG 2024/08/06 15:59:46 !! session: chargeRater.chargedEnergy=0.0 - chargedAtStartup=0.0
Aug 06 15:59:46 raspberrypi evcc[31822]: [lp-2  ] DEBUG 2024/08/06 15:59:46 charge total import: 0.797kWh
Aug 06 15:59:46 raspberrypi evcc[31822]: [lp-2  ] DEBUG 2024/08/06 15:59:46 charger status: B
Aug 06 15:59:46 raspberrypi evcc[31822]: [lp-2  ] DEBUG 2024/08/06 15:59:46 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 1p charger)
Aug 06 15:59:46 raspberrypi evcc[31822]: [lp-2  ] DEBUG 2024/08/06 15:59:46 pv charge current: 19.5A = 0A + 19.5A (-4485W @ 1p)
```

### What type of operating system are you running?

Docker container

### Version

0.129.0

StevieC121176, 2024-08-08:
Ich habe das gleiche beobachtet.

Fahrzeug: Tesla Model Y, über die UI angelegt. Klimatisieren wird nicht erkannt.

Logs:
[evcc-20240803-100631-trace.log](https://github.com/user-attachments/files/16540866/evcc-20240803-100631-trace.log)
[evcc-20240803-100641-trace.log](https://github.com/user-attachments/files/16540867/evcc-20240803-100641-trace.log)
[evcc-20240803-100649-trace.log](https://github.com/user-attachments/files/16540868/evcc-20240803-100649-trace.log)
[evcc-20240803-100659-trace.log](https://github.com/user-attachments/files/16540869/evcc-20240803-100659-trace.log)

Fahrzeugladung abgeschlossen:

[image]

Klimatisierung läuft nach abgeschlossener Ladung mit ca. 800W (3phasig) , wird aber nicht als Klimatisierung angezeigt.

[image]

closed 2024-08-20 by github-actions[bot]

Lightmanster, 2024-09-12:
@StevieC121176 Does climate function still work for you ? I do get power in pv mode but dont see icons or whatever

StevieC121176, 2024-09-12:
> @StevieC121176 Does climate function still work for you ? I do get power in pv mode but dont see icons or whatever

Here the same . Energy for climate is displayed but without icon.