evcc-io/evcc #15278: Energy meter does not respond after 4 minutes

lamemate, 2024-08-06:
### Describe the bug

I configured a site with the following components as addon for Home Assistant.

- SMA Home Manager 2.0
- go e Gemini 11 kw charger

The initial reading of the Home Manager is working and I get data for about 4 minutes. It then suddenly cannot be read anymore.
The configuration and debug log are attached. I also reproduced this issue with a Windows executable with the exact same behaviour.

The only workaround I found was to restart the evcc addon and it works again for 4 minutes. I also tried pinging the Home Manager and there is no disruption.

### Steps to reproduce

1. Install Home Assistant Operating System on a Pi 5
2. Add evcc addon version v0.129.0
3. Save the configuration
4. Start evcc
5. Wait 4 minutes

### Configuration details

```yaml
plant: uniqueidxxx
interval: 30s

tariffs:
  feedin:
    type: fixed
    price: 0.0811 # [currency]/kWh

vehicles:
  - name: skoda
    type: template
    template: enyaq
    # [...]

chargers:
  - name: wallbox
    type: template
    template: go-e-v3
    host: xxx.xxx.xxx.187

meters:
  - name: my_grid
    type: template
    template: sma-home-manager
    usage: grid
    host: xxx.xxx.xxx.92

site:
  - title: Zuhause
    meters:
      grid: my_grid
    residualPower: 100

loadpoints:
  - title: Wallbox
    charger: wallbox
    vehicle: skoda
    mode: pv

sponsortoken: ofcourse
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/06 22:13:41 evcc 0.129.0
[main  ] INFO 2024/08/06 22:13:41 using config file: /config/evcc.yaml
[db    ] INFO 2024/08/06 22:13:41 using sqlite database: /data/evcc.db
[main  ] INFO 2024/08/06 22:13:41 listening at :7070
[lp-1  ] DEBUG 2024/08/06 22:13:45 charge total import: 63.999kWh
[site  ] INFO 2024/08/06 22:13:45 site config:
[site  ] INFO 2024/08/06 22:13:45   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2024/08/06 22:13:45     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/08/06 22:13:45   vehicles:
[site  ] INFO 2024/08/06 22:13:45     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/08/06 22:13:45 loadpoint 1:
[lp-1  ] INFO 2024/08/06 22:13:45   mode:        pv
[lp-1  ] INFO 2024/08/06 22:13:45   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/08/06 22:13:45   meters:      charge ✓
[lp-1  ] INFO 2024/08/06 22:13:45     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/08/06 22:13:45 set smart cost limit: 0.2
[lp-1  ] DEBUG 2024/08/06 22:13:45 !! active phases: 3p = min(0p measured 0p vehicle 0p physical 0p charger)
[lp-1  ] DEBUG 2024/08/06 22:13:45 phase timer inactive
[lp-1  ] DEBUG 2024/08/06 22:13:45 pv timer inactive
[lp-1  ] INFO 2024/08/06 22:13:45 vehicle updated: unknown -> Enyaq
[lp-1  ] DEBUG 2024/08/06 22:13:45 !! active phases: 3p = min(0p measured 0p vehicle 0p physical 0p charger)
[site  ] DEBUG 2024/08/06 22:13:45 ----
[lp-1  ] DEBUG 2024/08/06 22:13:45 charge power: 0W
[lp-1  ] DEBUG 2024/08/06 22:13:45 charge currents: [0 0 0]A
[site  ] DEBUG 2024/08/06 22:13:45 grid meter: 369W
[site  ] DEBUG 2024/08/06 22:13:45 grid powers: [199 130 41]W
[site  ] DEBUG 2024/08/06 22:13:45 grid currents: [1.8 1.23 0.34]A
[site  ] DEBUG 2024/08/06 22:13:45 pv power: 0W
[site  ] DEBUG 2024/08/06 22:13:45 site power: 469W
[lp-1  ] DEBUG 2024/08/06 22:13:45 vehicle odometer: 6209km
[lp-1  ] DEBUG 2024/08/06 22:13:45 charge total import: 63.999kWh
[lp-1  ] DEBUG 2024/08/06 22:13:45 charger status: A
[lp-1  ] INFO 2024/08/06 22:13:45 car disconnected
[lp-1  ] DEBUG 2024/08/06 22:13:45 !! active phases: 3p = min(0p measured 0p vehicle 0p physical 0p charger)
[lp-1  ] DEBUG 2024/08/06 22:13:45 !! active phases: 3p = min(0p measured 0p vehicle 0p physical 0p charger)
[lp-1  ] DEBUG 2024/08/06 22:13:45 set session soc limit: 0
[lp-1  ] DEBUG 2024/08/06 22:13:45 set session energy limit: 0
[site  ] DEBUG 2024/08/06 22:14:15 ----
[lp-1  ] DEBUG 2024/08/06 22:14:15 charge power: 0W
[lp-1  ] DEBUG 2024/08/06 22:14:15 charge currents: [0 0 0]A
[site  ] DEBUG 2024/08/06 22:14:15 grid meter: 358W
[site  ] DEBUG 2024/08/06 22:14:15 grid powers: [198 119 41]W
[site  ] DEBUG 2024/08/06 22:14:15 grid currents: [1.8 1.21 0.34]A
[site  ] DEBUG 2024/08/06 22:14:15 pv power: 0W
[site  ] DEBUG 2024/08/06 22:14:15 site power: 458W
[lp-1  ] DEBUG 2024/08/06 22:14:15 vehicle odometer: xxxxkm
[lp-1  ] DEBUG 2024/08/06 22:14:15 charge total import: 63.999kWh
[lp-1  ] DEBUG 2024/08/06 22:14:15 charger status: A
[site  ] DEBUG 2024/08/06 22:14:45 ----
[lp-1  ] DEBUG 2024/08/06 22:14:45 charge power: 0W
[lp-1  ] DEBUG 2024/08/06 22:14:45 charge currents: [0 0 0]A
[site  ] DEBUG 2024/08/06 22:14:45 grid meter: 367W
[site  ] DEBUG 2024/08/06 22:14:45 grid powers: [198 127 41]W
[site  ] DEBUG 2024/08/06 22:14:45 grid currents: [1.8 1.23 0.341]A
[site  ] DEBUG 2024/08/06 22:14:45 pv power: 0W
[site  ] DEBUG 2024/08/06 22:14:45 site power: 467W
[lp-1  ] DEBUG 2024/08/06 22:14:45 charge total import: 63.999kWh
[lp-1  ] DEBUG 2024/08/06 22:14:45 charger status: A

[...] # Always the same but then

[site  ] DEBUG 2024/08/06 22:17:45 ----
[lp-1  ] DEBUG 2024/08/06 22:17:45 charge power: 0W
[lp-1  ] DEBUG 2024/08/06 22:17:45 charge currents: [0 0 0]A
[site  ] DEBUG 2024/08/06 22:17:45 grid meter: 428W
[site  ] DEBUG 2024/08/06 22:17:45 grid powers: [201 186 41]W
[site  ] DEBUG 2024/08/06 22:17:45 grid currents: [1.81 1.35 0.341]A
[site  ] DEBUG 2024/08/06 22:17:45 pv power: 0W
[site  ] DEBUG 2024/08/06 22:17:45 site power: 528W
[lp-1  ] DEBUG 2024/08/06 22:17:45 charge total import: 63.999kWh
[lp-1  ] DEBUG 2024/08/06 22:17:45 charger status: A
[sma   ] ERROR 2024/08/06 22:18:06 energy meter does not respond
[sma   ] ERROR 2024/08/06 22:18:11 energy meter does not respond
[site  ] DEBUG 2024/08/06 22:18:15 ----
[lp-1  ] DEBUG 2024/08/06 22:18:15 charge power: 0W
[lp-1  ] DEBUG 2024/08/06 22:18:15 charge currents: [0 0 0]A
[site  ] ERROR 2024/08/06 22:18:16 grid meter: outdated
[sma   ] ERROR 2024/08/06 22:18:16 energy meter does not respond
[sma   ] ERROR 2024/08/06 22:18:21 energy meter does not respond
[sma   ] ERROR 2024/08/06 22:18:26 energy meter does not respond
[sma   ] ERROR 2024/08/06 22:18:31 energy meter does not respond
[sma   ] ERROR 2024/08/06 22:18:36 energy meter does not respond
[sma   ] ERROR 2024/08/06 22:18:41 energy meter does not respond
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

v0.129.0

premultiply, 2024-08-06:
This just sounds like an IGMP multicast configuration issue with your ethernet switch.

No issue of evcc.

lamemate, 2024-08-06:
Thanks for the hint. That was indeed the problem.

closed 2024-08-06 by lamemate