evcc-io/evcc #15286: ellipro: evcc gepaired connected - steuert aber die Wallbox nicht

guggiguggo, 2024-08-07:
### Describe the bug

Seit kurzem teste ich evcc, installiert in einer Proxmox VM läuft es auf Debian 12, evcc hat version 0.129.0+1723019085.
Freigeschalten wird die Wallbox id charger connect aufgerüstet mit abb Zähler per rfid karte, die Box beginnt nach Freischaltung sofort zu laden, obwohl kein PV-überschuss vorhanden war.
Wechselrichter ist ein Fronius gen24 6.0 hybrid samt Fronius Smartmeter und der Batteriespeicher stammt von BYD deren Werte über die solar-api ausgelesen werden.
Klicke ich aufgrund des ungenügend vorhandenen PV-Überschusses im Webinterface von evcc auf aus, lädt die Wallbox das angeschlossene Fahrzeug trotzdem weiter auf. Das Unterbrechen des Ladevorgangs ist erst durch Beenden über die weconnect app am hy möglich. 
Am Webinterface der Wallbox wird angezeigt das das HEMS evcc erfolgreich gepaired wurde und zeigte während des gesamten Vorgangs den Status connected an.
Nachdem ich durch das Lesen einiger issues nun darüber informiert bin, dass elli die Firmware nur schlecht als recht wartet, hoffe ich nun zwischenzeitlich - bis zum Kauf einer neuen Wallbox - das Überschussladen noch erfolgreich einzurichten und über den Winter zu kommen.

### Steps to reproduce

1. Auto angeschlossen und per rfid freigeschalten
2. Auto lädt obwohl des ungenügenden PV-Überschusses, Haus Led Orange die anderen zwei grün.
3. Im Webinterfafe von evcc war PV eingesellt, der Timer für PV läuft ab - Ladung bleibt aktiv - ich klicke auf aus, das Auto wird aber weiter aufgeladen. Im Webinterface der ellipro wird connected mit evcc angezeigt.
4. Der Ladevorgang kann nur über die weconnect app am hy unterbrochen werden durch das Beenden des Ladevorgangs im Menü laden zu Hause.
5. Nach absteckem des Ladekabel am Auto waren alle Leds an der Wallbox alle drei grün.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace 
levels:
  cache: error

# unique installation id
plant: 0bfb18c62138af2c2d2a42dada772d1efa10ff4ae3c6e2932c55122fad48fcf3

interval: 30s # control cycle interval

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: 192.168.11.157  
  user: ***  
  password: ***  
  name: grid1
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.11.157  
  user: ***
  password: ***
  name: pv2
- type: template
  template: fronius-solarapi-v1 
  usage: battery  
  host: 192.168.11.157  
  user: ***  
  password: ***
  name: battery3

chargers:
- type: template
  template: ellipro 
  ski: '***'  
  host: 192.168.11.253
  name: wallbox5

vehicles:
- type: template
  template: vw 
  title: VW ID 3  
  user: ***
  password: ***
  vin: ***
  capacity: 58  
  timeout: 10s  
  name: ev4

loadpoints:
- title: Parkplatz
  charger: wallbox5
  mode: pv

site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 600
  maxGridSupplyWhileBatteryCharging: 50

eebus:
  shipid: evcc.home.lan 
  interfaces:
  - ens18
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          -----END CERTIFICATE-----
```

### Log details

```text
trace log:
https://0.jaegers.net/e45a7967-c46e-4be4-bc3a-72b6de2f17bf
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.129.0 (b4dea20f)

guggiguggo, 2024-08-07:
[trace.log](https://github.com/user-attachments/files/16530469/trace.log)

DerAndereAndi, 2024-08-07:
This issue should already be solved in the next nightly build. The data provided does not conform to the EEBUS specification and I had to add a workaround for that.

guggiguggo, 2024-08-07:
Thank you for your fast response, will try again tomorrow and let you know.

guggiguggo, 2024-08-08:
root@evcc:~# evcc --version
evcc version 0.129.0 (17d75c59)

leider immer noch selbes verhalten, trace im Anhang.
[evcc-20240808-084249-trace.log](https://github.com/user-attachments/files/16538780/evcc-20240808-084249-trace.log)

DerAndereAndi, 2024-08-08:
Danke für das Log, leider wieder ein falsches Verhalten der Elli Wallbox. Sie schickt weder das aktuell eingestellte Limit noch den möglichen Wertebereich.

Wie ist die Wallbox angeschlossen (1 oder 3 phasig) und wie sind die DIP Schalter gesetzt (welche max. A) ?

guggiguggo, 2024-08-08:
Die Wallbox ist 1phasig angeschlossen und das Ladelimit per Phase auf 15A gestellt im Anhang der Screenshot des Webinterfacres der Wallbox.
[image]

guggiguggo, 2024-08-08:
Danke leider komme ich erst am Samstag zum testen, werde berichten.

guggiguggo, 2024-08-10:
evcc version 0.129.0 (ee2284a6)

Klicke ich im Webinterface aus, bleit der Ladevorgang leider immer noch aktiv.
[evcc-20240810-122416-trace.log](https://github.com/user-attachments/files/16569234/evcc-20240810-122416-trace.log)

andig, 2024-08-10:
@DerAndereAndi es passiert

```

[lp-1  ] DEBUG 2024/08/10 12:22:40 set charge mode: off
[lp-1  ] DEBUG 2024/08/10 12:22:40 pv timer reset
[lp-1  ] DEBUG 2024/08/10 12:22:40 pv timer inactive
[site  ] DEBUG 2024/08/10 12:22:40 ----
[lp-1  ] DEBUG 2024/08/10 12:22:40 charge power: 1380W
[lp-1  ] DEBUG 2024/08/10 12:22:40 charge currents: [6 0 0]A
[site  ] DEBUG 2024/08/10 12:22:40 pv power: 2151W
[site  ] DEBUG 2024/08/10 12:22:41 battery soc: 100%
[site  ] DEBUG 2024/08/10 12:22:41 battery power: 5W
[site  ] DEBUG 2024/08/10 12:22:41 grid meter: 51W
[site  ] DEBUG 2024/08/10 12:22:41 site power: 656W
[lp-1  ] DEBUG 2024/08/10 12:22:41 detected active phases: 1p
[lp-1  ] DEBUG 2024/08/10 12:22:41 !! session: chargeRater.chargedEnergy=0.0 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/10 12:22:41 charger status: C
[lp-1  ] DEBUG 2024/08/10 12:22:41 vehicle api refresh
[main  ] DEBUG 2024/08/10 12:22:41 vehicle status: C (VW ID 3)
[lp-1  ] INFO 2024/08/10 12:22:41 vehicle updated: unknown -> VW ID 3
[lp-1  ] DEBUG 2024/08/10 12:22:41 vehicle status: C
[lp-1  ] DEBUG 2024/08/10 12:22:41 vehicle soc: 76%
[lp-1  ] DEBUG 2024/08/10 12:22:41 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/08/10 12:22:41 vehicle range: 270km
[eebus ] DEBUG 2024/08/10 12:22:41 !! api.StatusC
[eebus ] DEBUG 2024/08/10 12:22:41 no limits from the EVSE are provided: data not available
[eebus ] DEBUG 2024/08/10 12:22:41 Send: d:_i:47859_Elli-Wallbox-2020A0Q43V write 88 LoadControlLimitListData
[lp-1  ] DEBUG 2024/08/10 12:22:41 charger disable
[lp-1  ] DEBUG 2024/08/10 12:22:41 wake-up timer: stop
[eebus ] TRACE 2024/08/10 12:22:41 Send: 059d6b00543a2519c87eb18f4c7be3a719d0c076 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_n:EVCC_HEMS-evcc.home.lan"},{"entity":[1]},{"feature":7}]},{"addressDestination":[{"device":"d:_i:47859_Elli-Wallbox-2020A0Q43V"},{"entity":[1,1]},{"feature":10}]},{"msgCounter":88},{"cmdClassifier":"write"},{"ackRequest":true}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":0}]}],[{"limitId":1},{"isLimitActive":true},{"value":[{"number":0}]}],[{"limitId":2},{"isLimitActive":true},{"value":[{"number":0}]}]]}]}]]}]}]}}]}
[eebus ] TRACE 2024/08/10 12:22:41 Recv: 059d6b00543a2519c87eb18f4c7be3a719d0c076 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2020A0Q43V"},{"entity":[1,1]},{"feature":10}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-evcc.home.lan"},{"entity":[1]},{"feature":7}]},{"msgCounter":164},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":0},{"scale":0}]}],[{"limitId":1},{"isLimitActive":true},{"value":[{"number":0},{"scale":0}]}],[{"limitId":2},{"isLimitActive":true},{"value":[{"number":0},{"scale":0}]}]]}]}]]}]}]}}]}
[eebus ] DEBUG 2024/08/10 12:22:41 Recv: d:_i:47859_Elli-Wallbox-2020A0Q43V:LoadControl to LoadControl notify 164 LoadControlLimitListData
[eebus ] DEBUG 2024/08/10 12:22:41 ski 059d6b00543a2519c87eb18f4c7be3a719d0c076 event cem-opev-DataUpdateLimit
[eebus ] TRACE 2024/08/10 12:22:41 Recv: 059d6b00543a2519c87eb18f4c7be3a719d0c076 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2020A0Q43V"},{"entity":[1,1]},{"feature":10}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-evcc.home.lan"},{"entity":[1]},{"feature":7}]},{"msgCounter":165},{"msgCounterReference":88},{"cmdClassifier":"result"}]},{"payload":[{"cmd":[[{"resultData":[{"errorNumber":0}]}]]}]}]}}]}
[eebus ] DEBUG 2024/08/10 12:22:41 Recv: d:_i:47859_Elli-Wallbox-2020A0Q43V:LoadControl to LoadControl result 165 88 ResultData 0
[eebus ] DEBUG 2024/08/10 12:22:42 Send: d:_i:47859_Elli-Wallbox-2020A0Q43V notify 89 DeviceDiagnosisHeartbeatData
[eebus ] TRACE 2024/08/10 12:22:42 Send: 059d6b00543a2519c87eb18f4c7be3a719d0c076 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_n:EVCC_HEMS-evcc.home.lan"},{"entity":[1]},{"feature":8}]},{"addressDestination":[{"device":"d:_i:47859_Elli-Wallbox-2020A0Q43V"},{"entity":[1,1]},{"feature":1001}]},{"msgCounter":89},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"deviceDiagnosisHeartbeatData":[{"timestamp":"2024-08-10T10:22:42Z"},{"heartbeatCounter":58},{"heartbeatTimeout":"PT4S"}]}]]}]}]}}]}
[eebus ] DEBUG 2024/08/10 12:22:44 Send: d:_i:47859_Elli-Wallbox-2020A0Q43V notify 90 DeviceDiagnosisHeartbeatData
[eebus ] TRACE 2024/08/10 12:22:44 Send: 059d6b00543a2519c87eb18f4c7be3a719d0c076 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_n:EVCC_HEMS-evcc.home.lan"},{"entity":[1]},{"feature":8}]},{"addressDestination":[{"device":"d:_i:47859_Elli-Wallbox-2020A0Q43V"},{"entity":[1,1]},{"feature":1001}]},{"msgCounter":90},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"deviceDiagnosisHeartbeatData":[{"timestamp": [... cut ...] payload":[{"cmd":[[{"deviceDiagnosisHeartbeatData":[{"timestamp":"2024-08-10T10:23:20Z"},{"heartbeatCounter":77},{"heartbeatTimeout":"PT4S"}]}]]}]}]}}]}
[site  ] DEBUG 2024/08/10 12:23:20 ----
[lp-1  ] DEBUG 2024/08/10 12:23:20 charge power: 1380W
[lp-1  ] DEBUG 2024/08/10 12:23:20 charge currents: [6 0 0]A
```

Trotz `charger disable` scheint weiter geladen zu werden.

@guggiguggo kannst Du bitte id, db und http aus dem Log entfernen? Das brauchts nicht und macht es nur unnötig schwer zu lesen.

DerAndereAndi, 2024-08-10:
evcc sagt der Wallbox dass das Laden unterbrochen werden soll, die bestätigt das Limit, führt es dann aber wohl nicht aus.

Mir scheint dass das Fahrzeug bereits angesteckt war, bevor die Wallbox mit evcc eine Verbindung hatte. Das darf nicht sein, weil dann unterschiedliche Probleme kommen, so wie dieses. Also einmal abstecken, wieder anstecken.

guggiguggo, 2024-08-10:
systemctl stop evcc
auto abgesteckt
systemctl start evcc
alle leds grün
auto angesteckt
haus led orange
wallbox per rfid entsperrt
ladevorgang begint mit 1,4 kW
im Webinterface aus geklicht
Ladepunkt bleibt aktiv und zieht weiter ca. 1,4kW
[evcc-20240810-131236-trace.log](https://github.com/user-attachments/files/16569338/evcc-20240810-131236-trace.log)
](url)

andig, 2024-08-10:
Bitte: nimm http aus dem Log.

andig, 2024-08-10:
Limits werden gesetzt:

```
[lp-1  ] DEBUG 2024/08/10 13:11:31 charger disable
[lp-1  ] DEBUG 2024/08/10 13:11:31 wake-up timer: stop
[eebus ] TRACE 2024/08/10 13:11:31 Send: 059d6b00543a2519c87eb18f4c7be3a719d0c076 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_n:EVCC_HEMS-evcc.home.lan"},{"entity":[1]},{"feature":7}]},{"addressDestination":[{"device":"d:_i:47859_Elli-Wallbox-2020A0Q43V"},{"entity":[1,1]},{"feature":10}]},{"msgCounter":91},{"cmdClassifier":"write"},{"ackRequest":true}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":0}]}],[{"limitId":1},{"isLimitActive":true},{"value":[{"number":0}]}],[{"limitId":2},{"isLimitActive":true},{"value":[{"number":0}]}]]}]}]]}]}]}}]}
[eebus ] TRACE 2024/08/10 13:11:31 Recv: 059d6b00543a2519c87eb18f4c7be3a719d0c076 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2020A0Q43V"},{"entity":[1,1]},{"feature":10}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-evcc.home.lan"},{"entity":[1]},{"feature":7}]},{"msgCounter":168},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"function":"loadControlLimitListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":0},{"isLimitActive":true},{"value":[{"number":0},{"scale":0}]}],[{"limitId":1},{"isLimitActive":true},{"value":[{"number":0},{"scale":0}]}],[{"limitId":2},{"isLimitActive":true},{"value":[{"number":0},{"scale":0}]}]]}]}]]}]}]}}]}
```

Box lädt aus Sicht von evcc weiter:

```
[eebus ] DEBUG 2024/08/10 13:11:59 !! api.StatusC
[site  ] DEBUG 2024/08/10 13:12:28 ----
[lp-1  ] DEBUG 2024/08/10 13:12:28 charge power: 1380W
[lp-1  ] DEBUG 2024/08/10 13:12:28 charge currents: [6 0 0]A
```

@DerAndereAndi ich sehe allerdings nach setzen der Limits keine EEbus Kommunikation mehr ausser Heartbeats.

DerAndereAndi, 2024-08-10:
Ja, die Box liefert keine neuen Messwerte. Die letzten kommen hier:

```
[eebus ] TRACE 2024/08/10 13:10:20 Recv: 059d6b00543a2519c87eb18f4c7be3a719d0c076 {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2020A0Q43V"},{"entity":[1,1]},{"feature":11}]},{"addressDestination":[{"device":"d:_n:EVCC_HEMS-evcc.home.lan"},{"entity":[1]},{"feature":6}]},{"msgCounter":125},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"function":"measurementListData"},{"filter":[[{"cmdControl":[{"partial":[]}]}]]},{"measurementListData":[{"measurementData":[[{"measurementId":0},{"valueType":"value"},{"value":[{"number":6},{"scale":0}]}]]}]}]]}]}]}}]}
```

Vielleicht lädt das Auto gar nicht mehr und die Box aktualisiert die Daten einfach nicht? Wenn das so ist -> externe Strommessung nehmen.

guggiguggo, 2024-08-10:
> nimm

so? : cat evcc-20240810-131236-trace.log  | grep -v http  >  evcc-20240810-131236-trace1.log
[evcc-20240810-131236-trace1.log](https://github.com/user-attachments/files/16569377/evcc-20240810-131236-trace1.log)

DerAndereAndi, 2024-08-10:
Ach ja, würde eine Änderung auf "Now" überhaupt zu einem Effekt führen?

guggiguggo, 2024-08-10:
> Ja, die Box liefert keine neuen Messwerte. Die letzten kommen hier:
>
> ```
> [eebus ] TRACE 2024/08/10 13:10:20 Recv: 059d6b00543a2519c87eb18f4c7be3a719d0c076 [...]

Nach den Werten den Fronius WR und den sonsitigen Verbrauchen, lädt das Auto aber mit hoher Sicherheit weiter.
Im Moment, da noch alles neu, läuft alles mit Nulleinspeisung, könnte das zum Problem beitragen?

guggiguggo, 2024-08-10:
> Ach ja, würde eine Änderung auf "Now" überhaupt zu einem Effekt führen?

Du meinst schnell? da würde das Auto auch nur mit 1,4 kW weitergeladen.

DerAndereAndi, 2024-08-10:
Dann ignoriert die Wallbox einfach alles was von evcc kommt. Hat das jemals bei dir "besser" funktioniert?

guggiguggo, 2024-08-10:
> Dann ignoriert die Wallbox einfach alles was von evcc kommt. Hat das jemals bei dir "besser" funktioniert?

Ich hatte im evcc.yaml mal nur den Loadpoint, die Wallbox und das Auto, da konnte ich den Ladevorgang starten und stoppen.

DerAndereAndi, 2024-08-10:
Mit welcher Version von evcc?

andig, 2024-08-10:
> so? : cat evcc-20240810-131236-trace.log | grep -v http > evcc-20240810-131236-trace1.log

Da sind immer noch die hunderte Ergebniszeilen drin...

StefanSchoof, 2024-08-10:
Hast du nach jedem Neustart und/oder Update überprüft, ob die Wallbox in der Weboberfläche connected oder not connected anzeigt?

guggiguggo, 2024-08-10:
Kann gerade nicht mehr testen.
Nachdem ich den Fehlstromschutzschalter aus und wieder an geschalten habe um die Box neu zu starten, leuchtet jetzt die Hausled rot und die box meint die Dip-Schalter Konfiguration sei bei gleichzeitiger verwendung von CT-Spulen nicht kompatibel. Bin für jeden Tipp dankbar, da ein laden jetzt unmöglich ist.

guggiguggo, 2024-08-10:
> Hast du nach jedem Neustart und/oder Update überprüft, ob die Wallbox in der Weboberfläche connected oder not connected anzeigt?

War immer connected

DerAndereAndi, 2024-08-10:
Bitte keine CT Spulen verwenden. Das ist auch ein bekanntes Problem mit EEBUS!

guggiguggo, 2024-08-10:
> Bitte keine CT Spulen verwenden. Das ist auch ein bekanntes Problem mit EEBUS und dieser Box!

Danke für den hinweis! CT-Spulen für Lastmanagement verwende ich keine, DiP-Schalter Gruppe A alle 0

DerAndereAndi, 2024-08-10:
Der EEBUS Code in evcc hat keinen bekannten Fehler und funktioniert hier bei mir mit 2 verschiedenen anderen Wallboxen seit Wochen einwandfrei. Wir sind seit Wochen auf der Suche nach möglichen Workarounds um die Elli doch zum mitarbeiten zu bewegen. Wenn ein Ladelimit geschickt wird, dieses akzeptiert wird, und trotzdem nicht eingehalten wird, was soll man da noch tun !?

Die einzige Möglichkeit die ich sehe ist evtl. die Nadel im Heuhaufen zu finden, mit einem Vergleichslog von der V0.126.x. Irgendetwas hat ja laut euren Aussagen dort dazu geführt, dass es funktioniert hat.

Generell: Diese Wallbox ist nur in Ausnahmefällen wohl zum PV Laden geeignet. Der Allgemeinheit kann ich zu der Wallbox seit 2 Jahren nicht guten Gewissens raten, außer sie soll immer ungesteuert mit voller Leistung laden.

guggiguggo, 2024-08-12:
Hatte heute Zeit die Box aufzuschrauben. Die letzten zwei DIP-Schalter der Gruppe A waren auf 1 (bitte entschuldigt, das Webinterface der Wallbox hat aber eine richtige Position der Switches von Gruppe A alle 0  angezeigt, siehe angehängten Screenshot von vor 4 Tagen im Post oben!?). 
Nichtsdestrotrotz blieb die BOX mit version 129 und korrigierten DIP-Switch Gruppe A durch evcc unsteuerbar, Ladevorgänge konnten nur über die App und nicht über das Webinterface von evcc gestoppt werden.
Dann habe ich versucht evcc auf eine funktionierende Version, wie gebeten, downzugraden:

```
zcat /var/log/dpkg.log*.gz | grep evcc | grep inst
2024-06-25 20:35:31 status half-installed evcc:amd64 0.114.1
2024-06-25 20:37:09 status installed evcc:amd64 0.127.3
2023-03-26 08:40:45 status half-installed evcc:amd64 0.104.2
2023-03-26 08:42:22 status installed evcc:amd64 0.114.1
2022-10-02 17:09:58 status half-installed evcc:amd64 0.97.1
2022-10-02 17:12:04 status installed evcc:amd64 0.104.2
2022-07-28 22:08:34 status half-installed evcc:amd64 0.95.0
2022-07-28 22:08:36 status installed evcc:amd64 0.97.1
2022-06-20 14:35:50 install evcc:amd64 <none> 0.95.0
2022-06-20 14:35:50 status half-installed evcc:amd64 0.95.0
2022-06-20 14:35:51 status installed evcc:amd64 0.95.0

apt install evcc=0.127.3

```

Und siehe [evcc-20240812-111338-trace.log](https://github.com/user-attachments/files/16579636/evcc-20240812-111338-trace.log) im Anhang, bitte um Entschuldigung da ungefiltert, die Box war nun steuerbar durch evcc version 0.127.3 und wechselen der Lademodi von aus nach pv, pv+min und schnell und wieder zurück war möglich. Ich hoffe ich konnte helfen.

andig, 2024-08-12:
@guggiguggo und mit 129 meinst du die aktuelle nightly? Oder 129.0?

guggiguggo, 2024-08-12:
> @guggiguggo und mit 129 meinst du die aktuelle nightly? Oder 129.0?

ja, 2024-08-12 10:31:18 status installed evcc:amd64 0.129.0+1723429417
im moment geht mit 0.127.3 auch grad wieder nix.

DerAndereAndi, 2024-08-12:
Ich schau mir das Log gerade mit der v0.127 im Vergleich zum vorherigen mit 0.129 an.

Folgende Dinge fallen mir auf:
- In beiden Logs kommen von der Wallbox KEINE Messwerte! Trotzdem wird in v0.127 "geregelt"
- Die Wallbox erlaubt das Setzen von Werten kleiner als 6A, was technisch gar nicht stimmen kann. Und damit werden Ladeleistungen von 600W angezeigt, die auch nicht realistisch sind!
- Ich verstehe momentan nicht warum überhaupt in V0.127.3 eine Leistung gemeldet wird. Auf Messwerten basieren die Daten nicht, denn es kommen gar keine an.

Ich kann hier nur empfehlen einen externen Zähler anzuschließen, diesen am `loadpoint` zu Konfigurieren und hoffen dass es damit besser läuft.

Generell sollte dein Config angepasst werden:

```
loadpoints:
- title: Parkplatz
  charger: wallbox5
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 15
```

andig, 2024-08-12:
> Ich verstehe momentan nicht warum überhaupt in V0.127.3 eine Leistung gemeldet wird. Auf Messwerten basieren die Daten nicht, denn es kommen gar keine an.

Charge power ist im Zweifel 230V x Phasen x Freigabestrom. Das ist bei allen Chargern ohne Meter der Fallback seit Anfangszeiten. Implementierung ist im Loadpoint, nicht im Charger.

Olf1337, 2024-08-12:
bei mir grad wieder gleiches verhalten mit der 0.129.0. LP steht auf AUS...lädt aber fleißig weiter:
[image]

[evcc-20240812-143844-debug.log](https://github.com/user-attachments/files/16581747/evcc-20240812-143844-debug.log)
[evcc-20240812-143828-trace.log](https://github.com/user-attachments/files/16581749/evcc-20240812-143828-trace.log)
[evcc-20240812-143815-trace.log](https://github.com/user-attachments/files/16581751/evcc-20240812-143815-trace.log)
[evcc-20240812-143855-debug.log](https://github.com/user-attachments/files/16581752/evcc-20240812-143855-debug.log)

ich traue mich ja kaum noch LOGs anzuhängen....hoffe aber sie sind hilfreich.

Am anderen LP geht es übrigens mit dem Steuern.

DerAndereAndi, 2024-08-12:
@andig ich finde aber den Code nicht wo quasi dann der alte 6A Wert invalidiert wird.

@Olf1337 mit welcher Config? Könntest du bitte ein eigenes Ticket machen, oder dein altes eigenes nutzen? So verliere ich den Überblick wenn alles durcheinander ist.

guggiguggo, 2024-08-12:
> Ich schau mir das Log gerade mit der v0.127 im Vergleich zum vorherigen mit 0.129 an.
>
> Folgende Dinge fallen mir auf:
>
>     * In beiden Logs kommen von der Wallbox KEINE Messwerte! Trotzdem wird in [...]

Hatte ich mit Version 0.129 im Webinterface konfiguriert und jetzt mit 0.127.3 der evcc.yaml hinzugefügt danke für den Hinweis.

> ```
> 
> Ach ja: vertraue erst mal den angezeigten Ladewerten nicht, außer das Auto zeigt die auch nahezu identisch an. Die Wallbox sagt evcc jedenfalls nur einmal am Anfang es würde mit 6A laden. Das kann ich nicht glauben dass genau mit 6A geladen werden.

DerAndereAndi, 2024-08-12:
Es gibt hier nichts an EEBUS dass ich sehe was zu so etwas führen kann. Und bei dir ist es ja jetzt auch in der V0.127.3 nicht immer wie erwartet gewesen. Die Software hat zu viele Bugs und wie im Elli Template schon immer steht:

> Wichtig: Die möglichst reibungslose Funktionalität ist aufgrund von Software-Fehlern in der Wallbox nur mit einem externen Energiezähler und ohne Stromwandlerspulen möglich! Eine LAN Anbindung wird sehr empfohlen.

Mehr kann ich nicht tun.

guggiguggo, 2024-08-12:
Tinkerforge Warp3 ist in engerer Auswahl.

guggiguggo, 2024-08-12:
> Können wir empfehlen. So traurig das auch ist- Issue hier schließen?

Jo, völlig ok.