evcc-io/evcc #15288: Warn: charger out of sync: expected disabled, got enabled

TobiasHuber1980, 2024-08-07:
### Describe the bug

Nach dem Anstecken folgt immer dieser Warnhinweis:
`charger out of sync: expected disabled, got enabled`

### Steps to reproduce

1. Auto anstecken, Ladeung per RFID freigeben
2. Welcomecharge wg. Keba ist aktiviert
3. warten bis Warnhinweis kommt.
4. Trace-Log (Uhrzeit: 17:06:15)
[evcc-20240807-174119-trace.log](https://github.com/user-attachments/files/16531016/evcc-20240807-174119-trace.log)

...

### Configuration details

```yaml
[main  ] INFO 2024/08/07 08:38:56 evcc 0.129.0 (b4dea20f)
[main  ] INFO 2024/08/07 08:38:56 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/07 08:38:56 using sqlite database: /home/pi/.evcc/evcc.db
[circuit-main] DEBUG 2024/08/07 08:39:01 validation of max phase current disabled
[circuit-loadpoints] DEBUG 2024/08/07 08:39:01 validation of max power disabled
[circuit-loadpoints] DEBUG 2024/08/07 08:39:01 validation of max phase current disabled
[lp-1  ] DEBUG 2024/08/07 08:39:01 charge total import: 130.313kWh
[lp-2  ] DEBUG 2024/08/07 08:39:01 charge total import: 3629.874kWh
[lp-3  ] DEBUG 2024/08/07 08:39:01 charge total import: 463.889kWh
[lp-4  ] DEBUG 2024/08/07 08:39:01 charge total import: 354.434kWh

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

##################################

site:
  - title: Name
    circuit: main
    meters:
      grid: grid
      pv:
        - pv
        
##################################     
   
circuits:
- name: main
  title: Main Circuit
  maxPower: 35000
  #maxCurrent: 100
  meter: grid
  parent:
- name: loadpoints
  #maxCurrent: 120
  parent: main        

##################################
[... 313 lines cut ...]
[circuit-loadpoints] DEBUG 2024/08/07 17:07:35 power: 3994.7W
[circuit-loadpoints] DEBUG 2024/08/07 17:07:35 current: 6.06A
[circuit-main] DEBUG 2024/08/07 17:07:35 power: -10040W
[circuit-main] DEBUG 2024/08/07 17:07:35 current: 0A
[site ] DEBUG 2024/08/07 17:07:35 pv power: 36360W
[site ] DEBUG 2024/08/07 17:07:35 grid meter: -10040W
[site ] DEBUG 2024/08/07 17:07:35 site power: -10040W
[lp-4 ] DEBUG 2024/08/07 17:07:35 detected active phases: 3p
[lp-4 ] DEBUG 2024/08/07 17:07:35 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-4 ] DEBUG 2024/08/07 17:07:35 charge total import: 354.620kWh
[lp-4 ] DEBUG 2024/08/07 17:07:35 charger status: C
[lp-4 ] DEBUG 2024/08/07 17:07:35 pv charge current: 20.6A = 6A + 14.6A (-10040W @ 3p)
[lp-4 ] DEBUG 2024/08/07 17:07:35 max charge current: 20.6A
[site ] DEBUG 2024/08/07 17:07:55 ----
[lp-4 ] DEBUG 2024/08/07 17:07:55 charge power: 10729W
[lp-4 ] DEBUG 2024/08/07 17:07:55 charge currents: [16 16.1 16]A
```

### What type of operating system are you running?

Linux

### Version

0.129.0 (b4dea20f

andig, 2024-08-08:
> Nach dem Anstecken folgt (immer) dieser Warnhinweis:

Und?

TobiasHuber1980, 2024-08-10:
ok, dann stellt sich mir die Frage, warum fängt die Box von alleine an zu laden? 
Aber ich denke ich weiss warum: Die RFID-Funktion in der Keba gibt ja schon die Ladung frei und evcc kommt erst danach regelnd zum Einsatz.

closed 2024-08-10 by TobiasHuber1980

andig, 2024-08-10:
Genau so ist es- und wird dann wieder eingebremst