evcc-io/evcc #15290: Anzeigeerweiterung für Megawatt

mobbl, 2024-08-07:
### Describe the bug

Die CO² Anzeige ist sehr dicht an der Verbrauchsanzeige bei einer PV Anlage größer als 1MW

### Steps to reproduce

1. PV größer als 1000kWp

### Configuration details

```yaml
No YAML
```

### Log details

```text
No Logs
```

### What type of operating system are you running?

Linux

### Version

0.129

naltatis, 2024-08-07:
Idee wäre hier, wie auch schon beim `W > kW` automatisch auf `MW` Einheiten umzuschalten.

premultiply, 2024-08-07:
Nächstes Jahr kommt einer mit GW 🤠

julred, 2024-08-08:
Bei solchen Themen fragt man sich, ob evcc noch das richtige Vehicle für die Anforderungen sind.
Rein interessehalber, welche Art von Gewerbe steck hinter einem solchen Verbrauch?

premultiply, 2024-08-08:
Erst im Großen zeigt sich ob alles richtig skaliert und wo man ggf. noch nachsteuern sollte. Manche Effekte sieht man wie hier erst dann wenn man es "übertreibt".

mobbl, 2024-08-08:
https://github.com/evcc-io/evcc/issues/15290#issuecomment-2274461203

Gigawatt wird leider nichts auf einem Firmengelände auf dem Dorf 🫣

mobbl, 2024-08-08:
> Nächstes Jahr kommt einer mit GW 🤠

Gigawatt wird leider nichts auf einem Firmengelände auf dem Dorf 🫣

mobbl, 2024-08-08:
> Bei solchen Themen fragt man sich, ob evcc noch das richtige Vehicle für die Anforderungen sind. Rein interessehalber, welche Art von Gewerbe steck hinter einem solchen Verbrauch?

Da muss ich @premultiply Recht geben :)

Das ist aktuell noch eine kleine Machbarkeitsstudie, da noch einige Anlagen als Volleinspeiser laufen. Wenn die rausfliegen müssen wir entweder auf 0 abregeln aufgrund von zu viel Überschuss oder halt die Energie sinnvoll nutzen.
Es handelt sich um ein großes Firmengelände in der Holzindustrie mit ca. 17 Hallen die alle mit PV bestückt sind.

Wir werden nach und nach Wallboxen installieren ggf. auch 50kW CCS Säulen. 

Aufgrund von Planungssicherheit beim Stromeinkauf, Lastspitzen, Einspeisestrafen etc hat das alles so seine Berechtigung ;)

Maschga, 2024-08-30:
@naltatis Can I take over this issue?

Maschga, 2024-08-30:
> Currently, we keep all units in sync.

Can you show me the code where you do this?

[I already have a branch that supports MW](https://github.com/Maschga/evcc/tree/shorten-numbers-by-using-larger-units) but I don't know how I can make sure units stay in sync.
I can create a PR draft if that helps.

naltatis, 2024-08-30:
This is the logic that switches everything:
https://github.com/evcc-io/evcc/blob/master/assets/js/components/Energyflow/Energyflow.vue#L295-L297

naltatis, 2024-08-30:
Looking ad your change I'd suggest changing the computed `powerInKw` property to something like `powerUnit`

PR #15818 (2024-09-04): Add support for megawatt
Hi,
this PR adds support for the unit `megawatt` and its display on the screen.
If support for `gigawatt` is wanted I can add this, too.

I also changed the tests to work with the new enum.

I was unsure how I should export the enum.
If the current solution is not correct, please let me know so that I can learn from it and improve it.

Also, I was unsure if I should rename [this function](https://github.com/Maschga/evcc/blob/26d74ec66320a2c71854f26765fec497abe7c58a/assets/js/components/Energyflow/EnergyflowEntry.vue#L88-L93):
```go
powerInKw(newVal, oldVal) {
	// force [... cut]
closed 2024-09-04 by naltatis