evcc-io/evcc #15292: shelly-authentification doesen't support special characters

wachschaf, 2024-08-07:
### Describe the bug

if special characters are used in a shelly-password, evcc throws an error (even if they are supported by shelly) 

**config**
```typescript
meters:
- type: template
  template: shelly-3em
  usage: grid
  host: 192.168.101.100
  user: admin
  password: 'yY7k0X&^>=PQ]0s(J{TD.</_'
  name: grid1
``` 

**commandline-output**
```shell
Power:          parse "http://admin:yY7k0X&^>=PQ]0s(J{TD.&lt;/_@192.168.101.100/status": invalid port ":yY7k0X&^>=PQ]0s(J{TD.&lt;" after host
Energy:         parse "http://admin:yY7k0X&^>=PQ]0s(J{TD.&lt;/_@192.168.101.100/status": invalid port ":yY7k0X&^>=PQ]0s(J{TD.&lt;" after host
Current L1..L3: parse "http://admin:yY7k0X&^>=PQ]0s(J{TD.&lt;/_@192.168.101.100/emeter/0": invalid port ":yY7k0X&^>=PQ]0s(J{TD.&lt;" after host
Voltage L1..L3: parse "http://admin:yY7k0X&^>=PQ]0s(J{TD.&lt;/_@192.168.101.100/emeter/0": invalid port ":yY7k0X&^>=PQ]0s(J{TD.&lt;" after host
Power L1..L3:   parse "http://admin:yY7k0X&^>=PQ]0s(J{TD.&lt;/_@192.168.101.100/emeter/0": invalid port ":yY7k0X&^>=PQ]0s(J{TD.&lt;" after host
```  

additional issue: evcc seems to convert to convert characters into html-elements like & lt; (not sure if this is an issue too)

--> other platforms like home-assistant and the shelly-app don't have issues with these kind of characters 

### Steps to reproduce

1. use special characters like ^ or < or / , ...

### Configuration details

```yaml
meters:
- type: template
  template: shelly-3em
  usage: grid
  host: 192.168.101.100
  user: admin
  password: 'yY7k0X&^>=PQ]0s(J{TD.</_'
  name: grid1
```

### Log details

```text
Power:          parse "http://admin:yY7k0X&^>=PQ]0s(J{TD.&lt;/_@192.168.101.100/status": invalid port ":yY7k0X&^>=PQ]0s(J{TD.&lt;" after host
Energy:         parse "http://admin:yY7k0X&^>=PQ]0s(J{TD.&lt;/_@192.168.101.100/status": invalid port ":yY7k0X&^>=PQ]0s(J{TD.&lt;" after host
Current L1..L3: parse "http://admin:yY7k0X&^>=PQ]0s(J{TD.&lt;/_@192.168.101.100/emeter/0": invalid port ":yY7k0X&^>=PQ]0s(J{TD.&lt;" after host
Voltage L1..L3: parse "http://admin:yY7k0X&^>=PQ]0s(J{TD.&lt;/_@192.168.101.100/emeter/0": invalid port ":yY7k0X&^>=PQ]0s(J{TD.&lt;" after host
Power L1..L3:   parse "http://admin:yY7k0X&^>=PQ]0s(J{TD.&lt;/_@192.168.101.100/emeter/0": invalid port ":yY7k0X&^>=PQ]0s(J{TD.&lt;" after host
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.129.0

andig, 2024-08-07:
/cc @thierolm

closed 2024-08-08 by andig

wachschaf, 2024-08-09:
👍 the fix works like a charm... thanks