evcc-io/evcc #15294: SMA SBS 2.5 Modbus Battery loading request

wittmannchris, 2024-08-07:
Even with the Field-Test Firmeware of the Sunny Home manager, and full control of the battery inverter, a loading via 40149 and 40151 with the SMA SBS 2.5 is not possible.

sma-sbs-modbus.yaml uses Register 44039 for Battery Charge and Discharge

sma-si-modbus.yaml uses Register 40149 and 40151. 

SBS 2.5 has also the register 44039 and you can use this for loading and unloading. Tested right now. 

Is there a possibility to change this template as well to bring back battery loading possibility to the SBS2.5 as well @premultiply ?

The Bug with 40149 and 40151 also not works with direct modbus control and is reported to SMA field tester as well.

Thanks

wittmannchris, 2024-08-09:
Alternaly (and tested) 👍 

For Battery Loading:
40236 ->2289
40793 -> 2500
40795 -> 2500
40797 -> 0
40799 -> 2650
40801 -> 0

For 0-Usage Loading:

40236 ->2424
40793 -> 0
40795 -> 0
40797 -> 0
40799 -> 0
40801 -> 0

Normal : 

40236 ->2424
40793 -> 0
40795 -> 2500
40797 -> 0
40799 -> 2650
40801 -> 0

All6 Registers must be written within 60 seconds and than cyclical

Flash is not getting damaged as marked as a "ANlagensteuerobjekt" as well details in the documentation
[image]

wittmannchris, 2024-08-18:
> verhält sich mit -2147483648 abnorm.
> Mit -2147483647 lädt er dann mit maximaler Leistung.

Aus dem Smart Energy Issue.  Hier das gleiche, funktioniert also auch mit 40149 und 40151

Sobald also SMA den FIx Anfang September macht wäre es nett, wenn hier der SBS 2.5 auch überarbeitet wird. Mit welcher Version auch immer. (Ich würde die 6 Register bevorzugen da offiziell von SMA so empfohlen.

closed 2024-08-19 by andig

wittmannchris, 2024-08-20:
Warum auch immer, mit IoBroker kann ich die Register 40149 und 40151 steuern ( wenn auch ziemlich unnormal …) , mit EVCC und dem Fix funktioniert es aber dennoch nicht. 

Daher meine Bitte an @premultiply dennoch auf die Vorgabe von SMA einzugehen und den SBS 2.5nochmals zu ändern. 

die beiden Register 40149 und 40151 sind auch keine offiziellen in der aktuellen Modbuslist. 

Evtl. Werden diese beiden in Zukunft exklusiv vom SHM verwendet und der alternative Steuerung über Modbus wie von mir oben beschrieben. So sagte es zumindest auch der Feldtestservice. 

Bitte wenn mal Zeit ist ändern. Wäre schade um die Funktion….

Danke

premultiply, 2024-08-20:
Ehrlich gesagt hab ich jetzt den Überblick verloren was du möchtest.

- Stand heute gibt es kein Template für den SBS2.5 ?
- Sowohl SI als auch der STP-SE sind im Vergleich zum SBS3.7/5.0/6.0 bzgl. der Register eingeschränkt. Die schöne Variante über die beiden Leistungsgrenzen in Prozent vom SBS gibt es bei diesen nicht. Dort geht nur direkte Wirkleistungsvorgabe.
- Alle verwendeten Register stammen aus der offiziellen Modbus-Doku der entsprechenden Geräte

wittmannchris, 2024-08-21:
Sowie ich es verstanden habe ist für den SBs2.5 dieses Template zuständig : templates/definition/meter/sma-si-modbus.yaml

dieser ist wohl für den Sunny Island und den SBS 2.5 gleichermaßen . 

Nur funktionieren eben die Register 40149 und 40151 beim SBS2.5 nicht ( mehr?) 

https://github.com/evcc-io/evcc/issues/15294#issuecomment-2278588953 Wäre die offizielle Anwendung wie von SMA vorgegeben und auch in der Doku ersichtlich . 

Mit der Bitte u, Erstellung eines zusätzlichen Templates für den Sunny Boy Storage 1.5/2.0/2.5 um im Winter die Batterie auch mit diesen Geräten laden zu können.

wittmannchris, 2024-09-02:
> Sowie ich es verstanden habe ist für den SBs2.5 dieses Template zuständig : templates/definition/meter/sma-si-modbus.yaml
> 
> dieser ist wohl für den Sunny Island und den SBS 2.5 gleichermaßen .
> 
> Nur funktionieren eben die Register 40149 und 40151 beim SBS2.5 nicht ( mehr?)
> 
> [#15294 (comment)](https://github.com/evcc-io/evcc/issues/15294#issuecomment-2278588953) Wäre die offizielle Anwendung wie von SMA vorgegeben und auch in der Doku ersichtlich .
> 
> Mit der Bitte u, Erstellung eines zusätzlichen Templates für den Sunny Boy Storage 1.5/2.0/2.5 um im Winter die Batterie auch mit diesen Geräten laden zu können.

@premultiply gibts die Chance für die Erstellung eines eigenen Templates für den SBS2.5?
Die oben genannten Register sind genau diese aus der offiziellen Modbus Datei.
SMA verteilt ja nun auch die neue Software für den SHM2
[Modbus - Parameter und Messwerte.pdf](https://github.com/user-attachments/files/16837050/Modbus.-.Parameter.und.Messwerte.pdf)

S64 SMA_Modbus-TI-de-15

[SMA_Modbus-TI-de-15.pdf](https://github.com/user-attachments/files/16837061/SMA_Modbus-TI-de-15.pdf)

Danke für die Mühen