evcc-io/evcc #15295: Add Sungrow Ocpp Template, remove Modbus

andig, 2024-08-07:
Ich habs via OCPP hinbekommen. Wenn jemand das gleich Problem hat gerne anschreiben.

_Originally posted by @ruma007 in https://github.com/evcc-io/evcc/discussions/15079#discussioncomment-10268598_

premultiply, 2024-08-07:
Geht leider noch nicht, da bei der Box via OCPP die Phasenumschaltung noch nicht funktioniert.

closed 2024-08-07 by andig

ruma007, 2024-08-07:
> Geht leider noch nicht, da bei der Box via OCPP die Phasenumschaltung noch nicht funktioniert.

Wie meinst Du das? Ich lade zZ eh nur einen MG4 Standard. Der kann nur einphasig. Kann ich das nicht nutzen?

premultiply, 2024-08-07:
Doch, natürlich aber die Phasenumschaltung lässt sich via OCPP noch nicht ansprechen, via Modbus schon.
Du kannst die Sungrow-Box aber natürlich auch via OCPP nutzen wenn du dieses spezielle Feature für deinen Einsatzzweck nicht brauchst.

ruma007, 2024-08-07:
Achso ok. Wird sich das bald noch ändern?

premultiply, 2024-08-07:
Ääh das müsstest du SunGrow fragen?

ruma007, 2024-08-08:
Ein Frage noch,...soll ist in diesem Fall in den Einstellungen "3-phasig" verwenden oder ist es besser auf "1-phasig" zu stellen?

premultiply, 2024-08-08:
Völlig egal.
Hauptsache es ist am Fahrzeug richtig konfiguriert.