evcc-io/evcc #15317: Anzeige Ladevorgänge / CSV export: Einzelne Datensätze fehlen, die in Datenbank enthalten sind

An-Wi, 2024-08-09:
### Describe the bug

Eigentlich eher zufällig beim Korrigieren sehr alter Ladevorgangsdatensätze fiel mir auf, dass von meinen insgesamt 294 Datensätzen, die in der db-Datei gespeichert sind, nur 287 Datensätze in die CSV exportiert werden. Ein Abgleich mit der UI-Ansicht "Ladevorgänge" zeigt, auch dort fehlen diese 7 Ladevorgänge!

Exemplarisch füge ich den letzten fehlenden Eintrag hier mal per als Screenshots (db & UI) ein:
[image]
[image]

Woran kann das liegen? Für mich sieht der fehlende Datensatz "normal" aus. 

### Steps to reproduce

1. evcc.db (sessions table) auslesen 
2. CSV-Export (gesamt) herunterladen über UI
3. Anzahl Datenzeilen vergleichen...

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Windows

### Version

0.129.0

naltatis, 2024-08-09:
Ja, Ladeeinträge mit sehr kleinen Energiemengen (< 50 Wh) werden aktuell bewusst rausgefiltert. 

https://github.com/evcc-io/evcc/blob/1eeeebdadd09fe6fc152c86451ab480ea60d7a2c/server/http_session_handler.go#L61

naltatis, 2024-08-09:
Der Grund ist meist, dass beim Anstecken kurz geladen wurde, evcc dann mangels Überschuss schnell ausschaltet und auch in der Ansteckdauer kein "richtiger" Ladevorgang mehr zustande gekommen ist. Im Auto-Kontext und Heizungskontext ist das Limit glaube ich total ok. Für andere kleinere Verbraucher (Werkzeugakkus oder so) brauchen wir ohnehin eine Auswertung die anders arbeitet: https://github.com/evcc-io/evcc/issues/13050

closed 2024-08-09 by naltatis

An-Wi, 2024-08-09:
Alles klar - danke für die schnelle Info.

Ich dachte zwar, dass bei mir im letzten Jahr auch 1-2 Einträge mit größerer Energiemenge dabei war, die fehlen, aber hab es nochmal geprüft: dem ist doch nicht so. Passt also! 👍
Diese Differenzen, die ich hier noch im Kopf hatte, stammt aus unvollständigen Datensätzen, die es früher gelegentlich gab. 
Das ist jedoch in den letzten Monaten nur noch minimal der Fall (sichtbar auch nur, wenn man Anfangszählerstand mit Endzählerstand und geladener Energie gegenrechnet) - daher kein Issue wert. 😉