evcc-io/evcc #15324: Setting byte order (bigendian) for energy meter via Kostal inverter

ramikeller, 2024-08-09:
### Describe the bug

On my Kostal Plenticore, I need to set the byteorder to big-endian (ABCD) Sunspec (to be able to communicate with the my WARP3 Tinkerforge wallbox). 

However, then reading from evcc the grid values, it produces incorrect grid values.

While the endianness can be set for the kostal-plenticore, it cannot be defined in kostal-ksem-inverter.

### Steps to reproduce

Setting the byte order to little-endian (CDAB) Standard Modbus in Kostal generates the correct value.

Suggestion:

Modify templates/definition/meter/kostal-ksem-inverter.yaml as follows:

params:
  - name: usage
    choice: ["grid"]
  - name: modbus
    choice: ["tcpip"]
    port: 1502
    id: 71
  - name: endianness
    description:
      de: Byte-Reihenfolge (Little/Big)
      en: Endianness (Little/Big)
    validvalues: ["big", "little"]
    default: little
    advanced: true
render: |
  type: custom
  power:
    source: modbus # use ModBus plugin
    {{- include "modbus" . | indent 2 }}
    register: # manual non-sunspec register configuration
      address: 252 # (see ba_kostal_interface_modbus-tcp_sunspec.pdf)
      type: holding
      decode: {{ if (eq .endianness "big") }}float32{{ else }}float32s{{ end }}

### Configuration details

```yaml
meters:
- type: template
  template: kostal-plenticore
  id: 71
  host: kostal.home
  port: 1502
  usage: pv
  modbus: tcpip
  endianness: big
  minsoc: 20
  watchdog: 60s
  name: pv1
- type: template
  template: kostal-ksem-inverter
  id: 71
  host: kostal.home
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid3
```

### Log details

```text
When setting big-endian (ABCD) Sunspec in the Kostal inverter:

evcc meter --log trace

grid3
-----
[modbus] TRACE 2024/08/09 22:13:58 modbus: send 00 2e 00 00 00 06 47 03 00 fc 00 02
[modbus] TRACE 2024/08/09 22:13:59 modbus: recv 00 2e 00 00 00 07 47 03 04 44 c1 79 ea
Power: 152048994973195403960265013513420800W

When setting little-endian (CDAB) Standard Modbus

grid3
-----
[modbus] TRACE 2024/08/09 22:14:19 modbus: send 00 2e 00 00 00 06 47 03 00 fc 00 02
[modbus] TRACE 2024/08/09 22:14:19 modbus: recv 00 2e 00 00 00 07 47 03 04 e8 8c 44 c3
Power: 1567W

(correct value)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.129.0

andig, 2024-08-09:
We‘ve had this before. I don‘t think we want to re-add this setting.

ramikeller, 2024-08-09:
Hi Andi, can you give a suggestion on how to address this issue? 

Note that I need to set the endianness on the Kostal to big-endian (ABCD) Sunspec such that the WARP3 can read the grid values. Big-endian is the Sunspec standard and the WARP3 only supports big-endian.

Currently, with evcc I can read the PV values (as I can set the byte order to big endian) but not the grid value (as I cannot set it to big endian).

Note that there is a comment in kostal-ksem-inverter.yaml that some devices need big endian.

decode: float32s # may be float32 on specific firmware/devices

premultiply, 2024-08-10:
Just read it as SunSpec Meter?

ramikeller, 2024-08-10:
I've tried to read the grid values through the inverter as follows:

meters:
- type: template
  template: kostal-plenticore
  id: 71
  host: kostal.home
  port: 1502
  usage: pv
  modbus: tcpip
  endianness: big
  minsoc: 20
  watchdog: 60s
  name: pv1
- type: template
  template: sunspec-inverter
  id: 71
  host: kostal.home
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid3

# evcc -l debug meter
pv1
---
Power: 4569W  # correct

grid3
-----
Power: -147930742784W  # incorrect, should be read as bigendian

premultiply, 2024-08-10:
SunSpec has specified byte and word order so there is nothing to fix here.
If the decoding is wrong, the encoding on Kostal side MUST be wrong.

StevieC121176, 2024-08-10:
Das Problem besteht doch schon immer bei Kostal und liegt nicht an EVCC. Sobald man im Kostal WR auf BigEndian stellt werden die Werte komplett falsch angezeigt. Auch im KSEM (Kostal Smart Meter)