evcc-io/evcc #15332: Hallo, seit einigen Wochen bekomme ich den Cupra Born von meinem Sohn nicht mehr angemeldet. Hat nichts mit den letzten Versionen zu tun, hat sich mit durchgezogen. Zur Zeit ist 0.129 installiert. Weder im Auto noch in der Cupra APP ist irgendetwas zu bestätigen.  Die komplette Fehlermeldung ist "Request failed with status code 400.cannot create vehicle type 'template': cannot create vehicle type 'cupra': missing id_token.POST http://192.168.1.112:7070/api/config/test/vehicle" Ich versuche die Einstellung über die UI. Was kann ich versuchen um den Fehler zu beseitigen? Viele Grüße, Ralf

rbeudel, 2024-08-10:
Hallo, seit einigen Wochen bekomme ich den Cupra Born von meinem Sohn nicht mehr angemeldet. Hat nichts mit den letzten Versionen zu tun, hat sich mit durchgezogen. Zur Zeit ist 0.129 installiert. Weder im Auto noch in der Cupra APP ist irgendetwas zu bestätigen.  Die komplette Fehlermeldung ist "Request failed with status code 400.cannot create vehicle type 'template': cannot create vehicle type 'cupra': missing id_token.POST http://192.168.1.112:7070/api/config/test/vehicle" Ich versuche die Einstellung über die UI. Was kann ich versuchen um den Fehler zu beseitigen? Viele Grüße, Ralf

_Originally posted by @rbeudel in https://github.com/evcc-io/evcc/discussions/15273_

closed 2024-08-10 by andig

rbeudel, 2024-08-10:
Ich verstehe das nicht? Der Fehler ist nun mal da und ich weis nicht was ich machen soll.

andig, 2024-08-10:
Dann bitte das Issuetemplate ausfüllen.