evcc-io/evcc #15333: Add Growatt API for Growatt Inverters

nacho26, 2024-08-10:
**Is your feature request related to a problem? Please describe.**
I have a working setup with a Zaptec GO Wallbox and a Tibber Pulse Meter but it would be nice to have the information from my two Growatt Inverters also in evcc. I read that there should be an option to flash the Growatt Inverter WLAN Modules with an alternate Firmware that is then able to communicate with evcc but I would loose the connection to the Growatt Server and also the collection of the Inverter statistics there. On the other hand only one of my inverters is on the same Network like the evcc Server, that isn't a complete and easy solution for my setup and also not easy for none IT Pro's.

**Describe the solution you'd like**
I would be nice if you can intergrade the current Energy Production of my Growatt Inverters via the Growatt API. The Inverters send their energy production to the Growatt Cloud any 5 Minutes. That is enough from my point of view and I can retain the statistics there, don't need to flash the WLAN Modules of the Inverter. There are already working implementations like this https://github.com/indykoning/PyPi_GrowattServer or this https://github.com/Sjord/growatt_api_client. Even Tibber has integrated it into their App.

**Describe alternatives you've considered**
I'm not aware of any other alternatives.

**Additional context**
I'm able to contribute my credentials for the Growatt Server for testing if needed.

Regards and Thanks Marc

andig, 2024-08-10:
Why would you not use the local Growatt implementation?

nacho26, 2024-08-10:
Like I wrote I would lose the statistics collected by the Growatt Cloud and have to take care by myself to collect them. I you not an IT Pro it is not easy to flash the WLAN Adapter of the inverter. On the other hand I own two Inverters that are connected to the same Grid but not connected to same Network. The evcc can only contact one of the inverters directly not a complete solution. My goal is to see the own consumption more clearly. And on a day like today with no clouds the 5 min values are ok to measure the own consumption more "correctly".

andig, 2024-08-10:
I don't understand. We support modbus. What does this have to do with flashing the WLAN adapter?

nacho26, 2024-08-10:
I you buy a Growatt Inverter you get a WLAN Stick called Shine Wifi Stick. This Stick doesn't speak Modbus except you flash that with an custom firmware, but then you lose the collection of statistics by the Growatt Cloud. 
Currently with evcc + Tibber + Zaptec GO and Cupra Born I'm 100% API connected and it doesn't matter where my evcc instance is running - that is really cool
If I need to connect local to both inverters in my case I need additional hardware of some kind because one Inverter isn't on the same Network.
I can understand that you say, we do evcc in our spare time and don't want to wast time for certain api integrations on the other hand you did that already for more then 40 Inverter types regarding you website.
And I thought a developer can easily adapt an already developed integration by other GitHub Projects. :-)

nacho26, 2024-08-10:
now I see the "wontfix" Label and it looks like you already made your decision

andig, 2024-08-10:
First thing would be an api description. Having that you could check if this can already be done using plugins.

closed 2024-08-10 by andig

nacho26, 2024-08-11:
based on that Project https://github.com/indykoning/PyPi_GrowattServer/tree/master, I was able to build a small python script that outputs also the current power (power) value of my inverters via the Growatt API.
Now my idea is to use the script plugin to read the power values to evcc but I don't know how to read the power value. Is there some sort of documentation?

here is a sample output from the script:

[{'deviceType': 'tlx', 'powerStr': '0.09kW', 'type2': '0', 'bdc2Soc': 0, 'bdc1Soc': 0, 'type': 0, 'deviceSn': 'RGL1CGE0FT', 'bMerterConnectFlag': -1, 'deviceStatus': '1', 'eToday': '19', 'lost': False, 'datalogSn': 'XGD6CGR0S4', 'location': '', 'power': '94.1', 'prePto': '-1', 'deviceAilas': 'RGL1CGE0FT', 'energy': '5032.4', 'eTodayStr': '19kWh'}, {'deviceType': 'tlx', 'powerStr': '0.19kW', 'type2': '0', 'bdc2Soc': 0, 'bdc1Soc': 0, 'type': 0, 'deviceSn': 'FRJ7CH10JY', 'bMerterConnectFlag': -1, 'deviceStatus': '1', 'eToday': '7.7', 'lost': False, 'datalogSn': 'XGD6CJ159V', 'location': '', 'power': '189.9', 'prePto': '-1', 'deviceAilas': 'FRJ7CH10JY', 'energy': '1032.8', 'eTodayStr': '7.7kWh'}]

andig, 2024-08-11:
You need to create a `custom` meter, use the `http` or `script` plugin as `source` for `power` and parse the json using the `jq` property.