evcc-io/evcc #15336: `evcc checkconfig --config evcc.yaml` reports `config valid`, but evcc then refuses to start due to invalid config

carygravel, 2024-08-10:
### Describe the bug

1. Under `vehicles` I added a `targetSoC: 80` item
2. Then confirmed that the config was valid using `evcc checkconfig --config evcc.yaml`
3. But then refused to start, complaining about the `targetSoC: 80` line.

### Steps to reproduce

1. Under `vehicles` I added a `targetSoC: 80` item
2. Then confirmed that the config was valid using `evcc checkconfig --config evcc.yaml`
3. But then refused to start, complaining about the `targetSoC: 80` line.

### Configuration details

```yaml
See above
```

### Log details

```text
Aug 10 18:40:27 janus evcc[2525]: [main  ] FATAL 2024/08/10 18:40:27 cannot create vehicle 'ev1': cannot create vehicle type 'template': invalid key: targetsoc
Aug 10 18:40:27 janus evcc[2525]: [main  ] FATAL 2024/08/10 18:40:27 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

### Version

0.129.0

naltatis, 2024-08-10:
`targetSoc` ist eine alte Konfiguration. Das wird inzwischen über die UI eingestellt. Wo hast du die denn gefunden? In der Doku sollte die nicht mehr stehen.

naltatis, 2024-08-10:
@andig was genau prüft `checkconfig`? Hab den noch nie verwendet :D

andig, 2024-08-10:
Ob die Config (noch) legal ist

andig, 2024-08-11:
> Then confirmed that the config was valid using evcc checkconfig --config evcc.yaml

Ist das denn die gleiche Config? Der Service nutzt die unter /etc?

carygravel, 2024-08-11:
Ich habe unter `~` editiert, mit `evcc checkconfig --config evcc.yaml` geprüft, kopiert auf `/etc`, und dann den Service neu gestartet.

carygravel, 2024-08-11:
> `targetSoc` ist eine alte Konfiguration. Das wird inzwischen über die UI eingestellt. Wo hast du die denn gefunden? In der Doku sollte die nicht mehr stehen.

Ich kann in der UI 80% einstellen und es funktioniert, aber es springt irgendwann wieder auf 100%. Ich wollte der Default auf 80% setzen, da wir so gut wie nie auf 100% laden wollen. Du hast recht, dass es nicht mehr in der Doku steht, ich habe das in alten Issues/Diskussionen gesehen.

carygravel, 2024-08-11:
Übrigens - ich sehe, dass Seiten wie https://docs.evcc.io/en/docs/features/vehicle noch keine Übersetzung haben. Wenn neues zum Übersetzen gibt, könnt Ihr mich gern anpingen.

andig, 2024-08-11:
Checkconfig ist leider recht limitiert. Es findet nur grobe Fehler in der Gesamtstruktur, kann aber nicht in die einzelnen Geräte rein schauen. Insofern "works as designed".

closed 2024-08-11 by andig

carygravel, 2024-08-11:
> Checkconfig ist leider recht limitiert. Es findet nur grobe Fehler in der Gesamtstruktur, kann aber nicht in die einzelnen Geräte rein schauen. Insofern "works as designed".

Könnte dann die Info von --help mit den obigen 2 Sätzen ergänzt werden?

carygravel, 2024-08-11:
See https://github.com/evcc-io/evcc/pull/15348