evcc-io/evcc #15343: Vehicle charge limit currently defaults to 100%. Please allow an alternative default in `evcc.yaml`

carygravel, 2024-08-11:
**Is your feature request related to a problem? Please describe.**
Vehicle charge limit currently defaults to 100%. Most of the time, we'd prefer 80%. It is too complicated for the non-techy members of my family to adjust the charge limit in the UI down from 100->80% before charging. 

**Describe the solution you'd like**
So I would like to specify a default, e.g. in `evcc.yml` vehicles/defaulttargetsoc Then, on average once a year, when we need 100%, I can talk whoever needs it through the UI.

Alternatively, if the value in the UI was *much* stickier, that would be OK, too.

VolkerK62, 2024-08-11:
Choose the vehicle at the loadpoint.
click on "plan"
choose the tab "arrival"
there you can define the default limit
this is stored within the database

closed 2024-08-11 by andig

carygravel, 2024-08-11:
Where is "plan"? Or does the vehicle have to connected to see "plan"?

[image]

VolkerK62, 2024-08-11:
in the middle of the bottom line. click on "none"

carygravel, 2024-08-11:
Oh wow. The colour scheme meant that I missed that completely. I would suggest changing the colour.

Thanks for the pointer.

The "Arrival" and "Departure" tabs seem strangely named to me. The plan itself is only on "Arrival". How about renaming that "Plan" or "Times"? The "Departure" tab seems to have general options. How about renaming that "General options"?

andig, 2024-08-11:
/cc @naltatis

naltatis, 2024-08-11:
Yes, MinSoC and Limit should (and will) be moved to the vehicle entry in config ui once that's ready. So we can remove the tabs in the plan modal.