evcc-io/evcc #15346: Keba P30C & S10: Does not automatically switch between 3p and 1p in version 0.129.0

rockeyracoon69, 2024-08-11:
### Describe the bug

In version 1.129.0 no automatic switch to 1p happens: It worked before. Only loading with 3p.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
evcc.yaml:
loadpoints:
- title: Garage
  charger: wallbox4
  mode: off
  phases: 0
  mincurrent: 6
  maxcurrent: 16
```

### Log details

```text
Aug 11 14:51:28 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:28 charge power: 0W
Aug 11 14:51:28 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:28 charge currents: [0 0 0]A
Aug 11 14:51:28 raspberrypi evcc[55154]: [site  ] DEBUG 2024/08/11 14:51:28 pv power: 3948W
Aug 11 14:51:28 raspberrypi evcc[55154]: [site  ] DEBUG 2024/08/11 14:51:28 battery soc: 71%
Aug 11 14:51:28 raspberrypi evcc[55154]: [site  ] DEBUG 2024/08/11 14:51:28 battery power: -2209W
Aug 11 14:51:28 raspberrypi evcc[55154]: [site  ] DEBUG 2024/08/11 14:51:28 grid meter: -1511W
Aug 11 14:51:28 raspberrypi evcc[55154]: [site  ] DEBUG 2024/08/11 14:51:28 site power: -3620W
Aug 11 14:51:28 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:28 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.0
Aug 11 14:51:28 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:28 charge total import: 5891.640kWh
Aug 11 14:51:28 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:28 charger status: B
Aug 11 14:51:28 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:28 !! active phases: 3p = min(3p measured 3p vehicle 3p physical 0p charger)
Aug 11 14:51:28 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:28 !! active phases: 3p = min(3p measured 3p vehicle 3p physical 0p charger)
Aug 11 14:51:28 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:28 pv charge current: 5.25A = 0A + 5.25A (-3620W @ 3p)
Aug 11 14:51:42 raspberrypi evcc[55154]: [site  ] DEBUG 2024/08/11 14:51:42 ----
Aug 11 14:51:42 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:42 charge power: 0W
Aug 11 14:51:42 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:42 charge currents: [0 0 0]A
Aug 11 14:51:50 raspberrypi evcc[55154]: [site  ] DEBUG 2024/08/11 14:51:50 pv power: 3951W
Aug 11 14:51:50 raspberrypi evcc[55154]: [site  ] DEBUG 2024/08/11 14:51:50 battery soc: 71%
Aug 11 14:51:50 raspberrypi evcc[55154]: [site  ] DEBUG 2024/08/11 14:51:50 battery power: -2424W
Aug 11 14:51:50 raspberrypi evcc[55154]: [site  ] DEBUG 2024/08/11 14:51:50 grid meter: -1291W
Aug 11 14:51:50 raspberrypi evcc[55154]: [site  ] DEBUG 2024/08/11 14:51:50 site power: -3615W
Aug 11 14:51:50 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:50 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.0
Aug 11 14:51:50 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:50 charge total import: 5891.640kWh
Aug 11 14:51:50 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:50 charger status: B
Aug 11 14:51:50 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:50 !! active phases: 3p = min(3p measured 3p vehicle 3p physical 0p charger)
Aug 11 14:51:50 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:50 !! active phases: 3p = min(3p measured 3p vehicle 3p physical 0p charger)
Aug 11 14:51:50 raspberrypi evcc[55154]: [lp-1  ] DEBUG 2024/08/11 14:51:50 pv charge current: 5.24A = 0A + 5.24A (-3615W @ 3p)
```

### What type of operating system are you running?

Linux

andig, 2024-08-11:
Please provide a meaningful logfile showing the error. Happy to reopen when available.

closed 2024-08-11 by andig

rockeyracoon69, 2024-08-11:
There is no error. It does not switch to 1p automatically anymore. I have to switch to PV&Battery to load my car

rockeyracoon69, 2024-08-11:
I mean there is no error in the log reported. But is does not behave as expected.

andig, 2024-08-11:
Still needs a log

rockeyracoon69, 2024-08-11:
[evcc.log](https://github.com/user-attachments/files/16573911/evcc.log)

andig, 2024-08-11:
Was sagt

    evcc charger

rockeyracoon69, 2024-08-12:
evcc charger
[main  ] INFO 2024/08/12 16:48:54 evcc 0.129.0
[main  ] INFO 2024/08/12 16:48:54 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/12 16:48:54 using sqlite database: /home/wolfgang/.evcc/evcc.db
Power:          4003W
Energy:         5915.5kWh
Current L1..L3: 5.76A 5.72A 5.84A
Charge status:  C
Enabled:        true
Phases:         3

Was mich wundert:
Wenn ich die Keba manuell per Modus Befehl auf 1-phasig umschalte und evcc wieder starte, beginnt er einphasig zu laden, allerdings folgendes Warning im Log. Er switch ziemlich bald zurück nach 3p.

Aug 12 16:28:27 raspberrypi evcc[4453]: [lp-1  ] WARN 2024/08/12 16:28:27 charger logic error: phases mismatch (got 1, expected 3)
Aug 12 16:28:27 raspberrypi evcc[4453]: [lp-1  ] DEBUG 2024/08/12 16:28:27 !! active phases: 1p = min(0p measured 3p vehicle 1p physical 0p charger)
Aug 12 16:28:27 raspberrypi evcc[4453]: [lp-1  ] DEBUG 2024/08/12 16:28:27 switched phases: 3p
Aug 12 16:28:27 raspberrypi evcc[4453]: [lp-1  ] DEBUG 2024/08/12 16:28:27 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)

andig, 2024-08-12:
Also: laut `evcc charger` wird Phasenumschaltung erkannt.

> Er switch ziemlich bald zurück nach 3p.

Kann man aus dem Schnipsel nicht bewerten. Wenn genug Leistung vorhanden ist das ja auch richtig.

Long story short: hier ist kein Problem erkennbar.

rockeyracoon69, 2024-08-12:
Guten Abend,
Leider schaltet er nicht mehr in den 1p-Betrieb. Damit verliere ich Energie durch Rückspeissung an den Energieversorger.
Er hat das schon mal perfekt gemacht.

Die Warning Meldung finde ich seltsam:
Aug 12 16:28:27 raspberrypi evcc[4453]: [lp-1 ] WARN 2024/08/12 16:28:27 charger logic error: phases mismatch (got 1, expected 3)
Aug 12 16:28:27 raspberrypi evcc[4453]: [lp-1 ] DEBUG 2024/08/12 16:28:27 !! active phases: 1p = min(0p measured 3p vehicle 1p physical 0p charger)
Aug 12 16:28:27 raspberrypi evcc[4453]: [lp-1 ] DEBUG 2024/08/12 16:28:27 switched phases: 3p
Aug 12 16:28:27 raspberrypi evcc[4453]: [lp-1 ] DEBUG 2024/08/12 16:28:27 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)

Brauchst du noch mehr Logs ?

andig, 2024-08-12:
Ja. Ein vollständiges Log. 5 Zeilen sind leider nutzlos.

rockeyracoon69, 2024-08-12:
Gestern hatte ich eine große Logdatei hier hochgeladen: evcc.log. 
Ich kann das gerne aber nochmals tun.

rockeyracoon69, 2024-08-13:
[Archiv.zip](https://github.com/user-attachments/files/16594815/Archiv.zip)
Guten Morgen, 
anbei die Logs von Juli und August. Im Juli hatte evcc noch selbstständig die Wallbox in 1p-Modus geschalten. Im August nicht mehr.
Danke für deine Unterstützung.

andig, 2024-08-13:
Es braucht bitte mal _ein_ Log ab Start mit dem Timestamp wo Du das Problem siehst. 

> Im Juli hatte evcc noch selbstständig die Wallbox in 1p-Modus geschalten. Im August nicht mehr.

Was hast Du denn geändert?

rockeyracoon69, 2024-08-13:
Ich habe nur EVCC Versionsupgrades durchgeführt.

rockeyracoon69, 2024-08-13:
Das hat immer problemlos geklappt bis einschließlich 01.128.0.
Mit einer 0.128.x, ich meine 0.128.7 (?) hatte es Fehler gehagelt, evcc konnte nicht mehr starten.
Downgrade auf 0.128.0 hat es wieder gefixt. Dann nach Upgrade auf 0.129.0 keine Syntaxfehler, aber auch kein 1p-Mode mehr.

andig, 2024-08-13:
Und gibts jetzt _ein_ Logfile?

rockeyracoon69, 2024-08-13:
Ich habe doch heute Früh das komplette Juli- und August-Log hochgeladen, vor zwei Stunden.
?

rockeyracoon69, 2024-08-13:
Im Juli hat er noch selbstständig 1p geladen, im August nicht mehr.
Ist alles drin in den Logs.
Verstehe daher den Ton jetzt nicht.

andig, 2024-08-13:
> Es braucht bitte mal ein Log ab Start mit dem Timestamp wo Du das Problem siehst.

Letzer Versuch meinerseits: bei welchem Timestamp besteht das Problem? Ich habe keine Zeit mir das raus zu suchen!

> Im Juli hat er noch selbstständig 1p geladen, im August nicht mehr.

Hier sind alle Änderungen zwischen den Versionen. Nicht eine einzige davon betrifft Keba: https://github.com/evcc-io/evcc/compare/0.128.3...0.129.0

rockeyracoon69, 2024-08-13:
Bin in der Arbeit.
Ich versuche die Frage heute Abend zu beantworten.

rockeyracoon69, 2024-08-13:
[evcc_Auswertung.docx](https://github.com/user-attachments/files/16603110/evcc_Auswertung.docx)
[image]

In der Worddatei ist zu sehen, dass nach Upgrade auf 0.128.4 evcc beim Start abgestürzt ist. Nach Downgrade auf 0.128.0 bzw. Upgrade auf 0.129.0 startet evcc wieder fehlerfrei.

Ich habe möglicherweise die Ursache gefunden, ich kann nur vermuten. Siehe Screenshot Weboberfläche bzw. Worddatei.
"Batterieunterstütztes Laden" war eingestellt auf "nur mit PV-Überschuss" bzw. im Log "Buffer start SOC: 0"
Seitdem ich es auf einen prozentualen Wert gestellt habe, lädt er wieder, sobald der Prozentwert erreicht wird.

Ggf. kannst du mir noch bestätigen, dass das die Ursache war und mir erklären, was es mit den Schwellwerten im Screenshot auf sich hat. er lädt jetzt 3p mit/ohne Batterieunterstützung, 1p bisher noch nicht.